espressif/arduino-esp32 #10588: Core dump data check failed on XIAO ESP32-S3

SuomiKP, 2024-11-09:
### Board

XIAO ESP32S3

### Device Description

None

### Hardware Configuration

XIAO ESP32S3 Sense card slot occupies 14 GPIOs with SD card and camera modul
[image]

### Version

v3.0.6

### IDE Name

Ardiuno IDE

### Operating System

Window 10

### Flash frequency

DIO 80MHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

When encountering Core dump data check failed, it shows that ESP32-S3 has a core dump error.
Tried the following but couldn't improve it

Try lowering the resolution or JPEG quality of FRAMESIZE_UXGA to FRAMESIZE_SXGA or FRAMESIZE_XGA and setting jpeg_quality to between 12-15.

Set the framebuffer to fb_count = 1.

Reduce config.xclk_freq_hz = 20000000 to 10000000,

### Sketch

```cpp
#include "esp_camera.h"
#include "Arduino.h"
#include "FS.h"                // SD Card ESP32
#include "SD_MMC.h"            // SD Card ESP32
#include "soc/soc.h"           // Disable brownout problems
#include "soc/rtc_cntl_reg.h"  // Disable brownout problems
#include <EEPROM.h>            // read and write from flash memory

// define the number of bytes you want to access
#define EEPROM_SIZE 1

// Pin definition for CAMERA_MODEL_XIAO_ESP32S3
#define PWDN_GPIO_NUM  -1
#define RESET_GPIO_NUM -1
#define XCLK_GPIO_NUM  10
#define SIOD_GPIO_NUM  40
#define SIOC_GPIO_NUM  39

#define Y9_GPIO_NUM    48
#define Y8_GPIO_NUM    11
#define Y7_GPIO_NUM    12
#define Y6_GPIO_NUM    14
#define Y5_GPIO_NUM    16
#define Y4_GPIO_NUM    18
#define Y3_GPIO_NUM    17
#define Y2_GPIO_NUM    15
#define VSYNC_GPIO_NUM 38
#define HREF_GPIO_NUM  47
#define PCLK_GPIO_NUM  13

int pictureNumber = 0;

void setup() {
    WRITE_PERI_REG(RTC_CNTL_BROWN_OUT_REG, 0); // Disable brownout detector

    Serial.begin(115200);
    delay(2000);  // Add delay to stabilize the power before initializing

    // Initialize camera
    camera_config_t config;
    config.ledc_channel = LEDC_CHANNEL_0;
    config.ledc_timer = LEDC_TIMER_0;
    config.pin_d0 = Y2_GPIO_NUM;
    config.pin_d1 = Y3_GPIO_NUM;
    config.pin_d2 = Y4_GPIO_NUM;
    config.pin_d3 = Y5_GPIO_NUM;
    config.pin_d4 = Y6_GPIO_NUM;
    config.pin_d5 = Y7_GPIO_NUM;
    config.pin_d6 = Y8_GPIO_NUM;
    config.pin_d7 = Y9_GPIO_NUM;
    config.pin_xclk = XCLK_GPIO_NUM;
    config.pin_pclk = PCLK_GPIO_NUM;
    config.pin_vsync = VSYNC_GPIO_NUM;
    config.pin_href = HREF_GPIO_NUM;
    config.pin_sscb_sda = SIOD_GPIO_NUM;
    config.pin_sscb_scl = SIOC_GPIO_NUM;
    config.pin_pwdn = PWDN_GPIO_NUM;
    config.pin_reset = RESET_GPIO_NUM;
    config.xclk_freq_hz = 20000000; // Reduce XCLK frequency to prevent instability
    config.pixel_format = PIXFORMAT_JPEG;

  // Set frame size and quality for highest possible resolution with PSRAM support
    config.frame_size = FRAMESIZE_UXGA;
    config.jpeg_quality = 15;           // Slightly reduce quality to prevent memory issues
    config.fb_count = 1;                // Use one frame buffer to prevent memory overflow
    config.fb_location = CAMERA_FB_IN_PSRAM;

    if (config.pixel_format == PIXFORMAT_JPEG) {
      if (psramFound()) {
        config.jpeg_quality = 10;
        config.fb_count = 2;
        config.grab_mode = CAMERA_GRAB_LATEST;
      } else {
  // Limit the frame size when PSRAM is not available
        config.frame_size = FRAMESIZE_SVGA;
        config.fb_location = CAMERA_FB_IN_DRAM;
      }
      } else {
  // Best option for face detection/recognition
        config.frame_size = FRAMESIZE_240X240;
        #if CONFIG_IDF_TARGET_ESP32S3
          config.fb_count = 2;
        #endif
    }

    #if defined(CAMERA_MODEL_ESP_EYE)
      pinMode(13, INPUT_PULLUP);
      pinMode(14, INPUT_PULLUP);
    #endif
    // Init Camera
    esp_err_t err = esp_camera_init(&config);
    if (err != ESP_OK) {
        Serial.printf("Camera init failed with error 0x%x\n", err);
        return;
    }

    // Initialize SD Card
    if (!SD_MMC.begin()) {
        Serial.println("SD Card Mount Failed");
        return;
    }

    uint8_t cardType = SD_MMC.cardType();
    if (cardType == CARD_NONE) {
        Serial.println("No SD Card attached");
        return;
    }

    // Continuous capture loop
    //while (true) {
        // Capture a picture
        camera_fb_t *fb = nullptr;
        for (int retries = 0; retries < 5; retries++) {
            fb = esp_camera_fb_get();
            if (fb) {
                break; // Success
            }
            Serial.println("Camera capture failed, retrying...");
            delay(500); // Wait before retrying
        }

        if (!fb) {
            Serial.println("Camera capture failed after multiple retries");
            delay(2000); // Wait before next attempt
            return;
        }

        // Initialize EEPROM
        EEPROM.begin(EEPROM_SIZE);
        pictureNumber = EEPROM.read(0) + 1;
        // Path where new picture will be saved in SD Card
        String path = "/picture" + String(pictureNumber) + ".jpg";

        fs::FS &fs = SD_MMC; 
        Serial.printf("Picture file name: %s\n", path.c_str());
        
        File file = fs.open(path.c_str(), FILE_WRITE);
        if (!file) {
            Serial.println("Failed to open file in writing mode");
        } else {
            file.write(fb->buf, fb->len); // Write image data
            Serial.printf("Saved file to path: %s\n", path.c_str());
            //EEPROM.write(0, pictureNumber);
            //EEPROM.commit();
        }
        file.close();
        esp_camera_fb_return(fb); 

        delay(5000);  // Delay before taking the next picture to provide stability
    //}
    if(pictureNumber > 100){
      Serial.println("Going to sleep （￣︶￣）↗　");
      esp_deep_sleep_start();
      Serial.println("This will never be printed");
    }
    

}

void loop() {
}
```

### Debug Message

```plain
E (136) esp_core_dump_flash: Core dump data check failed:
Calculated checksum='a503c70d'
Image checksum='ffffffff'
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40377258
SPIWP:0xee
mode:DIO, clock div:1
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-11-09:
it's OK if you get this error. it should not prevent your sketch to run. You can erase flash to get rid of it

SuomiKP, 2024-11-10:
> it's OK if you get this error. it should not prevent your sketch to run. You can erase flash to get rid of it

**I tried wiping the flash to get rid of it, but the ESP still reboots.**

ESP-ROM:esp32s3-2021032
Build:Mar 27 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40377258
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0xf88
load:0x403c9700,len:0x4
load:0x403c9704,len:0xa14
load:0x403cc700,len:0x2d64
entry 0x403c988c

me-no-dev, 2024-11-13:
please compile with core debug level Verbose. There is another reason why your board reboots

chrish987, 2024-11-16:
I have a similar issue - probably the same issue.

My update routine copies a compiled firmware file into LittleFS and then uses the Update library to upgrade the module software from this file. Update code is pretty much identical to the examples included with the library.

```
void updateFirmware() {
	events.send("Firmware upgrade started.", "messagecr", millis());
	File file = LittleFS.open("/firmware.bin", "r");

	if (file) {
		size_t updatesize = file.size();

		if (Update.begin(updatesize)) {
			size_t written = Update.writeStream(file);
			if (Update.end()) {
				if (Update.isFinished()) {
					events.send("Firmware update completed successfully. Click 'Reboot' to restart.", "messagecr", millis());
				}
				else {
					events.send("ERROR: Update not finished - something went wrong!", "messagecr", millis());
				}
			}
			else {
				events.send("ERROR: Error Occurred. Error #: " + Update.getError(), "messagecr", millis());
			}
		}
		else {
			events.send("ERROR: Not enough space in flash.", "messagecr", millis());
		}
	}
	else {
		events.send("ERROR: Can't open firmware file.", "messagecr", millis());
	}
	file.close();
}
```

The upgrade works fine. The sketch runs fine. No issues.

However I get the following on boot:

```
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:4832
load:0x40078000,len:16460
load:0x40080400,len:4
load:0x40080404,len:3504
entry 0x400805cc
E (579) esp_core_`fW}���͡� Core dump data check failed:
Calculated checksum='852ee0a8'
Image checksum='ffffffff'
```

If I update over serial and use the 'erase flash' option then the issue goes away. But once i do my code update it comes back.

My partition table is:
```
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 0x330000,
app1,     app,  ota_1,   0x340000,0x330000,
spiffs,   data, spiffs,  0x670000,0x360000,
coredump, data, coredump,0x9d0000,0x10000,
```

Jason2866, 2025-03-13:
The behaviour is as designed. There is no checksum check.

closed 2025-03-13 by Jason2866

Sujjiiiya, 2025-05-22:
Hello, did the above problem get solved ?
I am facing a similar error and I am using Arduino IDE 2.3.6 , the error is like this

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x29 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x118c
load:0x403c8700,len:0x4
load:0x403c8704,len:0xc20
load:0x403cb700,len:0x30e0
entry 0x403c88b8
E (215) esp_core_dump_flash: Core dump data check failed:
Calculated checksum='48e0ac35'
Image checksum='ffffffff'

and these are my settings for the camera webserver code

[image]

Sujjiiiya, 2025-05-23:
@SuomiKP Thank you so much for your reply^^

I will check all the things mentioned, but also I am really confused if I am uncommenting the right board or not and if the pin configuration is correct or not, I will paste a **picture of my board here if anyone knows which line to uncomment for this board's name and the correct pin configuration** please share it with me it will be very helpful.    :)

Here is an image of my board and I have been trying to run the camera for so long but it just wont work I am doubtful that I am using wrong camera name and also pins might be wrong TT

[image]

chrish987, 2025-05-23:
the issue went away for me - and of course I did change a few things - so not sure what fixed it.

I did have this:
spiffs,   data, spiffs,  0x670000,0x360000,

which I changed to this (which I suspect was the most likely thing that fixed it):
spiffs,   data, littlefs,  0x670000,0x360000,

Sujjiiiya, 2025-05-23:
@chrish987 Thank you I tried that also.

Sujjiiiya, 2025-05-23:
Hello again , it looks like the board name I am using is correct and now I am getting this in my serial monitor , and the baud rate is 115200.

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x2a (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x118c
load:0x403c8700,len:0x4
load:0x403c8704,len:0xc20
load:0x403cb700,len:0x30e0
entry 0x403c88b8

and thats all comes in the serial  monitor and noting like wifi connecting or a link further

Sujjiiiya, 2025-05-23:
Final Update — RESOLVED

Hi everyone! Just wanted to share that my issue is finally resolved 

After trying many things (pin remapping, Wi-Fi switching, PSRAM tweaks, minimal test sketches, etc.), the real fix turned out to be:

 **Changing the "Upload Mode" setting in Arduino IDE to USB OTG CDC / TinyUSB**

Previously, I was using UART / Hardware CDC and even though uploads were successful, the code wasn't running afterward — only the ESP-ROM bootloader logs appeared. No camera output, no Wi-Fi connection, and no debug prints. After switching to USB OTG CDC, everything started working instantly!

Big thanks to everyone who gave suggestions — your help really kept me going :)