espressif/arduino-esp32 #10589: 3.1.0-RC2 Constant reboot

Arielhh, 2024-11-09:
### Board

lilygo t-display s3 touch esp32s3

### Device Description

3.1.0-RC1 is working just fine but when moving to 3.1.0-RC2 the micro controller go into Constant reboots

lilygo display t3 touch s3 - esp32s3

### Hardware Configuration

I2S max98357

### Version

latest development Release Candidate (RC-X)

### IDE Name

arduino ide

### Operating System

windows 11

### Flash frequency

40mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

3.1.0-RC1 is working just fine but when moving to 3.1.0-RC2 the micro controller go into Constant reboots

### Sketch

```cpp
the sketch is very large
```

### Debug Message

```plain
3.1.0-RC1 is working just fine but when moving to 3.1.0-RC2 the micro controller go into Constant reboots
```

### Other Steps to Reproduce

3.1.0-RC1 is working just fine but when moving to 3.1.0-RC2 the micro controller go into Constant reboots

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

TD-er, 2024-11-09:
Can the reboot be PSRAM related?
So if you disable PSRAM, does it still reboot?

Arielhh, 2024-11-09:
I need the psram to be enabled as I store there info which is required for
my product. I cannot run the code if psram is not enabled.

Jason2866, 2024-11-09:
Without providing a sample sketch which fails, it is guess work.
There is no general issue, I have several S3 boards with connected displays and enabled PSRAM perfectly fine with RC2

Arielhh, 2024-11-09:
Is the new version larger or occupied/ requires a larger ram memory space?
Maybe the issue is not enough ram space?

SuGlider, 2024-11-09:
@Arielhh - As pointed by others, this issue has no information about the Guru Mediation message, neither a minimum sketch that can be used to confirm and debug the issue. If this is a memory problem, please turn the DEBUG Level to Verbose, run your sketch and check the logs.
 
Make sure that in your code it is testing all returned pointers from memory allocation in order to avoid NULL pointers.

Arielhh, 2024-11-09:
Thanks, I will check and update.

Arielhh, 2024-11-11:
I set the DEBUG Level to Verbose but there was no errors reported. Still
when using the same code that i am using with 3.1.0-r1 and works with that
version very well it fails with R2 version causing the edp32s3 to reboot

SuGlider, 2024-11-11:
@Arielhh - Could you please provide a minimum Arduino Sketch that we can use to reproduce this issue?

emhave, 2025-01-08:
The smallest code that gives the error:
_____________________________________________________________________________________________________
#include "ESPAsyncWebServer.h" //

// Replace with your network credentials
const char* ssid = "NETGEAR04";
const char* password = "ancientshoe808";

void connectWiFi(void){  // Station 
  int Try_no = 0 ;
  Serial.printf("Connecting to: \r\n", ssid);
  Serial.println(ssid);
  WiFi.begin(ssid, password);
  while ((WiFi.status() != WL_CONNECTED) && (Try_no <= 15)) {
    delay(500);
    Serial.print(".");
    Try_no = Try_no +1 ;
    Serial.print(Try_no);
  }
  Serial.println("");
  if (WiFi.status() == WL_CONNECTED) {  
  Serial.print("WiFi connected, ");
  Serial.printf("IP address: \r\n",  WiFi.localIP() );
  Serial.println(WiFi.localIP());  
  }
}

// Create AsyncWebServer object on port 80
AsyncWebServer server(80);

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  connectWiFi();

  // Start server
  server.begin();
}

void loop() {
  // put your main code here, to run repeatedly:

}
____________________________________________________________________________________________________________
And the output on the Serial Monitor:
________________________________________________
22:18:45.085 -> ⸮⸮⸮ESP-ROM:esp32s3-20210327
22:18:45.085 -> Build:Mar 27 2021
22:18:45.085 -> rst:0xc (RTC_SW_CPU_RST),boot:0x2b (SPI_FAST_FLASH_BOOT)
22:18:45.085 -> Saved PC:0x4037a84a
22:18:45.085 -> SPIWP:0xee
22:18:45.085 -> mode:DIO, clock div:1
22:18:45.085 -> load:0x3fce2820,len:0x1188
22:18:45.085 -> load:0x403c8700,len:0x4
22:18:45.085 -> load:0x403c8704,len:0xbf0
22:18:45.085 -> load:0x403cb700,len:0x30e4
22:18:45.120 -> entry 0x403c88ac
22:18:45.323 -> Connecting to: 
22:18:45.323 -> NETGEAR04
22:18:45.867 -> .1.2.3.4
22:18:47.369 -> WiFi connected, IP address: 
22:18:47.369 -> 172.16.0.95
22:18:47.369 -> 
22:18:47.402 -> assert failed: tcp_alloc /IDF/components/lwip/lwip/src/core/tcp.c:1851 (Required to lock TCPIP core functionality!)
22:18:47.402 -> 
22:18:47.402 -> 
22:18:47.402 -> Backtrace: 0x403764c2:0x3fca9530 0x4037cb45:0x3fca9550 0x40383539:0x3fca9570 0x4201fbc7:0x3fca96a0 0x4201fd31:0x3fca96c0 0x42007628:0x3fca96e0 0x42006b91:0x3fca9730 0x42002aad:0x3fca9750 0x4200c14b:0x3fca9780 0x4037d832:0x3fca97a0
22:18:47.436 -> 
22:18:47.436 -> 
22:18:47.436 -> 
22:18:47.436 -> 
22:18:47.436 -> ELF file SHA256: 9de3629eb
22:18:47.436 -> 
22:18:47.537 -> Rebooting...
22:18:47.537 -> ⸮⸮⸮ESP-ROM:esp32s3-20210327
_______________________________________________________________________________________________________
AFAIK the issue is the line that starts with: assert failed:

On a ESP32 WROOM DA Module i get the same error .

lbernstone, 2025-01-09:
https://github.com/espressif/arduino-esp32/issues/10781#issuecomment-2561371032