espressif/arduino-esp32 #10590: Unable to advertise binary payload with BLE server

bilogic, 2024-11-10:
### Board

ESP32

### Device Description

WROOM32

### Hardware Configuration

Nothing else.

### Version

latest master (checkout manually)

### IDE Name

PlatformIO on VSCode

### Operating System

Windows 10

### Flash frequency

40Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

https://github.com/espressif/arduino-esp32/blob/09a6770320b75c219053aa19d630afe1a7c61147/libraries/BLE/src/BLEAdvertising.cpp#L472

My payload is truncated when it contains `0x0` because expected type of string is a null character terminated type.

### Sketch

```cpp
char buf[3];
buf[0] = 1;
buf[1] = 0;
buf[2] = 1;

pAdvertistmentDataa->setServiceData(BLEUUID(0x181D), buf)); 
// only 1 character set as payload instead of 3 because buf[1] is null

```
```

### Debug Message

```plain
N/A
```

### Other Steps to Reproduce

Can we please add a length parameter?

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2024-11-13:
There is no issue. Just use it as suggested below:

``` cpp
char buf[3];
buf[0] = 1;
buf[1] = 0;
buf[2] = 1;

// String is just a binary holder created from a byte array
pAdvertistmentDataa->setServiceData(BLEUUID(0x181D), String(buf, sizeof(buf))); 
```

bilogic, 2024-11-13:
ok thanks

closed 2024-11-13 by bilogic