espressif/arduino-esp32 #10594: ESP32 network problem

tianzehui666, 2024-11-11:
### Board

ESP32-S3-WROOM-1-N16R8

### Device Description

7.3inch Spectra 6 Eink Display Mainboard . Produck number: OK-DEV01C

### Hardware Configuration

connected to a elink display

### Version

v3.0.2

### IDE Name

PlatformIO

### Operating System

Windows11

### Flash frequency

40MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

1> After the WiFi connection is running, the ESP32 successfully obtains the correct IP address, subnet mask and gateway through DHCP.
2> At this time, the computer with normal network conditions under the same LAN performs a long ping: it cannot be reached.
3> It took quite some time for the ping to be able to ping, but the ICMP Echo took a lot of time to reply to the packets and the network quality was poor.

### Sketch

```cpp
//HERE IS MY SOURCE CODE:
#include "WiFi.h"
#include "WebServer.h"
#include "HTTPClient.h"
void setup()
{
    Serial.begin(115200);
    // scan wifi signal near by
    WIFI_Scan();
    // wifi contection 
    WIFI_Connection();
    server.on("/",handelRoot);
    server.begin();
    if (WiFi.status() == WL_CONNECTED) // if wifi conect success
    {
        wl_status_t status = WiFi.status();
        log_e("WiFi Status: %d", status);
        configTime(60*60*8, 0, "ntp.ntsc.ac.cn", "ntp1.aliyun.com", "ntp.tuna.tsinghua.edu.cn");// 时间同步

        for (int i = 0; i < 300; i++) 
        {
            if (getLocalTime(&timeinfo)) 
            {
                log_d("Time obtained successfully");
                break;
[... 120 lines cut ...]
␛[0;31m[112144][E][main.cpp:171] setup(): Failed to obtain time, retrying...␛[0m
␛[0;31m[118161][E][main.cpp:171] setup(): Failed to obtain time, retrying...␛[0m
␛[0;31m[124178][E][main.cpp:171] setup(): Failed to obtain time, retrying...␛[0m
␛[0;31m[130195][E][main.cpp:171] setup(): Failed to obtain time, retrying...␛[0m
␛[0;31m[136212][E][main.cpp:171] setup(): Failed to obtain time, retrying...␛[0m
␛[0;31m[142229][E][main.cpp:171] setup(): Failed to obtain time, retrying...␛[0m
␛[0;31m[148246][E][main.cpp:171] setup(): Failed to obtain time, retrying...␛[0m
␛[0;31m[154263][E][main.cpp:171] setup(): Failed to obtain time, retrying...␛[0m
␛[0;31m[160280][E][main.cpp:171] setup(): Failed to obtain time, retrying...␛[0m
␛[0;31m[166297][E][main.cpp:171] setup(): Failed to obtain time, retrying...␛[0m
␛[0;31m[172314][E][main.cpp:171] setup(): Failed to obtain time, retrying...␛[0m
␛[0;31m[178331][E][main.cpp:171] setup(): Failed to obtain time, retrying...␛[0m
␛[0;31m[184348][E][main.cpp:171] setup(): Failed to obtain time, retrying...␛[0m
␛[0;31m[190365][E][main.cpp:171] setup(): Failed to obtain time, retrying...␛[0m
␛[0;31m[196382][E][main.cpp:171] setup(): Failed to obtain time, retrying...␛[0m
␛[0;31m[202399][E][main.cpp:171] setup(): Failed to obtain time, retrying...␛[0m
␛[0;31m[208416][E][main.cpp:171] setup(): Failed to obtain time, retrying...␛[0m
␛[0;31m[214433][E][main.cpp:171] setup(): Failed to obtain time, retrying...␛[0m
␛[0;31m[220450][E][main.cpp:171] setup(): Failed to obtain time, retrying...␛[0m
␛[0;36m[222487][D][main.cpp:168] setup(): Time obtained successfully␛[0m
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-11-11:
are you pinging after setup is done (connected, time obtained and so on)? And what are you doing in the loop?

tianzehui666, 2024-11-12:
> are you pinging after setup is done (connected, time obtained and so on)? And what are you doing in the loop?

thx for reply. I pinging esp32 after connected(wifi status == WL_CONNECTED) ,before time obtained.
here is the code in loop():
void loop() 
{
    lv_task_handler();
    server.handleClient();
    vTaskDelay(pdMS_TO_TICKS(20));
}

SuGlider, 2024-11-19:
Does it work if the application is built using Arduino Core 2.0.17?

SuGlider, 2024-11-19:
In order to alllow us to help you, we need a minimum Arduino Sketch that can be used to reproduce the issue and debug it.
Please describe in details (if necessary with step to verify it) what is wrong and what was expected.

lbernstone, 2024-11-20:
Looks like the expected result if you have very poor signal.  Make sure the antenna end of your dev board is fully exposed above and below and does not have interference from wires or components nearby.

Poopi, 2025-01-13:
Isn't this being caused by configTime calling esp_netif_init() in the first place? According to the SDK documentation this function can only be called once. It is already being called by Network Manager. I removed it and I don't have any networking problems anymore with SNTP.

me-no-dev, 2025-01-13:
`esp_netif_init()` will return an error, but will not be taken into account, so it does not matter. It's probably only important that Network calls it first.