espressif/arduino-esp32 #10598: HTTPClient big lag on end on HTTP not on HTTPS

rvbc1, 2024-11-12:
### Board

ESP32-S3-DevKitC

### Device Description

DevKitC

### Hardware Configuration

No external hardware.

### Version

v3.0.6

### IDE Name

Platformio

### Operating System

Linux - Ubuntu 24.04

### Flash frequency

40MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

My pio:
platform = https://github.com/pioarduino/platform-espressif32/releases/download/51.03.07/platform-espressif32.zip
board = esp32-s3-devkitc-1
framework = arduino

Using HTTPClient over HTTP results in a significant delay on the last packet (around 5 seconds), whereas this delay doesn’t occur over HTTPS. In general, HTTPS is also noticeably faster, which is surprising since encryption is expected to add some overhead.

The file being downloaded is the same in both cases, from the same server, and the only difference in the URL is whether it starts with HTTP or HTTPS.

Could someone help me understand if I’m doing something wrong, or if this might be a bug?”

### Sketch

```cpp
#pragma once

#include "Drivers/String/Driver.h"
#include "Drivers/Timer/Software/Driver.h"
#include "HTTPClient.h"
#include "NetworkClientSecure.h"

namespace Drivers {
    namespace HttpDownloader {
        class Donwloader {
           private:
            auto &getStream() { return *http.getStreamPtr(); }
            void  end() {
                http.end();
                client.stop();
                clientSecure.stop();
            }

[... 244 lines cut ...]
[.pio/libdeps/hw_v1_debug/ESP32-Drivers/Drivers/HttpDownloader/Downloader.h:48] Download retry: 2
[.pio/libdeps/hw_v1_debug/ESP32-Drivers/Drivers/HttpDownloader/Downloader.h:45] Downloaded: 4096 | 265424/306743 | 1ms
[.pio/libdeps/hw_v1_debug/ESP32-Drivers/Drivers/HttpDownloader/Downloader.h:45] Downloaded: 4096 | 269520/306743 | 0ms
[.pio/libdeps/hw_v1_debug/ESP32-Drivers/Drivers/HttpDownloader/Downloader.h:45] Downloaded: 4096 | 273616/306743 | 0ms
[.pio/libdeps/hw_v1_debug/ESP32-Drivers/Drivers/HttpDownloader/Downloader.h:45] Downloaded: 4096 | 277712/306743 | 0ms
[.pio/libdeps/hw_v1_debug/ESP32-Drivers/Drivers/HttpDownloader/Downloader.h:48] Download retry: 0
[.pio/libdeps/hw_v1_debug/ESP32-Drivers/Drivers/HttpDownloader/Downloader.h:48] Download retry: 1
[.pio/libdeps/hw_v1_debug/ESP32-Drivers/Drivers/HttpDownloader/Downloader.h:48] Download retry: 2
[.pio/libdeps/hw_v1_debug/ESP32-Drivers/Drivers/HttpDownloader/Downloader.h:48] Download retry: 3
[.pio/libdeps/hw_v1_debug/ESP32-Drivers/Drivers/HttpDownloader/Downloader.h:48] Download retry: 4
[.pio/libdeps/hw_v1_debug/ESP32-Drivers/Drivers/HttpDownloader/Downloader.h:45] Downloaded: 4096 | 281808/306743 | 0ms
[.pio/libdeps/hw_v1_debug/ESP32-Drivers/Drivers/HttpDownloader/Downloader.h:45] Downloaded: 4096 | 285904/306743 | 0ms
[.pio/libdeps/hw_v1_debug/ESP32-Drivers/Drivers/HttpDownloader/Downloader.h:45] Downloaded: 4096 | 290000/306743 | 0ms
[.pio/libdeps/hw_v1_debug/ESP32-Drivers/Drivers/HttpDownloader/Downloader.h:45] Downloaded: 4096 | 294096/306743 | 0ms
[.pio/libdeps/hw_v1_debug/ESP32-Drivers/Drivers/HttpDownloader/Downloader.h:48] Download retry: 0
[.pio/libdeps/hw_v1_debug/ESP32-Drivers/Drivers/HttpDownloader/Downloader.h:48] Download retry: 1
[.pio/libdeps/hw_v1_debug/ESP32-Drivers/Drivers/HttpDownloader/Downloader.h:45] Downloaded: 4096 | 298192/306743 | 1ms
[.pio/libdeps/hw_v1_debug/ESP32-Drivers/Drivers/HttpDownloader/Downloader.h:45] Downloaded: 4096 | 302288/306743 | 0ms
[.pio/libdeps/hw_v1_debug/ESP32-Drivers/Drivers/HttpDownloader/Downloader.h:45] Downloaded: 4096 | 306384/306743 | 0ms
[.pio/libdeps/hw_v1_debug/ESP32-Drivers/Drivers/HttpDownloader/Downloader.h:45] Downloaded: 359 | 306743/306743 | 0ms
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2024-11-13:
Default HTTPClient timeut is 5000 ms (5 seconds).
It can be fixed by adding `setTimeout(time_ms);`

``` cpp
           NetworkClient &getClient(Drivers::string url) {
                client       = NetworkClient();
                client.setTimeout(100);  // 100ms reading timeout
                clientSecure = NetworkClientSecure();
                clientSecure.setInsecure();
                if (url.compare(0, 8, "https://") == 0) { return clientSecure; }
                return client;
            }
```

Timeout affects `Stream` class, here:
`auto downloaded = stream.readBytes(const_cast<uint8_t *>(buffer), bufferSize);`

rvbc1, 2024-11-15:
> Default HTTPClient timeut is 5000 ms (5 seconds). It can be fixed by adding `setTimeout(time_ms);`
> 
> ```c++
>            NetworkClient &getClient(Drivers::string url) {
>                 client       = NetworkClient();
>                 client.setTimeout(100);  // 100ms reading timeout
>                 clientSecure = NetworkClientSecure();
>                 clientSecure.setInsecure();
>                 if (url.compare(0, 8, "https://") == 0) { return clientSecure; }
>                 return client;
>             }
> ```
> 
> Timeout affects `Stream` class, here: `auto downloaded = stream.readBytes(const_cast<uint8_t *>(buffer), bufferSize);`

Unfortunately, the proposed solution did not help. That said, isn’t it a bug that when connecting via HTTP, the stream, after downloading the entire resource, waits for a timeout instead of terminating on its own? It seems it should automatically close after the resource is fully downloaded. Especially since this works perfectly fine over HTTPS

SuGlider, 2024-11-16:
@rvbc1 - I'll take a deeper look into it.

Jason2866, 2024-12-20:
Please try latest release core 3.1.0
```
platform = https://github.com/pioarduino/platform-espressif32/releases/download/53.03.10/platform-espressif32.zip
```

Jason2866, 2025-03-13:
Closing since no feedback. Feel free to ask to reopen if the issue still exists with actual Arduino core

closed 2025-03-13 by Jason2866