espressif/arduino-esp32 #10601: Zigbee_Temp_Hum_Sensor_Sleepy.ino : Temperature and Humidity value not correct reported in Home Assistant

phiphi060, 2024-11-13:
### Board

ESP32H2 Dev Module

### Device Description

Only the Chip module

### Hardware Configuration

No external connections

### Version

latest master (checkout manually)

### IDE Name

Arduino

### Operating System

Windows 11

### Flash frequency

921600

### PSRAM enabled

yes

### Upload speed

115200

### Description

The Temperature value is ever reported in Home Assistant with -327.7 °C
The Humidity value is ever reported in Home Assistant with 655.4 %

I have test to give fixed temperature and humidity value but the reported values in Home Assistant stay the sames

### Sketch

```cpp
Original example sketch "Zigbee_Temp_Hum_Sensor_Sleepy.ino" without modification
```

### Debug Message

```plain
no errors by Flashing
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-11-13:
@P-R-O-C-H-Y both number represent float from 16bit 0xFFFF divided by 100. One signed and one unsigned.

phiphi060, 2024-11-14:
Hey Prochy,
Thank you for your help and your detailled answer.
I'm a beginner with zigbee under Arduino. I'm set the Core Debug Level to Verbose but i don't no where can i see the log file.
In Serial Monitor is Nothing (only Chip Model : ESP-ROM:esp32h2-20221101)
In Output : i send you 2 files, one by compilation and one by flashing
[Zigbee_Temp_Hum_Sensor_Sleepy_compile.txt](https://github.com/user-attachments/files/17754936/Zigbee_Temp_Hum_Sensor_Sleepy_compile.txt)
[Zigbee_Temp_Hum_Sensor_Sleepy_flash.txt](https://github.com/user-attachments/files/17754981/Zigbee_Temp_Hum_Sensor_Sleepy_flash.txt)
... and an image File from my Home Assistant
[image]

My Home Assistant run as Stand-Alone HA OS on a Raspberry Pi5 8GB with SSD
My Zigbee Gateway is the Home Assistant Sky Connect and all my Zigbee Devices are connected under ZHA (Zigbee 3.0)

How i have say in my precedent message, i have test to set the temperature value at 22°C directly in Sketch but without success
The temperature and humidity values are not reported to Home Assistant and stay to default values (-327.7 °C and 655.4 %)

The exemple Sketch "Zigbee_Temperature_Sensor" from Library august 2024 run perfectly but this sketch has no humidity Sensor !

phiphi060, 2024-11-14:
When i set the battery level to 80 % this is correct reported in Home Assistant :)

P-R-O-C-H-Y, 2024-11-15:
Are you using the UART port or the USB port of the ESP32-H2? Will help you to get the logs.

So re are running different gateways, so maybe there will be a change needed in the delay, that's the only thing I can think of.
What you can try first is to comment out the line `62: esp_deep_sleep_start();` and replace it with `delay(5000)`. We can ensure that the problem is with the sleep this way.

michapr, 2024-11-15:
Hi,
I have/had same issue with ESP32C6. Using Zigbee2MQTT.

First problem - controller may remember about old configuration, if other sketch was loaded before (like RGB lamp etc.) So controller must be restarted after removing the device.

Second - something looks not right while join process.
In most cases (could not find out when) the join will fail:
```
[2024-11-15 13:53:17] debug:    zh:ember:ezsp: ezspTrustCenterJoinHandler(): callback called with: [newNodeId=9507], [newNodeEui64=0x543204fffe32f364], [status=STANDARD_SECURITY_SECURED_REJOIN], [policyDecision=NO_ACTION], [parentOfNewNodeId=55811]
[2024-11-15 13:53:17] debug:    zh:controller: Device '0x543204fffe32f364' joined
[2024-11-15 13:53:17] debug:    zh:controller: Device '0x543204fffe32f364' accepted by handler
[2024-11-15 13:53:17] debug:    zh:controller: New device '0x543204fffe32f364' joined
[2024-11-15 13:53:17] debug:    zh:controller: Creating device '0x543204fffe32f364'
[2024-11-15 13:53:17] info:     zh:controller: Interview for '0x543204fffe32f364' started
[2024-11-15 13:53:17] debug:    zh:controller:device: Interview - start device '0x543204fffe32f364'
[2024-11-15 13:53:17] info:     z2m: Device '0x543204fffe32f364' joined
[2024-11-15 13:53:17] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x543204fffe32f364","ieee_address":"0x543204fffe32f364"},"type":"device_joined"}'
[2024-11-15 13:53:17] info:     z2m: Starting interview of '0x543204fffe32f364'
[2024-11-15 13:53:17] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x543204fffe32f364","ieee_address":"0x543204fffe32f364","status":"started"},"type":"device_interview"}'
[2024-11-15 13:53:22] debug:    zh:controller:device: Interview - node descriptor request failed for '0x543204fffe32f364', attempt 1
[2024-11-15 13:53:24] debug:    zh:ember:ezsp: ezspTrustCenterJoinHandler(): callback called with: [newNodeId=9507], [newNodeEui64=0x543204fffe32f364], [status=STANDARD_SECURITY_SECURED_REJOIN], [policyDecision=NO_ACTION], [parentOfNewNodeId=55811]
[2024-11-15 13:53:24] debug:    zh:controller: Device '0x543204fffe32f364' joined
[2024-11-15 13:53:24] debug:    zh:controller: Device '0x543204fffe32f364' accepted by handler
[2024-11-15 13:53:24] debug:    zh:controller: Not interviewing '0x543204fffe32f364', completed 'false', in progress 'true'
[2024-11-15 13:53:27] debug:    zh:controller:device: Interview - node descriptor request failed for '0x543204fffe32f364', attempt 2
[2024-11-15 13:53:32] debug:    zh:controller:device: Interview - node descriptor request failed for '0x543204fffe32f364', attempt 3
[2024-11-15 13:53:36] debug:    zh:controller:device: Interview - node descriptor request failed for '0x543204fffe32f364', attempt 4
[2024-11-15 13:53:42] debug:    zh:controller:device: Interview - node descriptor request failed for '0x543204fffe32f364', attempt 5
[2024-11-15 13:53:42] debug:    zh:ember:ezsp: ezspTrustCenterJoinHandler(): callback called with: [newNodeId=9507], [newNodeEui64=0x543204fffe32f364], [status=STANDARD_SECURITY_SECURED_REJOIN], [policyDecision=NO_ACTION], [parentOfNewNodeId=55811]
```
I have seen cases, where the manufacturer and device type were shown, but no values were reported.

Have replaced the sleep with delay(10000), but in this case the setup() procedure will not be executed more.
Anyway the ESP will reboot now (about 2500ms after `zbTempSensor.reportHumidity();` - but is not reporting...

So - something maybe wrong....

BTW: the sample "Zigbee_Color_Dimmable_Light" is working (more or less, can switch colors, brightness, on/off, but cannot set the startup value as sample - but this is an other problem)

phiphi060, 2024-11-15:
Hi Prochy,
I have comment out the line 62: esp_deep_sleep_start(); and replace it with delay(5000) --> The temperature and humidity are correct reported :)
I have tested with smaller delay value (step by step) and i have see that with value >= 550 the temperature and humidity values are correct reported in Home Assistant. With a delay from 500, they are not reported !
[image]
Thank you for your Help

michapr, 2024-11-15:
@P-R-O-C-H-Y 
If I add hum/batt to the temperature sensor example code, it will work fine.
Reporting is working, request values also...
BUT: an other problem - if remove power - device is rebooting and will not report more any value (temp sketch).

```
[2024-11-15 18:32:54] debug:    zh:controller: Device '0x543204fffe32f364' joined
[2024-11-15 18:32:54] debug:    zh:controller: Device '0x543204fffe32f364' accepted by handler
[2024-11-15 18:32:54] debug:    zh:controller: Not interviewing '0x543204fffe32f364', completed 'true', in progress 'false'
[2024-11-15 18:33:12] debug:    zh:ember:ezsp: ezspTrustCenterJoinHandler(): callback called with: [newNodeId=15335], [newNodeEui64=0x543204fffe32f364], [status=STANDARD_SECURITY_SECURED_REJOIN], [policyDecision=NO_ACTION], [parentOfNewNodeId=55811]
[2024-11-15 18:33:12] debug:    zh:controller: Device '0x543204fffe32f364' joined
[2024-11-15 18:33:12] debug:    zh:controller: Device '0x543204fffe32f364' accepted by handler
[2024-11-15 18:33:12] debug:    zh:controller: Not interviewing '0x543204fffe32f364', completed 'true', in progress 'false'
[2024-11-15 18:33:21] debug:    zh:ember:ezsp: ezspTrustCenterJoinHandler(): callback called with: [newNodeId=15335], [newNodeEui64=0x543204fffe32f364], [status=STANDARD_SECURITY_SECURED_REJOIN], [policyDecision=NO_ACTION], [parentOfNewNodeId=55811]
```

michapr, 2024-11-16:
Additional debug log:
Have seen different lines before reboot:
`[  7919][E][ZigbeeCore.cpp:219] esp_zb_app_signal_handler(): Failed to initializeESP-ROM:esp32c6-20220919`
or
`[  7921][E][ZigbeeCore.cpp:219] esp_zb_app_signal_handler(): Failed to initialize Zigbee stack (status: ESP_FAIL)`
or nothing special...
```

[  7919][E][ZigbeeCore.cpp:219] esp_zb_app_signal_handler(): Failed to initializeESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0xc (SW_CPU),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x4001975a
SPIWP:0xee
mode:DIO, clock div:2
load:0x4086c410,len:0xcf8
load:0x4086e610,len:0x2e30
load:0x40875728,len:0x113c
entry 0x4086c410
[   252][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x42005d82
[   263][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x42005d46
[   275][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x42005d0a
[   286][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x42005cce
[   297][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x42005d82
[   309][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x42005d46
[   320][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x42005d0a
[   331][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x42005cce
[   343][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type USB_DM (38) successfully set to 0x420025ac
[   354][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type USB_DP (39) successfully set to 0x420025ac
[   365][V][ZigbeeEP.cpp:18] ZigbeeEP(): Endpoint: 10
[   375][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 17 successfully set to type UART_RX (2) with bus 0x4080e52c
[   386][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 16 successfully set to type UART_TX (3) with bus 0x4080e52c
[   396][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 12 successfully set to type USB_DM (38) with bus 0x40810394
[   407][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 13 successfully set to type USB_DP (39) with bus 0x40810394
=========== Before Setup Start ===========
Chip Info:
------------------------------------------
  Model             : ESP32-C6
  Package           : 1
  Revision          : 0.01
  Cores             : 1
  CPU Frequency     : 160 MHz
  XTAL Frequency    : 40 MHz
  Features Bitfield : 0x00000052
  Embedded Flash    : No
  Embedded PSRAM    : No
  2.4GHz WiFi       : Yes
  Classic BT        : No
  BT Low Energy     : Yes
  IEEE 802.15.4     : Yes
------------------------------------------
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   449996 B ( 439.4 KB)
  Free Bytes        :   414204 B ( 404.5 KB)
  Allocated Bytes   :    28856 B (  28.2 KB)
  Minimum Free Bytes:   414204 B ( 404.5 KB)
  Largest Free Block:   393204 B ( 384.0 KB)
------------------------------------------
Flash Info:
------------------------------------------
  Chip Size         :  4194304 B (4 MB)
  Block Size        :    65536 B (  64.0 KB)
  Sector Size       :     4096 B (   4.0 KB)
  Page Size         :      256 B (   0.2 KB)
  Bus Speed         : 40 MHz
  Bus Mode          : QIO
------------------------------------------
Partitions Info:
------------------------------------------
                nvs : addr: 0x00009000, size:    20.0 KB, type: DATA, subtype: NVS
[... 188 lines cut ...]
[   633][D][ZigbeeTempSensor.cpp:110] setHumidity(): Setting humidity to 3700
[   634][V][ZigbeeCore.cpp:301] esp_zb_app_signal_handler(): ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL
[   635][I][ZigbeeCore.cpp:191] esp_zb_app_signal_handler(): Zigbee stack initialized
=========== After Setup Start ============
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   449996 B ( 439.4 KB)
  Free Bytes        :   389260 B ( 380.1 KB)
  Allocated Bytes   :    52864 B (  51.6 KB)
  Minimum Free Bytes:   389260 B ( 380.1 KB)
  Largest Free Block:   368628 B ( 360.0 KB)
------------------------------------------
GPIO Info:
------------------------------------------
  GPIO : BUS_TYPE[bus/unit][chan]
  --------------------------------------
     9 : GPIO
    12 : USB_DM
    13 : USB_DP
    16 : UART_TX[0]
    17 : UART_RX[0]
============ After Setup End =============
```

If you need longer log file or something else tell me....

michapr, 2024-11-17:
@phiphi060 
Only for interest - what happen in your case if you remove power from sensor and start it again?
All values will be reported?

I have seen that there maybe a problem with battery powered devices, will open a new issue.
So you can close this issue here if all is fine for you.

phiphi060, 2024-11-17:
@michapr 
After comment line 62 and add a delay at line 63,iI have the same problem : if i remove the power - device is rebooting every 3,5 secs !

phiphi060, 2024-11-17:
I have updated the ESP32 Arduino Library to 3.1.0-RC3 and updated the Home Assistant Core to the latest version but the problem stay the same.

P-R-O-C-H-Y, 2024-11-18:
@phiphi060 are you using H2 or C6?
I have no issue with H2, but C6 temperature sensor (with/without humidity) is not working. It crashes after calling `reportTemperature` or `reportHumidity`. In the 3.0.0-RC3 we are using esp-zigbee-sdk 1.6.0, and I found out if I roll back to 1.5.1 it works again. So I assume the problem is there, but I am still testing if I can somehow fix it.

P-R-O-C-H-Y, 2024-11-19:
So after first reboot (waking up from deep sleep), the chip is always rebooting?

P-R-O-C-H-Y, 2024-11-19:
> @P-R-O-C-H-Y I have tested with the PR#10623 modifications but the problem stay the same : After the first reboot, the ESP32H2 reboot endless(direct after flashing) (original sketch "Zigbee_Temp_hum_Sensor_Sleepy.ino" without modification) [163015.852 - ESP-ROMesp32h2-2022110.txt](https://github.com/user-attachments/files/17817709/163015.852.-.ESP-ROMesp32h2-2022110.txt)

@phiphi060 please set debug level to Verbose in the Arduino IDE Tools menu. Flash again, and post the logs then.
Here is an empty log just with bootloader info.

phiphi060, 2024-11-20:
@P-R-O-C-H-Y  I have set the debug level to Verbose but i don't see more in serial monitor. 
I have see that the serial monitor don't print something with ESP32H2 chip. The command Serial.print run fine with Arduino R4 but not with ESP32H2. I see only the bootloader of the Chip. Do i making something special in configuration from Arduino IDE ?
My Serial Monitor is set to 115200 baud. I have set Show verbose output during "compile" and "upload" to true in Arduino IDE Preferences.
With the sketch "Zigbee_Temperature_Sensor2.ino", based of the example sketch "Zigbee_Temperature_Sensor.ino" where i have added a Humidity Cluster, all run fine but without Battery Level and Deepsleep. The Temperature and Humidity values are correct reported, after Power down and reboot too.

// Copyright 2024 Espressif Systems (Shanghai) PTE LTD
//
// Licensed under the Apache License, Version 2.0 (the "License");
// you may not use this file except in compliance with the License.
// You may obtain a copy of the License at
//
//     http://www.apache.org/licenses/LICENSE-2.0
//
// Unless required by applicable law or agreed to in writing, software
// distributed under the License is distributed on an "AS IS" BASIS,
// WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
// See the License for the specific language governing permissions and
// limitations under the License.

/**
 * @brief This example demonstrates Zigbee temperature sensor.
 *
 * The example demonstrates how to use Zigbee library to create a end device temperature sensor.
 * The temperature sensor is a Zigbee end device, which is controlled by a Zigbee coordinator.
 *
 * Proper Zigbee mode must be selected in Tools->Zigbee mode
 * and also the correct partition scheme must be selected in Tools->Partition Scheme.
 *
 * Please check the README.md for instructions and more detailed description.
 *
 * Created by Jan Procházka (https://github.com/P-R-O-C-H-Y/)
 */

#ifndef ZIGBEE_MODE_ED
#error "Zigbee coordinator mode is not selected in Tools->Zigbee mode"
#endif

#include "Zigbee.h"

#define BUTTON_PIN                  9  //Boot button for C6/H2
#define TEMP_SENSOR_ENDPOINT_NUMBER 10

ZigbeeTempSensor zbTempSensor = ZigbeeTempSensor(TEMP_SENSOR_ENDPOINT_NUMBER);

/************************ Temp_hum sensor *****************************/
static void temp_hum_sensor_value_update(void *arg) {
  for (;;) {
    // Read temperature sensor value
    float tsens_value = temperatureRead();
    log_v("Temperature sensor value: %.2f°C", tsens_value);
    // Update temperature value in Temperature sensor EP
    zbTempSensor.setTemperature(tsens_value);
    // Read humidity sensor value
    float hsens_value = 55;   //tsens_value
    log_v("Humidity sensor value: %.2f%", hsens_value);
    // Update Humidity value in Temperature sensor EP
    zbTempSensor.setHumidity(hsens_value);
    delay(1000);
  }
}

/********************* Arduino functions **************************/
void setup() {

  Serial.begin(115200);
  while (!Serial) {
    delay(10);
  }

  // Init button switch
  pinMode(BUTTON_PIN, INPUT_PULLUP);

  // Optional: set Zigbee device name and model
  zbTempSensor.setManufacturerAndModel("Espressif", "ZigbeeTempHumSensor");

  // Set minimum and maximum temperature measurement value (10-50°C is default range for chip temperature measurement)
  zbTempSensor.setMinMaxValue(10, 50);

  // Set tolerance for temperature measurement in °C (lowest possible value is 0.01°C)
  zbTempSensor.setTolerance(0.1);

  // Add humidity cluster to the temperature sensor device with min, max and tolerance values
  zbTempSensor.addHumiditySensor(0, 100, 1);

  // Add endpoint to Zigbee Core
  Zigbee.addEndpoint(&zbTempSensor);

  // When all EPs are registered, start Zigbee in End Device mode
  Zigbee.begin();

  // Start Temperature sensor reading task
  xTaskCreate(temp_hum_sensor_value_update, "temp_hum_sensor_update", 2048, NULL, 10, NULL);

  // Set reporting interval for temperature measurement in seconds, must be called after Zigbee.begin()
  // min_interval and max_interval in seconds, delta (temp change in °C)
  // if min = 1 and max = 0, reporting is sent only when temperature changes by delta
  // if min = 0 and max = 10, reporting is sent every 10 seconds or temperature changes by delta
  // if min = 0, max = 10 and delta = 0, reporting is sent every 10 seconds regardless of temperature change
  zbTempSensor.setReporting(1, 0, 0.1);
}

void loop() {
  // Checking button for factory reset
  if (digitalRead(BUTTON_PIN) == LOW) {  // Push button pressed
    // Key debounce handling
    delay(100);
    int startTime = millis();
    while (digitalRead(BUTTON_PIN) == LOW) {
      delay(50);
      if ((millis() - startTime) > 3000) {
        // If key pressed for more than 3secs, factory reset Zigbee and reboot
        Serial.printf("Resetting Zigbee to factory settings, reboot.\n");
        Zigbee.factoryReset();
      }
    }
    zbTempSensor.reportTemperature();
    zbTempSensor.reportHumidity();
  }
  delay(100);
}

P-R-O-C-H-Y, 2024-11-20:
@phiphi060 are you using UART or USB port on the devkit?

phiphi060, 2024-11-20:
@P-R-O-C-H-Y  I am using USB Port on the devkit. I don't have an USB --> UART interface

P-R-O-C-H-Y, 2024-11-20:
@phiphi060 So can you please also set this option in the Arduino IDE->Tools menu:
`USB CDC on Boot -> Enabled`
Let me know if you are getting any logs then, don't forget to keep the debug level on verbose.

phiphi060, 2024-11-22:
@P-R-O-C-H-Y Now i have installed the driver for the CP210x USB to UART Bridge and connected the ESP32H2 over UART.
Here the Log File at the Flash Process
[Log_flash_Zigbee_Temp_Hum_Sensor_Sleepy.txt](https://github.com/user-attachments/files/17874707/Log_flash_Zigbee_Temp_Hum_Sensor_Sleepy.txt)
and here the Log after joining Home Assistant
[Log_Zigbee_Temp_Hum_Sensor_Sleepy_join_Home_Assistant.txt](https://github.com/user-attachments/files/17874719/Log_Zigbee_Temp_Hum_Sensor_Sleepy_join_Home_Assistant.txt)
This Log repeat unendless !

michapr, 2024-11-23:
yes, right, it is not working. Can confirm it for ESP32C6, even with last PR https://github.com/espressif/arduino-esp32/pull/10636 no changes, is rebooting...

Nawor3565, 2024-11-24:
Somehow managed to not see this thread before making #10644. I was able to provide some more detailed logs in that issue that would probably be of help.

michapr, 2024-11-24:
@Nawor3565 
Please check https://github.com/espressif/arduino-esp32/issues/10612 - I think this is all the same issue.

Device does not like to reconnect after reboot/restart in most cases.
Here you find a lot of logs... 

I think if device will stable join again after reboot / restart, then the sleeping device also will work

P-R-O-C-H-Y, 2024-11-25:
@phiphi060 you can give a try with changes from #10651 :)

P-R-O-C-H-Y, 2024-11-26:
Hi @phiphi060, can you please use the `release/v3.1.x` branch instead of the PR checkout? All the changes are already merged :)

phiphi060, 2024-11-26:
Hi @P-R-O-C-H-Y , how can i install the release/v3.1.x in Arduino ?
I have tested to copy it in the directory from board manager : C:\Users\phili\AppData\Local\Arduino15\packages\esp32\hardware\esp32 but without success. The v3.1.x is not listed in the board manager from Arduino. I have tested too to replace all Files in the subDirectory "3.1.0-RC3", but after i have compile errors in Arduino !

P-R-O-C-H-Y, 2024-11-27:
@phiphi060  The best option is to use the "manual" installation method. You can check the documentation on how to install it here: https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html#windows-manual-installation

Afte cloning you can checkout the `release/v3.1.x` and after that run the `get.exe`.

michapr, 2024-11-27:
@phiphi060 - you can change in Arduino preferences - board manager the esp32 URL:
https://espressif.github.io/arduino-esp32/package_esp32_dev_index.json

Then you can select the available versions...
[image]

P-R-O-C-H-Y, 2024-11-27:
> @phiphi060 - you can change in Arduino preferences - board manager the esp32 URL:
> 
> https://espressif.github.io/arduino-esp32/package_esp32_dev_index.json
> 
> 
> 
> Then you can select the available versions...
> 
> [image]
> 
> 

@michapr but he needs to use the branch instead of the release to get the latest changes. So should use the manual install.

michapr, 2024-11-27:
> but he needs to use the branch instead of the release to get the latest changes. So should use the manual install.

ok,... but he asked 
>how can i install the release/v3.1.x in Arduino ?

That's why suggested this way ;)

phiphi060, 2024-11-27:
Hi @P-R-O-C-H-Y ,
Thank you for the Help.
I have following the steps 1 to 5 listed in manual installation. After that i have checkout the release/v3.1.x branch and after that run the get.exe.
But the release v3.1.x is not listed in Arduino Bord Manager !?
[image]
[image]

P-R-O-C-H-Y, 2024-11-27:
@phiphi060 You don't need to install anything in the Arduino IDE now.
In the Tools menu in Board selection you should see now 2 ESP32 Arduino, and one have "(in Sketchbook)" in the name. That's the one you installed manually. The other (without the in sketchbook), is the one installed through the board manager.
Check the screenshot. So now you can just select the one you installed manually and you should be ready to go :) 
[image]

phiphi060, 2024-11-27:
Hi @P-R-O-C-H-Y ,
I have now selected the ESP32H2 Dev MOdule in ESP32 Arduino (in sketchbook.
After that i have a compile error : 

FQBN: espressif:esp32:esp32h2:CDCOnBoot=cdc,PartitionScheme=zigbee,DebugLevel=verbose,EraseFlash=all,ZigbeeMode=ed
Using board 'esp32h2' from platform in folder: C:\Users\phili\OneDrive\Documents\Arduino\hardware\espressif\esp32
Using core 'esp32' from platform in folder: C:\Users\phili\OneDrive\Documents\Arduino\hardware\espressif\esp32

cmd /c if exist "C:\\Users\\phili\\OneDrive\\Documents\\Arduino\\hardware\\espressif\\esp32\\libraries\\Zigbee\\examples\\Zigbee_Temp_Hum_Sensor_Sleepy\\partitions.csv" COPY /y "C:\\Users\\phili\\OneDrive\\Documents\\Arduino\\hardware\\espressif\\esp32\\libraries\\Zigbee\\examples\\Zigbee_Temp_Hum_Sensor_Sleepy\\partitions.csv" "C:\\Users\\phili\\AppData\\Local\\Temp\\arduino\\sketches\\D0093BA3F3883866074E17186B3CCDE4\\partitions.csv"
cmd /c if not exist "C:\\Users\\phili\\AppData\\Local\\Temp\\arduino\\sketches\\D0093BA3F3883866074E17186B3CCDE4\\partitions.csv" if exist "C:\\Users\\phili\\OneDrive\\Documents\\Arduino\\hardware\\espressif\\esp32\\variants\\esp32h2\\partitions.csv" COPY "C:\\Users\\phili\\OneDrive\\Documents\\Arduino\\hardware\\espressif\\esp32\\variants\\esp32h2\\partitions.csv" "C:\\Users\\phili\\AppData\\Local\\Temp\\arduino\\sketches\\D0093BA3F3883866074E17186B3CCDE4\\partitions.csv"
cmd /c if not exist "C:\\Users\\phili\\AppData\\Local\\Temp\\arduino\\sketches\\D0093BA3F3883866074E17186B3CCDE4\\partitions.csv" COPY "C:\\Users\\phili\\OneDrive\\Documents\\Arduino\\hardware\\espressif\\esp32\\tools\\partitions\\zigbee.csv" "C:\\Users\\phili\\AppData\\Local\\Temp\\arduino\\sketches\\D0093BA3F3883866074E17186B3CCDE4\\partitions.csv"
        1 fichier(s) copi�(s).
cmd /c IF EXIST "C:\\Users\\phili\\OneDrive\\Documents\\Arduino\\hardware\\espressif\\esp32\\libraries\\Zigbee\\examples\\Zigbee_Temp_Hum_Sensor_Sleepy\\bootloader.bin" ( COPY /y "C:\\Users\\phili\\OneDrive\\Documents\\Arduino\\hardware\\espressif\\esp32\\libraries\\Zigbee\\examples\\Zigbee_Temp_Hum_Sensor_Sleepy\\bootloader.bin" "C:\\Users\\phili\\AppData\\Local\\Temp\\arduino\\sketches\\D0093BA3F3883866074E17186B3CCDE4\\Zigbee_Temp_Hum_Sensor_Sleepy.ino.bootloader.bin" ) ELSE ( IF EXIST "C:\\Users\\phili\\OneDrive\\Documents\\Arduino\\hardware\\espressif\\esp32\\variants\\esp32h2\\bootloader.bin" ( COPY "C:\\Users\\phili\\OneDrive\\Documents\\Arduino\\hardware\\espressif\\esp32\\variants\\esp32h2\\bootloader.bin" "C:\\Users\\phili\\AppData\\Local\\Temp\\arduino\\sketches\\D0093BA3F3883866074E17186B3CCDE4\\Zigbee_Temp_Hum_Sensor_Sleepy.ino.bootloader.bin" ) ELSE ( "C:\\Users\\phili\\OneDrive\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/esptool\\esptool.exe" --chip esp32h2 elf2image --flash_mode dio --flash_freq 48m --flash_size 4MB -o "C:\\Users\\phili\\AppData\\Local\\Temp\\arduino\\sketches\\D0093BA3F3883866074E17186B3CCDE4\\Zigbee_Temp_Hum_Sensor_Sleepy.ino.bootloader.bin" "C:\\Users\\phili\\OneDrive\\Documents\\Arduino\\hardware\\espressif\\esp32\\tools\\esp32-arduino-libs\\esp32h2\\bin\\bootloader_qio_64m.elf" ) )
Le chemin d'acc�s sp�cifi� est introuvable.
exit status 1

Compilation error: exit status 1

After control, i have seen that i have no repository named : ...\\esp32h2\\bin

After a good moment search, i have seen that the get.exe (run as administrator in CMD) report an error too :

C:\Users\phili\OneDrive\Documents\Arduino\hardware\espressif\esp32\tools>get.exe
System: Windows, Bits: 64, Info: Windows-10-10.0.22631-SP0
Platform: x86_64-mingw32

Downloading and extracting tools...

Tool esp32-arduino-libs-idf-release_v5.3-a0f798cf.zip already downloaded
 > Verify archive...  Extracting archive...
Removing existing esp32-arduino-libs ...
Traceback (most recent call last):
  File "get.py", line 481, in <module>
    if not get_tool(tool, force_download, force_extract):
  File "get.py", line 373, in get_tool
    return unpack(local_path, ".", force_extract, checksum)
  File "get.py", line 254, in unpack
    cfile.extractall(destination)
  File "zipfile.py", line 1647, in extractall
  File "zipfile.py", line 1701, in _extract_member
FileNotFoundError: [Errno 2] No such file or directory: 'esp32-arduino-libs\\esp32\\include\\espressif__esp_matter\\connectedhomeip\\connectedhomeip\\src\\app\\clusters\\boolean-state-configuration-server\\BooleanStateConfigurationTestEventTriggerHandler.h'
[PYI-21308:ERROR] Failed to execute script 'get' due to unhandled exception!

SuGlider, 2024-11-27:
@phiphi060 - could you please run `python --version` and report back?

SuGlider, 2024-11-27:
@phiphi060 - My suggestion would be to install IDF 5.3 framework and run `install.bat` from the `esp-idf` folder.
It will install/update all necessary tools, including Python, esp_tool.py, etc.

Installing IDF may take some time and also it needs about 4GB filesystem space. You may want to consider that.

phiphi060, 2024-11-28:
hi @SuGlider My Win11Pro has Python 3.10.5

SuGlider, 2024-11-28:
> hi @SuGlider My Win11Pro has Python 3.10.5

Could you please try `python get.py` ?
This should run the same as `get.exe`

phiphi060, 2024-11-29:
Hi @SuGlider,
I have the same error with python get.py

C:\Users\phili\OneDrive\Documents\Arduino\hardware\espressif\esp32\tools>python get.py
System: Windows, Bits: 64, Info: Windows-10-10.0.22631-SP0
Platform: x86_64-mingw32

Downloading and extracting tools...

Tool esp32-arduino-libs-idf-release_v5.3-a0f798cf.zip already downloaded
 > Verify archive...  Extracting archive...
Removing existing esp32-arduino-libs ...
Traceback (most recent call last):
  File "C:\Users\phili\OneDrive\Documents\Arduino\hardware\espressif\esp32\tools\get.py", line 481, in <module>
    if not get_tool(tool, force_download, force_extract):
  File "C:\Users\phili\OneDrive\Documents\Arduino\hardware\espressif\esp32\tools\get.py", line 373, in get_tool
    return unpack(local_path, ".", force_extract, checksum)
  File "C:\Users\phili\OneDrive\Documents\Arduino\hardware\espressif\esp32\tools\get.py", line 254, in unpack
    cfile.extractall(destination)
  File "C:\Program Files\Inkscape\lib\python3.10\zipfile.py", line 1645, in extractall
    self._extract_member(zipinfo, path, pwd)
  File "C:\Program Files\Inkscape\lib\python3.10\zipfile.py", line 1699, in _extract_member
    open(targetpath, "wb") as target:
FileNotFoundError: [Errno 2] No such file or directory: 'esp32-arduino-libs\\esp32\\include\\espressif__esp_matter\\connectedhomeip\\connectedhomeip\\src\\app\\clusters\\boolean-state-configuration-server\\BooleanStateConfigurationTestEventTriggerHandler.h'

C:\Users\phili\OneDrive\Documents\Arduino\hardware\espressif\esp32\tools>

me-no-dev, 2024-11-29:
looks like Windows path length issue all over again

phiphi060, 2024-11-29:
Compile Error Arduino : i have no directory named "**esp32h2**"

I have tested to uninstall and install v3.1.x but without success !

C:\\Users\\phili\\OneDrive\\Documents\\Arduino\\hardware\\espressif\\esp32\\tools\\esp32-arduino-libs\\**esp32h2\\bin\\bootloader_qio_64m.elf'**
[28120] Failed to execute script 'esptool' due to unhandled exception!

SuGlider, 2024-11-29:
> looks like Windows path length issue all over again

I just tested it changing my local path to a longer one. `get.exe` didn't fail.

SuGlider, 2024-11-29:
> Compile Error Arduino : i have no directory named "**esp32h2**"
>
> I have tested to uninstall and install v3.1.x but without success ! [...]

OK. Let start by a simple check: Is the `ESP32 Arduino (in sketchbook)` selected as the main Board?

[image]

Then you can select the H2 board from `ESP32 Arduino (in sketchbook)` option.

SuGlider, 2024-11-29:
> Hi @SuGlider, I have the same error with python get.py
> 
> C:\Users\phili\OneDrive\Documents\Arduino\hardware\espressif\esp32\tools>python get.py System: Windows, Bits: 64, Info: Windows-10-10.0.22631-SP0 Platform: x86_64-mingw32
> 
> Downloading and extracting tools...
> 
> Tool esp32-arduino-libs-idf-release_v5.3-a0f798cf.zip already downloaded
> 
> > Verify archive...  Extracting archive...
> > Removing existing esp32-arduino-libs ...
> > Traceback (most recent call last):
> > File "C:\Users\phili\OneDrive\Documents\Arduino\hardware\espressif\esp32\tools\get.py", line 481, in 
> > if not get_tool(tool, force_download, force_extract):
> > File "C:\Users\phili\OneDrive\Documents\Arduino\hardware\espressif\esp32\tools\get.py", line 373, in get_tool
> > return unpack(local_path, ".", force_extract, checksum)
> > File "C:\Users\phili\OneDrive\Documents\Arduino\hardware\espressif\esp32\tools\get.py", line 254, in unpack
> > cfile.extractall(destination)
> > File "C:\Program Files\Inkscape\lib\python3.10\zipfile.py", line 1645, in extractall
> > self._extract_member(zipinfo, path, pwd)
> > File "C:\Program Files\Inkscape\lib\python3.10\zipfile.py", line 1699, in _extract_member
> > open(targetpath, "wb") as target:
> > FileNotFoundError: [Errno 2] No such file or directory: 'esp32-arduino-libs\esp32\include\espressif__esp_matter\connectedhomeip\connectedhomeip\src\app\clusters\boolean-state-configuration-server\BooleanStateConfigurationTestEventTriggerHandler.h'
> 
> C:\Users\phili\OneDrive\Documents\Arduino\hardware\espressif\esp32\tools>

First things first: 
>> > Removing existing esp32-arduino-libs ...
This where the problem happens
> > FileNotFoundError: [Errno 2] No such file or directory: 'esp32-arduino-libs\esp32\include\espressif__esp_matter\connectedhomeip\connectedhomeip\src\app\clusters\boolean-state-configuration-server\BooleanStateConfigurationTestEventTriggerHandler.h'

> > looks like Windows path length issue all over again
> 
> I just tested it changing my local path to a longer one. `get.exe` didn't fail.

ALL Fine with long path... using Win CMD

```
C:\Users\esp\Documents\Arduino\hardware\espressif\path_very_long_00000000000000000000000000000000000000000000000000000000000000000000000000000000_testing\orig\tools>python get.py -e -v
System: Windows, Bits: 64, Info: Windows-10-10.0.22631-SP0
Platform: x86_64-mingw32

Downloading and extracting tools...

Tool esp32-arduino-libs-idf-release_v5.3-a0f798cf.zip already downloaded
 Forcing extraction
Removing existing esp32-arduino-libs ...

Verification passed; completed in 00:03.70
 Files extracted successfully.

Tool xtensa-esp-elf-13.2.0_20240530-x86_64-w64-mingw32_hotfix.zip already downloaded
 Forcing extraction
Removing existing xtensa-esp-elf ...

Verification passed; completed in 00:00.19
 Files extracted successfully.

Tool xtensa-esp-elf-gdb-14.2_20240403-x86_64-w64-mingw32.zip already downloaded
 Forcing extraction
Removing existing xtensa-esp-elf-gdb ...

Verification passed; completed in 00:00.00
 Files extracted successfully.

Tool riscv32-esp-elf-13.2.0_20240530-x86_64-w64-mingw32.zip already downloaded
 Forcing extraction
Removing existing riscv32-esp-elf ...

Verification passed; completed in 00:00.18
 Files extracted successfully.

Tool riscv32-esp-elf-gdb-14.2_20240403-x86_64-w64-mingw32.zip already downloaded
 Forcing extraction
Removing existing riscv32-esp-elf-gdb ...

Verification passed; completed in 00:00.01
 Files extracted successfully.

Tool openocd-esp32-win64-0.12.0-esp32-20241016.zip already downloaded
 Forcing extraction
Removing existing openocd-esp32 ...

Verification passed; completed in 00:00.10
 Files extracted successfully.

Tool esptool-v4.9.dev1-win64.zip already downloaded
 Forcing extraction
Removing existing esptool ...
Renaming esptool-win64 to esptool ...

Verification passed; completed in 00:00.00
 Files extracted successfully.

Tool x86_64-w64-mingw32.mklittlefs-c41e51a.200706.zip already downloaded
 Forcing extraction
Removing existing mklittlefs ...

Verification passed; completed in 00:00.00
 Files extracted successfully.

Tool mkspiffs-0.2.3-arduino-esp32-win32.zip already downloaded
 Forcing extraction
Removing existing mkspiffs ...
Renaming mkspiffs-0.2.3-arduino-esp32-win32 to mkspiffs ...

Verification passed; completed in 00:00.00
 Files extracted successfully.

Platform Tools Installed
```

phiphi060, 2024-11-29:
Hi @SuGlider , i have Administrator rights with this User. I can create files in this Folder (i have tested it with your instructions).

When i reinstall Arduino ESP32 Libs, i have ever an error ! (same error as before)

C:\Users\phili\OneDrive\Documents\Arduino\hardware\espressif\esp32\tools>python get.py -e -v
System: Windows, Bits: 64, Info: Windows-10-10.0.22631-SP0
Platform: x86_64-mingw32

Downloading and extracting tools...

Tool esp32-arduino-libs-idf-release_v5.3-a0f798cf.zip already downloaded
 Forcing extraction
Traceback (most recent call last):
  File "C:\Users\phili\OneDrive\Documents\Arduino\hardware\espressif\esp32\tools\get.py", line 481, in <module>
    if not get_tool(tool, force_download, force_extract):
  File "C:\Users\phili\OneDrive\Documents\Arduino\hardware\espressif\esp32\tools\get.py", line 373, in get_tool
    return unpack(local_path, ".", force_extract, checksum)
  File "C:\Users\phili\OneDrive\Documents\Arduino\hardware\espressif\esp32\tools\get.py", line 254, in unpack
    cfile.extractall(destination)
  File "C:\Program Files\Inkscape\lib\python3.10\zipfile.py", line 1645, in extractall
    self._extract_member(zipinfo, path, pwd)
  File "C:\Program Files\Inkscape\lib\python3.10\zipfile.py", line 1699, in _extract_member
    open(targetpath, "wb") as target:
**FileNotFoundError: [Errno 2] No such file or directory: 'esp32-arduino-libs\\esp32\\include\\espressif__esp_matter\\connectedhomeip\\connectedhomeip\\src\\app\\clusters\\boolean-state-configuration-server\\BooleanStateConfigurationTestEventTriggerHandler.h'**

SuGlider, 2024-11-29:
> > FileNotFoundError: [Errno 2] No such file or directory: 'esp32-arduino-libs\esp32\include\espressif__esp_matter\connectedhomeip\connectedhomeip\src\app\clusters\boolean-state-configuration-server\BooleanStateConfigurationTestEventTriggerHandler.h'
> 
> Humm... this could be the issue with Windows long path. Please check it: Problem Description: https://learn.microsoft.com/en-us/windows/win32/fileio/maximum-file-path-limitation?tabs=registry Fix Procedure: https://www.howtogeek.com/266621/how-to-make-windows-10-accept-file-paths-over-260-characters/

@phiphi060 - 
`C:\Users\phili\OneDrive\Documents\Arduino\hardware\espressif\esp32\tools\esp32-arduino-libs\esp32\include\espressif__esp_matter\connectedhomeip\connectedhomeip\src\app\clusters\boolean-state-configuration-server\BooleanStateConfigurationTestEventTriggerHandler.h` **has 263 character, over the default limit of 260 characters.**

phiphi060, 2024-11-29:
Hi @SuGlider ,
Thank you for the Help 🥇 I have applied the fix procedure in Regedit (accept path over 260 character)
Now, the get.py is run without error and all files are extracted at the right place.
In Arduino IDE, i have no more compile errors with the sketch Zigbee_Temp_Hum_Sensor_Sleepy 👍 
Now, i must testing if ESP32H2 after flashing can rejoin Home Assistant ...

phiphi060, 2024-11-29:
Hi @P-R-O-C-H-Y ,
Now, i have flashed the ESP32H2 with the sketch "Zigbee_Temp_Hum_Sensor_Sleepy.ino" from the branche v3.1.x
Unfortunely the ESP32H2 can not joining Home Assistant.
Flash process without problem but after that nothing !
The ESP32 stay as sleeping ?!

18:34:15.936 -> ESP-ROM:esp32h2-20221101
18:34:15.936 -> Build:Nov  1 2022
18:34:15.936 -> rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
18:34:15.936 -> SPIWP:0xee
18:34:15.936 -> mode:DIO, clock div:1
18:34:15.936 -> load:0x408460e0,len:0x112c
18:34:15.936 -> load:0x4083cad0,len:0xe18
18:34:15.936 -> load:0x4083efd0,len:0x2d04
18:34:15.936 -> entry 0x4083cad0
18:34:16.111 -> [     1][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x42005e7c
18:34:16.144 -> [    12][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x42005e40
18:34:16.144 -> [    24][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x42005e04
18:34:16.144 -> [    35][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x42005dc8
18:34:16.177 -> [    47][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x42005e7c
18:34:16.177 -> [    58][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x42005e40
18:34:16.177 -> [    69][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x42005e04
18:34:16.212 -> [    81][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x42005dc8
18:34:16.212 -> [    92][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type USB_DM (38) successfully set to 0x420026ac
18:34:16.212 -> [   104][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type USB_DP (39) successfully set to 0x420026ac
18:34:16.244 -> [   115][V][ZigbeeEP.cpp:18] ZigbeeEP(): Endpoint: 10
18:34:16.244 -> [   130][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 23 successfully set to type UART_RX (2) with bus 0x4080f3d8
18:34:16.244 -> [   141][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 24 successfully set to type UART_TX (3) with bus 0x4080f3d8
18:34:16.277 -> [   151][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 26 successfully set to type USB_DM (38) with bus 0x408113f0
18:34:16.277 -> [   162][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 27 successfully set to type USB_DP (39) with bus 0x408113f0

phiphi060, 2024-12-02:
Hi @P-R-O-C-H-Y ,
Here the log with Tools set to "Zigbee ED (end device) - Debug"
`18:04:15.926 -> ESP-ROM:esp32h2-20221101
18:04:15.926 -> Build:Nov  1 2022
18:04:15.926 -> rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
18:04:15.926 -> SPIWP:0xee
18:04:15.926 -> mode:DIO, clock div:1
18:04:15.926 -> load:0x408460e0,len:0x112c
18:04:15.926 -> load:0x4083cad0,len:0xe18
18:04:15.926 -> load:0x4083efd0,len:0x2d04
18:04:15.926 -> entry 0x4083cad0
18:04:16.177 -> [     1][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x42005e7c
18:04:16.212 -> [    12][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x42005e40
18:04:16.212 -> [    24][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x42005e04
18:04:16.212 -> [    35][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x42005dc8
18:04:16.212 -> [    47][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x42005e7c
18:04:16.246 -> [    58][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x42005e40
18:04:16.246 -> [    69][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x42005e04
18:04:16.246 -> [    81][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x42005dc8
18:04:16.280 -> [    92][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type USB_DM (38) successfully set to 0x420026ac
18:04:16.280 -> [   104][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type USB_DP (39) successfully set to 0x420026ac
18:04:16.280 -> [   115][V][ZigbeeEP.cpp:18] ZigbeeEP(): Endpoint: 10
18:04:16.314 -> ZBOSS: zcl/zcl_common.c:215   >> zb_zcl_get_attribute_size: attr_type 0x20 attr_value 0x4084ef0f
18:04:16.314 -> ZBOSS: zcl/zcl_common.c:348   << zb_zcl_get_attribute_size: result 1
18:04:16.314 -> ZBOSS: zcl/zcl_common.c:215   >> zb_zcl_get_attribute_size: attr_type 0x30 attr_value 0x4084ef0e
18:04:16.314 -> ZBOSS: zcl/zcl_common.c:348   << zb_zcl_get_attribute_size: result 1
18:04:16.314 -> ZBOSS: zcl/zcl_common.c:215   >> zb_zcl_get_attribute_size: attr_type 0x21 attr_value 0x4084ef0e
18:04:16.348 -> ZBOSS: zcl/zcl_common.c:348   << zb_zcl_get_attribute_size: result 2
18:04:16.348 -> ZBOSS: zcl/zcl_common.c:215   >> zb_zcl_get_attribute_size: attr_type 0x29 attr_value 0x4084ef0e
18:04:16.348 -> ZBOSS: zcl/zcl_common.c:348   << zb_zcl_get_attribute_size: result 2
18:04:16.348 -> ZBOSS: zcl/zcl_common.c:215   >> zb_zcl_get_attribute_size: attr_type 0x29 attr_value 0x4084ef0c
18:04:16.381 -> ZBOSS: zcl/zcl_common.c:348   << zb_zcl_get_attribute_size: result 2
18:04:16.381 -> ZBOSS: zcl/zcl_common.c:215   >> zb_zcl_get_attribute_size: attr_type 0x29 attr_value 0x4084ef0a
18:04:16.381 -> ZBOSS: zcl/zcl_common.c:348   << zb_zcl_get_attribute_size: result 2
18:04:16.381 -> [   206][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 23 successfully set to type UART_RX (2) with bus 0x4080f420
18:04:16.381 -> [   218][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 24 successfully set to type UART_TX (3) with bus 0x4080f420
18:04:16.414 -> [   229][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 26 successfully set to type USB_DM (38) with bus 0x40811440
18:04:16.414 -> [   240][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 27 successfully set to type USB_DP (39) with bus 0x40811440`

michapr, 2024-12-08:
@P-R-O-C-H-Y I cannot get any sleeping example code to work with Z2M.
Have tried to modify the simple temperature sensor code - without success.

I think here are different problems:
First: Sensor will not be full configured after first join, it is going to sleep before. After wakeup the configuration will not be done.
(Have added a delay of 60 seconds before loop - then configuration was finished) So would be great if we could detect this state - is sensor full configured or not.
-> in result the right values will be shown in Z2M interface

Second: I still get in this case after wakeup the low lqi issue - but only here....
```
ZBOSS: nwk/nwk_join.c:629   Rejoin failure: no dev with xpanid 3a:63:5d:94:1c:56:10:7e or can't choose best parent
ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:549   COMMISSIONING_STOP: app signal 6 comm status 3
[ 14761][W][ZigbeeCore.cpp:250] esp_zb_app_signal_handler(): Commissioning failed, trying again...
```

Will try to add the `esp_zb_secur_network_min_join_lqi_set(0);`

michapr, 2024-12-09:
> Will try to add the `esp_zb_secur_network_min_join_lqi_set(0);`

did not help...

But - in Homeassistant the time for join/configure is ok, here we do not see the problem, that device will not be configured.

But after wakeup device will not report more values (here last original sleep example with H2 module):

```
ZBOSS: common/zb_init_default.c:225   Configurable mem build, use ZBOSS lib defaults = 0
ZBOSS: common/zb_init_default.c:231   ZB_IOBUF_POOL_SIZE 80 ZB_NWK_IN_Q_SIZE 40 ZB_MAC_PENDING_QUEUE_SIZE 20 ZB_APS_DST_BINDING_TABLE_SIZE 16 ZB_APS_BIND_TRANS_TABLE_SIZE 20 ZB_N_APS_RETRANS_ENTRIES 27
ZBOSS: common/zb_init_default.c:261   ZB_N_APS_KEY_PAIR_ARR_MAX_SIZE 8 ZB_IEEE_ADDR_TABLE_SIZE 64 ZB_NEIGHBOR_TABLE_SIZE 64 ZB_NWK_ROUTING_TABLE_SIZE 0 ZB_APS_DUPS_TABLE_SIZE 32
[ 43369][D][ZigbeeCore.cpp:129] zigbeeInit(): Register all Zigbee EPs in list
ZBOSS: zdo/af_descriptor.c:357   ep_count 1
[ 43372][I][ZigbeeCore.cpp:137] zigbeeInit(): List of registered Zigbee EPs:
[ 43373][I][ZigbeeCore.cpp:139] zigbeeInit(): Device type: Temperature Sensor, Endpoint: 10, Device ID: 0x0302
ZBOSS: common/zb_nvram.c:3190   ds_ver == 2
ZBOSS: zdo/zdo_app_prod_conf.c:144   no production config block found
ZBOSS: zdo/zdo_app_prod_conf.c:781   no valid production configuration found, signaling to application
[ 43389][V][ZigbeeCore.cpp:332] esp_zb_app_signal_handler(): ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL
[ 43390][I][ZigbeeCore.cpp:218] esp_zb_app_signal_handler(): Zigbee stack initialized
ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:349   dev type 2, joined 1, ext_pan_id 1, authenticated 1, tclk_valid 1
ZBOSS: nwk/nwk_join.c:629   Rejoin failure: no dev with xpanid e7:32:5a:3d:f0:c0:7b:b2 or can't choose best parent
ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:549   COMMISSIONING_STOP: app signal 6 comm status 3
[ 52707][W][ZigbeeCore.cpp:250] esp_zb_app_signal_handler(): Commissioning failed, trying again...
ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:349   dev type 2, joined 1, ext_pan_id 1, authenticated 1, tclk_valid 1
ZBOSS: nwk/nwk_join.c:629   Rejoin failure: no dev with xpanid e7:32:5a:3d:f0:c0:7b:b2 or can't choose best parent
ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:549   COMMISSIONING_STOP: app signal 6 comm status 3
[ 62523][W][ZigbeeCore.cpp:250] esp_zb_app_signal_handler(): Commissioning failed, trying again...
ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:349   dev type 2, joined 1, ext_pan_id 1, authenticated 1, tclk_valid 1
ZBOSS: nwk/nwk_join.c:629   Rejoin failure: no dev with xpanid e7:32:5a:3d:f0:c0:7b:b2 or can't choose best parent
ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:549   COMMISSIONING_STOP: app signal 6 comm status 3
[ 72339][W][ZigbeeCore.cpp:250] esp_zb_app_signal_handler(): Commissioning failed, trying again...
ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:349   dev type 2, joined 1, ext_pan_id 1, authenticated 1, tclk_valid 1
[ 73386][E][ZigbeeCore.cpp:43] begin(): ZigbeeCore begin timeout
Zigbee failed to start!
Rebooting...
```

P-R-O-C-H-Y, 2024-12-09:
I will set up the Z2M in few days once the my RPi arrives. So I will test it using that also.

michapr, 2024-12-09:
>  You may just adjust that or also try to adjust the zigbeeConfig.nwk_cfg.zed_cfg.keep_alive = 10000; You can test with different values.

But I think this is not the right way... if a sleeping device - I want to save energy, and must minimize the live time.
So I do not want in such a case any long delay, right?
I think the standard Zigbee devices will not wait such a long time for reporting a value.
(Not sure about it in this time how in Zigbee, know it only from Z-Wave, BLE, LoRaWAN...)

P-R-O-C-H-Y, 2024-12-09:
I agree, will have to take a look again and check if there is anything I can do to remove the delay.
But the testing is not that simple, the best way is using a current measurement, where you see a current peak if the data is reported. It will take some time.

michapr, 2024-12-09:
What does this mean?
```
ZBOSS: zdo/zdo_app_prod_conf.c:144   no production config block found
ZBOSS: zdo/zdo_app_prod_conf.c:781   no valid production configuration found, signaling to application
```

michapr, 2024-12-09:
Adjust any delay or keep_alive did not help... I think main problem after wakeup is here:

```
[  1519][I][ZigbeeCore.cpp:218] esp_zb_app_signal_handler(): Zigbee stack initialized
ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:349   dev type 2, joined 1, ext_pan_id 1, authenticated 1, tclk_valid 1
ZBOSS: nwk/nwk_join.c:629   Rejoin failure: no dev with xpanid e7:32:5a:3d:f0:c0:7b:b2 or can't choose best parent
ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:549   COMMISSIONING_STOP: app signal 6 comm status 3
[ 10836][W][ZigbeeCore.cpp:250] esp_zb_app_signal_handler(): Commissioning failed, trying again...
```

P-R-O-C-H-Y, 2024-12-09:
For me when I changed the daly to 1s, its still working...

michapr, 2024-12-09:
>I need to set-up the Z2M to check if its somehow related to it.

This is Homeassistant now, with Dongle-P

P-R-O-C-H-Y, 2024-12-09:
HomeAssistant with ZHA integration??

michapr, 2024-12-09:
I use the docker installation and the ZHA integration, yes....
And here is now only one additional device included (because I use in normal case Z2M + MQTT from other machine)

P-R-O-C-H-Y, 2024-12-09:
So the only difference is I have the Dongle-E.

P-R-O-C-H-Y, 2024-12-09:
I guess now it would be great if you can manage to sniff the Zigbee communication to see what is happening there.

[37 more replies]

closed 2025-03-13 by Jason2866