espressif/arduino-esp32 #10602: 3.1.0-RC2 integrity failure

lyusupov, 2024-11-14:
### Board

ESP32-P4

### Device Description

devkit

### Hardware Configuration

default

### Version

latest development Release Candidate (RC-X)

### IDE Name

Arduino CLI

### Operating System

Linux

### Flash frequency

any

### PSRAM enabled

yes

### Upload speed

any

### Description

Installation of 3.1.0-RC2 with CLI is broken today. 3.1.0-RC1 is fine.

### Sketch

```cpp
Not applicable
```

### Debug Message

```plain
Error during install: Cannot install tool esp32:esp32-arduino-libs@idf-release_v5.3-a0f798cf: testing local archive integrity: testing archive size: fetched archive size differs from size specified in index: 339750940 != 344062217
```

### Other Steps to Reproduce

This is the command line that gives a failure:

```
$  arduino-cli core install esp32:esp32@3.1.0-RC2
Tool arduino:dfu-util@0.11.0-arduino5 already installed
Downloading packages...
esp32:esp32-arduino-libs@idf-release_v5.3-a0f798cf downloaded
esp32:esp-x32@2405 esp32:esp-x32@2405 already downloaded
esp32:xtensa-esp-elf-gdb@14.2_20240403 esp32:xtensa-esp-elf-gdb@14.2_20240403 already downloaded
esp32:esp-rv32@2405 esp32:esp-rv32@2405 already downloaded
esp32:riscv32-esp-elf-gdb@14.2_20240403 esp32:riscv32-esp-elf-gdb@14.2_20240403 already downloaded
esp32:openocd-esp32@v0.12.0-esp32-20241016 esp32:openocd-esp32@v0.12.0-esp32-20241016 already downloaded
esp32:esptool_py@4.8.1 esp32:esptool_py@4.8.1 already downloaded
esp32:mkspiffs@0.2.3 esp32:mkspiffs@0.2.3 already downloaded
esp32:mklittlefs@3.0.0-gnu12-dc7f933 esp32:mklittlefs@3.0.0-gnu12-dc7f933 already downloaded
esp32:esp32@3.1.0-RC2 esp32:esp32@3.1.0-RC2 already downloaded
Installing esp32:esp32-arduino-libs@idf-release_v5.3-a0f798cf...
Error during install: Cannot install tool esp32:esp32-arduino-libs@idf-release_v5.3-a0f798cf: testing local archive integrity: testing archive size: fetched archive size differs from size specified in index: 339750940 != 344062217
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2024-11-15:
@lucasssvaz - PTAL. Thanks!

me-no-dev, 2024-11-15:
I have an idea why that happened...

me-no-dev, 2024-11-15:
Fixed by releasing RC3

closed 2024-11-15 by me-no-dev

lyusupov, 2024-11-15:
RC3 works fine to me:
https://github.com/lyusupov/SoftRF/actions/runs/11858801381/job/33050267338

Nezaemmy, 2024-11-15:
@me-no-dev my computer use AMD cpu. is the cpu be an issue?
before I use Intel and I didn't have an issue to install RC2

Nezaemmy, 2024-11-16:
> hmmm... please delete all previously downloaded files. That should not happen on RC3. The CI tested the release

it worked. thank you