espressif/arduino-esp32 #10606: Server name returned by scan advertisement call back can contain padded NULLs causing problems.

tvonderhaar, 2024-11-15:
### Board

XIAO ESP32S3

### Device Description

0.49" OLED only.

### Hardware Configuration

GPIO 5(SDA),6(SCL) are connected as I2C interface to OLED.

### Version

latest development Release Candidate (RC-X)

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80MHZ

### PSRAM enabled

yes

### Upload speed

921,600

### Description

   The Owon B35T+ multimeter name string is padded with 12 NULLs so the "advertisedDevice.toString().toString().c_str()"
        call results in only showing the name. The "toSrting()" function appends the device
        advertisment information as a C++ String to the device name. Converting this to a C
        string via "c_str()" retains the NULLs in the combined string effectively
        terminating the C string at first padded NULL. Using "c_str()" on the function
        "getName()" directly will give you a correctly terminated string but will still
        have padded NULLs stored in the C strings. Printing "advertisedDevice.toString()"
        as a C++ String using C++ "println())" function prints the NULLs and the appended
        advertisement information by knowing the padded NULLs don't indicate the end of
        the C++ String (The C++ String class maintains the string length.)

### Sketch

```cpp
Just replace the class MyAdvertisedDeviceCallbacks code in the github
example found at: 
https://github.com/espressif/arduino-esp32/blob/master/libraries/BLE/examples/Scan/Scan.ino

with the following code:

/*Scan for BLE servers and print advertised information.*/
class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks
{
    /*Called for each advertising BLE server.*/
    void onResult(BLEAdvertisedDevice advertisedDevice)
    {
      //This will print the name only for the Owon  B35T+ multimeter.
      Serial.printf("Advertised Device: %s \n",
                     advertisedDevice.toString().c_str());

      //This will print the name,address... for the Owon B35T+ multimeter.
      Serial.println(advertisedDevice.toString());
}; // MyAdvertisedDeviceCallbacks
```

### Debug Message

```plain
No error message expose this problem.
```

### Other Steps to Reproduce

You may need a Owon BT35,BT41, or clamp meter capable of BLE  to reproduce this problem. Any server device that pads its Name with NULLs might reproduce this problem;Unless, the padding with NULLs is a software bug in the library and not related to receiving advertising name information  with these padded NULLS.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

P-R-O-C-H-Y, 2024-11-15:
Hi @tvonderhaar, can you please set Debug level to "Verbose" and post the whole log from serial monitor? Thanks

SuGlider, 2024-11-15:
Bluetooth Library will get all bytes of the advertised device name, including padding '\0' and store it as a byte array inside a `String` object.

`String` is used just as a byte container buffer. It has all bytes and a length.
BLE Library doesn't use `String` as a regular C string. We shall think about it as a byte array Class.

The way `String` (as a byte array) object obtained from `advertisedDevice.toString()` is used is up to the application.
 
 The application could parse the result from `advertisedDevice.toString()` looking for '\0', 
 split it into many pieces, replace '\0' by ' ' (0x20 - ASCII for space) or, for instance, just do something like this:
``` cpp
/*Scan for BLE servers and print advertised information.*/
class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks
{
    /*Called for each advertising BLE server.*/
    void onResult(BLEAdvertisedDevice advertisedDevice)
    {

    // get all bytes from Advertised BLE devices
    for (uint8_t i = 0;  i < advertisedDevice.toString().length(); i++) {
      Serial.print(advertisedDevice.toString()[i], HEX);
      Serial.print(" ");
    }
    Serial.println();
  }
}
```

SuGlider, 2024-11-15:
@tvonderhaar - It is intended to work like that (as explained above). Threefore, there is no issue.

tvonderhaar, 2024-11-16:
Why would you intentionally want the address and optional information not to be printed out from this git's  example "scan.ino" program. The output confused me for awhile thinking that the DMM was not sending its address. If BLE names should not have NULLs appended to its name according to the standard, I can understand why this situation would not be expected. Now knowing some devices will pad their names with NULLs maybe the library function that stores the received name could trim the NULLs off from the string so others aren't mislead by the example program's output. I don't need this fix, but I thought taking the time to register this issue would help others.

closed 2024-11-16 by tvonderhaar

SuGlider, 2024-11-16:
Thanks @tvonderhaar.

Actually the NUL character is fully valid in UTF-8, although not a printable one.

The Bluetooth standard says the following:

>Advertising data
Can contain multiple data segments, each with a one octet length field and a one octet type field. The advertising data is limited to 31 octets.

>The Bluetooth Device Name can be up to 248 bytes (see [Vol 2] Part C, Section 4.3.5). It shall be encoded according to UTF-8 (therefore the name entered on the UI level may be restricted to as few as 62 characters if codepoints outside the range U+0000 to U+007F are used).

It here explicitly lists U+0000 (and all other non-printable characters) as codepoints.

This is most likely an implementation bug where NUL-terminated C strings are used somewhere in the pipeline and not a Bluetooth specification issue.

SuGlider, 2024-11-16:
>BT35,BT4 

It seems that the Device tries to send a double name, each ended with a '\0'.
This is a particularity about how ` Owon B35T+ multimeter` has decided to implement its own naming in advertising.
Most BLE devices won't do that. There is nothing in BLE specification that prevents it.

tvonderhaar, 2024-11-17:
Thanks for the information @SuGlider. I used the code below and got the following output:
 

> scanning:
> .....
> Found Owon DMM:BDM
> name[len=15]:     "BDM\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00"
> add[len=72]:"Name: BDM\x00\x00\x00\x0\x00\x00\x00\x00\x00\x00\x00, Addres0:6e:49:19, ter: 0, rssi: -90"
> Connecting to device BDM@b0:10:a0:6e:49:19
> Connected to server BDM.
> 21301l 004.9 DC mV
> 21952l 004.9 DC mV
> 22527l 004.9 DC mV
> 23127l 004.9 DC mV
> 

For some reason printing output from this call back can corrupt the output. Notice some output characters are missing, but the lengths of the test strings are always the same.

I have a brand new Owon B35T+ from Amazon. Apparently the current version of this DMM sends the name with 12 padded NULLs.  I have read on other sites, if you name the meter via setting one of the BLE characteristic values, the meter advertises both with a name of the default "BDM" and with the characteristic value set name.  I assumed these names would be sent as two separate advertisements, but maybe it combines the default and set name as you describe. 

```
/* If this function determines that the advertising BLE server is a Owon DMM,
   it will stop the current scan and setup for a connect to the DMM in order
   to collect and display DMM measurements. 
     Note: This function is Called for each advertising BLE server.
*/
class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks
{
   void onResult(BLEAdvertisedDevice advertisedDevice)
   {
   #ifdef DEBUG
      Serial.print("BLE Advertised Device found: ");
      /*The Owon name string is padded with 12 NULLs so the "advertisedDevice.toString().c_str()"
        call results in only showing the name. The "toSrting()" function appends the device
        advertisment information as a C++ String to the device name. Converting this to a C
        string via "c_str()" retains the NULLs in the combined string effectively
        terminating the C string at first padded NULL. Using "c_str()" on the function
        "getName()" directly will give you a correctly terminated string but will still
        have padded NULLs stored in the C strings. Printing "advertisedDevice.toString()"
        as a C++ String using C++ "println())" function prints the NULLs and the appended
        advertisement information by knowing the padded NULLs don't indicate the end of
        the C++ String (The C++ String class maintains the string length.)
      */
      Serial.println(advertisedDevice.toString());
   #endif

      /*Get the name of the device found.*/
      const char *deviceName=advertisedDevice.getName().c_str();

      /*Is this a Owon DMM.*/
      if(deviceName && !strcmp(BT_METER_NAME,deviceName)){//Yes.
         Serial.printf("\nFound Owon DMM:%s\n",deviceName);
         #define DEBUG//Uncomment to execute debug statements.
         #ifdef DEBUG
         {
            /*Display all bytes from BDM "getName()".*/
            String name=advertisedDevice.getName();
            int    len=name.length();
            Serial.flush();
            Serial.printf("name[len=%d]:     \"",len);
            for(uint8_t i=0;i<len;i++){
               if(isprint(name[i]))
                  Serial.printf("%c",name[i]);
               else
                  Serial.printf("\\x%02X",name[i]);
            }
            Serial.printf("\"\n");
            Serial.flush();

            /*Display all bytes from BDM Advertisement.*/
            String add=advertisedDevice.toString();//Get a copy of toSring().
            len=add.length();
            Serial.printf("add[len=%d]:\"",len);
            for(uint8_t i=0;i<len;i++){
               if(isprint(add[i]))
                  Serial.printf("%c",add[i]);
               else
                  Serial.printf("\\x%02X",add[i]);
            }
            Serial.printf("\"\n");
            Serial.flush();
         }
         #endif

         doScan=false;   //Don't rescan until disconnect occurs.
         BLEDevice::getScan()->stop();//Stop current scan.
         myDevice=new BLEAdvertisedDevice(advertisedDevice);//Get ptr to device.
         doConnect=true; //Signal "loop()" to connect.
         scanDotColumn=0;//Reset scan serial dot output column count.
      }
      else{
         Serial.printf(".");
         if(++scanDotColumn>=79){
            Serial.printf("\n");
            scanDotColumn=0;//Reset scan serial dot output column count.
         }
 	       Serial.flush();
      }
   }//onResult()
};//MyAdvertisedDeviceCallbacks

```

SuGlider, 2024-11-17:
@tvonderhaar - One way to comapre outputs and behaviours, whould by using an Android/iOS APP called [nRF Connect for Mobile](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp).

This appliction can show the Raw Advertising data from any BLE device.
It can be used to compare the data you see there with what ESP BLE Interface is returning.

You can set characteristics after connecting to it. It may help you aldo to test and check if the ESP32 application works in the same way as viewed within `nRF Connect` APP.

tvonderhaar, 2024-11-22:
@SuGlider - I downloaded the nRF IOS app and ran the app. I did not find how to look at raw advertising data.  I was able to connect my B35T+ and display services and characteristic UUIDs.  Also, I was able to see the BT35T+ notify data which was as I expected(6 bytes of coded measurement values.)  I have finished my project  based on the instructables  project at the following link:
https://www.instructables.com/BLE-Data-Glasses/?utm_source=newsle

I used a XIAO ESP32S3 and a .49" OLED to display a HUD(Heads Up Display) of the Owon BT35T+ measurement values for all functions of the DMM.

SuGlider, 2024-11-22:
>I did not find how to look at raw advertising data.

While in the nRF Connect SCAN window, click on RAW option of the target BLE device to see it.

tvonderhaar, 2024-11-23:
@SuGlider you must have the Android version of nRF app. The IOS version 2.7.x does not have the raw option on the Scanner window. In fact, their user interfaces differ considerably and the Android version appears to have many more capabilities.

SuGlider, 2024-11-23:
Yes, I have the Android version. Good to know that iOS version is different. 
Maybe someone close to you, that has Android, may want to help it.