espressif/arduino-esp32 #10608: [3.1.0-RC3] Compilation error: 'ZigbeeTempSensor' does not name a type

lboue, 2024-11-15:
### Board

ESP32-C6-WROOM-1

### Device Description

ESP32-C6-WROOM-1

### Hardware Configuration

N/A

### Version

latest development Release Candidate (RC-X)

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

N/A

### PSRAM enabled

no

### Upload speed

N/A

### Description

```
FQBN: esp32:esp32:esp32c6:PartitionScheme=zigbee,ZigbeeMode=ed
Using board 'esp32c6' from platform in folder: C:\Users\ludov\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.0-RC3
Using core 'esp32' from platform in folder: C:\Users\ludov\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.0-RC3

cmd /c if exist "C:\\Users\\ludov\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.1.0-RC3\\libraries\\Zigbee\\examples\\Zigbee_Temp_Hum_Sensor_Sleepy\\partitions.csv" COPY /y "C:\\Users\\ludov\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.1.0-RC3\\libraries\\Zigbee\\examples\\Zigbee_Temp_Hum_Sensor_Sleepy\\partitions.csv" "C:\\Users\\ludov\\AppData\\Local\\Temp\\arduino\\sketches\\14611BEED82E1F00EE6288BBC9A3DD21\\partitions.csv"
cmd /c if not exist "C:\\Users\\ludov\\AppData\\Local\\Temp\\arduino\\sketches\\14611BEED82E1F00EE6288BBC9A3DD21\\partitions.csv" if exist "C:\\Users\\ludov\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.1.0-RC3\\variants\\esp32c6\\partitions.csv" COPY "C:\\Users\\ludov\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.1.0-RC3\\variants\\esp32c6\\partitions.csv" "C:\\Users\\ludov\\AppData\\Local\\Temp\\arduino\\sketches\\14611BEED82E1F00EE6288BBC9A3DD21\\partitions.csv"
cmd /c if not exist "C:\\Users\\ludov\\AppData\\Local\\Temp\\arduino\\sketches\\14611BEED82E1F00EE6288BBC9A3DD21\\partitions.csv" COPY "C:\\Users\\ludov\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.1.0-RC3\\tools\\partitions\\zigbee.csv" "C:\\Users\\ludov\\AppData\\Local\\Temp\\arduino\\sketches\\14611BEED82E1F00EE6288BBC9A3DD21\\partitions.csv"
        1 fichier(s) copi�(s).
cmd /c IF EXIST "C:\\Users\\ludov\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.1.0-RC3\\libraries\\Zigbee\\examples\\Zigbee_Temp_Hum_Sensor_Sleepy\\bootloader.bin" ( COPY /y "C:\\Users\\ludov\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.1.0-RC3\\libraries\\Zigbee\\examples\\Zigbee_Temp_Hum_Sensor_Sleepy\\bootloader.bin" "C:\\Users\\ludov\\AppData\\Local\\Temp\\arduino\\sketches\\14611BEED82E1F00EE6288BBC9A3DD21\\Zigbee_Temp_Hum_Sensor_Sleepy.ino.bootloader.bin" ) ELSE ( IF EXIST "C:\\Users\\ludov\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.1.0-RC3\\variants\\esp32c6\\bootloader.bin" ( COPY "C:\\Users\\ludov\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.1.0-RC3\\variants\\esp32c6\\bootloader.bin" "C:\\Users\\ludov\\AppData\\Local\\Temp\\arduino\\sketches\\14611BEED82E1F00EE6288BBC9A3DD21\\Zigbee_Temp_Hum_Sensor_Sleepy.ino.bootloader.bin" ) ELSE ( "C:\\Users\\ludov\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esptool_py\\4.9.dev1\\esptool.exe" --chip esp32c6 elf2image --flash_mode dio --flash_freq 80m --flash_size 4MB -o "C:\\Users\\ludov\\AppData\\Local\\Temp\\arduino\\sketches\\14611BEED82E1F00EE6288BBC9A3DD21\\Zigbee_Temp_Hum_Sensor_Sleepy.ino.bootloader.bin" "C:\\Users\\ludov\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.3-a0f798cf\\esp32c6\\bin\\bootloader_qio_80m.elf" ) )
esptool.py v4.8.1
Creating esp32c6 image...
Merged 2 ELF sections
Successfully created esp32c6 image.
cmd /c if exist "C:\\Users\\ludov\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.1.0-RC3\\libraries\\Zigbee\\examples\\Zigbee_Temp_Hum_Sensor_Sleepy\\build_opt.h" COPY /y "C:\\Users\\ludov\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.1.0-RC3\\libraries\\Zigbee\\examples\\Zigbee_Temp_Hum_Sensor_Sleepy\\build_opt.h" "C:\\Users\\ludov\\AppData\\Local\\Temp\\arduino\\sketches\\14611BEED82E1F00EE6288BBC9A3DD21\\build_opt.h"
cmd /c if not exist "C:\\Users\\ludov\\AppData\\Local\\Temp\\arduino\\sketches\\14611BEED82E1F00EE6288BBC9A3DD21\\build_opt.h" type nul > "C:\\Users\\ludov\\AppData\\Local\\Temp\\arduino\\sketches\\14611BEED82E1F00EE6288BBC9A3DD21\\build_opt.h"
cmd /c type nul > "C:\\Users\\ludov\\AppData\\Local\\Temp\\arduino\\sketches\\14611BEED82E1F00EE6288BBC9A3DD21/file_opts"
cmd /c COPY /y "C:\\Users\\ludov\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.3-a0f798cf\\esp32c6\\sdkconfig" "C:\\Users\\ludov\\AppData\\Local\\Temp\\arduino\\sketches\\14611BEED82E1F00EE6288BBC9A3DD21\\sdkconfig"
        1 fichier(s) copi�(s).
Detecting libraries used...
C:\Users\ludov\AppData\Local\Arduino15\packages\esp32\tools\esp-rv32\2405/bin/riscv32-esp-elf-g++ -c @C:\Users\ludov\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.3-a0f798cf\esp32c6/flags/cpp_flags -w -Os -Werror=return-type -w -x c++ -E -CC -DF_CPU=160000000L -DARDUINO=10607 -DARDUINO_ESP32C6_DEV -DARDUINO_ARCH_ESP32 -DARDUINO_BOARD="ESP32C6_DEV" -DARDUINO_VARIANT="esp32c6" -DARDUINO_PARTITION_zigbee -DARDUINO_HOST_OS="windows" -DARDUINO_FQBN="esp32:esp32:esp32c6:UploadSpeed=921600,CDCOnBoot=default,CPUFreq=160,FlashFreq=80,FlashMode=qio,FlashSize=4M,PartitionScheme=zigbee,DebugLevel=none,EraseFlash=none,JTAGAdapter=default,ZigbeeMode=ed" -DESP32=ESP32 -DCORE_DEBUG_LEVEL=0 -DARDUINO_USB_MODE=1 -DARDUINO_USB_CDC_ON_BOOT=0 -DZIGBEE_MODE_ED @C:\Users\ludov\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.3-a0f798cf\esp32c6/flags/defines -IC:\Users\ludov\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.0-RC3\libraries\Zigbee\examples\Zigbee_Temp_Hum_Sensor_Sleepy -iprefix C:\Users\ludov\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.3-a0f798cf\esp32c6/include/ @C:\Users\ludov\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.3-a0f798cf\esp32c6/flags/includes -IC:\Users\ludov\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.3-a0f798cf\esp32c6/qio_qspi/include -IC:\Users\ludov\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.0-RC3\cores\esp32 -IC:\Users\ludov\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.0-RC3\variants\esp32c6 @C:\Users\ludov\AppData\Local\Temp\arduino\sketches\14611BEED82E1F00EE6288BBC9A3DD21/build_opt.h @C:\Users\ludov\AppData\Local\Temp\arduino\sketches\14611BEED82E1F00EE6288BBC9A3DD21/file_opts C:\Users\ludov\AppData\Local\Temp\arduino\sketches\14611BEED82E1F00EE6288BBC9A3DD21\sketch\Zigbee_Temp_Hum_Sensor_Sleepy.ino.cpp -o nul
Alternatives for Zigbee.h: [Zigbee@3.0.5]
ResolveLibrary(Zigbee.h)
  -> candidates: [Zigbee@3.0.5]
C:\Users\ludov\AppData\Local\Arduino15\packages\esp32\tools\esp-rv32\2405/bin/riscv32-esp-elf-g++ -c @C:\Users\ludov\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.3-a0f798cf\esp32c6/flags/cpp_flags -w -Os -Werror=return-type -w -x c++ -E -CC -DF_CPU=160000000L -DARDUINO=10607 -DARDUINO_ESP32C6_DEV -DARDUINO_ARCH_ESP32 -DARDUINO_BOARD="ESP32C6_DEV" -DARDUINO_VARIANT="esp32c6" -DARDUINO_PARTITION_zigbee -DARDUINO_HOST_OS="windows" -DARDUINO_FQBN="esp32:esp32:esp32c6:UploadSpeed=921600,CDCOnBoot=default,CPUFreq=160,FlashFreq=80,FlashMode=qio,FlashSize=4M,PartitionScheme=zigbee,DebugLevel=none,EraseFlash=none,JTAGAdapter=default,ZigbeeMode=ed" -DESP32=ESP32 -DCORE_DEBUG_LEVEL=0 -DARDUINO_USB_MODE=1 -DARDUINO_USB_CDC_ON_BOOT=0 -DZIGBEE_MODE_ED @C:\Users\ludov\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.3-a0f798cf\esp32c6/flags/defines -IC:\Users\ludov\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.0-RC3\libraries\Zigbee\examples\Zigbee_Temp_Hum_Sensor_Sleepy -iprefix C:\Users\ludov\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.3-a0f798cf\esp32c6/include/ @C:\Users\ludov\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.3-a0f798cf\esp32c6/flags/includes -IC:\Users\ludov\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.3-a0f798cf\esp32c6/qio_qspi/include -IC:\Users\ludov\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.0-RC3\cores\esp32 -IC:\Users\ludov\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.0-RC3\variants\esp32c6 -IC:\Users\ludov\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.0-RC3\libraries\Zigbee\src @C:\Users\ludov\AppData\Local\Temp\arduino\sk [... cut ...]
Using library Zigbee at version 3.0.5 in folder: C:\Users\ludov\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.0-RC3\libraries\Zigbee 
exit status 1

Compilation error: 'ZigbeeTempSensor' does not name a type
```

### Sketch

```cpp
https://github.com/espressif/arduino-esp32/blob/release/v3.1.x/libraries/Zigbee/examples/Zigbee_Temp_Hum_Sensor_Sleepy/Zigbee_Temp_Hum_Sensor_Sleepy.ino
```

### Debug Message

```plain
N/A
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lboue, 2024-11-15:
I don't understand why I am getting this error because I can see the Zigbee files:

[image]

michapr, 2024-11-17:
Have you tried to compile other example code like Temperature sensor or Color_Dimmable Lamp?
(don't forget to set all settings like "erase all flash" etc.)

P-R-O-C-H-Y, 2024-11-18:
Hi @lboue, I have installed the 3.0.0-RC3 through the Board Manager in the Arduino IDE and the example compiles without any issues. Can you try to do a clean install of the Arduino core?

P-R-O-C-H-Y, 2024-11-18:
@lboue Did you use manual install of RC-3 or you use the Board manager?
If you use manual install, don't forget to get the new libs by get.py.

lboue, 2024-11-18:
For my first test I had done an update between the 2 RC versions. This time I uninstalled completely then reinstalled and it works fine.

P-R-O-C-H-Y, 2024-11-18:
So I am closing this issue as solved :)

closed 2024-11-18 by P-R-O-C-H-Y