espressif/arduino-esp32 #10611: claimInterface no longer works when using USB.h on an ESP32-S3

iKK001, 2024-11-17:
### Board

ESP32S3 Dev Module

### Device Description

plain module on breadboard

### Hardware Configuration

no special HW configuration

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

macOS 15.1

### Flash frequency

80 Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

I am running a WebUSB web-page. It all works if I use the standard `ESP32S3 Dev Module` out of the box.

However, if I try to customize the `productName` and `manufacturerName` of my ESP32S3 module, then the CLAIM INTERFACE of my web-application does no longer work.

Inside my Web-application (i.e. not so important whtat it does), the interace-claim is important in order to send or receive cmds to or from the ESP32 module.

Normally the interface-Nr is equal to 1 for my ESP32S3 Dev Module.  But as soon as I mutate prod- and manuf-Names, then my Web-Application can't claim the interface no more.
```
await usbDevice.claimInterface(pairedDevice, x);   // normally x=1   ....but not if I mutate prod- and manuf-Names...
```
I tried x=0....100
All with failure.....

I run this Arduino code to customize my productionName and manufacturerName

```
#include "USB.h"

void setup() {

    USB.PID(0x1001);
    USB.VID(0x303a);
    USB.productName("MyProduct");
    USB.manufacturerName("MyCompany");    
    USB.begin();
    
    Serial.begin(115200);
    while (!Serial) {
        delay(10);
    }
}

void loop() { 
  delay(1000);
}
```

Any idea how to get the CLAIM INTERFACE running again? Is there more USB cmds that I can apply to maybe define the interface ?

### Sketch

```cpp
#include "USB.h"

void setup() {

    USB.PID(0x1001);
    USB.VID(0x303a);
    USB.productName("MyProduct");
    USB.manufacturerName("MyCompany");    
    USB.begin();
    
    Serial.begin(115200);
    while (!Serial) {
        delay(10);
    }
}

void loop() { 
  delay(1000);
}
```
```

### Debug Message

```plain
My Web-Application throws an error saying something like "The interface of the usb-device cannot be claimed"
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2024-11-17:
Changing VID, PID, ProductName and VendorName is recognized by the HOST Operating System?
Can you see those ids in the USB devices list (usbls or some other command/GUI)?

iKK001, 2024-11-23:
No, after I run my setup as shown above, my ESP32S3 recognized by a webUSB website. 

However, I can no longer use it to program it anymore!  The port never shows again.

Any idea what to do ?

Is there a better way to change VID, PID, ProductName and MnufacturerName ??