espressif/arduino-esp32 #10612: problem with battery powered devices after power off/on

michapr, 2024-11-17:
### Board

ESP32C6

### Device Description

DevKit

### Hardware Configuration

No external connections, only modul. 

### Version

latest development Release Candidate (RC-X)

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

Using Zigbee2MQTT battery powered device will be recognized and values will be reported as expected.
After power on/off device (remove power) the device will not more join and send values to the network and reboots regularly.

Have seen that this is happen only for battery powered devices.

If adding some delay after `Zigbee.addEndpoint(&zbTempSensor);` and after `Zigbee.begin();` device will not reboot, but will also not join again / report values (some other debug messages).

### Sketch

```cpp
example Zigbee_Temperature_Sensor.ino modified:

static void temp_sensor_value_update(void *arg) {
  for (;;) {
    // Read temperature sensor value
    float tsens_value = temperatureRead();
    log_v("Temperature sensor value: %.2f°C", tsens_value);
    // Update temperature value in Temperature sensor EP
    zbTempSensor.setTemperature(tsens_value);
    zbTempSensor.setHumidity(tsens_value);
    delay(1000);
  }
}

....
  // Optional: set Zigbee device name and model
  zbTempSensor.setManufacturerAndModel("Espressif", "ZigbeeTempSensor");

  zbTempSensor.setPowerSource(ZB_POWER_SOURCE_BATTERY, 100);
  zbTempSensor.addHumiditySensor(0, 100, 1);

  // Set minimum and maximum temperature measurement value (10-50°C is default range for chip temperature measurement)
[... 437 lines cut ...]

ELF file SHA256: 5542d8f16
```

### Other Steps to Reproduce

Removing from `esp_zb_task(void *pvParameters)` in ZigbeeCore.cpp the workaround device will join and report values after power off/on (without described delay device will reboot before stable work 2-3 times...) :

```
static void esp_zb_task(void *pvParameters) {
  /* initialize Zigbee stack */
  ESP_ERROR_CHECK(esp_zb_start(false));

  //NOTE: This is a workaround to make battery powered devices to be discovered as battery powered
  if (((zigbee_role_t)Zigbee.getRole() == ZIGBEE_END_DEVICE) && edBatteryPowered) {
   // zb_set_ed_node_descriptor(0, 0, 0);    <<<=====
  }

  esp_zb_stack_main_loop();
}
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

P-R-O-C-H-Y, 2024-11-18:
Hi @michapr, Thanks for the report. I will take a look and try to reproduce.

michapr, 2024-11-18:
Today have seen that my "solution" is not working stable - after some time reporting stop with error:
`[6548110][V][ZigbeeCore.cpp:301] esp_zb_app_signal_handler(): ZDO signal: ZDO Device Unavailable (0x3c), status: ESP_OK`

log looks like:
```
[6547533][V][Zigbee_Temperature_Sensor.ino:45] temp_sensor_value_update(): Temperature sensor value: 31.00°C
[6547534][V][ZigbeeTempSensor.cpp:68] setTemperature(): Updating temperature sensor value...
[6547534][D][ZigbeeTempSensor.cpp:70] setTemperature(): Setting temperature to 3100
[6547535][V][ZigbeeTempSensor.cpp:107] setHumidity(): Updating humidity sensor value...
[6547535][D][ZigbeeTempSensor.cpp:109] setHumidity(): Setting humidity to 3100
[6548110][V][ZigbeeCore.cpp:301] esp_zb_app_signal_handler(): ZDO signal: ZDO Device Unavailable (0x3c), status: ESP_OK
[6548535][V][Zigbee_Temperature_Sensor.ino:45] temp_sensor_value_update(): Temperature sensor value: 31.00°C
[6548536][V][ZigbeeTempSensor.cpp:68] setTemperature(): Updating temperature sensor value...
[6548536][D][ZigbeeTempSensor.cpp:70] setTemperature(): Setting temperature to 3100
[6548537][V][ZigbeeTempSensor.cpp:107] setHumidity(): Updating humidity sensor value...
[6548537][D][ZigbeeTempSensor.cpp:109] setHumidity(): Setting humidity to 3100
[6549537][V][Zigbee_Temperature_Sensor.ino:45] temp_sensor_value_update(): Temperature sensor value: 31.00°C
```

And - in the morning the device after power on (with changed code) was still rebooting again for some time - then was starting reporting.
So, is not stable...

Thanks for checking!

P-R-O-C-H-Y, 2024-11-18:
I was able to reproduce it with C6, however with ESP32-H2 there is no issue. Will continue in the investigation.

btw. the message `esp_zb_app_signal_handler(): ZDO signal: ZDO Device Unavailable (0x3c), status: ESP_OK` happened to me when I removed the device from the HomeAssistant but I did not reset the device to factory setting, so its still trying to send data to the network. I will also check that later :)

P-R-O-C-H-Y, 2024-11-18:
As the example with/without selecting power source works on H2, I assume there must be some error in the esp-zigbee-sdk for C6. It may take more time to fix. I will keep this issue updated when I get any news.

P-R-O-C-H-Y, 2024-11-19:
Hi @michapr, can you test please with the changes from this PR: 
https://github.com/espressif/arduino-esp32/pull/10623 Thanks

P-R-O-C-H-Y, 2024-11-19:
I assume you removed the device from your system and do a factory reset.

michapr, 2024-11-19:
I removed, erased flash while programming, restarted Zigbee2MQTT (in database it is deleted, nothing more in memory) - so it should be new device...

P-R-O-C-H-Y, 2024-11-19:
Time to time that happens to me also, but lets say 1 out of 10 times.
For you its every single reboot?

P-R-O-C-H-Y, 2024-11-19:
Can you try to flash other ESP32C6/H2 with Thermostat example, to check if its related to the Zigbee2MQTT?

P-R-O-C-H-Y, 2024-11-19:
I have installed the Zigbee2MQTT add-on in my HomeAssistant. 
Adding the device was fine, event I am getting message that device is not supported, due to manufacturer/name.
The temperature reporting is fine, but when I send the message to report the temp the device reboots.
But After reboot the device is correctly started again. No crash reboots on my side.

michapr, 2024-11-19:
Yes,... I have tried several times, the device is not reporting values more after power off/on...

I do not have a second ESP32C6 in this time, wanted to test at first how Zigbee is working with ESP32 ;)
All other work with ESP32 last time was with BLE and LoRaWAN devices ....

As far as I understood the thermostat sample is not an end device, but is waiting for any other temperature sensor device right? 

In my Zigbee2MQTT installation are 14 other devices connected, sensors, plugs and thermostats (valves) - they do not have such problems, so I hope, that Z2M installation is not the problem.

As I wrote, the color_dimmable_light is working fine, even after power off/on.

Any other idea?

P-R-O-C-H-Y, 2024-11-19:
Does the board reboots after the stack fails to initialize? That's the behavior I added to reset the board after
`[  3316][E][ZigbeeCore.cpp:219] esp_zb_app_signal_handler(): Failed to initialize Zigbee stack (status: ESP_FAIL)`

I am really not sure what is happening on your side, this is not happening to me. I will try to investigate more but its hard when I am not able to replicate it.

P-R-O-C-H-Y, 2024-11-19:
IF you can try to sniff the packets going between the coordinator and the ESP that can help.

michapr, 2024-11-19:
No, board does not reboot in my case (with temperature sketch). It only does not report more values - looks like does not join again.

What I just have tried - I added again a dely between some lines:
```
  // Set tolerance for temperature measurement in °C (lowest possible value is 0.01°C)
  zbTempSensor.setTolerance(1);
 delay(500);
  // Add endpoint to Zigbee Core
  Zigbee.addEndpoint(&zbTempSensor);
delay(500);
  // When all EPs are registered, start Zigbee in End Device mode
  Zigbee.begin();
delay(500);
  // Start Temperature sensor reading task
  xTaskCreate(temp_sensor_value_update, "temp_sensor_update", 2048, NULL, 10, NULL);
delay(500);

```
Now it "can happen" that device is reporting value after power off/on.

If not reporting - I do not see any packets in Z2M debug log from the device.

michapr, 2024-11-19:
looks like with this settings (delay) device is reporting after power off/on every time:
```
  // Set tolerance for temperature measurement in °C (lowest possible value is 0.01°C)
  zbTempSensor.setTolerance(1);
 delay(500);
  // Add endpoint to Zigbee Core
  Zigbee.addEndpoint(&zbTempSensor);
delay(500);
  // When all EPs are registered, start Zigbee in End Device mode
  Zigbee.begin();
delay(1000);
  // Start Temperature sensor reading task
  xTaskCreate(temp_sensor_value_update, "temp_sensor_update", 2048, NULL, 10, NULL);
delay(1000);
```
 Not sure at what place it is right/needed - maybe you have an idea ;)

P-R-O-C-H-Y, 2024-11-19:
Can you just add a bigger delay after the `Zigbee.begin()` Add like 2000-5000 ms. The other delays are not relevant from my point of view.

michapr, 2024-11-19:
> add a bigger delay after the `Zigbee.begin()`

this did not help at all - up to 5000ms

Maybe that reporting starts before the stack was initialized? 
I hope I will find tomorrow time to make more experiments with delays...

michapr, 2024-11-21:
ok,.... have two different debug logs - one for working, second for not working device.
Have used the previous delay settings (500/500/1000/1000)
Device was powered off/on in every case.

Difference is after the "After Setup End" section - maybe you know why there can be different result??

**_NOT WORKING:_**
```
=========== Before Setup Start ===========
Chip Info:
------------------------------------------
  Model             : ESP32-C6
  Package           : 1
  Revision          : 0.01
  Cores             : 1
  CPU Frequency     : 160 MHz
  XTAL Frequency    : 40 MHz
  Features Bitfield : 0x00000052
  Embedded Flash    : No
  Embedded PSRAM    : No
  2.4GHz WiFi       : Yes
  Classic BT        : No
  BT Low Energy     : Yes
  IEEE 802.15.4     : Yes
------------------------------------------
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   445324 B ( 434.9 KB)
  Free Bytes        :   413824 B ( 404.1 KB)
  Allocated Bytes   :    24516 B (  23.9 KB)
  Minimum Free Bytes:   409012 B ( 399.4 KB)
  Largest Free Block:   376820 B ( 368.0 KB)
------------------------------------------
Flash Info:
------------------------------------------
  Chip Size         :  4194304 B (4 MB)
  Block Size        :    65536 B (  64.0 KB)
  Sector Size       :     4096 B (   4.0 KB)
  Page Size         :      256 B (   0.2 KB)
  Bus Speed         : 40 MHz
  Bus Mode          : QIO
------------------------------------------
Partitions Info:
------------------------------------------
                nvs : addr: 0x00009000, size:    20.0 KB, type: DATA, subtype: NVS
            otadata : addr: 0x0000E000, size:     8.0 KB, type: DATA, subtype: OTA
               app0 : addr: 0x00010000, size:  1280.0 KB, type:  APP, subtype: OTA_0
               app1 : addr: 0x00150000, size:  1280.0 KB, type:  APP, subtype: OTA_1
             spiffs : addr: 0x00290000, size:  1388.0 KB, type: DATA, subtype: SPIFFS
         zb_storage : addr: 0x003EB000, size:    16.0 KB, type: DATA, subtype: FAT
             zb_fct : addr: 0x003EF000, size:     4.0 KB, type: DATA, subtype: FAT
           coredump : addr: 0x003F0000, size:    64.0 KB, type: DATA, subtype: COREDUMP
------------------------------------------
Software Info:
------------------------------------------
  Compile Date/Time : Nov 19 2024 14:37:54
  Compile Host OS   : windows
  ESP-IDF Version   : v5.3.1-638-ga0f798cfc4-dirty
  Arduino Version   : 3.1.0
------------------------------------------
Board Info:
------------------------------------------
  Arduino Board     : ESP32C6_DEV
  Arduino Variant   : esp32c6
  Arduino FQBN      : esp32:esp32:esp32c6:UploadSpeed=921600,CDCOnBoot=cdc,CPUFreq=160,FlashFreq=80,FlashMode=qio,FlashSize=4M,PartitionScheme=zigbee,DebugLevel=verbose,EraseFlash=all,JTAGAdapter=default,ZigbeeMode=ed
============ Before Setup End ============
[  1339][I][esp32-hal-periman.c:141] perimanSetPinBus(): Pin 12 already has type USB_DM (38) with bus 0x4081155c
[  1340][I][esp32-hal-periman.c:141] perimanSetPinBus(): Pin 13 already has type USB_DP (39) with bus 0x4081155c
[  1340][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type GPIO (1) successfully set to 0x42003236
[  1341][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 9 successfully set to type GPIO (1) with bus 0xa
[... 142 lines cut ...]
    17 : UART_RX[0]
============ After Setup End =============
[  5250][I][ZigbeeCore.cpp:197] esp_zb_app_signal_handler(): Device started up in non factory-reset mode
[  5250][I][ZigbeeCore.cpp:210] esp_zb_app_signal_handler(): Device rebooted
[  5887][V][Zigbee_Temperature_Sensor_test.ino:45] temp_sensor_value_update(): Temperature sensor value: 25.00°C
[  5888][V][ZigbeeTempSensor.cpp:68] setTemperature(): Updating temperature sensor value...
[  5888][D][ZigbeeTempSensor.cpp:70] setTemperature(): Setting temperature to 2500
[  5972][V][ZigbeeHandlers.cpp:134] zb_cmd_default_resp_handler(): Received default response: from address(0x0), src_endpoint(1) to dst_endpoint(10), cluster(0x402) with status 0x0
[  6888][V][Zigbee_Temperature_Sensor_test.ino:45] temp_sensor_value_update(): Temperature sensor value: 26.00°C
[  6889][V][ZigbeeTempSensor.cpp:68] setTemperature(): Updating temperature sensor value...
[  6889][D][ZigbeeTempSensor.cpp:70] setTemperature(): Setting temperature to 2600
[  7000][V][ZigbeeHandlers.cpp:134] zb_cmd_default_resp_handler(): Received default response: from address(0x0), src_endpoint(1) to dst_endpoint(10), cluster(0x402) with status 0x0
[  7889][V][Zigbee_Temperature_Sensor_test.ino:45] temp_sensor_value_update(): Temperature sensor value: 26.00°C
[  7890][V][ZigbeeTempSensor.cpp:68] setTemperature(): Updating temperature sensor value...
[  7890][D][ZigbeeTempSensor.cpp:70] setTemperature(): Setting temperature to 2600
[  8890][V][Zigbee_Temperature_Sensor_test.ino:45] temp_sensor_value_update(): Temperature sensor value: 26.00°C
[  8891][V][ZigbeeTempSensor.cpp:68] setTemperature(): Updating temperature sensor value...
[  8891][D][ZigbeeTempSensor.cpp:70] setTemperature(): Setting temperature to 2600
[  9891][V][Zigbee_Temperature_Sensor_test.ino:45] temp_sensor_value_update(): Temperature sensor value: 27.00°C
[  9892][V][ZigbeeTempSensor.cpp:68] setTemperature(): Updating temperature sensor value...
[  9892][D][ZigbeeTempSensor.cpp:70] setTemperature(): Setting temperature to 2700
[  9958][V][ZigbeeHandlers.cpp:134] zb_cmd_default_resp_handler(): Received default response: from address(0x0), src_endpoint(1) to dst_endpoint(10), cluster(0x402) with status 0x0
[ 10892][V][Zigbee_Temperature_Sensor_test.ino:45] temp_sensor_value_update(): Temperature sensor value: 27.00°C
[ 10893][V][ZigbeeTempSensor.cpp:68] setTemperature(): Updating temperature sensor value...
```

P-R-O-C-H-Y, 2024-11-21:
Hi @michapr, yeah there is nothing more to see then the failure of the Zigbee stack:
`[  4724][E][ZigbeeCore.cpp:219] esp_zb_app_signal_handler(): Failed to initialize Zigbee stack (status: ESP_FAIL)`

I am sometimes getting this error too, so that's why I added a reboot if this initialization fails.
I can't tell why it's sometimes happening. I get this like 1/10 reboots, and I am not using any delays.

michapr, 2024-11-21:
ok,.. last result....
**WITH** esp_restart() in line 219 of ZigbeeCore.cpp:

After about 5 minutes rebooting the device was responding with other state (interesting why.... I think it could be a timing problem):

```
============ After Setup End =============
[  1334][V][Zigbee_Temperature_Sensor_test.ino:45] temp_sensor_value_update(): Temperature sensor value: 37.00°C
[  1335][V][ZigbeeTempSensor.cpp:68] setTemperature(): Updating temperature sensor value...
[  1335][D][ZigbeeTempSensor.cpp:70] setTemperature(): Setting temperature to 3700
[  2335][V][Zigbee_Temperature_Sensor_test.ino:45] temp_sensor_value_update(): Temperature sensor value: 37.00°C
[  2336][V][ZigbeeTempSensor.cpp:68] setTemperature(): Updating temperature sensor value...
[  2336][D][ZigbeeTempSensor.cpp:70] setTemperature(): Setting temperature to 3700
[  2689][I][ZigbeeCore.cpp:197] esp_zb_app_signal_handler(): Device started up in non factory-reset mode
[  2689][I][ZigbeeCore.cpp:210] esp_zb_app_signal_handler(): Device rebooted
[  3336][V][Zigbee_Temperature_Sensor_test.ino:45] temp_sensor_value_update(): Temperature sensor value: 37.00°C
[  3337][V][ZigbeeTempSensor.cpp:68] setTemperature(): Updating temperature sensor value...
[  3337][D][ZigbeeTempSensor.cpp:70] setTemperature(): Setting temperature to 3700
```

all cycles before:
```
============ After Setup End =============
[  1336][V][Zigbee_Temperature_Sensor_test.ino:45] temp_sensor_value_update(): Temperature sensor value: 37.00°C
[  1337][V][ZigbeeTempSensor.cpp:68] setTemperature(): Updating temperature sensor value...
[  1337][D][ZigbeeTempSensor.cpp:70] setTemperature(): Setting temperature to 3700
[  2337][V][Zigbee_Temperature_Sensor_test.ino:45] temp_sensor_value_update(): Temperature sensor value: 37.00°C
[  2338][V][ZigbeeTempSensor.cpp:68] setTemperature(): Updating temperature sensor value...
[  2338][D][ZigbeeTempSensor.cpp:70] setTemperature(): Setting temperature to 3700
[  2665][E][ZigbeeCore.cpp:219] esp_zb_app_signal_handler(): Failed to initialize Zigbee stack (status: ESP_FAIL)
```
After that device is reporting again.

So something is wrong there "anywhere" ;)

michapr, 2024-11-21:
Last for today.... an interesting fact:

If I set in the IDE the CPU Frequency to 80MHz - looks like all is working, without reboots after power off/on ....

P-R-O-C-H-Y, 2024-11-21:
> Last for today.... an interesting fact:
> 
> If I set in the IDE the CPU Frequency to 80MHz - looks like all is working, without reboots after power off/on ....

That is an interesting fact. Nice investigation. I will report all the info and also check if there is anything suspicious why that "fixed" the issue. Thank you again, great work.

P-R-O-C-H-Y, 2024-11-21:
You can add to the code to wait for Zigbee stack to be ready.
```
  // Waint until Zigbee stack is ready
  while (!Zigbee.isStarted()) {
    delay(100);
  }
```

michapr, 2024-11-22:
> You can add to the code to wait for Zigbee stack to be ready. But I think we have tired that and it did not help at all. But can you give it a try?
> 
> ```
>   // Waint until Zigbee stack is ready
>   while (!Zigbee.isStarted()) {
>     delay(100);
>   }
> ```

This will help a bit - device is rebooting sometimes (in my case up to 16 times) and then is responding. (additional delay did not change anything)
But I think it is not the best solution for a sleeping sensor sketch, here we need short live time ;)

> Have you tried to wait for ESP_ZB_BDB_SIGNAL_DEVICE_FIRST_START/REBOOT before setting any attributes? 

How this could be done?

me-no-dev, 2024-11-22:
> How this could be done?

More or less the wait for started does it.

michapr, 2024-11-22:
Once have seen one reboot only. On second try power off/on  I have switched of after 45 reboots....
(all at 160MHz CPU clock)

```
Rebooting...
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0xc (SW_CPU),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x4001975a
SPIWP:0xee
mode:DIO, clock div:2
load:0x40875720,len:0x1190
load:0x4086c110,len:0xd98
load:0x4086e610,len:0x2e84
entry 0x4086c110
[     0][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x42005d50
[    12][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x42005d14
[    23][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x42005cd8
[    34][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x42005c9c
[    46][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x42005d50
[    57][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x42005d14
[    68][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x42005cd8
[    80][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x42005c9c
[    91][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type USB_DM (38) successfully set to 0x420024e8
[   102][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type USB_DP (39) successfully set to 0x420024e8
[   114][V][ZigbeeEP.cpp:18] ZigbeeEP(): Endpoint: 10
[   129][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 17 successfully set to type UART_RX (2) with bus 0x4080f0fc
[   140][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 16 successfully set to type UART_TX (3) with bus 0x4080f0fc
[   151][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 12 successfully set to type USB_DM (38) with bus 0x40811560
[   161][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 13 successfully set to type USB_DP (39) with bus 0x40811560
=========== Before Setup Start ===========
Chip Info:
------------------------------------------
  Model             : ESP32-C6
  Package           : 1
  Revision          : 0.01
  Cores             : 1
  CPU Frequency     : 160 MHz
  XTAL Frequency    : 40 MHz
  Features Bitfield : 0x00000052
  Embedded Flash    : No
  Embedded PSRAM    : No
  2.4GHz WiFi       : Yes
  Classic BT        : No
  BT Low Energy     : Yes
  IEEE 802.15.4     : Yes
------------------------------------------
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   445324 B ( 434.9 KB)
  Free Bytes        :   408872 B ( 399.3 KB)
  Allocated Bytes   :    29436 B (  28.7 KB)
  Minimum Free Bytes:   408872 B ( 399.3 KB)
  Largest Free Block:   376820 B ( 368.0 KB)
------------------------------------------
Flash Info:
------------------------------------------
  Chip Size         :  4194304 B (4 MB)
  Block Size        :    65536 B (  64.0 KB)
  Sector Size       :     4096 B (   4.0 KB)
  Page Size         :      256 B (   0.2 KB)
  Bus Speed         : 40 MHz
  Bus Mode          : QIO
------------------------------------------
Partitions Info:
------------------------------------------
                nvs : addr: 0x00009000, size:    20.0 KB, type: DATA, subtype: NVS
            otadata : addr: 0x0000E000, size:     8.0 KB, type: DATA, subtype: OTA
               app0 : addr: 0x00010000, size:  1280.0 KB, type:  APP, subtype: OTA_0
               app1 : addr: 0x00150000, size:  1280.0 KB, type:  APP, subtype: OTA_1
[... 9 lines cut ...]
  ESP-IDF Version   : v5.3.1-638-ga0f798cfc4-dirty
  Arduino Version   : 3.1.0
------------------------------------------
Board Info:
------------------------------------------
  Arduino Board     : ESP32C6_DEV
  Arduino Variant   : esp32c6
  Arduino FQBN      : esp32:esp32:esp32c6:UploadSpeed=921600,CDCOnBoot=cdc,CPUFreq=160,FlashFreq=80,FlashMode=qio,FlashSize=4M,PartitionScheme=zigbee,DebugLevel=verbose,EraseFlash=all,JTAGAdapter=default,ZigbeeMode=ed
============ Before Setup End ============
[   284][I][esp32-hal-periman.c:141] perimanSetPinBus(): Pin 12 already has type USB_DM (38) with bus 0x40811560
[   285][I][esp32-hal-periman.c:141] perimanSetPinBus(): Pin 13 already has type USB_DP (39) with bus 0x40811560
[   285][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type GPIO (1) successfully set to 0x4200340c
[   286][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 9 successfully set to type GPIO (1) with bus 0xa
[   286][D][ZigbeeCore.cpp:82] addEndpoint(): Endpoint: 10, Device ID: 0x0302
Starting Zigbee...
[   287][D][ZigbeeCore.cpp:119] zigbeeInit(): Initialize Zigbee stack
[   335][D][ZigbeeCore.cpp:126] zigbeeInit(): Register all Zigbee EPs in list
[   336][I][ZigbeeCore.cpp:134] zigbeeInit(): List of registered Zigbee EPs:
[   336][I][ZigbeeCore.cpp:136] zigbeeInit(): Device type: Temperature Sensor, Endpoint: 10, Device ID: 0x0302
[   344][V][ZigbeeCore.cpp:322] esp_zb_app_signal_handler(): ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL
[   344][I][ZigbeeCore.cpp:207] esp_zb_app_signal_handler(): Zigbee stack initialized
[  2671][E][ZigbeeCore.cpp:239] esp_zb_app_signal_handler(): Failed to initialize Zigbee stack (status: ESP_FAIL)
Zigbee failed to start!
Rebooting...
```

P-R-O-C-H-Y, 2024-11-22:
Thank you for testing. I have one more task for you if you can help.
Can you please get the last commits from the PR I posted before?
Restart the Arduino IDE, and then you will be able to see new Zigbee model with DEBUG option.
Can you please select the End Device - Debug and post the output log when if fails to start? Thanks!
[image]

michapr, 2024-11-22:
cannot get the options in this time - looks like IDE is caching old options anywhere.
I see it for H2, but not for C6 :(
In board.txt the options are available... Already have reinstalledd the RC3 (all the time with restarting IDE) and modified the board.txt again - no success. Any idea?

P-R-O-C-H-Y, 2024-11-22:
@michapr Are you using the old IDE or DE 2.0? in the new 2.0 ice there is some problem with caches. For old Ideas it should work. Or you can delete the cached files for sure :)

michapr, 2024-11-22:
Just for record.... Clear the cache for fixing the board.txt issue:
```
Windows:
    C:\Users\<user name>\AppData\Roaming\arduino-ide\
Linux:
    ~/.config/arduino-ide/
macOS:
~/Library/Application Support/arduino-ide/
```

Here final is the log - at the end after some reboots the device is reporting.
(Temperature sketch)

```
=========== Before Setup Start ===========
Chip Info:
------------------------------------------
  Model             : ESP32-C6
  Package           : 1
  Revision          : 0.01
  Cores             : 1
  CPU Frequency     : 160 MHz
  XTAL Frequency    : 40 MHz
  Features Bitfield : 0x00000052
  Embedded Flash    : No
  Embedded PSRAM    : No
  2.4GHz WiFi       : Yes
  Classic BT        : No
  BT Low Energy     : Yes
  IEEE 802.15.4     : Yes
------------------------------------------
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   445260 B ( 434.8 KB)
  Free Bytes        :   413940 B ( 404.2 KB)
  Allocated Bytes   :    24352 B (  23.8 KB)
  Minimum Free Bytes:   409128 B ( 399.5 KB)
  Largest Free Block:   376820 B ( 368.0 KB)
------------------------------------------
Flash Info:
------------------------------------------
  Chip Size         :  4194304 B (4 MB)
  Block Size        :    65536 B (  64.0 KB)
  Sector Size       :     4096 B (   4.0 KB)
  Page Size         :      256 B (   0.2 KB)
  Bus Speed         : 40 MHz
  Bus Mode          : QIO
------------------------------------------
Partitions Info:
------------------------------------------
                nvs : addr: 0x00009000, size:    20.0 KB, type: DATA, subtype: NVS
            otadata : addr: 0x0000E000, size:     8.0 KB, type: DATA, subtype: OTA
               app0 : addr: 0x00010000, size:  1280.0 KB, type:  APP, subtype: OTA_0
               app1 : addr: 0x00150000, size:  1280.0 KB, type:  APP, subtype: OTA_1
             spiffs : addr: 0x00290000, size:  1388.0 KB, type: DATA, subtype: SPIFFS
         zb_storage : addr: 0x003EB000, size:    16.0 KB, type: DATA, subtype: FAT
             zb_fct : addr: 0x003EF000, size:     4.0 KB, type: DATA, subtype: FAT
           coredump : addr: 0x003F0000, size:    64.0 KB, type: DATA, subtype: COREDUMP
------------------------------------------
Software Info:
------------------------------------------
  Compile Date/Time : Nov 22 2024 20:19:57
  Compile Host OS   : windows
  ESP-IDF Version   : v5.3.1-638-ga0f798cfc4-dirty
  Arduino Version   : 3.1.0
------------------------------------------
Board Info:
------------------------------------------
  Arduino Board     : ESP32C6_DEV
  Arduino Variant   : esp32c6
[... 510 lines cut ...]
Updated temperature sensor value to 27.00°C
[  3804][V][ZigbeeTempSensor.cpp:68] setTemperature(): Updating temperature sensor value...
[  3805][D][ZigbeeTempSensor.cpp:70] setTemperature(): Setting temperature to 2700
[  4276][V][ZigbeeHandlers.cpp:134] zb_cmd_default_resp_handler(): Received default response: from address(0x0), src_endpoint(1) to dst_endpoint(10), cluster(0x402) with status 0x0
Updated temperature sensor value to 27.00°C
[  4805][V][ZigbeeTempSensor.cpp:68] setTemperature(): Updating temperature sensor value...
[  4806][D][ZigbeeTempSensor.cpp:70] setTemperature(): Setting temperature to 2700
Updated temperature sensor value to 27.00°C
[  5806][V][ZigbeeTempSensor.cpp:68] setTemperature(): Updating temperature sensor value...
[  5807][D][ZigbeeTempSensor.cpp:70] setTemperature(): Setting temperature to 2700
Updated temperature sensor value to 27.00°C
[  6806][V][ZigbeeTempSensor.cpp:68] setTemperature(): Updating temperature sensor value...
[  6807][D][ZigbeeTempSensor.cpp:70] setTemperature(): Setting temperature to 2700
Updated temperature sensor value to 27.00°C
[  7806][V][ZigbeeTempSensor.cpp:68] setTemperature(): Updating temperature sensor value...
[  7807][D][ZigbeeTempSensor.cpp:70] setTemperature(): Setting temperature to 2700
Updated temperature sensor value to 28.00°C
[  8807][V][ZigbeeTempSensor.cpp:68] setTemperature(): Updating temperature sensor value...
[  8808][D][ZigbeeTempSensor.cpp:70] setTemperature(): Setting temperature to 2800
[  8888][V][ZigbeeHandlers.cpp:134] zb_cmd_default_resp_handler(): Received default response: from address(0x0), src_endpoint(1) to dst_endpoint(10), cluster(0x402) with status 0x0
Updated temperature sensor value to 28.00°C
[  9808][V][ZigbeeTempSensor.cpp:68] setTemperature(): Updating temperature sensor value...
[  9809][D][ZigbeeTempSensor.cpp:70] setTemperature(): Setting temperature to 2800
Updated temperature sensor value to 28.00°C
```

P-R-O-C-H-Y, 2024-11-22:
Thank you for the log :)
I may also add the clear cache to the docs as reference in case its needed in future for anyone :)

I am getting exact same error in the log. I have reported that and waiting for a reply what can we do with that. Once I get any news I will update here. Btw, if you set the frequency to 80MHz the issue is gone right?

michapr, 2024-11-23:
I must say, that I have seen now reboots even at 80MHz :(
It must be something fundamental...

But I have also seen a reboot while first joining - have not noticed about this before:
`assert failed: vPortExitCritical port.c:632 (port_uxCriticalNesting[0] > 0)`
```
============ Before Setup End ============
[  2404][I][esp32-hal-periman.c:141] perimanSetPinBus(): Pin 12 already has type USB_DM (38) with bus 0x408115a0
[  2405][I][esp32-hal-periman.c:141] perimanSetPinBus(): Pin 13 already has type USB_DP (39) with bus 0x408115a0
[  2406][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type GPIO (1) successfully set to 0x42003400
[  2408][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 9 successfully set to type GPIO (1) with bus 0xa
ZBOSS: zcl/zcl_common.c:215   >> zb_zcl_get_attribute_size: attr_type 0x42 attr_value 0x408198e4
ZBOSS: zcl/zcl_common.c:348   << zb_zcl_get_attribute_size: result 10
ZBOSS: zcl/zcl_common.c:215   >> zb_zcl_get_attribute_size: attr_type 0x42 attr_value 0x40819900
ZBOSS: zcl/zcl_common.c:348   << zb_zcl_get_attribute_size: result 17
ZBOSS: zcl/zcl_common.c:215   >> zb_zcl_get_attribute_size: attr_type 0x29 attr_value 0x4081e29c
ZBOSS: zcl/zcl_common.c:348   << zb_zcl_get_attribute_size: result 2
ZBOSS: zcl/zcl_common.c:215   >> zb_zcl_get_attribute_size: attr_type 0x29 attr_value 0x4081e29e
ZBOSS: zcl/zcl_common.c:348   << zb_zcl_get_attribute_size: result 2
ZBOSS: zcl/zcl_common.c:215   >> zb_zcl_get_attribute_size: attr_type 0x21 attr_value 0x4081e29e
ZBOSS: zcl/zcl_common.c:348   << zb_zcl_get_attribute_size: result 2
[  2416][D][ZigbeeCore.cpp:82] addEndpoint(): Endpoint: 10, Device ID: 0x0302
Starting Zigbee...
[  2417][D][ZigbeeCore.cpp:119] zigbeeInit(): Initialize Zigbee stack
ZBOSS: common/zb_init_default.c:181   ED build
ZBOSS: common/zb_init_default.c:187   sizes: g_zb 7576 sched 224 bpool 16 nwk 912 aps 3040 addr 284 zdo 840
ZBOSS: common/zb_init_default.c:196   sec 2
ZBOSS: common/zb_init_default.c:198   zcl 768
ZBOSS: common/zb_init_default.c:201   zll 960
ZBOSS: common/zb_init_default.c:207   nvram 140
ZBOSS: common/zb_init_default.c:210   buttons 200
ZBOSS: common/zb_init_default.c:212   err_ind 4
ZBOSS: common/zb_init_default.c:218   scheduler q size 80
ZBOSS: common/zb_init_default.c:221   g_mac 816 g_imac 28
ZBOSS: common/zb_init_default.c:225   Configurable mem build, use ZBOSS lib defaults = 0
ZBOSS: common/zb_init_default.c:231   ZB_IOBUF_POOL_SIZE 80 ZB_NWK_IN_Q_SIZE 40 ZB_MAC_PENDING_QUEUE_SIZE 20 ZB_APS_DST_BINDING_TABLE_SIZE 16 ZB_APS_BIND_TRANS_TABLE_SIZE 20 ZB_N_APS_RETRANS_ENTRIES 27
ZBOSS: common/zb_init_default.c:261   ZB_N_APS_KEY_PAIR_ARR_MAX_SIZE 8 ZB_IEEE_ADDR_TABLE_SIZE 64 ZB_NEIGHBOR_TABLE_SIZE 64 ZB_NWK_ROUTING_TABLE_SIZE 0 ZB_APS_DUPS_TABLE_SIZE 32
[  2477][D][ZigbeeCore.cpp:126] zigbeeInit(): Register all Zigbee EPs in list
ZBOSS: zdo/af_descriptor.c:357   ep_count 1
[  2480][I][ZigbeeCore.cpp:134] zigbeeInit(): List of registered Zigbee EPs:
[  2481][I][ZigbeeCore.cpp:136] zigbeeInit(): Device type: Temperature Sensor, Endpoint: 10, Device ID: 0x0302
ZBOSS: zdo/zdo_app_prod_conf.c:144   no production config block found
ZBOSS: zdo/zdo_app_prod_conf.c:781   no valid production configuration found, signaling to application
[  2648][V][ZigbeeCore.cpp:322] esp_zb_app_signal_handler(): ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL
[  2650][I][ZigbeeCore.cpp:207] esp_zb_app_signal_handler(): Zigbee stack initialized
ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:349   dev type 2, joined 0, ext_pan_id 0, authenticated 0, tclk_valid 1
ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:549   COMMISSIONING_STOP: app signal 5 comm status 0
[  2656][I][ZigbeeCore.cpp:213] esp_zb_app_signal_handler(): Device started up in  factory-reset mode
[  2656][I][ZigbeeCore.cpp:220] esp_zb_app_signal_handler(): Start network steering
ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:349   dev type 2, joined 0, ext_pan_id 0, authenticated 0, tclk_valid 1
Zigbee started successfully!
Connecting to network
.......................ZBOSS: nwk/nwk_join.c:186   low lqi 10
ZBOSS: nwk/nwk_join.c:186   low lqi 0
ZBOSS: nwk/nwk_join.c:186   low lqi 0
ZBOSS: nwk/nwk_join.c:186   low lqi 0
ZBOSS: nwk/nwk_join.c:186   low lqi 0
ZBOSS: nwk/nwk_join.c:251   No dev for join
ZBOSS: nwk/nwk_join.c:186   low lqi 10
ZBOSS: nwk/nwk_join.c:186   low lqi 0
ZBOSS: nwk/nwk_join.c:186   low lqi 0
ZBOSS: nwk/nwk_join.c:186   low lqi 0
ZBOSS: nwk/nwk_join.c:186   low lqi 0
ZBOSS: nwk/nwk_join.c:251   No dev for join
ZBOSS: nwk/nwk_join.c:186   low lqi 10
ZBOSS: nwk/nwk_join.c:186   low lqi 0
ZBOSS: nwk/nwk_join.c:186   low lqi 0
ZBOSS: nwk/nwk_join.c:186   low lqi 0
ZBOSS: nwk/nwk_join.c:186   low lqi 0
ZBOSS: nwk/nwk_join.c:251   No dev for join
[... 405 lines cut ...]
[  5863][V][ZigbeeTempSensor.cpp:68] setTemperature(): Updating temperature sensor value...
[  5864][D][ZigbeeTempSensor.cpp:70] setTemperature(): Setting temperature to 2200
[  6319][V][ZigbeeCore.cpp:322] esp_zb_app_signal_handler(): ZDO signal: NLME Status Indication (0x32), status: ESP_OK
Updated temperature sensor value to 22.00°C
[  6864][V][ZigbeeTempSensor.cpp:68] setTemperature(): Updating temperature sensor value...
[  6865][D][ZigbeeTempSensor.cpp:70] setTemperature(): Setting temperature to 2200
Updated temperature sensor value to 22.00°C
[  7866][V][ZigbeeTempSensor.cpp:68] setTemperature(): Updating temperature sensor value...
[... cut ...]
[ 10871][V][ZigbeeTempSensor.cpp:68] setTemperature(): Updating temperature sensor value...
[ 10872][D][ZigbeeTempSensor.cpp:70] setTemperature(): Setting temperature to 2300
Updated temperature sensor value to 23.00°C
[ 11873][V][ZigbeeTempSensor.cpp:68] setTemperature(): Updating temperature sensor value...
[ 11874][D][ZigbeeTempSensor.cpp:70] setTemperature(): Setting temperature to 2300
[ 12447][V][ZigbeeCore.cpp:322] esp_zb_app_signal_handler(): ZDO signal: NLME Status Indication (0x32), status: ESP_OK
```

michapr, 2024-11-23:
Additional....
Have seen that device will stop reporting values after some time (without any reboot)
```
[3756824][V][ZigbeeTempSensor.cpp:68] setTemperature(): Updating temperature sensor value...
[3756825][D][ZigbeeTempSensor.cpp:70] setTemperature(): Setting temperature to 3400
Updated temperature sensor value to 33.00°C
[3757825][V][ZigbeeTempSensor.cpp:68] setTemperature(): Updating temperature sensor value...
[3757826][D][ZigbeeTempSensor.cpp:70] setTemperature(): Setting temperature to 3300
[3757891][V][ZigbeeHandlers.cpp:134] zb_cmd_default_resp_handler(): Received default response: from address(0x0), src_endpoint(1) to dst_endpoint(10), cluster(0x402) with status 0x0
Updated temperature sensor value to 34.00°C
[3758827][V][ZigbeeTempSensor.cpp:68] setTemperature(): Updating temperature sensor value...
[3758828][D][ZigbeeTempSensor.cpp:70] setTemperature(): Setting temperature to 3400
[3758934][V][ZigbeeHandlers.cpp:134] zb_cmd_default_resp_handler(): Received default response: from address(0x0), src_endpoint(1) to dst_endpoint(10), cluster(0x402) with status 0x0
Updated temperature sensor value to 33.00°C
[3759829][V][ZigbeeTempSensor.cpp:68] setTemperature(): Updating temperature sensor value...
[3759830][D][ZigbeeTempSensor.cpp:70] setTemperature(): Setting temperature to 3300
[3759989][V][ZigbeeHandlers.cpp:134] zb_cmd_default_resp_handler(): Received default response: from address(0x0), src_endpoint(1) to dst_endpoint(10), cluster(0x402) with status 0x0
Updated temperature sensor value to 34.00°C
[3760831][V][ZigbeeTempSensor.cpp:68] setTemperature(): Updating temperature sensor value...
[3760832][D][ZigbeeTempSensor.cpp:70] setTemperature(): Setting temperature to 3400
[3761035][V][ZigbeeHandlers.cpp:134] zb_cmd_default_resp_handler(): Received default response: from address(0x0), src_endpoint(1) to dst_endpoint(10), cluster(0x402) with status 0x0
Updated temperature sensor value to 34.00°C
[3761833][V][ZigbeeTempSensor.cpp:68] setTemperature(): Updating temperature sensor value...
[3761834][D][ZigbeeTempSensor.cpp:70] setTemperature(): Setting temperature to 3400
Updated temperature sensor value to 34.00°C
...
...
[3875986][D][ZigbeeTempSensor.cpp:70] setTemperature(): Setting temperature to 3500
Updated temperature sensor value to 35.00°C
[3876986][V][ZigbeeTempSensor.cpp:68] setTemperature(): Updating temperature sensor value...
[3876987][D][ZigbeeTempSensor.cpp:70] setTemperature(): Setting temperature to 3500
Updated temperature sensor value to 34.00°C
[3877988][V][ZigbeeTempSensor.cpp:68] setTemperature(): Updating temperature sensor value...
[3877989][D][ZigbeeTempSensor.cpp:70] setTemperature(): Setting temperature to 3400
Updated temperature sensor value to 34.00°C
[3878989][V][ZigbeeTempSensor.cpp:68] setTemperature(): Updating temperature sensor value...
[3878990][D][ZigbeeTempSensor.cpp:70] setTemperature(): Setting temperature to 3400
[3879979][V][ZigbeeCore.cpp:322] esp_zb_app_signal_handler(): ZDO signal: ZDO Device Unavailable (0x3c), status: ESP_OK
Updated temperature sensor value to 34.00°C
[3879991][V][ZigbeeTempSensor.cpp:68] setTemperature(): Updating temperature sensor value...
[3879992][D][ZigbeeTempSensor.cpp:70] setTemperature(): Setting temperature to 3400
Updated temperature sensor value to 35.00°C
[3880993][V][ZigbeeTempSensor.cpp:68] setTemperature(): Updating temperature sensor value...
[3880994][D][ZigbeeTempSensor.cpp:70] setTemperature(): Setting temperature to 3500
Updated temperature sensor value to 34.00°C
[3881995][V][ZigbeeTempSensor.cpp:68] setTemperature(): Updating temperature sensor value...
[3881996][D][ZigbeeTempSensor.cpp:70] setTemperature(): Setting temperature to 3400
Updated temperature sensor value to 34.00°C
[3882996][V][ZigbeeTempSensor.cpp:68] setTemperature(): Updating temperature sensor value...
[3882997][D][ZigbeeTempSensor.cpp:70] setTemperature(): Setting temperature to 3400
```

michapr, 2024-11-23:
I have "fixed" it for me for further testing
`if (sig_type == 60) ESP.restart();`
in ZigbeeCore.cpp after line 322

ggtimtom, 2024-11-24:
I have a similar issue with H2:

this happens in:
  // Report temperature and humidity values
  zbTempSensor.reportTemperature();
  zbTempSensor.reportHumidity();

Debug:
============ After Setup End =============
[ 18758][V][ZigbeeTempSensor.cpp:68] setTemperature(): Updating temperature sensor value...
[ 18767][D][ZigbeeTempSensor.cpp:70] setTemperature(): Setting temperature to 2200
[ 18774][V][ZigbeeTempSensor.cpp:107] setHumidity(): Updating humidity sensor value...
[ 18783][D][ZigbeeTempSensor.cpp:109] setHumidity(): Setting humidity to 2200
Zigbee stack assertion failed zcl/zcl_general_commands.c:612

abort() was called at PC 0x42010e0d on core 0
Core  0 register dump:
MEPC    : 0x40801588  RA      : 0x408050ca  SP      : 0x4081a170  GP      : 0x4080c074  
TP      : 0x4081a330  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130  
S0/FP   : 0x4081a19c  S1      : 0x4081a19c  A0      : 0x4081a1a8  A1      : 0x4081a18a  
A2      : 0x00000000  A3      : 0x4081a1d5  A4      : 0x00000001  A5      : 0x40813000  
A6      : 0x00000000  A7      : 0x76757473  S2      : 0x00000002  S3      : 0x42070161  
S4      : 0x00000000  S5      : 0x4081ead0  S6      : 0x00000000  S7      : 0x00000000  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938

Cage-ctr, 2024-11-24:
esp32c6 the same error in the log after Zigbee stack assertion zbTempSensor.reportTemperature();
zbTempSensor.reportHumidity(); 
error :
failed zcl/zcl_general_commands.c:612

P-R-O-C-H-Y, 2024-11-25:
Hi @michapr, can you give it a try with latest changes in the PR https://github.com/espressif/arduino-esp32/pull/10651? Thanks!

Nawor3565, 2024-11-25:
Hi @P-R-O-C-H-Y, I tried the fix in #10651 but it doesn't seem to have made a difference. It looks like it just fails to connect 4 times now before rebooting.

`[   732][D][ZigbeeCore.cpp:129] zigbeeInit(): Register all Zigbee EPs in list
ZBOSS: zdo/af_descriptor.c:357   ep_count 1
[   736][I][ZigbeeCore.cpp:137] zigbeeInit(): List of registered Zigbee EPs:
[   736][I][ZigbeeCore.cpp:139] zigbeeInit(): Device type: Temperature Sensor, Endpoint: 10, Device ID: 0x0302
ZBOSS: common/zb_nvram.c:3190   ds_ver == 2
ZBOSS: zdo/zdo_app_prod_conf.c:144   no production config block found
ZBOSS: zdo/zdo_app_prod_conf.c:781   no valid production configuration found, signaling to application
[   760][V][ZigbeeCore.cpp:332] esp_zb_app_signal_handler(): ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL
[   762][I][ZigbeeCore.cpp:218] esp_zb_app_signal_handler(): Zigbee stack initialized
ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:349   dev type 2, joined 1, ext_pan_id 1, authenticated 1, tclk_valid 1
ZBOSS: mac/mac.c:1039   Oops - error beacon payload is too small buf_sz 15 - drop it
ZBOSS: nwk/nwk_join.c:629   Rejoin failure: no dev with xpanid bf:de:b3:e8:e6:a6:c9:f9 or can't choose best parent
ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:549   COMMISSIONING_STOP: app signal 6 comm status 3
[ 10079][W][ZigbeeCore.cpp:249] esp_zb_app_signal_handler(): Commissioning failed, trying again...
ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:349   dev type 2, joined 1, ext_pan_id 1, authenticated 1, tclk_valid 1
ZBOSS: mac/mac.c:1039   Oops - error beacon payload is too small buf_sz 13 - drop it
ZBOSS: nwk/nwk_join.c:629   Rejoin failure: no dev with xpanid bf:de:b3:e8:e6:a6:c9:f9 or can't choose best parent
ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:549   COMMISSIONING_STOP: app signal 6 comm status 3
[ 19896][W][ZigbeeCore.cpp:249] esp_zb_app_signal_handler(): Commissioning failed, trying again...
ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:349   dev type 2, joined 1, ext_pan_id 1, authenticated 1, tclk_valid 1
ZBOSS: mac/mac.c:1039   Oops - error beacon payload is too small buf_sz 15 - drop it
ZBOSS: nwk/nwk_join.c:629   Rejoin failure: no dev with xpanid bf:de:b3:e8:e6:a6:c9:f9 or can't choose best parent
ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:549   COMMISSIONING_STOP: app signal 6 comm status 3
[ 29713][W][ZigbeeCore.cpp:249] esp_zb_app_signal_handler(): Commissioning failed, trying again...
ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:349   dev type 2, joined 1, ext_pan_id 1, authenticated 1, tclk_valid 1
[ 30757][E][ZigbeeCore.cpp:43] begin(): ZigbeeCore begin timeout
Zigbee failed to start!
Rebooting...`

michapr, 2024-11-25:
@P-R-O-C-H-Y : I have seen the same,... additional before rebooting:
```
ZBOSS: zdo/zdo_app_prod_conf.c:144   no production config block found
ZBOSS: zdo/zdo_app_prod_conf.c:781   no valid production configuration found, signaling to application
[   431][V][ZigbeeCore.cpp:332] esp_zb_app_signal_handler(): ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL
[   432][I][ZigbeeCore.cpp:218] esp_zb_app_signal_handler(): Zigbee stack initialized
ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:349   dev type 2, joined 1, ext_pan_id 1, authenticated 1, tclk_valid 1
ZBOSS: nwk/nwk_join.c:186   low lqi 10
ZBOSS: nwk/nwk_join.c:186   low lqi 0
ZBOSS: nwk/nwk_join.c:186   low lqi 0
ZBOSS: nwk/nwk_join.c:186   low lqi 10
ZBOSS: zdo/zdo_app_join.c:771   CONGRATULATIONS! joined status 0, addr 0x5865, iface_id 0, ch 11, rejoin 1
ZBOSS: zdo/zdo_app_join.c:773   xpanid 3a:63:5d:94:1c:56:10:7e
[  4767][V][ZigbeeCore.cpp:332] esp_zb_app_signal_handler(): ZDO signal: NLME Status Indication (0x32), status: ESP_OK
ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:549   COMMISSIONING_STOP: app signal 6 comm status 14
[  4768][W][ZigbeeCore.cpp:249] esp_zb_app_signal_handler(): Commissioning failed, trying again...
ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:349   dev type 2, joined 1, ext_pan_id 1, authenticated 1, tclk_valid 1
E (5746) ESP_ZIGBEE_CORE: In BDB commissioning, an error occurred (for example: the device has already been running)
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x420009b2
file: "C:\Users\micha\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.0-RC3\libraries\Zigbee\src\ZigbeeCore.cpp" line 204
func: void bdb_start_top_level_commissioning_cb(uint8_t)
expression: esp_zb_bdb_start_to
abort() was called at PC 0x40808033 on core 0
Core  0 register dump:
MEPC    : 0x408015c8  RA      : 0x4080803e  SP      : 0x408243b0  GP      : 0x4080f864
```
For me device is still working a little bit better with 80MHz (less reboots...)

P-R-O-C-H-Y, 2024-11-25:
> Hi @P-R-O-C-H-Y, I tried the fix in #10651 but it doesn't seem to have made a difference. It looks like it just fails to connect 4 times now before rebooting.
>
> ```
> [   732][D][ZigbeeCore.cpp:129] [...]

@Nawor3565 I am seeing in your log something suspicious, which is:
`ZBOSS: mac/mac.c:1039   Oops - error beacon payload is too small buf_sz 13 - drop it`
I will take a look on this. Also I don't know how far is your Coordinator or parent for your device.
Can you try to place them close and try again? :)

me-no-dev, 2024-11-25:
can you please post what zigbee adapters you are using?

P-R-O-C-H-Y, 2024-11-25:
> @P-R-O-C-H-Y : I have seen the same,... additional before rebooting:
>
> ```
> ZBOSS: zdo/zdo_app_prod_conf.c:144   no production config block found
> ZBOSS: zdo/zdo_app_prod_conf.c:781   no valid [...]

@michapr what is the log from? I see it crashed. Can you use normal CPU speed?

michapr, 2024-11-25:
> @michapr what is the log from? I see it crashed. Can you use normal CPU speed?
> I never got such an error like this.

This is after restart (power off/on) with 160Mhz

Nawor3565, 2024-11-25:
@me-no-dev I'm using a Nabu Casa SkyConnect v1.0. I have a different Zigbee adapter I can try, but I think it uses the same chipset.

@P-R-O-C-H-Y I will try experimenting with the channel mask later today. While doing these tests, the coordinator was only a few feet from the ESP, although I will also try placing it directly next to each other. It seems weird that the initial connection would work fine at that distance though, and only the re-connect would fail.

michapr, 2024-11-25:
@me-no-dev /dev/serial/by-id/usb-Silicon_Labs_CP2105_Dual_USB_to_UART_Bridge_Controller_012BCB52-if01-port0

P-R-O-C-H-Y, 2024-11-25:
> @me-no-dev /dev/serial/by-id/usb-Silicon_Labs_CP2105_Dual_USB_to_UART_Bridge_Controller_012BCB52-if01-port0
Do you know the Manufacturer and model of the Zigbee adaptor? From this you can only see the USB-UART controller.

michapr, 2024-11-25:
@P-R-O-C-H-Y - I thought the chipset is important for you ;)
It's the Z-Wave.Me USB adapter Z-Wave/Zigbee (2 controllers inside) Z-Station

With modified sketch have 2 results - one without problems restarting device (not only once) - and with problem - see attached log (without changes in sketch, only power off/on)

have added only ` Zigbee.setScanDuration(1);`
```
[  1658][D][ZigbeeCore.cpp:129] zigbeeInit(): Register all Zigbee EPs in list
ZBOSS: zdo/af_descriptor.c:357   ep_count 1
[  1660][I][ZigbeeCore.cpp:137] zigbeeInit(): List of registered Zigbee EPs:
[  1660][I][ZigbeeCore.cpp:139] zigbeeInit(): Device type: Temperature Sensor, Endpoint: 10, Device ID: 0x0302
ZBOSS: common/zb_nvram.c:3190   ds_ver == 2
ZBOSS: zdo/zdo_app_prod_conf.c:144   no production config block found
ZBOSS: zdo/zdo_app_prod_conf.c:781   no valid production configuration found, signaling to application
[  1668][V][ZigbeeCore.cpp:332] esp_zb_app_signal_handler(): ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL
[  1669][I][ZigbeeCore.cpp:218] esp_zb_app_signal_handler(): Zigbee stack initialized
ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:349   dev type 2, joined 1, ext_pan_id 1, authenticated 1, tclk_valid 1
ZBOSS: nwk/nwk_join.c:186   low lqi 0
ZBOSS: nwk/nwk_join.c:629   Rejoin failure: no dev with xpanid 3a:63:5d:94:1c:56:10:7e or can't choose best parent
ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:549   COMMISSIONING_STOP: app signal 6 comm status 3
[  2526][W][ZigbeeCore.cpp:249] esp_zb_app_signal_handler(): Commissioning failed, trying again...
ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:349   dev type 2, joined 1, ext_pan_id 1, authenticated 1, tclk_valid 1
ZBOSS: nwk/nwk_join.c:186   low lqi 5
ZBOSS: nwk/nwk_join.c:186   low lqi 0
ZBOSS: nwk/nwk_join.c:629   Rejoin failure: no dev with xpanid 3a:63:5d:94:1c:56:10:7e or can't choose best parent
ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:549   COMMISSIONING_STOP: app signal 6 comm status 3
[  3884][W][ZigbeeCore.cpp:249] esp_zb_app_signal_handler(): Commissioning failed, trying again...
ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:349   dev type 2, joined 1, ext_pan_id 1, authenticated 1, tclk_valid 1
ZBOSS: nwk/nwk_join.c:186   low lqi 5
ZBOSS: nwk/nwk_join.c:186   low lqi 0
ZBOSS: nwk/nwk_join.c:629   Rejoin failure: no dev with xpanid 3a:63:5d:94:1c:56:10:7e or can't choose best parent
ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:549   COMMISSIONING_STOP: app signal 6 comm status 3
[  5242][W][ZigbeeCore.cpp:249] esp_zb_app_signal_handler(): Commissioning failed, trying again...
ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:349   dev type 2, joined 1, ext_pan_id 1, authenticated 1, tclk_valid 1
ZBOSS: nwk/nwk_join.c:186   low lqi 0
ZBOSS: nwk/nwk_join.c:186   low lqi 10
ZBOSS: nwk/nwk_join.c:629   Rejoin failure: no dev with xpanid 3a:63:5d:94:1c:56:10:7e or can't choose best parent
ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:549   COMMISSIONING_STOP: app signal 6 comm status 3
[  6600][W][ZigbeeCore.cpp:249] esp_zb_app_signal_handler(): Commissioning failed, trying again...
ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:349   dev type 2, joined 1, ext_pan_id 1, authenticated 1, tclk_valid 1
ZBOSS: nwk/nwk_join.c:186   low lqi 0
ZBOSS: nwk/nwk_join.c:186   low lqi 0
ZBOSS: nwk/nwk_join.c:186   low lqi 5
ZBOSS: nwk/nwk_join.c:629   Rejoin failure: no dev with xpanid 3a:63:5d:94:1c:56:10:7e or can't choose best parent
ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:549   COMMISSIONING_STOP: app signal 6 comm status 3
[  7958][W][ZigbeeCore.cpp:249] esp_zb_app_signal_handler(): Commissioning failed, trying again...
ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:349   dev type 2, joined 1, ext_pan_id 1, authenticated 1, tclk_valid 1
ZBOSS: nwk/nwk_join.c:186   low lqi 0
ZBOSS: nwk/nwk_join.c:186   low lqi 20
ZBOSS: nwk/nwk_join.c:629   Rejoin failure: no dev with xpanid 3a:63:5d:94:1c:56:10:7e or can't choose best parent
ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:549   COMMISSIONING_STOP: app signal 6 comm status 3
[  9316][W][ZigbeeCore.cpp:249] esp_zb_app_signal_handler(): Commissioning failed, trying again...
ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:349   dev type 2, joined 1, ext_pan_id 1, authenticated 1, tclk_valid 1
ZBOSS: nwk/nwk_join.c:186   low lqi 0
ZBOSS: nwk/nwk_join.c:186   low lqi 5
ZBOSS: nwk/nwk_join.c:629   Rejoin failure: no dev with xpanid 3a:63:5d:94:1c:56:10:7e or can't choose best parent
ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:549   COMMISSIONING_STOP: app signal 6 comm status 3
[ 10673][W][ZigbeeCore.cpp:249] esp_zb_app_signal_handler(): Commissioning failed, trying again...
ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:349   dev type 2, joined 1, ext_pan_id 1, authenticated 1, tclk_valid 1
ZBOSS: nwk/nwk_join.c:186   low lqi 0
ZBOSS: nwk/nwk_join.c:629   Rejoin failure: no dev with xpanid 3a:63:5d:94:1c:56:10:7e or can't choose best parent
ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:549   COMMISSIONING_STOP: app signal 6 comm status 3
[ 12031][W][ZigbeeCore.cpp:249] esp_zb_app_signal_handler(): Commissioning failed, trying again...
ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:349   dev type 2, joined 1, ext_pan_id 1, authenticated 1, tclk_valid 1
ZBOSS: nwk/nwk_join.c:186   low lqi 0
ZBOSS: nwk/nwk_join.c:186   low lqi 20
ZBOSS: nwk/nwk_join.c:629   Rejoin failure: no dev with xpanid 3a:63:5d:94:1c:56:10:7e or can't choose best parent
[... cut ...]
  GPIO : BUS_TYPE[bus/unit][chan]
  --------------------------------------
     9 : GPIO
    12 : USB_DM
    13 : USB_DP
    16 : UART_TX[0]
    17 : UART_RX[0]
============ After Setup End =============
[ 30130][V][ZigbeeHandlers.cpp:134] zb_cmd_default_resp_handler(): Received default response: from address(0x0), src_endpoint(1) to dst_endpoint(10), cluster(0x402) with status 0x0
Updated temperature sensor value to 29.00°C
```
[zigbee_esp32c6.log](https://github.com/user-attachments/files/17906247/zigbee_esp32c6.log)

michapr, 2024-11-25:
@Nawor3565 I have in this moment 14 other devices, 5 Temp/hum sensors, 4 thermostats (TRV), plugs and 2 repeaters
Using latest Zigbee2MQTT, and no,... no other problems with devices.

Nawor3565, 2024-11-25:
@me-no-dev Yup, both adapters use the EFR32MG21. Just tried both of them, no difference in behavior between the two. I only have a single other temp sensor on my network, which seems to work fine.

@P-R-O-C-H-Y I just tried using a scan duration of 1 and 4, with the ESP a few inches from the coordinator. The shorter distance seemed to get rid of the `error beacon payload is too small` messages, but it didn't help it to reconnect. I also tried using Zigbee2MQTT just to see if maybe Home Assistant was doing something weird, but it also made no difference. Here's the new logs for reference, although not much is different.

Out of curiosity, since it seems like it's reconnecting just fine for both of you, are your adapters using a different chipset?

```
[   436][D][ZigbeeCore.cpp:129] zigbeeInit(): Register all Zigbee EPs in list
ZBOSS: zdo/af_descriptor.c:357   ep_count 1
[   439][I][ZigbeeCore.cpp:137] zigbeeInit(): List of registered Zigbee EPs:
[   440][I][ZigbeeCore.cpp:139] zigbeeInit(): Device type: Temperature Sensor, Endpoint: 10, Device ID: 0x0302
ZBOSS: common/zb_nvram.c:3190   ds_ver == 2
ZBOSS: zdo/zdo_app_prod_conf.c:144   no production config block found
ZBOSS: zdo/zdo_app_prod_conf.c:781   no valid production configuration found, signaling to application
[   454][V][ZigbeeCore.cpp:332] esp_zb_app_signal_handler(): ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL
[   455][I][ZigbeeCore.cpp:218] esp_zb_app_signal_handler(): Zigbee stack initialized
ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:349   dev type 2, joined 1, ext_pan_id 1, authenticated 1, tclk_valid 1
ZBOSS: nwk/nwk_join.c:629   Rejoin failure: no dev with xpanid bf:de:b3:e8:e6:a6:c9:f9 or can't choose best parent
ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:549   COMMISSIONING_STOP: app signal 6 comm status 3
[  1317][W][ZigbeeCore.cpp:249] esp_zb_app_signal_handler(): Commissioning failed, trying again...
ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:349   dev type 2, joined 1, ext_pan_id 1, authenticated 1, tclk_valid 1
ZBOSS: nwk/nwk_join.c:629   Rejoin failure: no dev with xpanid bf:de:b3:e8:e6:a6:c9:f9 or can't choose best parent
ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:549   COMMISSIONING_STOP: app signal 6 comm status 3
[  2679][W][ZigbeeCore.cpp:249] esp_zb_app_signal_handler(): Commissioning failed, trying again...
ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:349   dev type 2, joined 1, ext_pan_id 1, authenticated 1, tclk_valid 1
ZBOSS: nwk/nwk_join.c:629   Rejoin failure: no dev with xpanid bf:de:b3:e8:e6:a6:c9:f9 or can't choose best parent
ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:549   COMMISSIONING_STOP: app signal 6 comm status 3
[  9057][W][ZigbeeCore.cpp:249] esp_zb_app_signal_handler(): Commissioning failed, trying again...
ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:349   dev type 2, joined 1, ext_pan_id 1, authenticated 1, tclk_valid 1
ZBOSS: nwk/nwk_join.c:629   Rejoin failure: no dev with xpanid bf:de:b3:e8:e6:a6:c9:f9 or can't choose best parent
ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:549   COMMISSIONING_STOP: app signal 6 comm status 3
[ 15433][W][ZigbeeCore.cpp:249] esp_zb_app_signal_handler(): Commissioning failed, trying again...
ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:349   dev type 2, joined 1, ext_pan_id 1, authenticated 1, tclk_valid 1
ZBOSS: nwk/nwk_join.c:629   Rejoin failure: no dev with xpanid bf:de:b3:e8:e6:a6:c9:f9 or can't choose best parent
ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:549   COMMISSIONING_STOP: app signal 6 comm status 3
[ 21809][W][ZigbeeCore.cpp:249] esp_zb_app_signal_handler(): Commissioning failed, trying again...
ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:349   dev type 2, joined 1, ext_pan_id 1, authenticated 1, tclk_valid 1
ZBOSS: nwk/nwk_join.c:629   Rejoin failure: no dev with xpanid bf:de:b3:e8:e6:a6:c9:f9 or can't choose best parent
ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:549   COMMISSIONING_STOP: app signal 6 comm status 3
[ 28185][W][ZigbeeCore.cpp:249] esp_zb_app_signal_handler(): Commissioning failed, trying again...
ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:349   dev type 2, joined 1, ext_pan_id 1, authenticated 1, tclk_valid 1
ZBOSS: nwk/nwk_join.c:629   Rejoin failure: no dev with xpanid bf:de:b3:e8:e6:a6:c9:f9 or can't choose best parent
ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:549   COMMISSIONING_STOP: app signal 6 comm status 3
[ 29546][W][ZigbeeCore.cpp:249] esp_zb_app_signal_handler(): Commissioning failed, trying again...
ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:349   dev type 2, joined 1, ext_pan_id 1, authenticated 1, tclk_valid 1
[ 30451][E][ZigbeeCore.cpp:43] begin(): ZigbeeCore begin timeout
Zigbee failed to start!
Rebooting...
```

P-R-O-C-H-Y, 2024-11-25:
@Nawor3565 Can you try to increase the TimeOut to see if that even connects anytime in your case?
Just change the `#define ZB_INIT_TIMEOUT 30000  // 30 seconds` at the top of the `ZigbeeCore.cpp` file.

I am using the Sonoff Zigbee USB Dongle Plus, and in HomeAssistant I use the direct Zigbee HA integration.

michapr, 2024-11-25:
@Nawor3565 - can you try to give the ESP32 a "better place" - near the gateway? 
I have changed the place - so that I have a lqi about 160 - 180 , now I have no problems more.
Before it was 135-145 (all values from Z2M) 
It could of course be a coincidence.... I will continue to watch it tomorrow ....
(all my other sensors have a lqi between 65 and 130....)

Nawor3565, 2024-11-25:
@P-R-O-C-H-Y Just tried increasing it to 5 minutes, still no dice. I even tried turning off my WiFi router completely just in case it was interfering. 

@michapr According to Home Assistant, my LQI on initial connection is always around 250. Did you swap to the Sonoff Dongle Plus? From what I can tell, there's a "P" version and an "E" version of that adapter, and the P version uses the Ti CC2652P chipset instead of the SI chipset. So if you changed to a different chipset and it started working, that's probably at the root of the problem.

P-R-O-C-H-Y, 2024-11-25:
@Nawor3565 Thanks for testing. To fulfill the info, my son off is "E" version.

michapr, 2024-11-25:
@Nawor3565 - no, have not switched in this time.
My gateway is about 3 meters away only... in same room, never had problems with any devices.
And - first join is working, it is only the rejoin, so I do not think that lqi difference should make sense.... It was only interesting for me.

P-R-O-C-H-Y, 2024-11-25:
One more thing you can test right now is to add `esp_zb_secur_network_min_join_lqi_set(0);` to set the minimum allowed joining LQI (default is 32) https://github.com/espressif/arduino-esp32/blob/0f3191e34f9cd0a496b3add66006a21326103559/libraries/Zigbee/src/ZigbeeCore.cpp#L95 here before   `ESP_ERROR_CHECK(esp_zb_start(false));`

Can you give it a shot? :) @Nawor3565 @michapr

me-no-dev, 2024-11-25:
please note that @SuGlider merged the PR by mistake, so you can switch to the `release/v3.1.x` branch for further testing

Nawor3565, 2024-11-25:
@P-R-O-C-H-Y Just tried it, no luck. However, I did try moving both the coordinator and the ESP around to different positions while it was trying to reconnect, and twice I got it to register *something* in Home Assistant. It logged an "identify" of the Zigbee device, but the ESP itself didn't seem to actually register a connection.

After some more digging, it seems like these "SuperMini" AliExpress boards have a terrible antenna, its just a ceramic SMD that may or may not even be properly impedance matched to the rest of the circuit. [Here](https://community.home-assistant.io/t/esp32-c3-super-mini-wifi-signal-booster/657661/11) are a [couple](https://sigmdel.ca/michel/ha/esp8266/super_mini_esp32c3_en.html#wifi_test) of [references](https://roryhay.es/blog/esp32-c3-super-mini-flaw).

I just ordered a Seeed Studio XIAO ESP32-C6 board on Amazon that should arrive in a couple days. This supposedly has a much better on-board antenna, and also has a jack for an external antenna which I also ordered.

*However*, this doesn't address why the initial connection seems to have no problem. If it was just really low signal coming from the ESP, I would expect it to not connect to the coordinator *at all*.

michapr, 2024-11-25:
@P-R-O-C-H-Y any progress.... Only once I have seen a reboot - all other tries with direct connection.

BTW: I'm using the DevBoard, not the Mini/SuperMini board

```
=========== Before Setup Start ===========
Chip Info:
------------------------------------------
  Model             : ESP32-C6
  Package           : 1
  Revision          : 0.01
  Cores             : 1
  CPU Frequency     : 160 MHz
  XTAL Frequency    : 40 MHz
  Features Bitfield : 0x00000052
  Embedded Flash    : No
  Embedded PSRAM    : No
  2.4GHz WiFi       : Yes
  Classic BT        : No
  BT Low Energy     : Yes
  IEEE 802.15.4     : Yes
------------------------------------------
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   445260 B ( 434.8 KB)
  Free Bytes        :   413940 B ( 404.2 KB)
  Allocated Bytes   :    24352 B (  23.8 KB)
  Minimum Free Bytes:   409128 B ( 399.5 KB)
  Largest Free Block:   376820 B ( 368.0 KB)
------------------------------------------
Flash Info:
------------------------------------------
  Chip Size         :  4194304 B (4 MB)
  Block Size        :    65536 B (  64.0 KB)
  Sector Size       :     4096 B (   4.0 KB)
  Page Size         :      256 B (   0.2 KB)
  Bus Speed         : 40 MHz
  Bus Mode          : QIO
------------------------------------------
Partitions Info:
------------------------------------------
                nvs : addr: 0x00009000, size:    20.0 KB, type: DATA, subtype: NVS
            otadata : addr: 0x0000E000, size:     8.0 KB, type: DATA, subtype: OTA
               app0 : addr: 0x00010000, size:  1280.0 KB, type:  APP, subtype: OTA_0
               app1 : addr: 0x00150000, size:  1280.0 KB, type:  APP, subtype: OTA_1
             spiffs : addr: 0x00290000, size:  1388.0 KB, type: DATA, subtype: SPIFFS
         zb_storage : addr: 0x003EB000, size:    16.0 KB, type: DATA, subtype: FAT
             zb_fct : addr: 0x003EF000, size:     4.0 KB, type: DATA, subtype: FAT
           coredump : addr: 0x003F0000, size:    64.0 KB, type: DATA, subtype: COREDUMP
------------------------------------------
Software Info:
------------------------------------------
  Compile Date/Time : Nov 25 2024 16:42:26
  Compile Host OS   : windows
  ESP-IDF Version   : v5.3.1-638-ga0f798cfc4-dirty
  Arduino Version   : 3.1.0
------------------------------------------
Board Info:
------------------------------------------
  Arduino Board     : ESP32C6_DEV
  Arduino Variant   : esp32c6
  Arduino FQBN      : esp32:esp32:esp32c6:UploadSpeed=921600,CDCOnBoot=cdc,CPUFreq=160,FlashFreq=80,FlashMode=qio,FlashSize=4M,PartitionScheme=zigbee,DebugLevel=verbose,EraseFlash=all,JTAGAdapter=default,ZigbeeMode=ed_debug
============ Before Setup End ============
[  1720][I][esp32-hal-periman.c:141] perimanSetPinBus(): Pin 12 already has type USB_DM (38) with bus 0x408115a0
[  1721][I][esp32-hal-periman.c:141] perimanSetPinBus(): Pin 13 already has type USB_DP (39) with bus 0x408115a0
[  1721][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type GPIO (1) successfully set to 0x420034a0
[  1722][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 9 successfully set to type GPIO (1) with bus 0xa
ZBOSS: zcl/zcl_common.c:215   >> zb_zcl_get_attribute_size: attr_type 0x42 attr_value 0x408198e4
ZBOSS: zcl/zcl_common.c:348   << zb_zcl_get_attribute_size: result 10
ZBOSS: zcl/zcl_common.c:215   >> zb_zcl_get_attribute_size: attr_type 0x42 attr_value 0x40819900
[... 272 lines cut ...]
Updated temperature sensor value to 24.00°C
[  5331][V][ZigbeeTempSensor.cpp:68] setTemperature(): Updating temperature sensor value...
[  5332][D][ZigbeeTempSensor.cpp:70] setTemperature(): Setting temperature to 2400
Updated temperature sensor value to 24.00°C
[  6332][V][ZigbeeTempSensor.cpp:68] setTemperature(): Updating temperature sensor value...
[  6333][D][ZigbeeTempSensor.cpp:70] setTemperature(): Setting temperature to 2400
Updated temperature sensor value to 25.00°C
[  7332][V][ZigbeeTempSensor.cpp:68] setTemperature(): Updating temperature sensor value...
[  7333][D][ZigbeeTempSensor.cpp:70] setTemperature(): Setting temperature to 2500
[  7419][V][ZigbeeHandlers.cpp:134] zb_cmd_default_resp_handler(): Received default response: from address(0x0), src_endpoint(1) to dst_endpoint(10), cluster(0x402) with status 0x0
Updated temperature sensor value to 25.00°C
[  8333][V][ZigbeeTempSensor.cpp:68] setTemperature(): Updating temperature sensor value...
[  8334][D][ZigbeeTempSensor.cpp:70] setTemperature(): Setting temperature to 2500
Updated temperature sensor value to 25.00°C
[  9334][V][ZigbeeTempSensor.cpp:68] setTemperature(): Updating temperature sensor value...
[  9335][D][ZigbeeTempSensor.cpp:70] setTemperature(): Setting temperature to 2500
Updated temperature sensor value to 25.00°C
[ 10334][V][ZigbeeTempSensor.cpp:68] setTemperature(): Updating temperature sensor value...
[ 10335][D][ZigbeeTempSensor.cpp:70] setTemperature(): Setting temperature to 2500
Updated temperature sensor value to 25.00°C
[ 11335][V][ZigbeeTempSensor.cpp:68] setTemperature(): Updating temperature sensor value...
[ 11336][D][ZigbeeTempSensor.cpp:70] setTemperature(): Setting temperature to 2500
```

What I have seen before - device was stop to send values again after some time (have reported before with log)

P-R-O-C-H-Y, 2024-11-26:
@Nawor3565 @michapr Can you try to test the example from esp-zigbee-sdk to check, if the issue is there also? 
https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_HA_sample/HA_temperature_sensor

Nawor3565, 2024-11-26:
@P-R-O-C-H-Y I can definitely give that a shot, although I don't think I have time to setup the tools to compile it right now - would it be possible for you to compile a binary for a ESP32-H2 board that I could flash with esptool? :)

[33 more replies]

PR #10704 (2024-12-11): fix(zigbee): Fix rejoin issue for battery powered devices
## Description of Change
TBD

## Tests scenarios
TBD

## Related links
Closes #10612
closed 2024-12-12 by P-R-O-C-H-Y