espressif/arduino-esp32 #10613: mDNS Resolution Issues on ESP32-S3

TobbeG, 2024-11-17:
### Board

ESP32-S3

### Device Description

Environment
Board: ESP32-S3
Arduino IDE Version: 2.3.3
ESP32 Core Version: 3.0.7
Library in Use: ESPmDNS
Operating System: Windows 11

### Hardware Configuration

TFT display (Elecrow ESP32 Terminal parallell)

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE 2.3.3

### Operating System

Windows 11

### Flash frequency

QIO 80MHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

Issue
I am encountering issues with mDNS resolution on the ESP32-S3 when using the MDNS.queryService() and MDNS.queryHost() functions. Here is what I observe:

Device Advertises mDNS Service Successfully:
The MDNS.begin() function completes successfully, and the device advertises its service (e.g., _http._tcp) with no errors.
Debug Output:

makefile
Kopiera kod
[INFO}: MDNS responder started: ss
Service Discovery Partially Works:
MDNS.queryService("http", "tcp") lists services as expected, but MDNS.queryHost() fails to resolve the IPs for all discovered services.
Debug Output:

lua
Kopiera kod
Found 19 services:
  shelly1.local IP resolution failed at port 80
  kitchen.local IP resolution failed at port 80
  ...
  ss.local IP resolution failed at port 80
Self-Hostname Query Fails:
Even querying the ESP32-S3's own hostname (e.g., ss.local) using MDNS.queryHost() consistently fails:

cpp
Kopiera kod
IPAddress selfIP;
if (MDNS.queryHost("ss.local", selfIP)) {
    Serial.println("Self-hostname resolved successfully.");
} else {
    Serial.println("Failed to resolve self-hostname.");
}
Output:

css
Kopiera kod
Failed to resolve self-hostname.
Code Snippet
Here’s the minimal code used for testing:

cpp
Kopiera kod
#include <ESPmDNS.h>
#include <WiFi.h>

void setup() {
  Serial.begin(115200);
  WiFi.begin("SSID", "PASSWORD");

  while (WiFi.status() != WL_CONNECTED) {
    delay(1000);
    Serial.println("Connecting to WiFi...");
  }
  Serial.println("Connected!");

  if (MDNS.begin("ss")) {
    Serial.println("[INFO}: MDNS responder started: ss");
    MDNS.addService("http", "tcp", 80);
  } else {
    Serial.println("[ERROR]: Failed to start MDNS responder");
  }
}

void loop() {
  int n = MDNS.queryService("http", "tcp");
  Serial.printf("Found %d services:\n", n);

  for (int i = 0; i < n; i++) {
    String hostname = MDNS.hostname(i) + ".local";
    IPAddress ip;
    if (MDNS.queryHost(hostname, ip)) {
      Serial.printf("Resolved %s -> %s\n", hostname.c_str(), ip.toString().c_str());
    } else {
      Serial.printf("Failed to resolve %s\n", hostname.c_str());
    }
  }

  // Self-check
  IPAddress selfIP;
  if (MDNS.queryHost("ss.local", selfIP)) {
    Serial.printf("Self-hostname resolved: %s\n", selfIP.toString().c_str());
  } else {
    Serial.println("Failed to resolve self-hostname.");
  }

  delay(60000); // Run every minute
}
What I Have Tried
Verified network multicast settings on the router (multicast is enabled).
Ensured all devices are on the same subnet.
Tested on different networks with the same results.
Checked Wireshark logs and confirmed mDNS traffic is being sent and received.
Updated ESP32 Core to the latest version (3.0.7).
Expected Behavior
MDNS.queryHost() should resolve the IP addresses for all discovered hostnames, including the ESP32-S3's own hostname (ss.local).
Actual Behavior
MDNS.queryHost() fails to resolve the IP address for all discovered services and the ESP32-S3 itself.
Request
Could you help investigate why MDNS.queryHost() consistently fails in this scenario? If this is a known issue with the library, is there a recommended workaround or a fix planned for future releases?

### Sketch

```cpp
#include <ESPmDNS.h>
#include <WiFi.h>

void setup() {
  Serial.begin(115200);
  WiFi.begin("SSID", "PASSWORD");

  while (WiFi.status() != WL_CONNECTED) {
    delay(1000);
    Serial.println("Connecting to WiFi...");
  }
  Serial.println("Connected!");

  if (MDNS.begin("ss")) {
    Serial.println("[INFO}: MDNS responder started: ss");
    MDNS.addService("http", "tcp", 80);
  } else {
    Serial.println("[ERROR]: Failed to start MDNS responder");
  }
}

void loop() {
  int n = MDNS.queryService("http", "tcp");
  Serial.printf("Found %d services:\n", n);

  for (int i = 0; i < n; i++) {
    String hostname = MDNS.hostname(i) + ".local";
    IPAddress ip;
    if (MDNS.queryHost(hostname, ip)) {
      Serial.printf("Resolved %s -> %s\n", hostname.c_str(), ip.toString().c_str());
    } else {
      Serial.printf("Failed to resolve %s\n", hostname.c_str());
    }
  }

  // Self-check
  IPAddress selfIP;
  if (MDNS.queryHost("ss.local", selfIP)) {
    Serial.printf("Self-hostname resolved: %s\n", selfIP.toString().c_str());
  } else {
    Serial.println("Failed to resolve self-hostname.");
  }

  delay(60000); // Run every minute
}
```

### Debug Message

```plain
Found 19 services:
  shelly1.local IP resolution failed at port 80
  kitchen.local IP resolution failed at port 80
  ...
  ss.local IP resolution failed at port 80
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2024-11-17:
[MDNS.queryHost returns an IPAddress](https://github.com/espressif/arduino-esp32/blob/master/libraries/ESPmDNS/src/ESPmDNS.h#L93).  Your use is more like the C-style where you would pass a reference pointer.  The second parameter is a timeout, so you can skip that unless needed.

TobbeG, 2024-11-17:
I try this code:

        String hostname = "s";
        IPAddress resolvedIP = MDNS.queryHost(hostname.c_str(), 2000); // 2000ms timeout
        if (resolvedIP != IPAddress(0, 0, 0, 0)) {
          Serial.print("[MDNS  ]: ");
          Serial.print(hostname);
          Serial.print(" OK - IP: ");
          Serial.println(resolvedIP);
        } else {
          Serial.print("[MDNS  ]: FAILED - for ");
          Serial.println(hostname);
        }
      }
         
but it always returns FAILED

s.local on my webrowsers works fine and I get the ESP32-S3 webpage

I want the ESP32-S3 to verify its mDNS is working once in a while and if not 
it initializes MDNS again with:   

MDNS.end() followed by MDNS.begin();

Don't understand what gets wrong here ...

TobbeG, 2024-11-17:
i now tested with 2 esp32-s3 on same network with same software 

device 1 = "s.local"
device 2 = "ss.local"

When device 1 resolves hostname "s.local" = it always fails returning 0.0.0.0
IPAddress resolvedIP = MDNS.queryHost(hostname.c_str(), 2000);

When device 1 reolves hostname "ss.local" = it always works fine !

**Is it not possible to resolve your own mDNS name to IP ?**

TobbeG, 2024-11-17:
Here is an example where s.local on 192.168.7.76
tries to resolve s.local which fails but 
works fine when resolving ss.local which is on 192.168.7.108

57	3.415883	192.168.7.76	224.0.0.251	MDNS	68	Standard query 0x0000 A ss.local, "QM" question
58	3.523002	192.168.7.108	224.0.0.251	MDNS	78	Standard query response 0x0000 A, cache flush 192.168.7.108
74	6.515743	192.168.7.76	224.0.0.251	MDNS	67	Standard query 0x0000 A s.local, "QM" question
83	7.614928	192.168.7.76	224.0.0.251	MDNS	67	Standard query 0x0000 A s.local, "QM" question

The purpose of resolving its own s.local is to make sure mdns is working properly on the local network,
and if not - restart it.

**Any work arounds available?**

TobbeG, 2024-11-17:
ping works on both - W11 Lenovo

C:\Users\tge>ping s.local

Pinging s.local [192.168.7.76] with 32 bytes of data:
Reply from 192.168.7.76: bytes=32 time=4ms TTL=64
Reply from 192.168.7.76: bytes=32 time=2ms TTL=64
Reply from 192.168.7.76: bytes=32 time=3ms TTL=64
Reply from 192.168.7.76: bytes=32 time=3ms TTL=64

Ping statistics for 192.168.7.76:
    Packets: Sent = 4, Received = 4, Lost = 0 (0% loss),
Approximate round trip times in milli-seconds:
    Minimum = 2ms, Maximum = 4ms, Average = 3ms

C:\Users\tge>ping ss.local

Pinging ss.local [192.168.7.108] with 32 bytes of data:
Reply from 192.168.7.108: bytes=32 time=5ms TTL=64
Reply from 192.168.7.108: bytes=32 time=4ms TTL=64
Reply from 192.168.7.108: bytes=32 time=2ms TTL=64
Reply from 192.168.7.108: bytes=32 time=2ms TTL=64

Ping statistics for 192.168.7.108:
    Packets: Sent = 4, Received = 4, Lost = 0 (0% loss),
Approximate round trip times in milli-seconds:
    Minimum = 2ms, Maximum = 5ms, Average = 3ms

me-no-dev, 2024-11-18:
Are you trying to resolve the IP of the same device that is sending the query? That would not work.

TobbeG, 2024-11-18:
I know now😳

closed 2024-11-22 by TobbeG