espressif/arduino-esp32 #10615: How to compile the idf-esp project into arduino-esp32 for use

allen4z, 2024-11-18:
### Related area

BT

### Hardware specification

ESP32

### Is your feature request related to a problem?

No related issues

### Describe the solution you'd like

Our project uses the classic Bluetooth function. Some Android phones cannot adjust the volume. We asked for help from the idf-esp project and provided a modification patch. It can be used normally in the idf environment.

However, we encountered problems when compiling idf-esp into the arduino environment. We don’t know how to correctly compile the idf-esp project required by arduino. At present, we only need to compile the libbt.a package. The size compiled by cmake is inconsistent with the size of the idf-esp associated with the directly downloaded arduino.

We tried the following versions:
· idf-esp v4.4.7 corresponding to arduino-esp32 v2.0.17
· idf-esp v5.3.1 corresponding to arduino-esp32 v3.1.0 RC3

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

me-no-dev, 2024-11-18:
It's not very clear what you want to achieve. Do you want to compile custom ESP-IDF libs for use in Arduino, or use Arduino as part of IDF?

- how to use as ESP-IDF component: https://docs.espressif.com/projects/arduino-esp32/en/latest/esp-idf_component.html
- how to use the lib-builder: https://docs.espressif.com/projects/arduino-esp32/en/latest/lib_builder.html

For Arduino v3.1.x you need to use the `release/v5.3` branch of the lib-builder. For Arduino v3.0.x you need to use the `release/v5.1` branch and for 2.0.17, you should use the `release/v4.4` branch

allen4z, 2024-11-19:
need to compile custom ESP-IDF libs for use in Arduino