espressif/arduino-esp32 #10641: (UART): ESP32 - Cannot change baud rate  from 230400 to 460800 with Serial.updateBaudRate()

deltecent, 2024-11-23:
### Board

ESP-WROOM-32

### Device Description

I have an ESP-WROOM-32 DEV KIT 1 with RX2 and TX2 (Serial2) connected to a MAX3380 RS232 Transceiver.

### Hardware Configuration

Nothing else attached to the board.

### Version

v3.0.6

### IDE Name

Arduino IDE 2.3.3

### Operating System

macOS 13.7.1

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

921600

### Description

#10172 corrects a problem with baud rates above 250K by setting the proper clock source. This functionality only works in `Serial.begin()` and does not work in `Serial.updateBaudRate()`.

If `Serial.begin()` is invoked with a baud rate under 250K, the source clock is set to `UART_SCLK_REF_TICK`, otherwise the source clock is set to `UART_SCLK_APB`.

If the source clock is set to `UART_SCLK_REF_TICK` during `Serial.begin()`, updating the baud rate with `Serial.updateBaudRate()` to one that requires `UART_SCLK_APB` will not work because the source clock isn't change.

### Sketch

```cpp
Serial2.begin(230400);
Serial.printf("%d\n", Serial2.baudRate());
// Clock is now UART_SCLK_REF_TICK

Serial2.updateBaudRate(460800);
Serial.printf("%d\n", Serial2.baudRate());
// Clock is still UART_SCLK_REF_TICK

Serial2.end();
Serial2.begin(460800);
Serial.printf("%d\n", Serial2.baudRate());
// Clock is now UART_SCLK_APB
```

### Debug Message

```plain
I do not have any debug output as I do not know how to access the current UART source clock setting from the Serial API. I just know Serial2.baudRate() returns baud rates based on the initial `Serial2.begin()` source clock setting.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2024-11-23:
Yes. correct. Issue confirmed.
It happens with ESP32 and ESP32-S2 only.
Other SoC activate UART using `UART_SCLK_XTAL` clock source are safe.

Fix on the way.

SuGlider, 2024-11-23:
@deltecent - Please test the PR #10642 and let me know.

deltecent, 2024-11-24:
@SuGlider I am very new to the Arduino IDE environment. Is there documentation on how to test a PR?

SuGlider, 2024-11-24:
@deltecent - In order to test a PR, it is necessary to create an Arduino boards from a sketchbook.
It is documented here: https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html#macos
In a few words, the whole process shall go like this:
1) create a `hardware/espressif/` folder under your Arduino Sketchbook folder (usually `~/Documents/Arduino`)
2) clone ESP32 Arduino into this folder
3) go to <cloned_arduino_folder>/tools and execute `get.py` to install all the tools and libraries
4) restart your Arduino IDE and go to `Menu->Board` option to see the `ESP32 Arduino (in sketchbook)` new board.
5) go to the  <cloned_arduino_folder> and execute `git fetch origin pull/PR_NUMBER/head:PR_BRANCH_NAME` using PR_NUMBER as 10643 and BRANCH_NAME as any name you prefer.
6) in to the  <cloned_arduino_folder> execute `git checkout PR_BRANCH_NAME` using the selected branch name from (5)
7) go to your Arduino IDE and build/upload again the testing sketch

deltecent, 2024-11-24:
@SuGlider Thank you for those instructions. They worked great. PR #10642 appears to have fixed this problem.

I want to point out that the comment for this PR in the test program above is in the wrong place. The PR corrects `Serial.updateBaudRate()`:

```cpp
void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  delay(500);
  
  Serial1.begin(230400);
  Serial.printf("%d\n", Serial1.baudRate());
  // Clock is UART_SCLK_REF_TICK as expected

  Serial1.updateBaudRate(460800);
  Serial.printf("%d\n", Serial1.baudRate());
  // Clock is still UART_SCLK_REF_TICK, should be UART_CLK_APB <<---- FIXED by the PR

  Serial1.end();
  Serial1.begin(460800);
  Serial.printf("%d\n", Serial1.baudRate());
  // Clock is now UART_SCLK_APB as expected
}

void loop() {
}
```
Thank you for the quick fix!

PR #10642 (2024-11-25): feat(uart): [IDF5.1] fixes HardwareSerial::updateBaudRate() using a baud rate higher 230400 - checks UART Clock Source
# EXCLUSIVE FOR IDF 5.1 - Arduino Core 3.0.x

## Description of Change

ESP32 and ESP32S2 use REF_TICK as UART clock source whenever the baud rate is lower then 250,000.
This is done within `HardwareSerial::begin(uint32_t baud)`
This prevents issues with chaning the APB clock to lower than 40MHz, specially when achieving lower power consumpition.

This PR also fixes it for the `HardwareSerial::updateBaudRate(uint32_t baud)` by checking the target baud rate and adjusting the UA [... cut]
PR #10643 (2024-11-25): feat(uart): [IDF 5.3] fixes HardwareSerial::updateBaudRate() using a baud rate higher 230400 - checks UART Clock Source
# EXCLUSIVE FOR IDF 5.3 - Arduino Core 3.1.x

## Description of Change

ESP32 and ESP32S2 use REF_TICK as UART clock source whenever the baud rate is lower then 250,000.
This is done within `HardwareSerial::begin(uint32_t baud)`
This prevents issues with chaning the APB clock to lower than 40MHz, specially when achieving lower power consumpition.

This PR also fixes it for the `HardwareSerial::updateBaudRate(uint32_t baud)` by checking the target baud rate and adjusting the U [... cut]
closed 2024-11-27 by SuGlider