espressif/arduino-esp32 #10645: Using sntp_set_time_sync_notification_cb() works only once and not regularly

hasenradball, 2024-11-24:
### Board

ESP32-C3

### Device Description

ESP32-C3 supermini Module

### Hardware Configuration

no GPIO used.

### Version

latest master (checkout manually)

### IDE Name

PlatformIO

### Operating System

MacOS

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

I ues the callback function `sntp_set_time_sync_notification_cb()` to get notified when time has synchronized.

I use a code where the exception is that every 1 hour the snap sync time with the timeserver (similar as in ESP8266).
But the theme seems to sync only one at the beginning and not regularly.

[image]

### Sketch

```cpp
/**
 * @brief Construct a new MESZ::MESZ object
 */
MESZ::MESZ() {
  sntp_set_time_sync_notification_cb(MESZ::set_time_synchronized_true);
  clear_tm();
}

void loop() {
   if (MESZ::get_time_synchronized()) {
      Serial.println("Time successfully syncronized!\n");
      MESZ::set_time_synchronized_false();
      Serial.printf("\n Zeit: %s", uhr.get_asctime());
   }
}
```

### Debug Message

```plain
No error, but also no regular update.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

sivar2311, 2024-11-24:
Whatever "MESZ" and "uhr" are.... The sntp callback works perfectly and gets called regularly:

Example:
```
#include <Arduino.h>
#include <WiFi.h>
#include <esp_sntp.h>

const char* WIFI_SSID = "YOUR_WIFI_SSID_HERE";
const char* WIFI_PASS = "YOUR_WIFI_PASS_HERE";

unsigned long lastUpdate = 0;

void setupWiFi() {
    Serial.print("Connecting Wifi");
    WiFi.begin(WIFI_SSID, WIFI_PASS);
    while (WiFi.status() != WL_CONNECTED) {
        Serial.print(".");
        delay(100);
    }
    Serial.println("connected");
}

void sntp_sync_cb(struct timeval* tv) {
    Serial.print("Got new timestamp: ");
    Serial.print(tv->tv_sec);

    if (lastUpdate != 0) {
        unsigned long diff = (millis() - lastUpdate) / 1000;
        Serial.print(" Last update was ");
        Serial.print(diff);
        Serial.println(" seconds ago.");
    } else {
        Serial.println(" This was the first received timestamp.");
    }
    lastUpdate = millis();
}

void setupSNTP() {
    const uint32_t EVERY_15_SECONDS = 15000;
    sntp_set_sync_interval(EVERY_15_SECONDS);

    sntp_set_time_sync_notification_cb(sntp_sync_cb);

    const char* BERLIN_TIMEZONE = "CET-1CEST,M3.5.0,M10.5.0/3";
    const char* EUROPE_NTP_SERVER = "europe.pool.ntp.org";
    configTzTime(BERLIN_TIMEZONE, EUROPE_NTP_SERVER);
}

void setup() {
    Serial.begin(115200);
    setupWiFi();
    setupSNTP();
}

void loop() {
}
```

Output:
```
Connecting Wifi..connected
Got new timestamp: 1732460988 This was the first update
Got new timestamp: 1732461003 Last update was 15 seconds ago.
Got new timestamp: 1732461018 Last update was 15 seconds ago.
Got new timestamp: 1732461033 Last update was 15 seconds ago.
Got new timestamp: 1732461048 Last update was 15 seconds ago.
Got new timestamp: 1732461063 Last update was 15 seconds ago.
Got new timestamp: 1732461078 Last update was 15 seconds ago.
Got new timestamp: 1732461093 Last update was 15 seconds ago.
Got new timestamp: 1732461108 Last update was 15 seconds ago.
Got new timestamp: 1732461123 Last update was 15 seconds ago.
Got new timestamp: 1732461138 Last update was 15 seconds ago.
Got new timestamp: 1732461153 Last update was 15 seconds ago.
Got new timestamp: 1732461168 Last update was 15 seconds ago.
```

hasenradball, 2024-11-24:
Hi @sivar2311 
thanks for watching my post.
MESZ is in German „Mitteleuropäische Sommerzeit“ a class which calcullates the Time for Germany.

In ESP8266 there i no need for setup  the sync interval.

But in your example forxESP32 it seems that is needed to setup the sync interval, right?

What is sntp doing whe I setup no sync interval?

sivar2311, 2024-11-24:
> MESZ is in German „Mitteleuropäische Sommerzeit“ a class which calcullates the Time for Germany.

I know the term MESZ. I'm from germany, too ;)
But what i don't know the the Object `MESZ` you use.
Where does it come from (Library)? What does it do... how is it implemented?
The same applies to the object `uhr`.

>In ESP8266 there i no need for setup the sync interval.

In the example above i set the interval to 15 seconds to get a quick recurring call of the callback function to prove it gets called regularly and exactly. 15 seconds is the minimal update intervall to set.

hasenradball, 2024-11-24:
@sivar2311 
The MESZ is a wrapper class written by me to setup a clock.
Uhr ( clock in english) is then th instance of MESZ.
[MESZ](https://github.com/hasenradball/ESP32-C3-Examples/tree/main/lib/cMESZ)

Here the complete example.
[Example Code](https://github.com/hasenradball/ESP32-C3-Examples/blob/main/examples/timeMESZ/timeMESZ.ino)

But let me ask a final question,
what happens if I do not call
`sntp_set_sync_interval(EVERY_15_SECONDS);`

Is there no default sync time defined?
😊

hasenradball, 2024-11-24:
@sivar2311 I think I could answer the Question by myself....

```
=== ESP32-c3 Example to use MESZ (Mean European Summertime) ===

Set WiFi mode to WIFI_STA - 1

 Connection with Wifi_Start() - successful <<<

        Verbunden mit:  topsecret-DSL
        IP address:     192.168.178.22
        gateway:        192.168.178.1
        subnet mask:    255.255.255.0
        DNS server:     192.168.178.1

>>> Init MESZ
Configured Timeserver-settings: ...
        TimesZone:   CET-1CEST,M3.5.0,M10.5.0/3
        Timeserver1: 192.168.178.1
        Timeserver2: ptbtime2.ptb.de
        Timeserver3: ptbtime3.ptb.de
Time successfully syncronized!

the sntp sync intervall is set to : 10800000 ms / 10800 s / 180 min / 3 h
Zeit: Sun Nov 24 18:00:56 2024
```

Thanks so much for the fast reply....
:-)

closed 2024-11-24 by hasenradball

sivar2311, 2024-11-24:
@hasenradball 

A call to

```
    Serial.printf("Default sync interval (seconds): %lu\r\n", sntp_get_sync_interval() / 1000);
```

revealed:

```
Default sync interval (seconds): 10800
```

So every 3 hours

hasenradball, 2024-11-24:
Nearly in time

sivar2311, 2024-11-24:
Sorry, i was out for a walk... ;)

hasenradball, 2024-11-24:
Mee too :-)

sivar2311, 2024-11-24:
@hasenradball If you are interested in german speaking realtime chat: https://discord.gg/egJXPbqm3Z

hasenradball, 2024-11-24:
@sivar2311 Thank you for the hint. Actually I think hav no further Questions for now.
Have a nice evening :-)

hasenradball, 2024-11-25:
Dear @sivar2311 ,

one question cane in mind regarding the ESP32.
From ESP8266 I have in mind that wifi goes off (for example overnight) the chip connects automatically in the morning.

But known from the past with ESP32 does this different.
You have actually actively reconnect the ESP32 by yourself.
Is this actually the normal behavior of ESP32?

hasenradball, 2024-11-25:
From the past my expirience was that itdid not work.
But I will try.
👍

sivar2311, 2024-11-25:
I never had an issue with auto reconnecting.
But you can utilize the WiFi events to detect a disconnect / IP loss and implement a reconnecting mechanism.

hasenradball, 2024-11-25:
I startet in 2016 with esp32 and had this issue all the time.
Therefore stopped working on ESP32 since years.
But I will check again.

The reconnect issue also confirmed by the developers.

sivar2311, 2024-11-25:
@hasenradball 
> The reconnect issue also confirmed by the developers.

Do you have a link and do you know if this also applies to Arduino 3.x?

hasenradball, 2024-11-25:
No the expirience I had was maybee with older versions.
Maybee with v3 there was an improvement.
With v2 this did not work.

sivar2311, 2024-11-25:
I noticed, you're using PlatformIO.
To get Espressif32 Arduino 3.x, try [pioarduino](https://github.com/pioarduino/platform-espressif32)

You'll find me on discord :)

hasenradball, 2024-11-25:
I have the version 3.x installed.
There is a small change in config needed.
😊

Poopi, 2025-01-24:
I've spend some time to get the sntp_set_sync_interval to work. Finally I got it working. No mater where I've tried to setup interval it was not used. I've got only one callback and then nothing. Seems like it cannot be set in the setup() function.I've got it to work from loop by calling esp_netif_sntp_deinit(), sntp_set_sync_interval(), sntp_restart(). It works like a charm.

hasenradball, 2025-01-24:
it worked for me!