espressif/arduino-esp32 #10646: Bound zigbee devices are lost after a reboot

AndunHH, 2024-11-24:
### Board

seeed xiao esp32-c6

### Device Description

The development board is used without any other attachments. Only a powering / data USB-C cable.
See https://wiki.seeedstudio.com/xiao_esp32c6_getting_started/# for more details.

### Hardware Configuration

nothing else attached

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE 2

### Operating System

Ubuntu 2024.04

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

Short: The esp32-c6 doesn't remember the zigbee bounds from the last reset.

Problem description:
I run the example Zigbee_on_off_switch, but removed the waiting period in the setup().
When I run the script for the first time and bind a new device can toggle the status of the device. Tested successfully with the light example on a second xiao esp32-c6 and an actuall ikea tradfri plug.
Now I reset / power cycle the switch.
The former established connections are forgotten: If I just press the button again, no devices are bound. I don't want to pair the plug or light again after a power reset of the switch.

Expected behavior:
When I reset the controller, I expect, that the controller remembers the bound devices from last time. A press on the button of the switch shall remember the bound devices from the last power cycle.

### Sketch

```cpp
// Copyright 2024 Espressif Systems (Shanghai) PTE LTD
//
// Licensed under the Apache License, Version 2.0 (the "License");
// you may not use this file except in compliance with the License.
// You may obtain a copy of the License at
//
//     http://www.apache.org/licenses/LICENSE-2.0
//
// Unless required by applicable law or agreed to in writing, software
// distributed under the License is distributed on an "AS IS" BASIS,
// WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
// See the License for the specific language governing permissions and
// limitations under the License.

/**
 * @brief This example demonstrates simple Zigbee light switch.
 *
[... 398 lines cut ...]
    16 : UART_TX[0]
    17 : UART_RX[0]
============ After Setup End =============
[  1150][I][ZigbeeCore.cpp:280] esp_zb_app_signal_handler(): Network(0x5dac) is open for 180 seconds[ 46677][E][ZigbeeSwitch.cpp:76] lightToggle(): Light not bound
[ 46678][I][ZigbeeEP.cpp:124] printBoundDevices(): Bound devices:
[ 47926][E][ZigbeeSwitch.cpp:76] lightToggle(): Light not bound
[ 49143][E][ZigbeeSwitch.cpp:76] lightToggle(): Light not bound
```

### Other Steps to Reproduce

I first mentioned this problem here: https://github.com/espressif/arduino-esp32/issues/10135#issuecomment-2481577417 

With the espressif SDK this problem seems already to be solved, regarding:
https://github.com/espressif/esp-idf/issues/11904

Therefore, I also tried the example from the [esp zigbee sdk ](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_HA_sample/HA_on_off_switch) which is NOT showing the problem.
If necessary, you can see [my fork here](https://github.com/AndunHH/esp-zigbee-sdk/tree/xiao-esp32-c6/examples/esp_zigbee_HA_sample/HA_on_off_switch), which is working on a xiao esp32-c6, because I needed to enable the internal antenna.

I think, the problem is that the arduino implementation is using the variable [_is_bound ](https://github.com/espressif/arduino-esp32/blob/e99f1fdc6d12e07324be6e956318496e6a0bfac3/libraries/Zigbee/src/ZigbeeEP.cpp#L10) and the [list of bound ](https://github.com/espressif/arduino-esp32/blob/e99f1fdc6d12e07324be6e956318496e6a0bfac3/libraries/Zigbee/src/ZigbeeEP.h#L122)devices which are not initialized based on remebered connections. 
-> How to "save" the list of bound_devices or releoad them after a power cycle?

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

P-R-O-C-H-Y, 2024-11-25:
Hi @AndunHH, thank you for a nice report. I will take a look soon :)

P-R-O-C-H-Y, 2024-12-04:
Hi @AndunHH, can you please test your case with the proposed changes in the PR I linked? 
The PR should fix the issue. Thank you for any feedback.

AndunHH, 2024-12-05:
Hi @P-R-O-C-H-Y, thanks for fixing this issue! 

After struggling with manually installing/updating the library to your branch I got it working. :)

I had some reboots during testing the connection but I'm not able to reproduce them. I doubt, that they are in relation with the newly introduced feature. Therefore, we can close this issue successfully! 

Thanks!

PR #10676 (2024-12-09): feat(Zigbee): Recall bounded devices after reboot + IEEE address option for commands
## Description of Change
This PR adds new feature/fix to recall bounded devices from the binding table after device reboot, with that also the isBound() will return correct value, as the devices will be found if there were some before rebooting.

Other updates:
- Fixes print of the IEEE address, which had reverted bytes order
- `printBoundDevices()` can now be called with Print parameter, so printing to serial instead of log_i is possible.
- Added missing Zigbee locks, fixing the `assert failed: vPortExitCriti [... cut]
closed 2024-12-05 by AndunHH