espressif/arduino-esp32 #10663: ESP32-C6 and Zigbee2MQTT - Sleepy Temperature Example rebooting constantly

HiFiJ, 2024-11-29:
### Board

M5Nano-C6 (M5Stack)

### Device Description

No additional hardware

### Hardware Configuration

No

### Version

latest development Release Candidate (RC-X)

### IDE Name

Arduino-IDE  2.3.3

### Operating System

Linux

### Flash frequency

40 or 80MHz

### PSRAM enabled

yes

### Upload speed

961600

### Description

Hi everyone, thanks to your great work!!

I updated to yesterday night's (28. to 29.11.2024) github master code using the manual method (download in the Sketchbook/Arduino folder). This went successfull, I could choose the esp32c6-dev in "esp 32 (Arduino Sketchbbok)".
I am using "Erase all flash".
I have zigbee2mqtt with a Zigstar Gateway connected via Ethernet, running a Zigbee network with around 10-15 devices.

I am using the unmodified "Zigbee_Temp_Hum_Sensor_Sleepy.ino" example.

After joining the network and getting a  "Inteview successfull" from Z2M, the end device / ESP32 reboots constantly approximately every 7 seconds.

NB: probably as the end point is not known/supported, I can only see a temperature/LQI in Z2m (no humidity), and this does not update (this is likely caused by the constant reboots, as no new data is sent).

Thanks everyone,

Joost

### Sketch

```cpp
Unmodified "Zigbee_Temp_Hum_Sensor_Sleepy.ino" example
```

### Debug Message

```plain
Temperature Sensor, Endpoint: 10, Device ID: 0x0302
[  2629][V][ZigbeeCore.cpp:301] esp_zb_app_signal_handler(): ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL
[  2630][I][ZigbeeCore.cpp:191] esp_zb_app_signal_handler(): Zigbee stack initialized
[  9966][E][ZigbeeCore.cpp:219] esp_zb_app_signal_handler(): Failed to initialize Zigbee stack (status: ESP_FAIL)
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0xc (SW_CPU),boot:0x1c (SPI_FAST_FLASH_BOOT)
[... 162 lines cut ...]
  Compile Date/Time : Nov 28 2024 22:10:31
  Compile Host OS   : linux
  ESP-IDF Version   : v5.1.4-828-g33fbade6b8-dirty
  Arduino Version   : 3.0.5
------------------------------------------
Board Info:
------------------------------------------
  Arduino Board     : ESP32C6_DEV
  Arduino Variant   : esp32c6
  Arduino FQBN      : espressif:esp32:esp32c6:UploadSpeed=921600,CDCOnBoot=cdc,CPUFreq=160,FlashFreq=80,FlashMode=qio,FlashSize=4M,PartitionScheme=zigbee,DebugLevel=verbose,EraseFlash=all,JTAGAdapter=default,ZigbeeMode=ed
============ Before Setup End ============
[   540][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type GPIO (1) successfully set to 0x4200236a
[   541][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 9 successfully set to type GPIO (1) with bus 0xa
[   542][D][ZigbeeCore.cpp:66] addEndpoint(): Endpoint: 10, Device ID: 0x0302
[   542][D][ZigbeeCore.cpp:103] zigbeeInit(): Initialize Zigbee stack
[   599][D][ZigbeeCore.cpp:110] zigbeeInit(): Register all Zigbee EPs in list
[   600][I][ZigbeeCore.cpp:118] zigbeeInit(): List of registered Zigbee EPs:
[   601][I][ZigbeeCore.cpp:120] zigbeeInit(): Device type: Temperature Sensor, Endpoint: 10, Device ID: 0x0302
[   607][V][ZigbeeCore.cpp:301] esp_zb_app_signal_handler(): ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL
[   608][I][ZigbeeCore.cpp:191] esp_zb_app_signal_handler(): Zigbee stack initialized
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

P-R-O-C-H-Y, 2024-11-29:
Hi @HiFiJ, please use `release/v3.1.x` branch for Zigbee. Master branch is missing latest changes.
Can you please test on this branch instead? Thanks

HiFiJ, 2024-11-29:
Hi, thanks a lot. I believe I changed the branch successfully (git chekcout release/v.3.1.x, then python get.py), and the reboots seem to be fixed!
For now it seems there is a new hurdle to overcome, as no values seem to get transmitted due to 2commisioning errors", followed by a reboot approx. every 30 seconds.

Booting...
19:02:24.974 -> [   281][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type GPIO (1) successfully set to 0x420035e2
19:02:24.974 -> [   282][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 9 successfully set to type GPIO (1) with bus 0xa
19:02:24.974 -> [   283][D][ZigbeeCore.cpp:83] addEndpoint(): Endpoint: 10, Device ID: 0x0302
19:02:24.974 -> [   283][D][ZigbeeCore.cpp:122] zigbeeInit(): Initialize Zigbee stack
19:02:25.070 -> [   352][D][ZigbeeCore.cpp:129] zigbeeInit(): Register all Zigbee EPs in list
19:02:25.070 -> [   354][I][ZigbeeCore.cpp:137] zigbeeInit(): List of registered Zigbee EPs:
19:02:25.070 -> [   354][I][ZigbeeCore.cpp:139] zigbeeInit(): Device type: Temperature Sensor, Endpoint: 10, Device ID: 0x0302
**_19:02:25.070 -> [   366][V][ZigbeeCore.cpp:332] esp_zb_app_signal_handler(): ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL
19:02:25.070 -> [   366][I][ZigbeeCore.cpp:218] esp_zb_app_signal_handler(): Zigbee stack initialized
19:02:39.395 -> [ 14678][W][ZigbeeCore.cpp:249] esp_zb_app_signal_handler(): Commissioning failed, trying again...
19:02:54.208 -> [ 29490][W][ZigbeeCore.cpp:249] esp_zb_app_signal_handler(): Commissioning failed, trying again...
19:02:55.076 -> [ 30363][E][ZigbeeCore.cpp:43] begin(): ZigbeeCore begin timeout
19:02:55.076 -> Zigbee failed to start!
19:02:55.076 -> Rebooting..._**
19:02:55.076 -> ESP-ROM:esp32c6-20220919
19:02:55.076 -> Build:Sep 19 2022
19:02:55.076 -> rst:0xc (SW_CPU),boot:0xc (SPI_FAST_FLASH_BOOT)
19:02:55.076 -> Saved PC:0x4001975a
19:02:55.076 -> SPIWP:0xee
19:02:55.076 -> mode:DIO, clock div:2
19:02:55.076 -> load:0x40875720,len:0x1190

HiFiJ, 2024-11-29:
I was reading this comment https://github.com/espressif/arduino-esp32/issues/10612#issuecomment-2508154081 regarding disabling the power source, and on first sight it seems like it does indeed improve the situation. Now I am getting the planned 1 minute deep sleep both immediately after flashing and also after powering down/repowering the device. Those are very preliminary observations, just running for a couple of sleep cycles for now.

19:19:47.002 -> [  2681][I][ZigbeeCore.cpp:139] zigbeeInit(): Device type: Temperature Sensor, Endpoint: 10, Device ID: 0x0302
_**19:19:47.002 -> [  2695][V][ZigbeeCore.cpp:332] esp_zb_app_signal_handler(): ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL**_
19:19:47.002 -> [  2695][I][ZigbeeCore.cpp:218] esp_zb_app_signal_handler(): Zigbee stack initialized
19:19:51.341 -> [  7029][I][ZigbeeCore.cpp:224] esp_zb_app_signal_handler(): Device started up in non factory-reset mode
19:19:51.341 -> [  7030][I][ZigbeeCore.cpp:237] esp_zb_app_signal_handler(): Device rebooted
19:19:51.341 -> Connecting to network
19:19:51.341 -> 
19:19:51.341 -> Successfully connected to Zigbee network
19:19:56.352 -> =========== After Setup Start ============
19:19:56.352 -> INTERNAL Memory Info:
19:19:56.352 -> ------------------------------------------
19:19:56.352 ->   Total Size        :   438836 B ( 428.6 KB)
19:19:56.352 ->   Free Bytes        :   380360 B ( 371.4 KB)
19:19:56.352 ->   Allocated Bytes   :    50692 B (  49.5 KB)
19:19:56.352 ->   Minimum Free Bytes:   380360 B ( 371.4 KB)
19:19:56.352 ->   Largest Free Block:   352244 B ( 344.0 KB)
19:19:56.352 -> ------------------------------------------
19:19:56.352 -> GPIO Info:
19:19:56.352 -> ------------------------------------------
19:19:56.352 ->   GPIO : BUS_TYPE[bus/unit][chan]
19:19:56.352 ->   --------------------------------------  
19:19:56.352 ->      9 : GPIO
19:19:56.352 ->     12 : USB_DM
19:19:56.352 ->     13 : USB_DP
19:19:56.352 ->     16 : UART_TX[0]
19:19:56.352 ->     17 : UART_RX[0]
19:19:56.352 -> ============ After Setup End =============
19:19:56.394 -> [ 12060][V][ZigbeeTempSensor.cpp:68] setTemperature(): Updating temperature sensor value...
19:19:56.394 -> [ 12060][D][ZigbeeTempSensor.cpp:70] setTemperature(): Setting temperature to 1800
19:19:56.394 -> [ 12061][V][ZigbeeTempSensor.cpp:108] setHumidity(): Updating humidity sensor value...
19:19:56.394 -> [ 12062][D][ZigbeeTempSensor.cpp:110] setHumidity(): Setting humidity to 1800
19:19:56.394 -> [ 12063][V][ZigbeeTempSensor.cpp:90] reportTemperature(): Temperature report sent
19:19:56.394 -> [ 12064][V][ZigbeeTempSensor.cpp:131] reportHumidity(): Humidity report sent
19:19:56.394 -> Reported temperature: 18.00°C, Humidity: 18.00%
19:19:56.394 -> Going to sleep now

HiFiJ, 2024-11-29:
On a side note, is ther info on what external converter would be needed for Z2M to show the values of the example?

KMTsvetanov, 2024-11-29:
I just started with the zigbee2mqtt on home assistant.
Can you say that everythink is working?
I have a EPS32-h2 and HA with Zigbee2Mqtt.

P-R-O-C-H-Y, 2024-12-05:
@HiFiJ I am not that familiar with Zigbee2MQTT. I am using HomeAssistant with ZHA.
You mean that you are not seeing the values in the Z2M?

michapr, 2024-12-08:
I have also problems with ESP32C6 and Z2M - it is not a converter problem...
You can get it work without sleeping - please check https://github.com/espressif/arduino-esp32/issues/10601 and https://github.com/espressif/arduino-esp32/issues/10663

michapr, 2024-12-12:
@HiFiJ - you can close this issue, it is solved - look here https://github.com/espressif/arduino-esp32/issues/10612

HiFiJ, 2024-12-13:
Thank you very much for the heads up - I'll test and get back with results this weekend I hope! 
Looking forward to getting this working reliably, 

Joost

HiFiJ, 2024-12-14:
Thanks, I managed to update and it looks like it is working now. Only battery level is not available.
Great to see this project improving so fast!
Joost

closed 2024-12-14 by HiFiJ

P-R-O-C-H-Y, 2024-12-16:
@HiFiJ I will also test soon with Z2M, and check the battery level if there is any issue. Using ZHA on HomeAssistant there is no issue with the battery level.