espressif/arduino-esp32 #10668: Change validation for Broadcast in _esp_now_rx_cb

PepeTux, 2024-12-02:
### Board

ESP32

### Device Description

ESP32-C3 Board

### Hardware Configuration

N/A

### Version

latest master (checkout manually)

### IDE Name

Arduino 2.3.3

### Operating System

Windows 10

### Flash frequency

80 MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

Configurinf two devices on ESP-Now, as Transmitter and Receiver,
when I send a message as Broadcast I got a "true" on my onreceive callback.
when I send a message with the Receiver MAC address I got a "false" on my Onreceive callback.

but if I send a Message with the destination address:
uint8_t dst_MAC[] = { 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x0F};
I receive also a false on my Onreceive callback, as it was an Unicast message.

I dont know if this address is an issue but I suggest change the validation from Broadcast to Unicast.

### Sketch

```cpp
// Multicast Address.
uint8_t broadcastAddress[] = { 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x0F};      

// Structure example to send data
// Must match the receiver structure
typedef struct struct_message {
  char a[32];
  int b;
  float c;
  bool d;
} struct_message;

// Create a struct_message called myData
struct_message myData;

void ESP_Now_Sender_Init(void)
{
  // Init ESP-NOW
  if (esp_now_init() != ESP_OK) {
    Serial.println("Error initializing ESP-NOW");
    return;
  }

  // Once ESPNow is successfully Init, we will register for Send CB to
  // get the status of Trasnmitted packet
  esp_now_register_send_cb(OnDataSent);
  
  // Register peer
  memcpy(peerInfo.peer_addr, broadcastAddress, 6);
  peerInfo.channel = 0;  
  peerInfo.encrypt = false;
  
  // Add peer        
  if (esp_now_add_peer(&peerInfo) != ESP_OK){
    Serial.println("Failed to add peer");
    return;
  }

  //Serial.printf("\r\nWifi Channel: %d\n\n", WiFi.channel());
}

void Send_Message(void)
{
  // Set values to send
  strcpy(myData.a, USER_NAME);
  myData.b = NUMERO_DE_CASA;      // Int
  myData.c = 3.141592f;           // Float
  myData.d = true;                // Bool
  
  // Send message via ESP-NOW
  esp_err_t result = esp_now_send(broadcastAddress, (uint8_t *) &myData, sizeof(myData));
  
  #ifdef DEBUG
    if (result == ESP_OK) {
      Serial.println("Sent with success");
    }
    else {
      Serial.println("Error sending the data");
    }
  #endif
}
```

### Debug Message

```plain
N/A
```

### Other Steps to Reproduce

Send an ESP-Now message with destination addres to:

uint8_t dst_MAC[] = { 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x0F};

and the Receiver will detect it as Unicast message.

I suggest validate for Unicast instead of Broadcast.

[image]

My suggestion is change the validation as follows:

uint8_t baseMac[6];
esp_wifi_get_mac(WIFI_IF_STA, baseMac);
// esp_wifi_get_mac(WIFI_IF_AP, baseMac);
  bool unicast = memcmp(info->des_addr, baseMac, ESP_NOW_ETH_ALEN) == 0;

The sample code for Receiver and Transmitter:
[ESP32_C3_307_ESP_Now_Receiver_01.zip](https://github.com/user-attachments/files/17973641/ESP32_C3_307
[ESP32_C3_307_ESP_Now_Transmitter_01.zip](https://github.com/user-attachments/files/17973642/ESP32_C3_307_ESP_Now_Transmitter_01.zip)
_ESP_Now_Receiver_01.zip)

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lucasssvaz, 2024-12-02:
The broadcast address being used is wrong. It should be all `0xFF` change this line in your code to:

```c
uint8_t broadcastAddress[] = { 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF};
```

PepeTux, 2024-12-02:
> The broadcast address being used is wrong. It should be all `0xFF` change this line in your code to:
> 
> ```c
> uint8_t broadcastAddress[] = { 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF};
> ```

Hello Lucas, thanks for your review.
I know the last byte is wrong, that is why I suggest to change the validation from:
`  bool broadcast = memcmp(info->des_addr, ESP_NOW.BROADCAST_ADDR, ESP_NOW_ETH_ALEN) == 0;`
in the [ESP32_NOW.cpp](https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP_NOW/src/ESP32_NOW.cpp) line 106.
to:
`uint8_t baseMac[6];
esp_wifi_get_mac(WIFI_IF_STA, baseMac);
// esp_wifi_get_mac(WIFI_IF_AP, baseMac);
bool unicast = memcmp(info->des_addr, baseMac, ESP_NOW_ETH_ALEN) == 0;`

Also, it's right the ESP-Now can receive that wrong address as a broadcast ?

I upload two zip files with the code for Tx/Rx for test.

yuvashrikarunakaran, 2024-12-04:
Transmitter code
#include <esp_now.h>
#include <WiFi.h>

// Broadcast Address
uint8_t broadcastAddress[] = { 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF };      

// Structure example to send data
typedef struct struct_message {
  char a[32];
  int b;
  float c;
  bool d;
} struct_message;

// Create a struct_message called myData
struct_message myData;

// Callback when data is sent
void OnDataSent(const uint8_t *mac_addr, esp_now_send_status_t status) {
  Serial.print("Send Status: ");
  Serial.println(status == ESP_NOW_SEND_SUCCESS ? "Success" : "Fail");
}

void setup() {
  // Initialize Serial Monitor
  Serial.begin(115200);
  
  // Set device as a Wi-Fi Station
  WiFi.mode(WIFI_STA);
  WiFi.disconnect();

  // Init ESP-NOW
  if (esp_now_init() != ESP_OK) {
    Serial.println("Error initializing ESP-NOW");
    return;
  }

  // Register callback
  esp_now_register_send_cb(OnDataSent);
  
  // Register peer
  esp_now_peer_info_t peerInfo;
  memcpy(peerInfo.peer_addr, broadcastAddress, 6);
  peerInfo.channel = 0;  
  peerInfo.encrypt = false;

  // Add peer
  if (esp_now_add_peer(&peerInfo) != ESP_OK) {
    Serial.println("Failed to add peer");
    return;
  }
}

void loop() {
  // Set values to send
  strcpy(myData.a, "ESP32-C3");   // String
  myData.b = 123;                // Int
  myData.c = 3.14;               // Float
  myData.d = true;               // Bool

  // Send message via ESP-NOW
  esp_err_t result = esp_now_send(broadcastAddress, (uint8_t *)&myData, sizeof(myData));
  
  if (result == ESP_OK) {
    Serial.println("Message sent successfully");
  } else {
    Serial.println("Error sending message");
  }

  delay(2000);
}
Receiver code
#include <esp_now.h>
#include <WiFi.h>

// Callback when data is received
void OnDataRecv(const uint8_t *mac, const uint8_t *incomingData, int len) {
  struct_message myData;
  memcpy(&myData, incomingData, sizeof(myData));

  // Print received data
  Serial.print("Bytes received: ");
  Serial.println(len);
  Serial.print("Message: ");
  Serial.println(myData.a);
  Serial.print("Integer: ");
  Serial.println(myData.b);
  Serial.print("Float: ");
  Serial.println(myData.c);
  Serial.print("Boolean: ");
  Serial.println(myData.d);

  // Check if the message is unicast or broadcast
  uint8_t baseMac[6];
  esp_wifi_get_mac(WIFI_IF_STA, baseMac);
  bool isUnicast = memcmp(mac, baseMac, 6) == 0;

  if (isUnicast) {
    Serial.println("Unicast Message Received");
  } else {
    Serial.println("Broadcast Message Received");
  }
}

void setup() {
  // Initialize Serial Monitor
  Serial.begin(115200);

  // Set device as a Wi-Fi Station
  WiFi.mode(WIFI_STA);
  WiFi.disconnect();

  // Init ESP-NOW
  if (esp_now_init() != ESP_OK) {
    Serial.println("Error initializing ESP-NOW");
    return;
  }

  // Register callback
  esp_now_register_recv_cb(OnDataRecv);
}

void loop() {
  // Do nothing, wait for ESP-NOW messages
}

closed 2025-01-17 by lucasssvaz