espressif/arduino-esp32 #10669: Zigbee ColorDimmableLight hue/sat

ilker-aktuna, 2024-12-02:
### Related area

Zigbee , ZHA

### Hardware specification

ESP32C6

### Is your feature request related to a problem?

I am trying to implement and test zigbee features on a ESP32C6
ZigbeeColorDimmableLight example works partially.
On/off and level commands are working fine but when my Zigbee hub sends a color change it does not work.
When I debug the code, I see that R,G,B values are all 255
I guess my Zigbee hub (Hubitat) sends only Hue&Sat values.
But the examples do not include anything with hue and sat. How can I make the device receive HSL ?

### Describe the solution you'd like

an example with HSL command.

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

P-R-O-C-H-Y, 2024-12-03:
Hi @ilker-aktuna, I will work on that soon to add a hue/sat support also. Thank you for the report.

ilker-aktuna, 2024-12-03:
thanks for your fast response. What would be the fastest way to be informed when you add hue/sat support ?

Also, there is another issue ; even if the device is paired with my hub , I believe the device stays in "pairing" mode. Whenever I try adding another zigbee device, I see the esp32c6 in pairing queue. How can I stop pairing ?

P-R-O-C-H-Y, 2024-12-03:
I will keep you updated in this issue when there are updates on this. 

About the "pairing" I did not experienced anything like that. Can you please send more informations about this issue? The more info and logs the best :)

ilker-aktuna, 2024-12-03:
ok. what logs can I get you ? 
Do you want the serial console logs when the device powers up ?

P-R-O-C-H-Y, 2024-12-03:
What system are you using for the controlling the Zigbee devices? HomeAssistant?

ilker-aktuna, 2024-12-03:
it is Hubitat

P-R-O-C-H-Y, 2024-12-03:
I see, what is the model of the hubitat?
You can send the logs from serial monitor if there is the connection done properly, but I assume as you are controlling the device, it should be fine. Are there any logs available form the hubitat? So we can search why the device is still in pairing mode.

P-R-O-C-H-Y, 2024-12-04:
@yuvashrikarunakaran May I ask you what is this code for?

ilker-aktuna, 2024-12-04:
@P-R-O-C-H-Y 

I removed the device and then re-paired with the Hubitat hub.
Sending you logs from both Arduino serial monitor and also from Hubitat.

I started pairing on Hubitat several times after the initial pairing process completed. As you will see in the summary log everytime Hubitat detected again the esp32c6 in pairing mode. This does not happen with any other zigbee devices that are already paired with Hubitat.

I also noticed that the "controlling" part is not stable. Sometimes, Hubitat does not detect the state of the light (on or off) and can't send the command.
This seems as if the pairing process did not complete.
[hubitat-summary.log](https://github.com/user-attachments/files/18014786/hubitat-summary.log)
[hubitat-full-log.txt](https://github.com/user-attachments/files/18014787/hubitat-full-log.txt)
[arduino-log.txt](https://github.com/user-attachments/files/18014788/arduino-log.txt)

ilker-aktuna, 2024-12-04:
> @yuvashrikarunakaran May I ask you what is this code for?

should I try this code ?
would it work with the current release of Zigbee library ?

P-R-O-C-H-Y, 2024-12-05:
> > @yuvashrikarunakaran May I ask you what is this code for?
> 
> should I try this code ? would it work with the current release of Zigbee library ?

@ilker-aktuna No you don't have to, its not related. I am removing the comment.

ilker-aktuna, 2024-12-05:
ok. you got my logs above ?
do you need anything else to analyze the situation ?

ilker-aktuna, 2024-12-05:
I am currently using a Hubitat C7 
But I also have an offline C8 and I tried the device also on this one. Results are similar.

P-R-O-C-H-Y, 2024-12-05:
Think you may need the `Zigbee 3.0`, which is only for C8 + C8 Pro.

P-R-O-C-H-Y, 2024-12-05:
Main thing is the log from the Arduino device. Your log says the device never joins the network.
```
01:04:46.091 -> [193222][I][ZigbeeCore.cpp:243] esp_zb_app_signal_handler(): Network steering was not successful (status: ESP_FAIL)
01:04:49.319 -> [196449][I][ZigbeeCore.cpp:243] esp_zb_app_signal_handler(): Network steering was not successful (status: ESP_FAIL)
01:04:52.561 -> [199677][I][ZigbeeCore.cpp:243] esp_zb_app_signal_handler(): Network steering was not successful (status: ESP_FAIL)
```
Can you use the latest version? Use manual install of Arduino core and use the `release/v3.1.x` branch please.

ilker-aktuna, 2024-12-05:
> Think you may need the `Zigbee 3.0`, which is only for C8 + C8 Pro.

well if that's the case, then esp32c6 has a serious problem. Because I have more than 100 devices connected to my hub and they all work fine. A lot of different vendors. 
I believe zigbee 3.0 devices have  backwards compatibility normally. If esp32-c6 doesn't have then it is a problem.
On the other hand I have used esp32-c6 with my hub using IDF code instead of Arduino. So it is not a hardware issue.

Also, the hub accepts the device as joined. And control it randomly. The issue is that EP does not accept itself as joined. I believe it is an implementation issue. How can we troubleshoot this ?

ilker-aktuna, 2024-12-05:
> Can you use the latest version? Use manual install of Arduino core and use the `release/v3.1.x` branch please.

How can I do that ?
if I search fore "core" in libraries  I don't see it.

ilker-aktuna, 2024-12-05:
Now I updated esp32 core to 3.1.0-RC3
Behaviour is similar.
When I put the hub in to pairing mode, it detects the c6
then it can control the device (on/off works)
after that hub goes out of pairin mode. At this point it can not control c6 anymore.
If I put the hub in pairing mode again, it detects the device and can control it again.

ilker-aktuna, 2024-12-05:
the device does not understand that it has joined the network.
how can we fix that ?

ilker-aktuna, 2024-12-09:
@P-R-O-C-H-Y 
any ideas to solve this issue ?

P-R-O-C-H-Y, 2024-12-17:
@ilker-aktuna Its in my backlog. It needs to be addition to the code in order to make it work with the hue/sat.

ilker-aktuna, 2025-02-09:
@P-R-O-C-H-Y 
I know this is in your backlog and I know you are working on many things so it may not be a priority for you.
I am also discussing this with Hubitat community and they are telling that Zigbee specification is mainly based on Hue&Sat and not RGB. I also checked on the specification here:
https://zigbeealliance.org/wp-content/uploads/2019/12/07-5123-06-zigbee-cluster-library-specification.pdf

It lists 2 color modes: XY and HueSat. RGB seems to be an option but not the primary one.
So, I really don't understand "why RGB was initially developed on Espressif , BEFORE HueSat ?" 
is there a reason ?

Do you have an estimation about when you can return to this subject ? Because Hubitat side doesn't really seem interested in RGB option as HueSat is primaryly supported by the specification.

P-R-O-C-H-Y, 2025-02-10:
Hi @ilker-aktuna, sorry for this taking so long, but as you mentioned I have been working on adding more features to the Zigbee library over last weeks. I will get to work on it this week, so you may expect a PR soon. I will ping you there and will be happy if you will give it a try before merging it :)

ilker-aktuna, 2025-02-11:
thanks for your work. I downloaded the new branch and will try it in a few minutes.
This is not merged into master yet, right ?

ilker-aktuna, 2025-02-11:
@P-R-O-C-H-Y 
I don't see the example sketch with HSV setting ?

ilker-aktuna, 2025-02-12:
@P-R-O-C-H-Y 
The ColorDimmableLight doesn't have any function to handle hue and sat values.
In any case, since you asked, I tried it now.
On/off and level commands work fine.
color change does not set any color. The led stays white but its level changes.

P-R-O-C-H-Y, 2025-02-12:
@ilker-aktuna I see what I have missed, the hue/sat attributes were not added to the cluster.
Let me add it and can you please try again then?

ilker-aktuna, 2025-02-12:
@P-R-O-C-H-Y 
sure , let me know.
shall I download a new commit ? Or maybe just let me know which file to edit , and I can modify locally

P-R-O-C-H-Y, 2025-02-12:
I will update the PR anyway, so for me it would be better if you can pull the latest commit from the PR branch / or update the changed files manually (for sure will be both cpp and header file of the ColorDimmableLight endpoint).

ilker-aktuna, 2025-02-13:
@P-R-O-C-H-Y 
I see a new commit , but you didn't write it here. So is it ready for test or not ?

P-R-O-C-H-Y, 2025-02-13:
@ilker-aktuna I have pushed the changes, which adds the HUE/SAT attributes and also updates the color capabilities to 9, which means both XY and HUE/SAT is supported by the device. Please give it a try :)

P-R-O-C-H-Y, 2025-02-13:
@ilker-aktuna I have tested the step_hue and step_saturation commands in HomeAsisstant and it worked fine :) So the PR is ready to be merged. It should work on your end too now.

closed 2025-02-13 by me-no-dev

ilker-aktuna, 2025-02-13:
@P-R-O-C-H-Y 

Thanks. It works.
On/off, level and color setting from Hubitat (with HSL) works fine.

ilker-aktuna, 2025-02-13:
@P-R-O-C-H-Y 

As this is now merged into master, can I use the esp32 drivers from release ? I don't see 3.1.2 update on the IDE

P-R-O-C-H-Y, 2025-02-13:
The 3.1.2 should be there. I will check. If the release is not showing for sure you can use master branch now :)

ilker-aktuna, 2025-02-13:
got it now.

PR #10959 (2025-02-13): feat(zigbee): Support HSV color commands for RGB light endpoint
## Description of Change
This PR adds support for HSV color commands.

## Tests scenarios
Tested XY color, please @ilker-aktuna test the HSV colors.

## Related links
Closes #10669