espressif/arduino-esp32 #10670: vendorRequestCallback not working in USBVendor.ino example

ms4sman, 2024-12-02:
### Board

ESP32-S3

### Device Description

DevKitC

### Hardware Configuration

Currently nothing else connected to any pins

### Version

latest development Release Candidate (RC-X)

### IDE Name

CLion with PlatformIO Plugin

### Operating System

Windows

### Flash frequency

40 MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

I'm trying to use the example sketch USBVendor.ino from the examples folder. Everything works good, I can connect with the demo HTML page and communicate both ways and it's good. But the vendorRequestCallback method never seems to be hit. None of the Serial logs from it ever appear in the console.

### Sketch

```cpp
#ifndef ARDUINO_USB_MODE
#error This ESP32 SoC has no Native USB interface
#elif ARDUINO_USB_MODE == 1
#warning This sketch should be used when USB is in OTG mode
void setup() {}
void loop() {}
#else
#include "USB.h"
#include "USBVendor.h"

USBVendor Vendor;
const int buttonPin = 0;

//CDC Control Requests
#define REQUEST_SET_LINE_CODING        0x20
#define REQUEST_GET_LINE_CODING        0x21
#define REQUEST_SET_CONTROL_LINE_STATE 0x22

//CDC Line Coding Control Request Structure
typedef struct __attribute__((packed)) {
  uint32_t bit_rate;
  uint8_t stop_bits;  //0: 1 stop bit, 1: 1.5 stop bits, 2: 2 stop bits
  uint8_t parity;     //0: None, 1: Odd, 2: Even, 3: Mark, 4: Space
  uint8_t data_bits;  //5, 6, 7, 8 or 16
} request_line_coding_t;

static request_line_coding_t vendor_line_coding = {9600, 0, 0, 8};

// Bit 0:  DTR (Data Terminal Ready), Bit 1: RTS (Request to Send)
static uint8_t vendor_line_state = 0;

//USB and Vendor events
static void usbEventCallback(void *arg, esp_event_base_t event_base, int32_t event_id, void *event_data) {
  if (event_base == ARDUINO_USB_EVENTS) {
    arduino_usb_event_data_t *data = (arduino_usb_event_data_t *)event_data;
    switch (event_id) {
      case ARDUINO_USB_STARTED_EVENT: Serial.println("USB PLUGGED"); break;
      case ARDUINO_USB_STOPPED_EVENT: Serial.println("USB UNPLUGGED"); break;
      case ARDUINO_USB_SUSPEND_EVENT: Serial.printf("USB SUSPENDED: remote_wakeup_en: %u\n", data->suspend.remote_wakeup_en); break;
      case ARDUINO_USB_RESUME_EVENT:  Serial.println("USB RESUMED"); break;

      default: break;
    }
  } else if (event_base == ARDUINO_USB_VENDOR_EVENTS) {
    arduino_usb_vendor_event_data_t *data = (arduino_usb_vendor_event_data_t *)event_data;
    switch (event_id) {
      case ARDUINO_USB_VENDOR_DATA_EVENT:
        Serial.printf("Vendor RX: len:%u\n", data->data.len);
        for (uint16_t i = 0; i < data->data.len; i++) {
          Serial.write(Vendor.read());
        }
        Serial.println();
        break;

      default: break;
    }
  }
}

static const char *strRequestDirections[] = {"OUT", "IN"};
static const char *strRequestTypes[] = {"STANDARD", "CLASS", "VENDOR", "INVALID"};
static const char *strRequestRecipients[] = {"DEVICE", "INTERFACE", "ENDPOINT", "OTHER"};
static const char *strRequestStages[] = {"SETUP", "DATA", "ACK"};

//Handle USB requests to the vendor interface
bool vendorRequestCallback(uint8_t rhport, uint8_t requestStage, arduino_usb_control_request_t const *request) {
  Serial.printf(
    "Vendor Request: Stage: %5s, Direction: %3s, Type: %8s, Recipient: %9s, bRequest: 0x%02x, wValue: 0x%04x, wIndex: %u, wLength: %u\n",
    strRequestStages[requestStage], strRequestDirections[request->bmRequestDirection], strRequestTypes[request->bmRequestType],
    strRequestRecipients[request->bmRequestRecipient], request->bRequest, request->wValue, request->wIndex, request->wLength
  );

  bool result = false;

  if (request->bmRequestDirection == REQUEST_DIRECTION_OUT && request->bmRequestType == REQUEST_TYPE_STANDARD
      && request->bmRequestRecipient == REQUEST_RECIPIENT_INTERFACE && request->bRequest == 0x0b) {
    if (requestStage == REQUEST_STAGE_SETUP) {
      // response with status OK
      result = Vendor.sendResponse(rhport, request);
    } else {
      result = true;
    }
  } else
    //Implement CDC Control Requests
    if (request->bmRequestType == REQUEST_TYPE_CLASS && request->bmRequestRecipient == REQUEST_RECIPIENT_DEVICE) {
      switch (request->bRequest) {

        case REQUEST_SET_LINE_CODING:  //0x20
          // Accept only direction OUT with data size 7
          if (request->wLength != sizeof(request_line_coding_t) || request->bmRequestDirection != REQUEST_DIRECTION_OUT) {
            break;
          }
          if (requestStage == REQUEST_STAGE_SETUP) {
            //Send the response in setup stage (it will write the data to vendor_line_coding in the DATA stage)
            result = Vendor.sendResponse(rhport, request, (void *)&vendor_line_coding, sizeof(request_line_coding_t));
          } else if (requestStage == REQUEST_STAGE_ACK) {
            //In the ACK stage the response is complete
            Serial.printf(
              "Vendor Line Coding: bit_rate: %lu, data_bits: %u, stop_bits: %u, parity: %u\n", vendor_line_coding.bit_rate, vendor_line_coding.data_bits,
              vendor_line_coding.stop_bits, vendor_line_coding.parity
            );
          }
          result = true;
          break;

        case REQUEST_GET_LINE_CODING:  //0x21
          // Accept only direction IN with data size 7
          if (request->wLength != sizeof(request_line_coding_t) || request->bmRequestDirection != REQUEST_DIRECTION_IN) {
            break;
          }
          if (requestStage == REQUEST_STAGE_SETUP) {
            //Send the response in setup stage (it will write the data to vendor_line_coding in the DATA stage)
            result = Vendor.sendResponse(rhport, request, (void *)&vendor_line_coding, sizeof(request_line_coding_t));
          }
          result = true;
          break;

        case REQUEST_SET_CONTROL_LINE_STATE:  //0x22
          // Accept only direction OUT with data size 0
          if (request->wLength != 0 || request->bmRequestDirection != REQUEST_DIRECTION_OUT) {
            break;
          }
          if (requestStage == REQUEST_STAGE_SETUP) {
            //Send the response in setup stage
            vendor_line_state = request->wValue;
            result = Vendor.sendResponse(rhport, request);
          } else if (requestStage == REQUEST_STAGE_ACK) {
            //In the ACK stage the response is complete
            bool dtr = (vendor_line_state & 1) != 0;
            bool rts = (vendor_line_state & 2) != 0;
            Serial.printf("Vendor Line State: dtr: %u, rts: %u\n", dtr, rts);
          }
          result = true;
          break;

        default:
          // stall unknown request
          break;
      }
    }

  return result;
}

void setup() {
  pinMode(buttonPin, INPUT_PULLUP);
  Serial.begin(115200);
  Serial.setDebugOutput(true);

  Vendor.onEvent(usbEventCallback);
  Vendor.onRequest(vendorRequestCallback);
  Vendor.begin();

  USB.onEvent(usbEventCallback);
  USB.webUSB(true);
  // Set the URL for your WebUSB landing page
  USB.webUSBURL("https://docs.espressif.com/projects/arduino-esp32/en/latest/_static/webusb.html");
  USB.begin();
}

void loop() {
  static int previousButtonState = HIGH;
  int buttonState = digitalRead(buttonPin);
  if (buttonState != previousButtonState) {
    previousButtonState = buttonState;
    if (buttonState == LOW) {
      Serial.println("Button Pressed");
      Vendor.println("Button Pressed");
      Vendor.flush();  //Without flushing the data will only be sent when the buffer is full (64 bytes)
    } else {
      Serial.println("Button Released");
      Vendor.println("Button Released");
      Vendor.flush();  //Without flushing the data will only be sent when the buffer is full (64 bytes)
    }
    delay(100);
  }

  while (Serial.available()) {
    size_t l = Serial.available();
    uint8_t b[l];
    l = Serial.read(b, l);
    Vendor.write(b, l);
    Vendor.flush();  //Without flushing the data will only be sent when the buffer is full (64 bytes)
  }
}
#endif /* ARDUINO_USB_MODE */
```

### Debug Message

```plain
I am not getting output from the method mentioned above, that's the problem.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

yuvashrikarunakaran, 2024-12-04:
#ifndef ARDUINO_USB_MODE
#error This ESP32 SoC has no Native USB interface
#elif ARDUINO_USB_MODE == 1
#warning This sketch should be used when USB is in OTG mode
void setup() {}
void loop() {}
#else
#include "USB.h"
#include "USBVendor.h"

USBVendor Vendor;
const int buttonPin = 0;

// CDC Control Requests
#define REQUEST_SET_LINE_CODING        0x20
#define REQUEST_GET_LINE_CODING        0x21
#define REQUEST_SET_CONTROL_LINE_STATE 0x22

typedef struct __attribute__((packed)) {
  uint32_t bit_rate;
  uint8_t stop_bits;
  uint8_t parity;
  uint8_t data_bits;
} request_line_coding_t;

static request_line_coding_t vendor_line_coding = {9600, 0, 0, 8};
static uint8_t vendor_line_state = 0;

static void usbEventCallback(void *arg, esp_event_base_t event_base, int32_t event_id, void *event_data) {
  if (event_base == ARDUINO_USB_EVENTS) {
    switch (event_id) {
      case ARDUINO_USB_STARTED_EVENT: Serial.println("USB PLUGGED"); break;
      case ARDUINO_USB_STOPPED_EVENT: Serial.println("USB UNPLUGGED"); break;
      case ARDUINO_USB_SUSPEND_EVENT: Serial.println("USB SUSPENDED"); break;
      case ARDUINO_USB_RESUME_EVENT: Serial.println("USB RESUMED"); break;
    }
  }
}

bool vendorRequestCallback(uint8_t rhport, uint8_t requestStage, arduino_usb_control_request_t const *request) {
  Serial.println("vendorRequestCallback triggered");
  Serial.printf(
    "Vendor Request: Stage: %s, Direction: %s, Type: %s, Recipient: %s, bRequest: 0x%02x, wValue: 0x%04x, wIndex: %u, wLength: %u\n",
    (requestStage == REQUEST_STAGE_SETUP) ? "SETUP" : "UNKNOWN",
    (request->bmRequestDirection == REQUEST_DIRECTION_OUT) ? "OUT" : "IN",
    (request->bmRequestType == REQUEST_TYPE_CLASS) ? "CLASS" : "OTHER",
    (request->bmRequestRecipient == REQUEST_RECIPIENT_DEVICE) ? "DEVICE" : "OTHER",
    request->bRequest, request->wValue, request->wIndex, request->wLength
  );

  if (request->bmRequestType == REQUEST_TYPE_CLASS && request->bRequest == REQUEST_SET_LINE_CODING) {
    if (requestStage == REQUEST_STAGE_SETUP) {
      Vendor.sendResponse(rhport, request, &vendor_line_coding, sizeof(vendor_line_coding));
    }
    return true;
  }

  return false; // Stall unknown requests
}

void setup() {
  pinMode(buttonPin, INPUT_PULLUP);
  Serial.begin(115200);

  Serial.println("Setting up USB Vendor...");
  Vendor.onEvent(usbEventCallback);
  Vendor.onRequest(vendorRequestCallback);
  if (!Vendor.begin()) {
    Serial.println("Failed to initialize USB Vendor");
  } else {
    Serial.println("USB Vendor initialized successfully");
  }

  USB.onEvent(usbEventCallback);
  USB.webUSB(true);
  USB.webUSBURL("https://docs.espressif.com/projects/arduino-esp32/en/latest/_static/webusb.html");
  USB.begin();
}

void loop() {
  if (digitalRead(buttonPin) == LOW) {
    Vendor.println("Button Pressed");
    Vendor.flush();
    delay(100);
  }

  while (Serial.available()) {
    Vendor.write(Serial.read());
    Vendor.flush();
  }
}
#endif /* ARDUINO_USB_MODE */

ms4sman, 2024-12-05:
I'm a bit confused. Is this meant to be an alternate code for me to try? If so, it doesn't even seem valid. the Vendor.begin method returns void as far as I can tell, both in 3.0.7 and 3.1.x. So that check thats assuming a bool isn't going to work...

ms4sman, 2024-12-05:
That said, I took out that check so that it would build, and flashed it onto my ESP32 and tried it. Still nothing at all in the console it looks like when receiving messages from the browser or anything else that should be triggering them.

It looks like I may have misspoken in my first post on this topic. As far as I know, on Core 3.0.7 I have never been able to receive a message back from the browser or any other incoming message that should get logged with the vendor request or vendor event callbacks.

ms4sman, 2024-12-13:
I have been continuing to mess with this while waiting for responses and I think I may just be even more confused than I was before. But here's some additional notes I have gathered so far.

I have been testing this using both the Arduino IDE and also PlatformIO with the same results. I have tried many different combinations of USB settings in Arduino IDE and build flags for PIO with varying results.

What I have narrowed down though is this:

- On Core 2.0.17 the example worked well to connect to the browser and send messages from the browser to the ESP32. However, there was a problem with the flush command on the Vendor that meant calling flush didn't actually work and it would only send from ESP32 to browser when the buffer filled. You can't manually flush it. 
- On Core 2.0.17 the vendorRequestCallback code never seems to get hit ever that I can find. None of the Serial logs in there ever appear and I can find no evidence that any of that is being hit.
- On Core 2.0.17 the ARDUINO_USB_EVENTS work good and get logged properly on plug and unplug.

- On Core 3.0.7 the example works different. Now sending from the ESP32 to the browser works, and the flush issue is fixed. But messages sent from the browser to the ESP32 never seem to be arriving. There's no error in the browser, it seems to think it was sent just fine. But the ESP32 never logs it as it should in the code under ARDUINO_USB_VENDOR_EVENTS. This did work in 2.0.17 however.
- On Core 3.0.7 the vendorRequestCallback stuff still does not work. Seems exactly the same as on 2.0.17. Never hit that I can find. I'd really like to use this though to determine when WebUSB actually connects.
- On Core 3.0.7 the ARDUINO_USB_EVENTS still work just fine as far as I can tell.

Hopefully these additional notes are helpful. I'd really like to get this figured out! Let me know if I can provide any other useful info to help in debugging this!

Csontikka, 2025-01-14:
Hi!
I’ve encountered the same issue with my s3 devkit c, and I wanted to ask if you’ve found a solution or maybe a workaround since then? Thank you in advance for your help!

ms4sman, 2025-01-15:
Hi @Csontikka, unfortunately I have not figured anything out yet. I have kind of put the project aside for now since I really need this in order for it to work. I hope they can resolve it soon so I can resume the project! If you find anything, let me know!

Ddvos, 2025-01-16:
Hi @ms4sman & @Csontikka I have also the same issue. I can send data from the ESP32 S3 to the browser. But the otherway around wont work. I also tried this library [Adafruit ](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/tree/master) but that one is just working the otherway around. Via that library i can only send data from the browser to the ESP32 S3.

I am looking for a solution to send data in both directions, you would think that woulndt be that hard?
With an Arduino Pro Micro which has an ATmega32U4 it's quit easy to setup USB communication

ms4sman, 2025-01-16:
> Hi [@ms4sman](https://github.com/ms4sman) & [@Csontikka](https://github.com/Csontikka) I have also the same issue. I can send data from the ESP32 S3 to the browser. But the otherway around wont [...]

I have had the exact same findings with the Adafruit library as well unfortunately. I hope that one or the other gets resolved soon!

ms4sman, 2025-01-16:
For myself, I only have the ESP32, no ATmega so I'll let one of the others fill in about that if they have one.

In my testing though, I have this problem just using the included example sketch (also included in my original post) and using the demo page that goes along with it. The problem is present also in my project but in the process of tracking it down I ended up stripping back to just the original example and found that it happens there too.

Thanks!

Ddvos, 2025-01-18:
@me-no-dev this library[WebUSB  ️Arduino](https://github.com/webusb/arduino/tree/gh-pages) is working with the ATmega32U4 and can send and receive data. You can find an example under demos/rgb/sketch/sketch.ino the working is almost the same with Serial.read() and Serial.write() , but this one is not working with the ESP32 s2 or s3 maybe you can use it as an example?

I think it is a small thing that should be changed in de library because the Adafruit version can send data from the browser to the ESP S3 it would be realy nice if you can help us with this

ms4sman, 2025-02-07:
Any progress on this? My project is kind of on hold in the meantime until I can get bi-directional communications working. 

If there's anything I can do to help, let me know! I just don't have any other devices besides the ESP32-S3 so I unfortunately can't help with testing on other hardware.

Ddvos, 2025-02-11:
I still have the same issue would be fine if someone could fix this...

ms4sman, 2025-06-19:
Has there by chance been any progress on this issue? My project is still on hold pending a solution for this.

@Ddvos did you find any sort of workaround either with this library or any other way to accomplish communications both ways?