espressif/arduino-esp32 #10675: assert failed: sntp_stop /IDF/components/lwip/lwip/src/apps/sntp/sntp.c:696 (Required to lock TCPIP core functionality!)

holgerlembke, 2024-12-03:
### Board

eps32 s3 dev module

### Device Description

yes

### Hardware Configuration

no

### Version

latest development Release Candidate (RC-X)

### IDE Name

Arduino-IDE

### Operating System

Win10

### Flash frequency

yes

### PSRAM enabled

yes

### Upload speed

yes

### Description

assert failed: sntp_stop /IDF/components/lwip/lwip/src/apps/sntp/sntp.c:696 (Required to lock TCPIP core functionality!) is raised.

### Sketch

```cpp
#include <WiFi.h>
#include <esp_sntp.h>

void setup() {
  configTzTime("CET-1CEST,M3.5.0,M10.5.0/3", "pool.ntp.org", "");
  configTzTime("CET-1CEST,M3.5.0,M10.5.0/3", "pool.ntp.org", "");
}

void loop() {}
```

### Debug Message

```plain
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4037b4ee
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x10f0
load:0x403c8700,len:0x4
load:0x403c8704,len:0xbec
load:0x403cb700,len:0x2fcc
entry 0x403c88ac
[     0][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type USB_DM (45) successfully set to 0x4200abe0
[    12][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type USB_DP (46) successfully set to 0x4200abe0
[    23][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x4200a568
[    34][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x4200a534
[    46][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x4200a500
[    57][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x4200a4cc
[    68][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x4200a568
[    80][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x4200a534
[    91][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x4200a500
[   102][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x4200a4cc
[   114][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x4200a568
[   125][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x4200a534
[   136][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x4200a500
[   148][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x4200a4cc
[   172][I][esp32-hal-psram.c:102] psramAddToHeap(): PSRAM added to the heap.
[   191][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 44 successfully set to type UART_RX (2) with bus 0x3fc9670c
[   202][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 43 successfully set to type UART_TX (3) with bus 0x3fc9670c
[   213][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 19 successfully set to type USB_DM (45) with bus 0x3fc9c45c
[   223][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 20 successfully set to type USB_DP (46) with bus 0x3fc9c45c
=========== Before Setup Start ===========
Chip Info:
------------------------------------------
  Model             : ESP32-S3
  Package           : 0
  Revision          : 0.01
  Cores             : 2
  CPU Frequency     : 240 MHz
  XTAL Frequency    : 40 MHz
  Features Bitfield : 0x00000012
  Embedded Flash    : No
  Embedded PSRAM    : No
  2.4GHz WiFi       : Yes
  Classic BT        : No
  BT Low Energy     : Yes
  IEEE 802.15.4     : No
------------------------------------------
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   358124 B ( 349.7 KB)
  Free Bytes        :   324372 B ( 316.8 KB)
  Allocated Bytes   :    28624 B (  28.0 KB)
  Minimum Free Bytes:   320568 B ( 313.1 KB)
  Largest Free Block:   258036 B ( 252.0 KB)
------------------------------------------
SPIRAM Memory Info:
------------------------------------------
------------------------------------------
Flash Info:
------------------------------------------
  Chip Size         :  4194304 B (4 MB)
  Block Size        :    65536 B (  64.0 KB)
  Sector Size       :     4096 B (   4.0 KB)
  Page Size         :      256 B (   0.2 KB)
  Bus Speed         : 80 MHz
  Bus Mode          : QIO
------------------------------------------
Partitions Info:
------------------------------------------
                nvs : addr: 0x00009000, size:    20.0 KB, type: DATA, subtype: NVS
            otadata : addr: 0x0000E000, size:     8.0 KB, type: DATA, subtype: OTA
               app0 : addr: 0x00010000, size:  1280.0 KB, type:  APP, subtype: OTA_0
               app1 : addr: 0x00150000, size:  1280.0 KB, type:  APP, subtype: OTA_1
               ffat : addr: 0x00290000, size:  1408.0 KB, type: DATA, subtype: FAT
           coredump : addr: 0x003F0000, size:    64.0 KB, type: DATA, subtype: COREDUMP
------------------------------------------
Software Info:
------------------------------------------
  Compile Date/Time : Dec  3 2024 19:52:04
  Compile Host OS   : windows
  ESP-IDF Version   : v5.3.1-638-ga0f798cfc4-dirty
  Arduino Version   : 3.1.0
------------------------------------------
Board Info:
------------------------------------------
  Arduino Board     : LOLIN_S3_MINI
  Arduino Variant   : lolin_s3_mini
  Arduino FQBN      : esp32:esp32:lolin_s3_mini:UploadSpeed=921600,USBMode=hwcdc,CDCOnBoot=cdc,MSCOnBoot=default,DFUOnBoot=default,UploadMode=cdc,CPUFreq=240,FlashMode=qio120,PartitionScheme=defaultffat,DebugLevel=verbose,LoopCore=1,EventsCore=1,EraseFlash=none
============ Before Setup End ============

assert failed: sntp_stop /IDF/components/lwip/lwip/src/apps/sntp/sntp.c:696 (Required to lock TCPIP core functionality!)

Backtrace: 0x40376586:0x3fca9cb0 0x4037dd91:0x3fca9cd0 0x40384801:0x3fca9cf0 0x4201cf1e:0x3fca9e20 0x42008917:0x3fca9e40 0x42003076:0x3fca9e60 0x4200898b:0x3fca9e80 0x4037eaa6:0x3fca9ea0

ELF file SHA256: 0a6c2ca4c
```

### Other Steps to Reproduce

no

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2024-12-03:
I'm unable to reproduce.  
Do you get it if you call `WiFi.mode(WIFI_STA)` before the configTzTime call?

me-no-dev, 2024-12-03:
which RC is this? Please write better titles for your issues and give some explanation of your build config

yuvashrikarunakaran, 2024-12-04:
#include <WiFi.h>
#include <esp_sntp.h>

// Function to initialize SNTP with proper checks
void initializeSNTP(const char* timezone, const char* server1, const char* server2 = "") {
  static bool sntp_initialized = false; // Track if SNTP is already initialized

  // Stop SNTP if it's already running
  if (sntp_initialized) {
    sntp_stop(); // Properly stop SNTP before reinitializing
    sntp_initialized = false;
  }

  // Set timezone and initialize SNTP
  configTzTime(timezone, server1, server2);

  // Mark SNTP as initialized
  sntp_initialized = true;
}

void setup() {
  Serial.begin(115200);
  WiFi.begin("your-SSID", "your-PASSWORD");

  // Wait for Wi-Fi connection
  while (WiFi.status() != WL_CONNECTED) {
    delay(1000);
    Serial.println("Connecting to WiFi...");
  }
  Serial.println("Connected to WiFi!");

  // Initialize SNTP once
  initializeSNTP("CET-1CEST,M3.5.0,M10.5.0/3", "pool.ntp.org");
  // If you need to update the time zone, call `initializeSNTP()` again
  // with the new parameters.
}

void loop() {
  // No operations in the loop
}

holgerlembke, 2024-12-04:
@lbernstone 

```WiFi.mode(WIFI_STA)```  does not change anything.

3.1.0-RC3

Board is a Lolin/Wemos S3 Mini (those small pink boards).

holgerlembke, 2024-12-04:
Thanks @yuvashrikarunakaran, I did almost the same code to avoid the assert.

But I think that is not the solution. There should not be an assert and configTzTime should handle that stuff internally. Confusingly @lbernstone can not reproduce it... did you get the assert?

me-no-dev, 2024-12-05:
@yuvashrikarunakaran please read on how to paste code in github comments. Just pasting it in the box does not work well

holgerlembke, 2024-12-17:
https://github.com/espressif/arduino-esp32/releases/tag/3.1.0 fixes this.

Is that by chance or intended? Anyone?

me-no-dev, 2024-12-17:
@holgerlembke a fix was merged :)

closed 2024-12-17 by me-no-dev

holgerlembke, 2024-12-17:
What? Who? Where?

Seriously, can you point me to the solution? Thanks.

me-no-dev, 2024-12-17:
https://github.com/espressif/arduino-esp32/pull/10725