espressif/arduino-esp32 #10679: ESP32 Read unique Flash ID

s54b32julian, 2024-12-04:
### Board

ESP32-S3 N16R8

### Device Description

plain modul

### Hardware Configuration

plain ÉSP32-S3 N16R8

### Version

v3.0.6

### IDE Name

VSCode

### Operating System

Linux 

### Flash frequency

40Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

I want to read the Unique flash ID with esp_flash_read_unique_chip_id() per SPI Flash API.
But as i get back 0x102, there must be sometime wrong with the chip parameter.
So what do i have to pass instead of NULL as esp_flash_t *chip parameter? I dont see it available anywhere?

### Sketch

```cpp
uint64t *CHIPuniqueID;

esp_flash_read_unique_chip_id(NULL, CHIPuniqueID);

Serial.println(esp_flash_read_unique_chip_id(NULL, CHIPuniqueID)); -->returns 258d
```

### Debug Message

```plain
see above
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

yuvashrikarunakaran, 2024-12-04:
#include "esp_system.h"
#include "esp_flash.h"
#include "esp_log.h"

void setup() {
  Serial.begin(115200);
  delay(1000); // Give some time for Serial to initialize
  
  uint64_t chipUniqueID = 0; // Variable to store the unique chip ID
  esp_err_t result = esp_flash_read_unique_chip_id(esp_flash_default_chip, &chipUniqueID);

  if (result == ESP_OK) {
    Serial.print("Unique Chip ID: ");
    Serial.println(chipUniqueID, HEX); // Print in hexadecimal format
  } else {
    Serial.print("Failed to read chip ID, error code: ");
    Serial.println(result);
  }
}

void loop() {
  // Nothing to do here
}

s54b32julian, 2024-12-04:
Super quick, thank you so much for solving this!

closed 2024-12-04 by s54b32julian