espressif/arduino-esp32 #10683: ESP32-MINI-1 (ESP32-U4WDH) identified by ESP.getChipModel() as ESP32-PICO-D2

mstegen, 2024-12-04:
### Board

ESP-MINI-1 module

### Device Description

ESP module on custom hardware

### Hardware Configuration

All I/O used

### Version

v2.0.17

### IDE Name

VSCode

### Operating System

Linux and Win10

### Flash frequency

40Mhz

### PSRAM enabled

no

### Upload speed

2000000

### Description

New hardware with instead of a ESP32-WROOM-32E a ESP32-MINI-1, gives me 2 extra IO pins which come in handy.
I want the existing software to detect the new module/chip with ESP.getChipModel(), so it can configure the correct I/O based on the detected chip.
When programming, the chip is correctly identified, and everything is working fine.

```Uploading .pio\build\release\firmware.bin
esptool.py v4.5.1
Serial port COM3
Connecting.........
Chip is ESP32-U4WDH (revision v3.1)
```

But when i use ESP.getChipModel() in my code, the wrong Chiptype is reported: ESP32-PICO-D2
When browsing through the Arduino code (Esp.cpp) i see that there is no mention anywhere of the ESP32-U4WDH chip..

### Sketch

```cpp
Serial.printf("ESP32 Chip model:%s\n",ESP.getChipModel());
```

### Debug Message

```plain
ESP32-WROOM-32 module:
ESP32 Chip model:ESP32-D0WD-V3 

ESP32-MINI-1 module:
ESP32 Chip model:ESP32-PICO-D2
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2024-12-05:
Older versions of the software don't know about newer chips.  What does v3.0.7 show for the chip model?

mstegen, 2024-12-06:
2.0.17 does have the correct chiptype in it's headers already.
A fix would be to replace
https://github.com/espressif/arduino-esp32/blob/c7022caeda46c216bb2a00081deb9810d66f33ed/cores/esp32/Esp.cpp#L277
with 
`case EFUSE_RD_CHIP_VER_PKG_ESP32U4WDH:   return "ESP32-U4WDH";`

EFUSE_RD_CHIP_VER_PKG_ESP32U4WDH is already defined here:
https://github.com/espressif/esp-idf/blob/master/components/soc/esp32/include/soc/efuse_defs.h

closed 2024-12-09 by SuGlider