espressif/arduino-esp32 #10684: Motion sensor example for Zigee

ilker-aktuna, 2024-12-04:
### Related area

Zigbee , ZHA

### Hardware specification

Support for motion sensor

### Is your feature request related to a problem?

I am working on a motion sensor. Is the motion sensor feature available in Zigbee stack ?
if yes, where can I find an example code ?

### Describe the solution you'd like

motion sensor zigbee example

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

P-R-O-C-H-Y, 2024-12-05:
Hi @ilker-aktuna, there is currently no PIR endpoint available in the Zigbee library.
The occupancy cluster (PIR) is available in esp-zigbee-sdk, so if you want you can create your own implementation of this endpoint type. There is currently no plan to add it. But I will have this in the list of feature requests for the Zigbee library.

ilker-aktuna, 2024-12-05:
ok. what is the correct path for creating my own implementation ?
For example, there is the temp. sensor example. Where can I find the code for temperature sensor ? maybe I can clone it and change the cluster IDs etc.

ilker-aktuna, 2024-12-05:
ok thank you.
Which source should I look to find options like "ESP_ZB_DEFAULT_TEMPERATURE_SENSOR_CONFIG" or "ESP_ZB_ZCL_CLUSTER_ID_REL_HUMIDITY_MEASUREMENT" ?

P-R-O-C-H-Y, 2024-12-06:
There is no default config for PIR motion sensor in esp-zigbee-sdk. You may need to create your own "default config", which means declaring the clusters it should contain.

To be honest it's not that simple to do a new EP type with all the stuff around. Maybe you can keep this feature request open and I will take a look into it and add the implementation in future updates.

ilker-aktuna, 2024-12-09:
ok. good.
In fact, if I had time, I would like to learn and add a new EP type as a contribution but I won't have time to learn soon.
I'll be waiting. But if I do something, I'll inform here.

P-R-O-C-H-Y, 2024-12-12:
Hi @ilker-aktuna, you can check the PR I linked. I have added the PIR motion sensor (occupancy) endpoint.
Feel free to test. I have been testing using C6 + HC-SR501 PIR module connecting to HomeAssistant.

michapr, 2024-12-12:
sorry for hijacking :-)) -> could you add environmental sensors too? CO2, VOC sensor?
(don't want to open new issue....)

P-R-O-C-H-Y, 2024-12-12:
In progress right now :) almost done with the CO2.

ilker-aktuna, 2024-12-12:
> Hi @ilker-aktuna, you can check the PR I linked. I have added the PIR motion sensor (occupancy) endpoint. Feel free to test. I have been testing using C6 + HC-SR501 PIR module connecting to HomeAssistant.

I could get the example from the PR but I also need the updated core library to test this.
how can I get the unmerged code for the library ?

ilker-aktuna, 2024-12-13:
@P-R-O-C-H-Y 
I have downloaded the branch with PR but I can't find any way to update the esp32-core on my Windows pc.
is that even possible ?

P-R-O-C-H-Y, 2024-12-13:
@ilker-aktuna Please follow the Manual installation part of documentation. You can use the branch you downloaded instead of cloning :) https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html#windows-manual-installation

ilker-aktuna, 2024-12-13:
ok I unzipped the folder to the Arduino/hardware/espressif/esp32 folder and then used the "get.exe" under tool folder.
 but on my PC , under Sketchbook Directory there was no "hardware" folder.
So now I believe I have 2 installations of arduino-esp32 on my PC (one is the default one which is accessible through "boards manager", the other one is your branch which I installed manually)
How shall I use the one that I manually installed ?

P-R-O-C-H-Y, 2024-12-13:
@ilker-aktuna If you go to the Arduino IDE Tools menu -> Board:
You can see 2 options for ESP32 Arduino. The one with `(in sketchbook)` is the manual installed core.
So you can see in the image, I am using the manually installed core for developing :)

[image]

ilker-aktuna, 2024-12-15:
ok. I tried that now , but I get the following error when I try to compile:

> The system cannot find the path specified.
> exit status 1
> 
> Compilation error: exit status 1
>

ilker-aktuna, 2024-12-15:
ok. installed and trying the example , but when I try to compile, I get this error:

> error: 'OCCUPANCY_SENSOR_ENDPOINT_NUMBER' was not declared in this scope
>    47 | ZigbeeOccupancySensor zbOccupancySensor = ZigbeeOccupancySensor(OCCUPANCY_SENSOR_ENDPOINT_NUMBER);
>       |                                                                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
> 
> exit status 1
> 
> Compilation error: 'OCCUPANCY_SENSOR_ENDPOINT_NUMBER' was not declared in this scope

ilker-aktuna, 2024-12-15:
ok. now I compiled and trying with my Hubitat hub.
This time with a Hubitat C8

The device starts , then I start pairing on my Hub.
Hub finds and says "Found a Zigbee device, initializing..."
But it gets stuck there.

On the device, I see that it is rebooting while tryign to connect to Zigbee network:

> 
> 13:03:00.130 -> Starting Zigbee...
> 13:03:00.161 -> [  2201][I][ZigbeeCore.cpp:138] zigbeeInit(): List of registered Zigbee EPs:
> 13:03:00.161 -> [  2208][I][ZigbeeCore.cpp:140] zigbeeInit(): Device type: Simple Sensor device, Endpoint: 10, Device ID: 0x000c
> 13:03:00.201 -> [  2221][I][ZigbeeCore.cpp:219] esp_zb_app_signal_handler(): Zigbee stack initialized
> 13:03:00.201 -> [  2230][I][ZigbeeCore.cpp:225] esp_zb_app_signal_handler(): Device started up in  factory-reset mode
> 13:03:00.201 -> [  2239][I][ZigbeeCore.cpp:232] esp_zb_app_signal_handler(): Start network steering
> 13:03:00.201 -> Zigbee started successfully!
> 13:03:00.201 -> Connecting to network
> 13:03:00.201 -> ..............................ESP-ROM:esp32c6-20220919
> 13:03:03.207 -> Build:Sep 19 2022
> 13:03:03.251 -> rst:0xc (SW_CPU),boot:0x8 (SPI_FAST_FLASH_BOOT)
> 13:03:03.251 -> Saved PC:0x4001975a
> 13:03:03.251 -> SPIWP:0xee
> 13:03:03.251 -> mode:DIO, clock div:2
> 13:03:03.251 -> load:0x40875720,len:0x1228
> 13:03:03.251 -> load:0x4086c110,len:0xd9c
> 13:03:03.251 -> load:0x4086e610,len:0x2f74
> 13:03:03.251 -> entry 0x4086c110
> 13:03:03.482 -> [0;31mE (274) ESP_ZIGBEE_CLUSTER: The requested add cluster ID:0x3 is already existed[0m
> 13:03:04.518 -> Zigbee TEST

ilker-aktuna, 2024-12-15:
@P-R-O-C-H-Y 

if I put the log level to "verbose" , I get this line before reboot:

> 13:08:27.383 -> Connecting to network
> 13:08:27.383 -> .............................[ 5289][V][ZigbeeCore.cpp:344] factoryReset(): Factory resetting Zigbee stack, device will reboot
> 13:08:30.300 -> .ESP-ROM:esp32c6-20220919
>

P-R-O-C-H-Y, 2024-12-15:
This `ZDO_SIGNAL_LEAVE` is recieved when device is removed from the network. This means the hubitat removes the device. 

If you want to investigate this further, you will need to sniff the Zigbee network and send us a pcap file from the WireShark. Do sou have another C6/H2 devkit available?

ilker-aktuna, 2024-12-15:
@P-R-O-C-H-Y 
I have another C6 
how do I sniff ?

P-R-O-C-H-Y, 2024-12-15:
@ilker-aktuna please follow this comment: https://github.com/espressif/arduino-esp32/issues/10601#issuecomment-2527619657

ilker-aktuna, 2024-12-15:
@P-R-O-C-H-Y 
is there an easier way of creating a sniffer ?
I have IDF 5.2 and I'M really not using it. I am not sure how to update to 5.3
is there a prebuilt binary maybe ?

ilker-aktuna, 2024-12-15:
@P-R-O-C-H-Y 
I compiled with 5.2 (attached)
I hope it works..
[esp_ot_rcp.zip](https://github.com/user-attachments/files/18141323/esp_ot_rcp.zip)

ilker-aktuna, 2024-12-15:
@P-R-O-C-H-Y 

I followed the tutorial and did everything. 
However, wireshark shows "no packet" after this command:

> 
> python sniffer.py -c 13 -u COM19 --crc --rssi -b 460800 | "C:\Program Files\Wireshark\Wireshark.exe" -k -i -
> 
> 
> Initializing sniffer...
>  ** (wireshark:13284) 20:34:44.119428 [Capture MESSAGE] -- Capture Start ...

on COM19 , I have the esp32-c6 with esp_ot_rcp binary
it is the correct port  , if disconnect the cable, wireshark says "end of file on pipe"

then what is missing ?

P-R-O-C-H-Y, 2024-12-16:
> @P-R-O-C-H-Y
> 
> I followed the tutorial and did everything. However, wireshark shows "no packet" after this command:
> 
> > python sniffer.py -c 13 -u COM19 --crc --rssi -b 460800 | "C:\Program Files\Wireshark\Wireshark.exe" -k -i -
> > Initializing sniffer...
> > ** (wireshark:13284) 20:34:44.119428 [Capture MESSAGE] -- Capture Start ...
> 
> on COM19 , I have the esp32-c6 with esp_ot_rcp binary it is the correct port , if disconnect the cable, wireshark says "end of file on pipe"
> 
> I have the key "5A:69:67:42:65:65:41:6C:6C:69:61:6E:63:65:30:39" on pre-configured keys list Wireshark protocol zigbee.
> 
> then what is missing ?

The `-c 13` in the command means your network channel. You need to check in your system on what channel is your Zigbee network running and change it to the proper channel.

P-R-O-C-H-Y, 2024-12-16:
I am closing this issue as it's solved by #10720. But let's continue here with the "debugging" on the Hubitat.

closed 2024-12-16 by P-R-O-C-H-Y

ilker-aktuna, 2024-12-16:
@P-R-O-C-H-Y 
I changed the parameter to 20 (my Hubitat is using channel 20 for zigbee)
but nothing changed.
no packets.
anything else that I should check ?

P-R-O-C-H-Y, 2024-12-17:
@ilker-aktuna I don't know honestly what may be wrong. I assume you set up everything correctly by following the manual.

ilker-aktuna, 2024-12-17:
@P-R-O-C-H-Y 
I believe so. Maybe my ot_rcp binary is not good. Can you share a working binary that I can upload to c6 using esptool ?

P-R-O-C-H-Y, 2024-12-17:
@ilker-aktuna Sharing a binaries for Zigbee (OpenThread) sniffer for both C6 and H2.
The C6 I build right now, and the H2 is the one I have flashed on my boards that I am using to sniff the network.
[Zigbee:OT Sniffer binaries.zip](https://github.com/user-attachments/files/18164168/Zigbee.OT.Sniffer.binaries.zip)

ilker-aktuna, 2024-12-17:
@P-R-O-C-H-Y 
now I tried on another laptop. Still no packet
I'm probably missing something. But what is it ?

ilker-aktuna, 2024-12-17:
This is the channel info when I scan with Hubitat:
[image]

ch 11 is Hue hub
ch 20 is Hubitat

P-R-O-C-H-Y, 2024-12-17:
Its hard to tell as I did not experienced that issue.
Can you just to be sure post a log from the ESP? Just open some serial monitor and reset the esp

ilker-aktuna, 2024-12-17:
@P-R-O-C-H-Y 
you mean the sniffer esp , right ?

I got the log on idf monitor:

> PS D:\esp\tools\Espressif\frameworks\esp-idf-v5.3.1\examples\openthread\ot_rcp> idf.py -p COM19 monitor
> Executing action: monitor
> Running idf_monitor in directory D:\esp\tools\Espressif\frameworks\esp-idf-v5.3.1\examples\openthread\ot_rcp
> Executing "D:\esp\tools\Espressif\python_env\idf5.2_py3.11_env\Scripts\python.exe D:\esp\esp-idf\tools/idf_monitor.py -p COM19 -b 115200 --toolchain-prefix riscv32-esp-elf- --target esp32c6 --revision 0 --decode-panic backtrace D:\esp\tools\Espressif\frameworks\esp-idf-v5.3.1\examples\openthread\ot_rcp\build\esp_ot_rcp.elf --force-color -m 'D:\esp\tools\Espressif\python_env\idf5.2_py3.11_env\Scripts\python.exe' 'D:\esp\esp-idf\tools\idf.py' '-p' 'COM19'"...
> --- WARNING: GDB cannot open serial ports accessed as COMx
> --- Using \\.\COM19 instead...
> --- esp-idf-monitor 1.4.0 on \\.\COM19 115200 ---
> --- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
> ESP-ROM:esp32c6-20220919
> Build:Sep 19 2022
> rst:0x15 (USB_UART_HPSYS),boot:0xc (SPI_FAST_FLASH_BOOT)
> Saved PC:0x40805088
> 0x40805088: clk_ll_bbpll_set_config at D:/esp/esp-idf/components/hal/esp32c6/include/hal/clk_tree_ll.h:329
>  (inlined by) rtc_clk_bbpll_configure at D:/esp/esp-idf/components/esp_hw_support/port/esp32c6/rtc_clk.c:168
>  (inlined by) rtc_clk_cpu_freq_set_config at D:/esp/esp-idf/components/esp_hw_support/port/esp32c6/rtc_clk.c:274
> 
> SPIWP:0xee
> mode:DIO, clock div:2
> load:0x40875720,len:0x1344
> load:0x4086c110,len:0xdbc
> load:0x4086e610,len:0x2b20
> entry 0x4086c11a
> I (23) boot: ESP-IDF v5.3.1-638-ga0f798cfc4 2nd stage bootloader
> I (23) boot: compile time Dec 17 2024 11:41:49
> I (24) boot: chip revision: v0.1
> I (27) boot.esp32c6: SPI Speed      : 80MHz
> I (32) boot.esp32c6: SPI Mode       : DIO
> I (37) boot.esp32c6: SPI Flash Size : 2MB
> I (41) boot: Enabling RNG early entropy source...
> I (47) boot: Partition Table:
> I (50) boot: ## Label            Usage          Type ST Offset   Length
> I (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000
> I (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000
> I (72) boot:  2 factory          factory app      00 00 00010000 00100000
> I (80) boot: End of partition table
> I (84) esp_image: segment 0: paddr=00010020 vaddr=42028020 size=0222ch (  8748) map
> I (95) esp_image: segment 1: paddr=00012254 vaddr=40800000 size=05dc4h ( 24004) load
> I (106) esp_image: segment 2: paddr=00018020 vaddr=42000020 size=26fd4h (159700) map
> I (142) esp_image: segment 3: paddr=0003effc vaddr=40805dc4 size=06618h ( 26136) load
> I (149) esp_image: segment 4: paddr=0004561c vaddr=4080c3e0 size=012a0h (  4768) load
> I (154) boot: Loaded app from partition at offset 0x10000
>

P-R-O-C-H-Y, 2024-12-17:
It looks OK, so its booting fine.
Are you able to get any logs from the Hubitat? I really don't know how to fix the sniffer if you went through the steps.

ilker-aktuna, 2024-12-17:
I will try on a Raspberry PI now. First I have to install wireshark and pyspinel
Does wireshark run on CLI ?

P-R-O-C-H-Y, 2024-12-17:
> I will try on a Raspberry PI now. First I have to install wireshark and pyspinel Does wireshark run on CLI ?

I am sorry but you will have to find out. Try searching it as you know what OS and RPi you are running.

ilker-aktuna, 2024-12-17:
@P-R-O-C-H-Y 
installed a fresh copy of raspbian. then installed wireshark and pyspinel
USB port is identified as /dev/ttyACM0
I run the command:
python sniffer.py -d 3 -c 20 -u /dev/ttyACM0 --crc --rssi -b 460800 | wireshark -k -i -

no errors
but also no packet received.

So I tried 2 different Windows installations, 1 raspberry pi installation
Also tried 2 different ot-rcp binaries (one self compiled, one from you)
Also tried 2 different esp32-c6 devices.

None of these scenarios received a packet.

I assume there is something wrong, or that I do miss.
Unfortunately I could not succeed on sniffing zigbee.

Btw, today I tried the occupancy sensor on my friend's home assistant setup with a zigbee dongle.
It also stuck on pairing process.
All I can say is that this does not work on Hubitat and Home Assistant.

How else can I troubleshoot ?

michapr, 2024-12-19:
@P-R-O-C-H-Y  I have tried to burn ESP32C6 with your bin files (Sniffer), but do not get any output.
what about the partition-table.bin? 
Have flashed with bootloader.bin at 0x0, esp_ot_rcp.bin at 0x10000 (and tried partion-table.bin from VSCode project)
No output at serial monitor (tried Arduino monitor) at all (different modules).
Any idea? Maybe USB CDC is disabled?

P-R-O-C-H-Y, 2024-12-20:
If you can folks install the ESP-IDF v5.3 and build/flash it on your own by using idf.py command. 

The main difference I there might be is a Board you are using and idk yours, but I sm using a UART port on the H2 Espressifs devkit. 

Can you tell me your setup?

ilker-aktuna, 2024-12-20:
> If you can folks install the ESP-IDF v5.3 and build/flash it on your own by using idf.py command.
> 
> The main difference I there might be is a Board you are using and idk yours, but I sm using a UART port on the H2 Espressifs devkit.
> 
> Can you tell me your setup?

I already compiled myself and flashed using idf.py before you shared the binaries. I had the same result.
My device is a esp32c6-zero with one USB port (uart + power)

michapr, 2024-12-20:
I was using the USB port before - every time no output.
Now have seen in esp_ot_config.h that UART is using (no idea how to change it to USB port,....)

I have connected an adapter UART-USB and here I receive data now....

But in this time (after adding Zigbee key and made setting from the link) do not receive valid data:
[image]

Something is still missing....

ilker-aktuna, 2024-12-20:
@P-R-O-C-H-Y 
update:
I got a "normal" C6 from a friend. Using the UART port it can sniff actually.

So I got the attached traces showing the issue on my zigbee network.
Before explaining further, let me remind you  that Hubitat actually detects the device and pairs successfully. But the device (esp32) reports it has failed.

So the first file is capture when my hub has already paired and shows a motion sensor.
Then I removed the sensor from zigbee network (Hubitat) and restarted capture (and esp32)  to record a full failing scenario.
That is the second capture file.  I hope it is clear for you and you can see the problem in these traces.

[pairing_2.zip](https://github.com/user-attachments/files/18211178/pairing_2.zip)

Btw, do you have any idea on how I can use sniffer on esp32-c6 zero which has no UART port ?

coolibre, 2024-12-23:
> There is no default config for PIR motion sensor in esp-zigbee-sdk. You may need to create your own "default config", which means declaring the clusters it should contain.
> 
> To be honest it's not that simple to do a new EP type with all the stuff around. Maybe you can keep this feature request open and I will take a look into it and add the implementation in future updates.
> 
> I am planning to add a Common sensor device type, where you will be able to add any clusters into it easily. So you will be able to create a single/multi sensor device.

Is this something that is already on the horizon, or to complex for a quickshot? Im currently failing on adapting TempSensor example and registering a cluster for an object speed measuring sensor. Basically I only need one float attribute speed but had yet no luck on figuring out the co working with esp zigbee SDK. https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/user-guide/zcl_custom.html

ilker-aktuna, 2024-12-24:
@P-R-O-C-H-Y 
Did you have any chance to look at my capture files ?

ilker-aktuna, 2025-01-10:
@P-R-O-C-H-Y 
I hope everything is fine, wishing you a happy new year.
Did you have any chance to analyze my sniff output ?

ilker-aktuna, 2025-02-02:
@P-R-O-C-H-Y 

Finally I had some time to test the motion sensor example.
On my Hubitat the device is recognized and I selected the "generic zigbee motion sensor" profile.
From the arduino skecth I am sending the following commands:
zbOccupancySensor.setOccupancy(true);
and
zbOccupancySensor.setOccupancy(false);

followed by 
zbOccupancySensor.report();

but on the hub "motion" stated does not change.
Is "occupancy" equivalent to "motion" ? I don't see a "occupancy sensor" profile on the hub.

P-R-O-C-H-Y, 2025-02-03:
Hi @ilker-aktuna, there is no issue with the Occupancy sensor when its connected to HomeAssistant.
It gets automatically recognized by the ZHA, that it is an occupancy sensor.
About the 2 endpoints, this also perfectly works on HomeAssistant. You can add even more than 2 endpoints, so you can combine whatever you want like:
EP1 - Temperature sensor
EP2 - ON/OFF Light
EP3 - Occupancy sensor
...

I don't have the hubitat device so Idk if I would be able to help you more on this.
I can just say it works perfectly using the HomeAssistant, and also if I program any Coordinator, which controls device with multiple endpoints.

ilker-aktuna, 2025-02-03:
Hi @P-R-O-C-H-Y ,

I understand that it works perfectly with HA but I am trying to make this run on a Hubitat hub. 2 EPs on 1 device might be a problem on Hubitat side. I will try to get help from Hubitat community on that (because their support is not really taking any feedback)
But if this is a limitation on Hubitat side, how can I overcome this ?
Can I create a single EP with both features ?

About the occupancy sensor; can you give me a specification on how it is designed on esp32 side ?
Maybe an occupancy sensor is not a motion sensor. If that is the case, I should give up because there is no occupancy sensor profile on Hubitat.
But if motion sensor and occupancy sensor are same thiğng with different names then I have to understand why it does not work.

P-R-O-C-H-Y, 2025-02-03:
The main thing is the Zigbee 3.0 Specification. So there are many defined Home Automation profiles, which defines what the device is doing. So each end device have its own type, only some few sensors are just "Simple Sensor" device type. This is how the systems recognizes the devices (also by check of the clusters). I think this is something not completely done on the Hubitat side.

You can try to create an endpoint device having custom clusters with extended functionality, but I am not sure if it will work well.
Try to create a custom end device, so you will have only 1 endpoint. Let me know if that worked for you.
But TBH, this is not how the things should be done the right way.

The occupancy is implemented as:
`ESP_ZB_HA_SIMPLE_SENSOR_DEVICE_ID                 = 0x000C,  /*!< Simple Sensor device */`

having this cluster:
`ESP_ZB_ZCL_CLUSTER_ID_OCCUPANCY_SENSING          = 0x0406U,     /*!< Occupancy sensing */`

P-R-O-C-H-Y, 2025-02-03:
@ilker-aktuna Good example of multiple functionality device in simple endpoint is the TemperatureSensor end device.
There is an option to add the humidity cluster, so you can have both TEMP+HUM in one endpoint.

ilker-aktuna, 2025-02-03:
> You can try to create an endpoint device having custom clusters with extended functionality, but I am not sure if it will work well.
> Try to create a custom end device, so you will have only 1 endpoint. Let me know if that worked for you.

how and where should I do that ? On the sketch itself or should I change the core esp32 code to add a new device ?

P-R-O-C-H-Y, 2025-02-06:
You can implement that in a separate cpp and header file directly in your sketch. Just don't forget to use correct includes in the files :)

ilker-aktuna, 2025-02-11:
@P-R-O-C-H-Y 
I learnt that on Hubitat 2 Endpoints in one Zigbee device is supported.
But I am confused. As I wrote in this thread, if I add 2 sensors to the zigbee core in Arduino sketch , the device is only recognized as the first cluster defines on Hubitat.
What is the correct implementation of adding 2 endpoints in one device on Arduino sketch ?

Also, as another option,  I don't understand how I should add the second cluster to the motion sensor.
On the Temp+Humidity sensor example, there is a "zbTempSensor.addHumiditySensor(0, 100, 1);" command which calls the "addHumiditySensor" function in the tempsensor code.
But there is no such "addTemperatureSensor" in the OccupancySensor code.

ilker-aktuna, 2025-02-13:
@P-R-O-C-H-Y 

did you notice my question above ?

P-R-O-C-H-Y, 2025-02-14:
Hi @ilker-aktuna,
sorry for no reply on that. It's too many issues/pr to check and time to time you can forget something. Good you pinged me :)

Only the Temperature sensor have the option to add a Humidity cluster, as the device type for HA is Temperature_Humidity_Sensor. SO there is the function addHumidity to add this cluster to the endpoint.

For all other devices, you can just combine multiple endpoints in the sketch itself with unique numbers. As example you can check the Pressure + Flow example : https://github.com/espressif/arduino-esp32/blob/62082398d322c636184ba4b4a385f9b618f61a77/libraries/Zigbee/examples/Zigbee_Pressure_Flow_Sensor/Zigbee_Pressure_Flow_Sensor.ino#L41

ilker-aktuna, 2025-02-26:
@P-R-O-C-H-Y 
Multiple-endpoints issue on Hubitat seems to be resolved. I just need to write my own driver on Hubitat for that. So with a custom driver this implementation works (PIR & temperature sensor for example)

I just have a side question:
Esp32-c6 seems to support coexistence of Zigbee and Wifi as explained here with some level of performance degradation:
https://github.com/espressif/esp-zigbee-sdk/issues/361
But when I try it on Arduino sketch, wifi connects , but loses connectivity when zigbee network is joined.
Do we need to implement something on Arduino side to enable coexistence of wifi and zigbee ?
on native code , there is this "esp_coex_wifi_i154_enable()" command.

P-R-O-C-H-Y, 2025-02-27:
@ilker-aktuna I am glad to hear that its now working also with the Hubitat.

About the C6 and WiFi+Zigbee coexistence, please refer to this issue https://github.com/espressif/arduino-esp32/issues/10871

ilker-aktuna, 2025-02-27:
@P-R-O-C-H-Y 
thanks.
how can I stop zigbee when I need wifi ?
also visa versa, how can I stop wifi and restart zigbee ?

ilker-aktuna, 2025-03-20:
@P-R-O-C-H-Y 

I have a problem with status reporting of the motion sensor.
If the zigbee coordinator sends a read command to the cluster 0x0406 , it can read the motion value reported.
But normally the code is sending updates on change of status with following commands:
            zbOccupancySensor.setOccupancy(true);
            zbOccupancySensor.report();

I believe the corrdinator shouldn't need to send a read command because the device will update status when needed.
So what is missing ?
is there another command to send the zigbee status report to the coordinator ?

P-R-O-C-H-Y, 2025-03-20:
Can you provide more info of what is your setup? Who is the coordinator and if you use any system like HA with ZHA or Z2M?

[5 more replies]

PR #10720 (2024-12-14): feat(zigbee): Add pressure, flow, ccupancy and carbon dioxide sensor
## Description of Change
This PR adds 4 new endpoint types:

- Pressure sensor
- Flow sensor
- Occupancy sensor
- Carbon dioxide (CO2)

There are 3 new examples: `Pressure+Flow sensor`, `Occupancy sensor`, `Carbon dioxide sensor`

+ Some minor enhancements.

## Tests scenarios
Tested using C6 + HomeAssistant

## Related links
Closes #10684