espressif/arduino-esp32 #10685: Dual I2C Channels not functioning on ESP32-C6

Legion-Cube, 2024-12-04:
### Board

ESP32-C6 Dev Module

### Device Description

DevKit on a bread board As well as wokwi simulation.

An I2C deivce connected on each I2C bus (3.3v) Breakours have own pull up resistors (Checked with OLED scrren, and TOF sensor) 
[image]

### Hardware Configuration

I2C_Bus1 : SDA Pin4, SLC Pin3
I2C_Bus1 : SDA Pin5, SLC Pin6

### Version

latest master (checkout manually)

### IDE Name

Ardunio IDE

### Operating System

Ubuntu

### Flash frequency

80MHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

The second I2C bus fails to initialise and propprely work 
I have checked Hardware connections and ruled them out using diffrent variations of pins and sensors corretly as well as checking indiviual I2C channels one at a time (both happy with own sets and issue chases the second initalised, not the GPIO pins)

But when attemping to run two the second I2C will fail to initialose and cause an error if attempting to read.

If done on another microcontroller (e.g. ESP32-S3) it dosnt seem to be an issue and it will show any I2C devices Ive put on as normal

After searching i found  [https://github.com/espressif/arduino-esp32/issues/6966](url) But belive this is a diffrent issue as the same code i have used works fine on the other ESP32's ive tried 

### Sketch

```cpp
#include <Wire.h>

TwoWire I2C_0 = TwoWire(0);  // First I2C channel
TwoWire I2C_1 = TwoWire(1);  // Second I2C channel

void setup() {
  Serial.begin(115200);

  // Initialize I2C channel 0 on pins SDA=21, SCL=22
  if (I2C_0.begin(4, 3, 100000)) {
    Serial.println("I2C_0 initialized successfully.");
  } else {
    Serial.println("I2C_0 initialization failed!");
  }

  // Initialize I2C channel 1 on pins SDA=25, SCL=26
  if (I2C_1.begin(5, 6, 100000)) {
    Serial.println("I2C_1 initialized successfully.");
  } else {
    Serial.println("I2C_1 initialization failed!");
  }
  scanI2C(I2C_0); 
  scanI2C(I2C_1); 

}

void loop() {
  // Regular communication logic here
}

void scanI2C(TwoWire &wire) {
 byte error, address;
 int nDevices = 0;
 Serial.println("Scanning I2C Bus...");
 for (address = 1; address < 127; address++) {
   wire.beginTransmission(address);
   error = wire.endTransmission();

   if (error == 0) {
     Serial.print("I2C device found at address 0x");
     if (address < 16) Serial.print("0");
     Serial.print(address, HEX);
     Serial.println(" !");
     nDevices++;
   } else if (error == 4) {
     Serial.print("Unknown error at address 0x");
     if (address < 16) Serial.print("0");
     Serial.println(address, HEX);
   }
 }

 if (nDevices == 0) {
   Serial.println("No I2C devices found.\n");
 } else {
   Serial.println("Done scanning.\n");
 }
}
```

### Debug Message

```plain
I2C_0 initialized successfully.
I2C_1 initialization failed!
Scanning I2C Bus...
I2C device found at address 0x3C !
Done scanning.

Scanning I2C Bus...
Unknown error at address 0x01
Unknown error at address 0x02
Unknown error at address 0x03
Unknown error at address 0x04
Unknown error at address 0x05
Unknown error at address 0x06
Unknown error at address 0x07
Unknown error at address 0x08
Unknown error at address 0x09
Unknown error at address 0x0A
Unknown error at address 0x0B
Unknown error at address 0x0C
Unknown error at address 0x0D
Unknown error at address 0x0E
Unknown error at address 0x0F
Unknown error at address 0x10
Unknown error at address 0x11
Unknown error at address 0x12
Unknown error at address 0x13
Unknown error at address 0x14
Unknown error at address 0x15
Unknown error at address 0x16
Unknown error at address 0x17
Unknown error at address 0x18
Unknown error at address 0x19
Unknown error at address 0x1A
Unknown error at address 0x1B
Unknown error at address 0x1C
Unknown error at address 0x1D
Unknown error at address 0x1E
Unknown error at address 0x1F
Unknown error at address 0x20
Unknown error at address 0x21
Unknown error at address 0x22
Unknown error at address 0x23
Unknown error at address 0x24
Unknown error at address 0x25
Unknown error at address 0x26
Unknown error at address 0x27
Unknown error at address 0x28
Unknown error at address 0x29
Unknown error at address 0x2A
Unknown error at address 0x2B
Unknown error at address 0x2C
Unknown error at address 0x2D
Unknown error at address 0x2E
Unknown error at address 0x2F
Unknown error at address 0x30
Unknown error at address 0x31
Unknown error at address 0x32
Unknown error at address 0x33
Unknown error at address 0x34
Unknown error at address 0x35
Unknown error at address 0x36
Unknown error at address 0x37
Unknown error at address 0x38
Unknown error at address 0x39
Unknown error at address 0x3A
Unknown error at address 0x3B
Unknown error at address 0x3C
Unknown error at address 0x3D
Unknown error at address 0x3E
Unknown error at address 0x3F
Unknown error at address 0x40
Unknown error at address 0x41
Unknown error at address 0x42
Unknown error at address 0x43
Unknown error at address 0x44
Unknown error at address 0x45
Unknown error at address 0x46
Unknown error at address 0x47
Unknown error at address 0x48
Unknown error at address 0x49
Unknown error at address 0x4A
Unknown error at address 0x4B
Unknown error at address 0x4C
Unknown error at address 0x4D
Unknown error at address 0x4E
Unknown error at address 0x4F
Unknown error at address 0x50
Unknown error at address 0x51
Unknown error at address 0x52
Unknown error at address 0x53
Unknown error at address 0x54
Unknown error at address 0x55
Unknown error at address 0x56
Unknown error at address 0x57
Unknown error at address 0x58
Unknown error at address 0x59
Unknown error at address 0x5A
Unknown error at address 0x5B
Unknown error at address 0x5C
Unknown error at address 0x5D
Unknown error at address 0x5E
Unknown error at address 0x5F
Unknown error at address 0x60
Unknown error at address 0x61
Unknown error at address 0x62
Unknown error at address 0x63
Unknown error at address 0x64
Unknown error at address 0x65
Unknown error at address 0x66
Unknown error at address 0x67
Unknown error at address 0x68
Unknown error at address 0x69
Unknown error at address 0x6A
Unknown error at address 0x6B
Unknown error at address 0x6C
Unknown error at address 0x6D
Unknown error at address 0x6E
Unknown error at address 0x6F
Unknown error at address 0x70
Unknown error at address 0x71
Unknown error at address 0x72
Unknown error at address 0x73
Unknown error at address 0x74
Unknown error at address 0x75
Unknown error at address 0x76
Unknown error at address 0x77
Unknown error at address 0x78
Unknown error at address 0x79
Unknown error at address 0x7A
Unknown error at address 0x7B
Unknown error at address 0x7C
Unknown error at address 0x7D
Unknown error at address 0x7E
No I2C devices found.
```

### Other Steps to Reproduce

As well as the physical ESP32-C6 I used the onlike wokwi to simulate it, same results. But when used on an ESP32-S3 it works fine

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2025-01-06:
This is because Low Power peripherals are not yet supported in Arduino. That is still to come. C6 has only one High Power I2C bus

tvixen, 2025-06-16:
Is there any default pin assignment for this High Power I2C bus?
Because I just spend 16hours coding to the second I2C bus which apperently is not there 🤷‍♂️

me-no-dev, 2025-06-16:
no default pins for I2C0, but there are for I2C1 https://github.com/espressif/arduino-esp32/blob/master/variants/esp32c6/pins_arduino.h#L37-L38

Closing the issue, because LP I2C has been added to v3.2.0

tvixen, 2025-06-17:
Yes thanks, I found out.. the hard way 😁
Also that ESP32-C6 can't operate i Slave mode ... Has anyone solved this yet?