espressif/arduino-esp32 #10688: #  error "Please configure IDF framework to include mbedTLS -> Enable pre-shared-key ciphersuites and activate at least one cipher"

davidbcf, 2024-12-06:
### Board

ESP32-C3

### Device Description

ESP32-C3

### Hardware Configuration

ESP32-C3

### Version

v2.0.6

### IDE Name

Arduino

### Operating System

windows

### Flash frequency

40

### PSRAM enabled

yes

### Upload speed

115200

### Description

#  error "Please configure IDF framework to include mbedTLS -> Enable pre-shared-key ciphersuites and activate at least one cipher"
 How to configue mbedTLS  in Arduino for ESP32  to use TLS1.2?

### Sketch

```cpp
#  error "Please configure IDF framework to include mbedTLS -> Enable pre-shared-key ciphersuites and activate at least one cipher"
```

### Debug Message

```plain
#  error "Please configure IDF framework to include mbedTLS -> Enable pre-shared-key ciphersuites and activate at least one cipher"
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2024-12-06:
mbedTLS is definitely configured and enabled.  
v2.0.6 is quite old now.  Please test with a newer version.
What code is producing this?

davidbcf, 2024-12-06:
V3.0.7 as the same result
examlpe : WiFiClientSecure
while using const char* test_client_key  && const char* test_client_cert
C:\Users\admini\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6\libraries\WiFiClientSecure\src\ssl_client.cpp:23:4

davidbcf, 2024-12-06:
V3.0.7  ichange the library of example to add _private_key_password. and  here is the log

15:47:02.367 -> Starting connection to server...
15:47:02.367 -> [  1840][V][ssl_client.cpp:68] start_ssl_client(): Free internal heap before TLS 222228
15:47:02.367 -> [  1852][V][ssl_client.cpp:75] start_ssl_client(): Starting socket (domain 2)
15:47:05.377 -> [  4832][E][ssl_client.cpp:146] start_ssl_client(): socket error on fd 48, errno: 104, "Connection reset by peer"
15:47:05.377 -> [  4843][I][NetworkClientSecure.cpp:155] connect(): Actual TLS start postponed.
15:47:05.377 -> [  4850][E][NetworkClientSecure.cpp:161] connect(): start_ssl_client: connect failed: -1
15:47:05.377 -> [  4858][V][ssl_client.cpp:360] stop_ssl_socket(): Cleaning SSL connection.
15:47:05.377 -> Connection failed!

davidbcf, 2024-12-06:
[ssl_starttls_handshake():313]: (-9984) X509 - Certificate verification failed, e.g. CRL, CA or signature check failed

sivar2311, 2024-12-06:
Unfortunately you did not provide a sketch to reproduce the issue.

davidbcf, 2024-12-06:
/*
  Wifi secure connection example for ESP32
  Running on TLS 1.2 using mbedTLS
  Supporting the following ciphersuites:
  "TLS_ECDHE_ECDSA_WITH_AES_256_GCM_SHA384","TLS_ECDHE_RSA_WITH_AES_256_GCM_SHA384","TLS_DHE_RSA_WITH_AES_256_GCM_SHA384","TLS_ECDHE_ECDSA_WITH_AES_256_CCM","TLS_DHE_RSA_WITH_AES_256_CCM","TLS_ECDHE_ECDSA_WITH_AES_256_CBC_SHA384","TLS_ECDHE_RSA_WITH_AES_256_CBC_SHA384","TLS_DHE_RSA_WITH_AES_256_CBC_SHA256","TLS_ECDHE_ECDSA_WITH_AES_256_CBC_SHA","TLS_ECDHE_RSA_WITH_AES_256_CBC_SHA","TLS_DHE_RSA_WITH_AES_256_CBC_SHA","TLS_ECDHE_ECDSA_WITH_AES_256_CCM_8","TLS_DHE_RSA_WITH_AES_256_CCM_8","TLS_ECDHE_ECDSA_WITH_CAMELLIA_256_GCM_SHA384","TLS_ECDHE_RSA_WITH_CAMELLIA_256_GCM_SHA384","TLS_DHE_RSA_WITH_CAMELLIA_256_GCM_SHA384","TLS_ECDHE_ECDSA_WITH_CAMELLIA_256_CBC_SHA384","TLS_ECDHE_RSA_WITH_CAMELLIA_256_CBC_SHA384","TLS_DHE_RSA_WITH_CAMELLIA_256_CBC_SHA256","TLS_DHE_RSA_WITH_CAMELLIA_256_CBC_SHA","TLS_ECDHE_ECDSA_WITH_AES_128_GCM_SHA256","TLS_ECDHE_RSA_WITH_AES_128_GCM_SHA256","TLS_DHE_RSA_WITH_AES_128_GCM_SHA256","TLS_ECDHE_ECDSA_WITH_AES_128_CCM","TLS_DHE_RSA_WITH_AES_128_CCM","TLS_ECDHE_ECDSA_WITH_AES_128_CBC_SHA256","TLS_ECDHE_RSA_WITH_AES_128_CBC_SHA256","TLS_DHE_RSA_WITH_AES_128_CBC_SHA256","TLS_ECDHE_ECDSA_WITH_AES_128_CBC_SHA","TLS_ECDHE_RSA_WITH_AES_128_CBC_SHA","TLS_DHE_RSA_WITH_AES_128_CBC_SHA","TLS_ECDHE_ECDSA_WITH_AES_128_CCM_8","TLS_DHE_RSA_WITH_AES_128_CCM_8","TLS_ECDHE_ECDSA_WITH_CAMELLIA_128_GCM_SHA256","TLS_ECDHE_RSA_WITH_CAMELLIA_128_GCM_SHA256","TLS_DHE_RSA_WITH_CAMELLIA_128_GCM_SHA256","TLS_ECDHE_ECDSA_WITH_CAMELLIA_128_CBC_SHA256","TLS_ECDHE_RSA_WITH_CAMELLIA_128_CBC_SHA256","TLS_DHE_RSA_WITH_CAMELLIA_128_CBC_SHA256","TLS_DHE_RSA_WITH_CAMELLIA_128_CBC_SHA","TLS_ECDHE_ECDSA_WITH_3DES_EDE_CBC_SHA","TLS_ECDHE_RSA_WITH_3DES_EDE_CBC_SHA","TLS_DHE_RSA_WITH_3DES_EDE_CBC_SHA","TLS_DHE_PSK_WITH_AES_256_GCM_SHA384","TLS_DHE_PSK_WITH_AES_256_CCM","TLS_ECDHE_PSK_WITH_AES_256_CBC_SHA384","TLS_DHE_PSK_WITH_AES_256_CBC_SHA384","TLS_ECDHE_PSK_WITH_AES_256_CBC_SHA","TLS_DHE_PSK_WITH_AES_256_CBC_SHA","TLS_DHE_PSK_WITH_CAMELLIA_256_GCM_SHA384","TLS_ECDHE_PSK_WITH_CAMELLIA_256_CBC_SHA384","TLS_DHE_PSK_WITH_CAMELLIA_256_CBC_SHA384","TLS_PSK_DHE_WITH_AES_256_CCM_8","TLS_DHE_PSK_WITH_AES_128_GCM_SHA256","TLS_DHE_PSK_WITH_AES_128_CCM","TLS_ECDHE_PSK_WITH_AES_128_CBC_SHA256","TLS_DHE_PSK_WITH_AES_128_CBC_SHA256","TLS_ECDHE_PSK_WITH_AES_128_CBC_SHA","TLS_DHE_PSK_WITH_AES_128_CBC_SHA","TLS_DHE_PSK_WITH_CAMELLIA_128_GCM_SHA256","TLS_DHE_PSK_WITH_CAMELLIA_128_CBC_SHA256","TLS_ECDHE_PSK_WITH_CAMELLIA_128_CBC_SHA256","TLS_PSK_DHE_WITH_AES_128_CCM_8","TLS_ECDHE_PSK_WITH_3DES_EDE_CBC_SHA","TLS_DHE_PSK_WITH_3DES_EDE_CBC_SHA","TLS_RSA_WITH_AES_256_GCM_SHA384","TLS_RSA_WITH_AES_256_CCM","TLS_RSA_WITH_AES_256_CBC_SHA256","TLS_RSA_WITH_AES_256_CBC_SHA","TLS_ECDH_RSA_WITH_AES_256_GCM_SHA384","TLS_ECDH_RSA_WITH_AES_256_CBC_SHA384","TLS_ECDH_RSA_WITH_AES_256_CBC_SHA","TLS_ECDH_ECDSA_WITH_AES_256_GCM_SHA384","TLS_ECDH_ECDSA_WITH_AES_256_CBC_SHA384","TLS_ECDH_ECDSA_WITH_AES_256_CBC_SHA","TLS_RSA_WITH_AES_256_CCM_8","TLS_RSA_WITH_CAMELLIA_256_GCM_SHA384","TLS_RSA_WITH_CAMELLIA_256_CBC_SHA256","TLS_RSA_WITH_CAMELLIA_256_CBC_SHA","TLS_ECDH_RSA_WITH_CAMELLIA_256_GCM_SHA384","TLS_ECDH_RSA_WITH_CAMELLIA_256_CBC_SHA384","TLS_ECDH_ECDSA_WITH_CAMELLIA_256_GCM_SHA384","TLS_ECDH_ECDSA_WITH_CAMELLIA_256_CBC_SHA384","TLS_RSA_WITH_AES_128_GCM_SHA256","TLS_RSA_WITH_AES_128_CCM","TLS_RSA_WITH_AES_128_CBC_SHA256","TLS_RSA_WITH_AES_128_CBC_SHA","TLS_ECDH_RSA_WITH_AES_128_GCM_SHA256","TLS_ECDH_RSA_WITH_AES_128_CBC_SHA256","TLS_ECDH_RSA_WITH_AES_128_CBC_SHA","TLS_ECDH_ECDSA_WITH_AES_128_GCM_SHA256","TLS_ECDH_ECDSA_WITH_AES_128_CBC_SHA256","TLS_ECDH_ECDSA_WITH_AES_128_CBC_SHA","TLS_RSA_WITH_AES_128_CCM_8","TLS_RSA_WITH_CAMELLIA_128_GCM_SHA256","TLS_RSA_WITH_CAMELLIA_128_CBC_SHA256","TLS_RSA_WITH_CAMELLIA_128_CBC_SHA","TLS_ECDH_RSA_WITH_CAMELLIA_128_GCM_SHA256","TLS_ECDH_RSA_WITH_CAMELLIA_128_CBC_SHA256","TLS_ECDH_ECDSA_WITH_CAMELLIA_128_GCM_SHA256","TLS_ECDH_ECDSA_WITH_CAMELLIA_128_CBC_SHA256","TLS_RSA_WITH_3DES_EDE_CBC_SHA","TLS_ECDH_RSA_WITH_3DES_EDE_CBC_SHA","TLS_ECDH_ECDSA_WITH_3DES_EDE_CBC_SHA","TLS_RSA_PSK_WITH_AES_256_GCM_SHA384","TLS_RSA_PSK_WITH_AES_256_CBC_SHA384","TLS_RSA_PSK_WITH_AES_256_CBC_SHA","TLS_RSA_PSK_WITH_CAMELLIA_256_GCM_SHA384","TLS_RSA_PSK_WITH_CAMELLIA_256_CBC_SHA384","TLS_RSA_PSK_WITH_AES_128_GCM_SHA256","TLS_RSA_PSK_WITH_AES_128_CBC_SHA256","TLS_RSA_PSK_WITH_AES_128_CBC_SHA","TLS_RSA_PSK_WITH_CAMELLIA_128_GCM_SHA256","TLS_RSA_PSK_WITH_CAMELLIA_128_CBC_SHA256","TLS_RSA_PSK_WITH_3DES_EDE_CBC_SHA","TLS_PSK_WITH_AES_256_GCM_SHA384","TLS_PSK_WITH_AES_256_CCM","TLS_PSK_WITH_AES_256_CBC_SHA384","TLS_PSK_WITH_AES_256_CBC_SHA","TLS_PSK_WITH_CAMELLIA_256_GCM_SHA384","TLS_PSK_WITH_CAMELLIA_256_CBC_SHA384","TLS_PSK_WITH_AES_256_CCM_8","TLS_PSK_WITH_AES_128_GCM_SHA256","TLS_PSK_WITH_AES_128_CCM","TLS_PSK_WITH_AES_128_CBC_SHA256","TLS_PSK_WITH_AES_128_CBC_SHA","TLS_PSK_WITH_CAMELLIA_128_GCM_SHA256","TLS_PSK_WITH_CAMELLIA_128_CBC_SHA256","TLS_PSK_WITH_AES_128_CCM_8","TLS_PSK_WITH_3DES_EDE_CBC_SHA","TLS_EMPTY_RENEGOTIATION_INFO_SCSV"]
  2017 - Evandro Copercini - Apache 2.0 License.
*/

#include <NetworkClientSecure.h>
#include <WiFi.h>

const char* ssid     = "TP";     // your network SSID (name of wifi network)
const char* password = "tvt-2018"; // your network password

const char*  server = "10.8.4.118";  // Server URL

// www.howsmyssl.com root certificate authority, to verify the server
[... cut ...]
    client.println("Host: www.howsmyssl.com");
    client.println("Connection: close");
    client.println();

    while (client.connected()) {
      String line = client.readStringUntil('\n');
      if (line == "\r") {
        Serial.println("headers received");
        break;
      }
    }
    // if there are incoming bytes available
    // from the server, read them and print them:
    while (client.available()) {
      char c = client.read();
      Serial.write(c);
    }

    client.stop();
  }
}

void loop() {
  // do nothing
}

davidbcf, 2024-12-06:
i changed the library API client.setPrivateKey in order to add password ,and some functions relate to the password

me-no-dev, 2024-12-06:
Log suggests that your connection is failing before SSL even starts: `socket error on fd 48, errno: 104, "Connection reset by peer"` On other hand, the example you shared looks like it would test against `www.howsmyssl.com` but you are connecting to some other server on port `8801`. Are you sure all your connect parameters are ok?

What do you mean by `i changed the library API client.setPrivateKey`?

davidbcf, 2024-12-06:
1.errono：104  this has been solved
2. the issue  is
> [ssl_starttls_handshake():313]: (-9984) X509 - Certificate verification failed, e.g. CRL, CA or signature check failed
3. We have our server(10.8.4.118，8801) with TLS1.2,just for test send that message ; we have our CA cert ,client_key(with password),client_cert. all info you can see in that codes;
4. in order to correct work with  client_key(with password), we need add an API to send password to function :mbedtls_pk_parse_key
5.V3.0.7 the original definition : void setPrivateKey(const char *private_key);
and i changed it to void setPrivateKey(const char *private_key, const char *private_key_password); 
in order  i can send password for  client_key in function: mbedtls_pk_parse_key
6. the attached file  is the APIs i had changed for my purpose; i rewrite some functions in class ssl_client and NetworkClientSecure. you can use the attached file replace the file  which show in picture.
7. now all  i need is slove this problem > [ssl_starttls_handshake():313]: (-9984) X509 - Certificate verification failed, e.g. CRL, CA or signature check failed.   it looks like something wrong before client get connection with server
thanks a lot 
[src.zip](https://github.com/user-attachments/files/18038702/src.zip)
[image]

davidbcf, 2024-12-09:
the error  is from file: ssl_client.cpp

davidbcf, 2024-12-09:
The error message is from file: ssl_client.cpp --- int ssl_starttls_handshake(sslclient_context *ssl_client)

Device log :
10:23:42.211 -> Starting connection to server...
10:23:42.211 -> [  1913][V][ssl_client.cpp:68] start_ssl_client(): Free internal heap before TLS 222208
10:23:42.211 -> [  1925][V][ssl_client.cpp:75] start_ssl_client(): Starting socket (domain 2)
10:23:42.211 -> [  1948][V][ssl_client.cpp:166] start_ssl_client(): Seeding the random number generator
10:23:42.211 -> [  1958][V][ssl_client.cpp:174] start_ssl_client(): Setting up the SSL/TLS structure...
10:23:42.259 -> [  1967][V][ssl_client.cpp:194] start_ssl_client(): Loading CA cert
10:23:42.259 -> [  1977][V][ssl_client.cpp:265] start_ssl_client(): Loading CRT cert
10:23:42.259 -> [  1988][V][ssl_client.cpp:274] start_ssl_client(): Loading private key
10:23:42.445 -> [  2162][V][ssl_client.cpp:288] start_ssl_client(): Setting hostname for TLS session...
10:23:42.445 -> [  2170][V][ssl_client.cpp:309] ssl_starttls_handshake(): Performing the SSL/TLS handshake...
10:23:42.539 -> [  2263][E][ssl_client.cpp:36] _handle_error(): [ssl_starttls_handshake():313]: (-9984) X509 - Certificate verification failed, e.g. CRL, CA or signature check failed
10:23:42.539 -> [  2278][E][NetworkClientSecure.cpp:161] connect(): start_ssl_client: connect failed: -9984
10:23:42.586 -> [  2286][V][ssl_client.cpp:360] stop_ssl_socket(): Cleaning SSL connection.
10:23:42.586 -> Connection failed!

davidbcf, 2024-12-10:
we used openssl  test these certs ,and it works.
[image]

lbernstone, 2024-12-10:
cn=iot.ax-end.com
You are connecting to the IP address.  You can either add the IP address as an alternative CN, or connect with the DNS name as specified in the certificate.

davidbcf, 2024-12-12:
thank you so much! we solved this problem with your suggestion.

closed 2024-12-12 by davidbcf