espressif/arduino-esp32 #10690: Example for WebServer using Ethernet

lbernstone, 2024-12-06:
### Related area

Ethernet / Network

### Hardware specification

all variants

### Is your feature request related to a problem?

Please write an example demonstrating how to use the new Network library setDefault to provide a WebServer instance running on Ethernet that does other activity (maybe an HTTPClient) on the WiFi adapter.

### Describe the solution you'd like

Please write an example demonstrating how to use the new Network library setDefault to provide a WebServer instance running on Ethernet that does other activity (maybe an HTTPClient) on the WiFi adapter.

### Additional context

This is a new feature in v3.x that is completely undocumented. It would be helpful to have an example that shows how to use it.

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

JAndrassy, 2024-12-06:
if you use the WebServer constructor with IP address it will listen only at that IP address
`WebServer(IPAddress addr, int port = 80);`

lbernstone, 2024-12-07:
WebServer has been redesigned around the Network library, but it would be nice to get an example of how to do this with something that is not so specific.

me-no-dev, 2024-12-09:
@lbernstone the solution that @JAndrassy proposes "should" work. Your `setDefault` should be STA if you want to use it with other things (and that is the default anyway). If you `setDefault` to ETH, then all outgoing traffic will go from there

Parsaabasi, 2025-03-13:
Hi @lbernstone 

Seems like this issue is resolved according to the comments above. Therefore, I close this for now. In case you need further assistance, please feel free to reopen or create a new issue. 

Thanks.