espressif/arduino-esp32 #10693: Zigbee: device is stopping send values after some time

michapr, 2024-12-07:
### Board

ESP32C6, ESP32H2

### Device Description

DevModule C6, SuperMini C6, SuperMini H2  (same issue at all devices)

### Hardware Configuration

nothing else connected

### Version

latest development Release Candidate (RC-X)

### IDE Name

Arduino IDE

### Operating System

Windows11

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

Device is sending values for a wile, but after some time (20 minutes about?) is stopping sending values until restart of device.

### Sketch

```cpp
Sample sketch Zigbee_Temperature_Sensor.ino with libraries from PR #10676
```

### Debug Message

```plain
from ESP32H2:

[07/12/24 - 15:35:49:504] [3828362][V][ZigbeeTempSensor.cpp:70] setTemperature(): Updating temperature sensor value...
[07/12/24 - 15:35:49:508] [3828363][D][ZigbeeTempSensor.cpp:72] setTemperature(): Setting temperature to 3700
[07/12/24 - 15:35:49:562] [3828424][V][ZigbeeHandlers.cpp:134] zb_cmd_default_resp_handler(): Received default response: from address(0x0), src_endpoint(1) to dst_endpoint(10), cluster(0x402) with status 0x0
[07/12/24 - 15:35:50:501] Updated temperature sensor value to 36.00°C
[07/12/24 - 15:35:50:504] [3829363][V][ZigbeeTempSensor.cpp:70] setTemperature(): Updating temperature sensor value...
[07/12/24 - 15:35:50:509] [3829364][D][ZigbeeTempSensor.cpp:72] setTemperature(): Setting temperature to 3600
[07/12/24 - 15:35:50:610] [3829471][V][ZigbeeHandlers.cpp:134] zb_cmd_default_resp_handler(): Received default response: from address(0x0), src_endpoint(1) to dst_endpoint(10), cluster(0x402) with status 0x0
[07/12/24 - 15:35:51:501] Updated temperature sensor value to 36.00°C
[07/12/24 - 15:35:51:506] [3830364][V][ZigbeeTempSensor.cpp:70] setTemperature(): Updating temperature sensor value...
[07/12/24 - 15:35:51:512] [3830365][D][ZigbeeTempSensor.cpp:72] setTemperature(): Setting temperature to 3600
...
...
[07/12/24 - 15:37:03:535] Updated temperature sensor value to 36.00°C
[07/12/24 - 15:37:03:539] [3902397][V][ZigbeeTempSensor.cpp:70] setTemperature(): Updating temperature sensor value...
[07/12/24 - 15:37:03:543] [3902398][D][ZigbeeTempSensor.cpp:72] setTemperature(): Setting temperature to 3600
[07/12/24 - 15:37:03:549] [3902399][V][ZigbeeCore.cpp:332] esp_zb_app_signal_handler(): ZDO signal: ZDO Device Unavailable (0x3c), status: ESP_OK
[07/12/24 - 15:37:04:536] Updated temperature sensor value to 36.00°C
[07/12/24 - 15:37:04:540] [3903397][V][ZigbeeTempSensor.cpp:70] setTemperature(): Updating temperature sensor value...
[07/12/24 - 15:37:04:544] [3903398][D][ZigbeeTempSensor.cpp:72] setTemperature(): Setting temperature to 3600
[07/12/24 - 15:37:05:536] Updated temperature sensor value to 36.00°C
...
...
[07/12/24 - 15:37:16:559] [3915405][V][ZigbeeTempSensor.cpp:70] setTemperature(): Updating temperature sensor value...
[07/12/24 - 15:37:16:565] [3915406][D][ZigbeeTempSensor.cpp:72] setTemperature(): Setting temperature to 3600
[07/12/24 - 15:37:16:569] p_zb_app_signal_handler(): ZDO signal: ZDO Device Unavailable (0x3c), status: ESP_OK
[07/12/24 - 15:37:17:545] Updated temperature sensor value to 36.00°C
[07/12/24 - 15:37:17:548] [3916406][V][ZigbeeTempSensor.cpp:70] setTemperature(): Updating temperature sensor value...
...
...
[07/12/24 - 15:39:59:630] [4078485][V][ZigbeeTempSensor.cpp:70] setTemperature(): Updating temperature sensor value...
[07/12/24 - 15:39:59:635] [4078486][D][ZigbeeTempSensor.cpp:72] setTemperature(): Setting temperature to 3600
[07/12/24 - 15:40:00:625] [4079483][V][ZigbeeCore.cpp:332] esp_zb_app_signal_handler(): ZDO signal: ZDO Device Unavailable (0x3c), status: ESP_OK
[07/12/24 - 15:40:00:634] Updated temperature sensor value to 36.00°C
[07/12/24 - 15:40:00:637] [4079486][V][ZigbeeTempSensor.cpp:70] setTemperature(): Updating temperature sensor value...
[07/12/24 - 15:40:00:646] [4079487][D][ZigbeeTempSensor.cpp:72] setTemperature(): Setting temperature to 3600
[07/12/24 - 15:40:01:628] Updated temperature sensor value to 37.00°C
...
...
[07/12/24 - 15:41:38:690] Updated temperature sensor value to 36.00°C
[07/12/24 - 15:41:38:694] [4177547][V][ZigbeeTempSensor.cpp:70] setTemperature(): Updating temperature sensor value...
[07/12/24 - 15:41:38:698] [4177548][D][ZigbeeTempSensor.cpp:72] setTemperature(): Setting temperature to 3600
[07/12/24 - 15:41:39:640] ZBOSS: zdo/af_rx.c:494   zb_zdo_device_is_unreachable: addr_ref 25
[07/12/24 - 15:41:39:692] Updated temperature sensor value to 36.00°C
[07/12/24 - 15:41:39:695] [4178548][V][ZigbeeTempSensor.cpp:70] setTemperature(): Updating temperature sensor value...
[07/12/24 - 15:41:39:699] [4178549][D][ZigbeeTempSensor.cpp:72] setTemperature(): Setting temperature to 3600
[07/12/24 - 15:41:40:692] Updated temperature sensor value to 36.00°C
...
...
[07/12/24 - 15:42:38:718] Updated temperature sensor value to 36.00°C
[07/12/24 - 15:42:38:722] [4237574][V][ZigbeeTempSensor.cpp:70] setTemperature(): Updating temperature sensor value...
[07/12/24 - 15:42:38:727] [4237575][D][ZigbeeTempSensor.cpp:72] setTemperature(): Setting temperature to 3600
[07/12/24 - 15:42:39:643] ZBOSS: zdo/af_rx.c:494   zb_zdo_device_is_unreachable: addr_ref 255
[07/12/24 - 15:42:39:719] Updated temperature sensor value to 36.00°C
[07/12/24 - 15:42:39:722] [4238575][V][ZigbeeTempSensor.cpp:70] setTemperature(): Updating temperature sensor value...
[07/12/24 - 15:42:39:725] [4238576][D][ZigbeeTempSensor.cpp:72] setTemperature(): Setting temperature to 3600
[07/12/24 - 15:42:40:719] Updated temperature sensor value to 36.00°C
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

michapr, 2024-12-07:
additional for ESP32H2:

```
[07/12/24 - 12:09:50:684] Chip Info:
[07/12/24 - 12:09:50:684] ------------------------------------------
[07/12/24 - 12:09:50:686]   Model             : ESP32-H2
[07/12/24 - 12:09:50:687]   Package           : 0
[07/12/24 - 12:09:50:688]   Revision          : 0.01
[07/12/24 - 12:09:50:690]   Cores             : 1
[07/12/24 - 12:09:50:690]   CPU Frequency     : 96 MHz
[07/12/24 - 12:09:50:692]   XTAL Frequency    : 32 MHz
[07/12/24 - 12:09:50:693]   Features Bitfield : 0x00000050
[07/12/24 - 12:09:50:695]   Embedded Flash    : No
[07/12/24 - 12:09:50:696]   Embedded PSRAM    : No
[07/12/24 - 12:09:50:697]   2.4GHz WiFi       : No
[07/12/24 - 12:09:50:699]   Classic BT        : No
[07/12/24 - 12:09:50:700]   BT Low Energy     : Yes
[07/12/24 - 12:09:50:701]   IEEE 802.15.4     : Yes
[07/12/24 - 12:09:50:702] ------------------------------------------
[07/12/24 - 12:09:50:704] INTERNAL Memory Info:
[07/12/24 - 12:09:50:705] ------------------------------------------
[07/12/24 - 12:09:50:706]   Total Size        :   240976 B ( 235.3 KB)
[07/12/24 - 12:09:50:714]   Free Bytes        :   211716 B ( 206.8 KB)
[07/12/24 - 12:09:50:716]   Allocated Bytes   :    24108 B (  23.5 KB)
[07/12/24 - 12:09:50:718]   Minimum Free Bytes:   206904 B ( 202.1 KB)
[07/12/24 - 12:09:50:723]   Largest Free Block:   196596 B ( 192.0 KB)
[07/12/24 - 12:09:50:726] ------------------------------------------
[07/12/24 - 12:09:50:729] Flash Info:
[07/12/24 - 12:09:50:729] ------------------------------------------
[07/12/24 - 12:09:50:732]   Chip Size         :  4194304 B (4 MB)
[07/12/24 - 12:09:50:734]   Block Size        :    65536 B (  64.0 KB)
[07/12/24 - 12:09:50:746]   Sector Size       :     4096 B (   4.0 KB)
[07/12/24 - 12:09:50:750]   Page Size         :      256 B (   0.2 KB)
[07/12/24 - 12:09:50:753]   Bus Speed         : 64 MHz
[07/12/24 - 12:09:50:755]   Bus Mode          : QIO
[07/12/24 - 12:09:50:756] ------------------------------------------
[07/12/24 - 12:09:50:758] Partitions Info:
[07/12/24 - 12:09:50:759] ------------------------------------------
[07/12/24 - 12:09:50:761]                 nvs : addr: 0x00009000, size:    20.0 KB, type: DATA, subtype: NVS
[07/12/24 - 12:09:50:765]             otadata : addr: 0x0000E000, size:     8.0 KB, type: DATA, subtype: OTA
[07/12/24 - 12:09:50:776]                app0 : addr: 0x00010000, size:  1280.0 KB, type:  APP, subtype: OTA_0
[07/12/24 - 12:09:50:780]                app1 : addr: 0x00150000, size:  1280.0 KB, type:  APP, subtype: OTA_1
[07/12/24 - 12:09:50:785]              spiffs : addr: 0x00290000, size:  1388.0 KB, type: DATA, subtype: SPIFFS
[07/12/24 - 12:09:50:788]          zb_storage : addr: 0x003EB000, size:    16.0 KB, type: DATA, subtype: FAT
[07/12/24 - 12:09:50:792]              zb_fct : addr: 0x003EF000, size:     4.0 KB, type: DATA, subtype: FAT
[07/12/24 - 12:09:50:796]            coredump : addr: 0x003F0000, size:    64.0 KB, type: DATA, subtype: COREDUMP
[07/12/24 - 12:09:50:799] ------------------------------------------
[07/12/24 - 12:09:50:809] Software Info:
[07/12/24 - 12:09:50:810] ------------------------------------------
[07/12/24 - 12:09:50:812]   Compile Date/Time : Dec  7 2024 11:53:43
[07/12/24 - 12:09:50:814]   Compile Host OS   : windows
[07/12/24 - 12:09:50:816]   ESP-IDF Version   : v5.3.1-638-ga0f798cfc4-dirty
[07/12/24 - 12:09:50:820]   Arduino Version   : 3.1.0
[07/12/24 - 12:09:50:822] ------------------------------------------
[07/12/24 - 12:09:50:824] Board Info:
[07/12/24 - 12:09:50:825] ------------------------------------------
[07/12/24 - 12:09:50:827]   Arduino Board     : ESP32H2_DEV
[07/12/24 - 12:09:50:834]   Arduino Variant   : esp32h2
[07/12/24 - 12:09:50:835]   Arduino FQBN      : espressif:esp32:esp32h2:UploadSpeed=921600,CDCOnBoot=cdc,FlashFreq=64,FlashMode=qio,FlashSize=4M,PartitionScheme=zigbee,DebugL
...
...
[07/12/24 - 12:42:52:093] [1981760][V][ZigbeeTempSensor.cpp:70] setTemperature(): Updating temperature sensor value...
[07/12/24 - 12:42:52:099] [1981760][D][ZigbeeTempSensor.cpp:72] setTemperature(): Setting temperature to 2700
[07/12/24 - 12:42:52:241] [1981911][V][ZigbeeHandlers.cpp:134] zb_cmd_default_resp_handler(): Received default response: from address(0x0), src_endpoint(1) to dst_endpoint(10), cluster(0x402) with status 0x0
[07/12/24 - 12:42:53:092] Updated temperature sensor value to 28.00°C
[07/12/24 - 12:42:53:094] [1982762][V][ZigbeeTempSensor.cpp:70] setTemperature(): Updating temperature sensor value...
[07/12/24 - 12:42:53:100] [1982762][D][ZigbeeTempSensor.cpp:72] setTemperature(): Setting temperature to 2800
[07/12/24 - 12:42:53:290] [1982960][V][ZigbeeHandlers.cpp:134] zb_cmd_default_resp_handler(): Received default response: from address(0x0), src_endpoint(1) to dst_endpoint(10), cluster(0x402) with status 0x0
[07/12/24 - 12:42:54:093] Updated temperature sensor value to 28.00°C
[07/12/24 - 12:42:54:096] [1983764][V][ZigbeeTempSensor.cpp:70] setTemperature(): Updating temperature sensor value...
[... 2 lines cut ...]
[07/12/24 - 12:42:55:099] [1984766][V][ZigbeeTempSensor.cpp:70] setTemperature(): Updating temperature sensor value...
[07/12/24 - 12:42:55:103] [1984766][D][ZigbeeTempSensor.cpp:72] setTemperature(): Setting temperature to 2800
...
,,,
[07/12/24 - 13:27:25:379] Updated temperature sensor value to 27.00°C
[07/12/24 - 13:27:25:382] [4655102][V][ZigbeeTempSensor.cpp:70] setTemperature(): Updating temperature sensor value...
[... cut ...]
[07/12/24 - 13:27:39:370] [4669092][V][ZigbeeCore.cpp:332] esp_zb_app_signal_handler(): ZDO signal: ZDO Device Unavailable (0x3c), status: ESP_OK
[07/12/24 - 13:27:39:407] Updated temperature sensor value to 26.00°C
[07/12/24 - 13:27:39:408] [4669130][V][ZigbeeTempSensor.cpp:70] setTemperature(): Updating temperature sensor value...
[07/12/24 - 13:27:39:413] [4669130][D][ZigbeeTempSensor.cpp:72] setTemperature(): Setting temperature to 2600
[07/12/24 - 13:27:40:409] Updated temperature sensor value to 26.00°C

```

P-R-O-C-H-Y, 2024-12-09:
@michapr Do you know if there is any error log or any other before the 1st `[V][ZigbeeCore.cpp:332] esp_zb_app_signal_handler(): ZDO signal: ZDO Device Unavailable (0x3c), status: ESP_OK`? That can help with the investigation, if there is any.

michapr, 2024-12-09:
Just have checked again,... there are no more debug messages in log files between these events.

P-R-O-C-H-Y, 2024-12-09:
In the HomeAssistant you still see the device as connected?

michapr, 2024-12-09:
it was at Z2M, here I have seen that "last seen" was long time ago. Then have tried same sketch with different modules with same result. Can test later with Homeassistant

michapr, 2024-12-10:
Maybe the `ZPS_eAplAfSendKeepAlive(void)` is missing or not optimal configured? (from Zigbee documentation)
Its related to the Timeout period.

> The parent must receive at least one keep-alive packet from the End Device within
> the timeout period defined using the function ZPS_bAplAfSetEndDeviceTimeout().
> Otherwise, the parent will assume that the child is no longer active and discard the
> child. 

If temperature will not be changed for a longer time (or reporting delta is to high) then device will not send any information to router and router can forget about the device. That's why may work if reporting every 10s...

Only an idea,... not sure of course...

michapr, 2024-12-11:
Have placed the sensor at other place where temperature is changing more - and here it is reporting values since 20 hours about without problems. 
Of course, this could also be a coincidence....

P-R-O-C-H-Y, 2024-12-12:
The keep alive should be sent every 3s by default. Should not be an issue if it's actually sent. I can test it using the sniffer.

P-R-O-C-H-Y, 2024-12-12:
Maybe good point there would be setting the reporting to repent on small delta + also every 1 minute for example. To have both options enabled for reporting.

tofurky, 2024-12-18:
I think I encountered the same thing on ESP32C6 (DFRobot Beetle C6 w/3.1.0 core) using various values for `zbTempSensor.setReporting()`; it wasn't behaving as expected.

Serial output showed that `setTemperature()` was called every second, with the expected values, but updates were not sent over Zigbee on intervals or delta. I figured perhaps Home Assistant was masking the values, but used a Zigbee sniffer to confirm nothing was being sent.

As a workaround, I've just been using my own intervals and calling e.g.:
```
    zbTempSensor.setTemperature(readingAverage);
    zbTempSensor.reportTemperature();
```

An aside, when I tried to completely remove the use of `setReporting()`, I'd get bad values on startup (showed up as -327.7 in Home Assistant, and as invalid 16-bit signed data 0x0080 in Wireshark). I ended up with `zbTempSensor.setReporting(1, 0, 100);`, which prevents the bad values shortly after startup, and prevents duplicates when calling `setTemperature()`.

tofurky, 2024-12-29:
I think I have figured out why the reporting intervals seem not to match up. In Home Assistant (HA) using the ZHA integration, the zigpy library has hard-coded reporting intervals that it uses:  https://github.com/zigpy/zha/blob/a9c9f8b5d28561782b9cdfccee132abc7dac3282/zha/zigbee/cluster_handlers/const.py#L6-L8

When initially adding the device or clicking "RECONFIGURE" in HA, it tells the ESP32C6 to use its own reporting interval, overriding what is set via `setReporting()` and `setHumidityReporting()` per the Arduino library.

Since this is set every time in my code at boot, just restarting the device is enough to have it start honoring what I specified in the code. Unsure if the ESP32C6 stack actually stores the reporting interval/delta values in NVRAM or just RAM.

[image]

So, it's not actually necessary to call e.g. `reportTemperature()` as long as HA hasn't messed with the bindings.

I guess a workaround would be to somehow add a callback some seconds after "Configure Reporting" is issued to the device by HA to call `setReporting()` again to undo the change. Calling it again in my code before `setTemperature()` is enough to restore the desired reporting interval, even after HA messes with the intervals.

razarahil, 2025-02-05:
Is this issue fixed? I'm also getting the same -327.7 reading on HA.

P-R-O-C-H-Y, 2025-02-06:
@razarahil Are you using the latest ESP32 Arduino release? This issue should be fixed already.

razarahil, 2025-02-08:
@P-R-O-C-H-Y I think I found the issue, I was waiting for 9 seconds for the data transmission before putting the device into deep sleep. Turn out 9 seconds is not enough and I've to put the 1 minute timeout.

closed 2025-02-08 by P-R-O-C-H-Y