espressif/arduino-esp32 #10709: SD_MMC: Incorrect pin settings for ESP32

broddo, 2024-12-10:
### Board

ESP32-DEVKITC-32E

### Device Description

SDCARD breakout board wired via breadboard

### Hardware Configuration

N/A

### Version

latest development Release Candidate (RC-X)

### IDE Name

VSCode

### Operating System

macOS 14.5

### Flash frequency

40 Mhz

### PSRAM enabled

yes

### Upload speed

460800

### Description

I recently updated one of my old projects that was using IDF v4 to IDF v5 and upgrade to the latest version of the Arduino library too. The hardware for this project includes an eMMC card that was functioning fine on the old libraries. 

The eMMC card no longer works when using the latest libraries. The reason is due to several calls to the `perimanSetPinBus` which result in this error: 

`[   177][E][esp32-hal-periman.c:122] perimanSetPinBus(): Invalid pin: 255` 

These calls fail because _pin_clk, _pin_cmd are indeed set to -1 if you are configuring for an ESP32. The solution is was to just add a `#ifndef CONFIG_IDF_TARGET_ESP32` to exclude these calls on ESP32 hardware. 

### Sketch

```cpp
#include <Arduino.h>
#include <SD_MMC.h>

void setup() {
  Serial.begin(115200);
  if (!SD_MMC.begin("/sdmmc")) {
    Serial.println("Failed to mount SDMMC module");
  }
}

void loop() { }
```

### Debug Message

```plain
[   177][E][esp32-hal-periman.c:122] perimanSetPinBus(): Invalid pin: 255
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2024-12-10:
In the constructor, [_pin_clk = SDMMC_SLOT1_IOMUX_PIN_NUM_CLK](https://github.com/espressif/arduino-esp32/blob/master/libraries/SD_MMC/src/SD_MMC.cpp#L49), so it should use the pins as defined in [soc/sdmmc_pins.h](https://github.com/espressif/esp-idf/blob/v5.3/components/soc/esp32/include/soc/sdmmc_pins.h).  What do you get if you print out SDMMC_SLOT1_IOMUX_PIN_NUM_CLK?

broddo, 2024-12-11:
D'oh - I missed the define `BOARD_HAS_SDMMC`. Sorry - I can't believe I missed that. It wasn't in my project originally as I using an older version of this library that didn't require it and I'm running custom hardware. I see that it is defined in some board variants but can you suggest a good place for me to define this? 

(I'm not using platformio - I'm using IDF v5.3.1 with Arduino ESP32 (v 3.1.0-RC2 804ee127) added as a component.)

me-no-dev, 2024-12-11:
It should not require `BOARD_HAS_SDMMC` to work. @P-R-O-C-H-Y PTAL

P-R-O-C-H-Y, 2024-12-16:
Closing as solved by PR #10713

PR #10713 (2024-12-11): fix(SDMMC): Remove need for BOARD_HAS_SDMMC for esp32
## Description of Change
This PR removed unnecessary check for define `BOARD_HAS_SDMMC`, which is causing that SDMMC can't be used if a board variant don't have this defined. No other SoC is affected by this PR.

## Tests scenarios
Tested with SDMMC example on ESP32-Wrover-kit

## Related links
Close #10709
closed 2024-12-16 by P-R-O-C-H-Y