espressif/arduino-esp32 #10721: Libraries not working with PlatformIO

Dhruvgoel3829, 2024-12-12:
### Board

ESP32 dev module

### Device Description

esp32 dev module is connected to 4 relays and 4 push buttons.

### Hardware Configuration

nothing else is connected to the dev board

### Version

latest master (checkout manually)

### IDE Name

VSCode with PlatformIO IDE

### Operating System

Windows 11

### Flash frequency

40mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

I am a beginner to esp32. Currently trying to build some standard projects to get a better understanding on how things actually works. I was working on a basic relay interfacing project which takes inputs from the web server as well as from four push buttons.
But my code is not compiling and I think its a problem with the libraries
ERROR I am getting (attached .txt file)
Note: These libraries are not included in the platformIO repo and I have to include them using the link:

https://github.com/espressif/arduino-esp32.git

in the platformio.ini file as lib_deps = https://github.com/espressif/arduino-esp32.git

### Sketch

```cpp
#include <Arduino.h>

#include <WiFi.h>
#include <WebServer.h>

// Wi-Fi access point credentials
const char* ssid = "ESP32_Hotspot";
const char* password = "12345678";

// Static IP configuration
IPAddress local_IP(192, 168, 4, 1);
IPAddress gateway(192, 168, 4, 1);
IPAddress subnet(255, 255, 255, 0);

// GPIO pins for relays (active low)
const int relayPins[] = {25, 26, 32,33};

// GPIO pins for push buttons
const int buttonPins[] = {4, 5, 16, 17};

// Relay states
bool relayStates[] = {HIGH, HIGH, HIGH, HIGH}; // Start with relays off

// Create a WebServer object on port 80
WebServer server(80);

// Toggle relay function
void toggleRelay(int relayIndex) {
  relayStates[relayIndex] = !relayStates[relayIndex];
  digitalWrite(relayPins[relayIndex], relayStates[relayIndex]);
}

// HTML content generation function with feedback and styling
String generateHTML() {
  String html = "<!DOCTYPE html><html><head><meta name=\"viewport\" content=\"width=device-width, initial-scale=1.0\">";
  html += "<style>";
  html += "body { font-family: Arial, sans-serif; max-width: 400px; margin: auto; text-align: center; }";
  html += "h1 { color: #333; }";
  html += ".button { display: inline-block; width: 80%; padding: 15px; margin: 10px; font-size: 18px; color: white; background-color: #4CAF50; border: none; border-radius: 5px; text-decoration: none; }";
  html += ".button.off { background-color: #f44336; }";
  html += ".refresh { background-color: #2196F3; padding: 10px 20px; margin: 15px; color: white; font-size: 18px; border: none; border-radius: 5px; }";
  html += ".footer { margin-top: 20px; color: #666; font-size: 14px; }";
  html += "</style></head><body>";
  html += "<h1>ESP32 Relay Control</h1>";
  html += "<h2>CSE(AI) Sec-B<br></h2";
  html += "<h2>Group 07<br></h2>";
  html += "<h2>Members</h2>";
  html += "<h3>Devansh Rai (25)</h3>";
  html += "<h3>Dhairya Goel (26)</h3>";
  html += "<h3>Dhruv Goel (27)</h3>";
  html += "<h3>Dhruv Kesarwani (28)</h3>";

  
  // Relay control buttons
  for (int i = 0; i < 4; i++) {
    html += "<p>Relay " + String(i+1) + ": " + (relayStates[i] == LOW ? "ON" : "OFF") + "</p>";
    html += "<form action=\"/toggle_relay_" + String(i) + "\" method=\"GET\">";
    html += "<button class=\"button " + String(relayStates[i] == LOW ? "off" : "") + "\" type=\"submit\">" 
            + String(relayStates[i] == LOW ? "Turn OFF" : "Turn ON") + " Relay " + String(i+1) + "</button>";
    html += "</form><br>";
  }

  // Refresh button
  html += "<form action=\"/refresh\" method=\"GET\">";
  html += "<button class=\"refresh\" type=\"submit\">Refresh</button>";
  html += "</form>";
  
  // Footer text
  html += "<div class=\"footer\">KIET GROUP OF INSTITUTIONS<br></div>";
  
  html += "</body></html>";
  return html;
}

// Setup server routes
void setupServerRoutes() {
  // Route for the main page
  server.on("/", HTTP_GET, []() {
    server.send(200, "text/html", generateHTML());
  });

  // Route for relay toggle control
  for (int i = 0; i < 4; i++) {
    int relayIndex = i;
    server.on(("/toggle_relay_" + String(i)).c_str(), HTTP_GET, [relayIndex]() {
      toggleRelay(relayIndex);
      server.send(200, "text/html", generateHTML()); // Update the page with current states
    });
  }

  // Route for refreshing the page without toggling relays
  server.on("/refresh", HTTP_GET, []() {
    server.send(200, "text/html", generateHTML());
  });
}

void setup() {
  Serial.begin(115200);
  
  

  // Initialize relay pins as output and set them to HIGH (inactive for active-low relays)
  for (int i = 0; i < 4; i++) {
    pinMode(relayPins[i], OUTPUT);
    digitalWrite(relayPins[i], HIGH); // Start with relays off
  }

  // Initialize button pins as input with pull-up resistors
  for (int i = 0; i < 4; i++) {
    pinMode(buttonPins[i], INPUT_PULLUP);
  }

  // Configure Wi-Fi with a fixed IP and set up as an access point
  WiFi.softAPConfig(local_IP, gateway, subnet);
  WiFi.softAP(ssid, password);
  Serial.print("Access Point IP: ");
  Serial.println(WiFi.softAPIP());

  // Set up web server routes
  setupServerRoutes();

  // Start the server
  server.begin();
  Serial.println("Web server started.");
}

void loop() {
  // Handle incoming client requests
  server.handleClient();

  // Manual button control
  for (int i = 0; i < 4; i++) {
    if (digitalRead(buttonPins[i]) == LOW) { // Button pressed (active low)
      delay(100); // Debounce delay
      toggleRelay(i); // Toggle the corresponding relay
      while (digitalRead(buttonPins[i]) == LOW); // Wait until button is released
      delay(100); // Debounce delay

    }
  }
 

}
```

### Debug Message

```plain
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (6.8.1) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20017.0 (2.0.17)
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 34 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ArduinoOTA @ 3.0.5+sha.fb6e977
|-- WebServer @ 2.0.0
|-- WiFi @ 2.0.0
Building in release mode
Compiling .pio\build\seeed_xiao_esp32s3\src\main.cpp.o
Building .pio\build\seeed_xiao_esp32s3\bootloader.bin
Generating partitions .pio\build\seeed_xiao_esp32s3\partitions.bin
Compiling .pio\build\seeed_xiao_esp32s3\lib528\ArduinoOTA\ArduinoOTA.cpp.o
esptool.py v4.5.1
Creating esp32 image...
Merged 1 ELF section
Successfully created esp32 image.
Compiling .pio\build\seeed_xiao_esp32s3\liba4e\WiFi\WiFi.cpp.o
Compiling .pio\build\seeed_xiao_esp32s3\liba4e\WiFi\WiFiAP.cpp.o
Compiling .pio\build\seeed_xiao_esp32s3\liba4e\WiFi\WiFiClient.cpp.o
Compiling .pio\build\seeed_xiao_esp32s3\liba4e\WiFi\WiFiGeneric.cpp.o
Compiling .pio\build\seeed_xiao_esp32s3\liba4e\WiFi\WiFiMulti.cpp.o
Compiling .pio\build\seeed_xiao_esp32s3\liba4e\WiFi\WiFiSTA.cpp.o
In file included from .pio/libdeps/seeed_xiao_esp32s3/ArduinoOTA/src/ArduinoOTA.cpp:19:
.pio/libdeps/seeed_xiao_esp32s3/ArduinoOTA/src/ArduinoOTA.h:18:10: fatal error: Network.h: No such file or directory

*****************************************************************
* Looking for Network.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Network.h"
* Web  > https://registry.platformio.org/search?q=header:Network.h
*
*****************************************************************

 #include "Network.h"
          ^~~~~~~~~~~
compilation terminated.
*** [.pio\build\seeed_xiao_esp32s3\lib528\ArduinoOTA\ArduinoOTA.cpp.o] Error 1
```

### Other Steps to Reproduce

I also tried including the libraries by copying the libraries folders to libs folder of my project and it resulted in a different error.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-12-12:
you have mixed versions of this repo. PlatformIO provides up to a very old now version 2.0.17. PlatformIO no longer supports the newer versions of ESP32 Arduino. @Jason2866 can help you switch to pioarduino where they are all supported currently

Jason2866, 2024-12-12:
I suggest you want to use actual stable espressif Arduino core 3.0.7?
So you have just to use the `platform` nothing else.
Change in your `platformio.ini` the entry for `platform` to:
```
platform = https://github.com/pioarduino/platform-espressif32/releases/download/51.03.07/platform-espressif32.zip
```
That's it

Dhruvgoel3829, 2024-12-12:
Thanks for the response. I guess it's the people like you which makes open source great.

closed 2024-12-29 by SuGlider