espressif/arduino-esp32 #10726: Zigbee: Multiple endpoints crash due to static endpoint var

FaBjE, 2024-12-13:
When adding a second Zigbee endpoint the CPU crashes.
I noticed that the same endpoint ID was displayed while I added both endpoints with separate ID's

After some digging I found that the ZigbeeEP class has the endpoint variable as static.
This is not possible as each instance (aka endpoint) must have a unique endpoint ID.
I removed the static and it started working: https://github.com/FaBjE/arduino-esp32/commit/f0566d822bb90af466e421fdb18bd41365c4d6fd

Unfortunately this breaks several classes as they rely on the endpoint and others being static. For testing I removed those classes so I could compile: https://github.com/FaBjE/arduino-esp32/commit/2c98584aff8e7574220891c5f88e68c6f81d6076

I think there is some awareness about this issue, but maybe not the severity when trying to use a second endpoint.
_Originally posted by @P-R-O-C-H-Y in https://github.com/espressif/arduino-esp32/issues/10135#issuecomment-2491740744_

P-R-O-C-H-Y, 2024-12-13:
Hi @FaBjE, what version of the esp-arduino core are you using?
I have no issues running multiple endpoints on last Zigbee library commits.
Can you try to use the v3.1.x branch?

FaBjE, 2024-12-13:
Hi @P-R-O-C-H-Y 
Thanks for your swift response!

I use the latest release V3.0.7

When using master / V3.1.x branch the issues are gone indeed :)

P-R-O-C-H-Y, 2024-12-13:
Its fixed in the 3.1.x, but we did not back port zigbee updates to 3.0.x branch.
We will release the 3.1.0 stable release soon. I am closing this issue as it's solved.
Fix was done in this PR: https://github.com/espressif/arduino-esp32/pull/10551

closed 2024-12-13 by P-R-O-C-H-Y