espressif/arduino-esp32 #10731: wire.requestFrom A large number of bytes

Y1hsiaochunnn, 2024-12-16:
### Board

Any ESP32-S3 core board

### Device Description

The LC76G is connected using I2C and connected via SDA(15) and SCL(14)

### Hardware Configuration

No more

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

40 Mhz

### PSRAM enabled

yes

### Upload speed

921600

### Description

I am using the development board of the ESP32-S3 chip to develop the code to drive the LC76G via I2C, and I have a good idea of how the LC76G should communicate and its register configuration.

When I use the wire library, I use the more common wire.requestfrom for data reading work, but due to the communication requirements of LC76G, the amount of NMEA data read at one time is relatively large, such as 1024 bytes. Then I should use wire.requestFrom(0x54,1024), but it will make an error. I tried a step-by-step transfer, but it didn't work.

Perhaps you can give me some suggestions to use other apis to read i2c data? I am not sure whether the performance of ESP32-S3 itself is enough, if not, could you give me some suggestions to use PSRAM or Nor Flash for caching? Below is some of my code

### Sketch

```cpp
#include <Wire.h>

void setup() {
  Serial.begin(115200);
  Wire.begin(15, 14);
  pinMode(0, INPUT_PULLUP);
}

void loop() {

  if (digitalRead(0) == LOW) {
    Serial.println("Button pressed! Sending I2C data...");
    writeDataToI2C();
    delay(500);
  }
}

void writeDataToI2C() {
  uint8_t deviceAddress = 0x50;
  uint8_t registerAddress = 0x08;
  uint8_t data[] = { 0x00, 0x51, 0xaa, 0x04, 0x00, 0x00, 0x00 };

  Wire.beginTransmission(deviceAddress);
  Wire.write(registerAddress);
  Wire.write(data, sizeof(data));
  byte error = Wire.endTransmission();

  if (error == 0) {
    Serial.println("Data written successfully!");
  } else {
    Serial.print("Error writing data: ");
    Serial.println(error);
  }

  delay(100);

  uint8_t deviceAddress_R = 0x54;
  uint8_t registerPointer = 0x00;
  Wire.beginTransmission(deviceAddress_R);
  Wire.write(registerPointer);
  uint8_t result = Wire.endTransmission(false);
  if (result != 0) {
    Serial.println("Device not responding.");
    delay(1000);
    return;
  }

  uint8_t readData[4];
  Wire.requestFrom(deviceAddress_R, 4);

  int i = 0;
  while (Wire.available()) {
    readData[i++] = Wire.read();
  }

  Serial.print("Data read: ");
  for (int i = 0; i < 4; i++) {
    Serial.print(readData[i]);
    Serial.print(" ");
  }
  Serial.println();

  uint32_t dataLength = (readData[0]) | (readData[1] << 8) | (readData[2] << 16) | (readData[3] << 24);
  if (dataLength > 65535) {
    return;
  }
  // Wire.endTransmission(true);

  delay(100);

  uint8_t data2[] = { 0x00, 0x20, 0x51, 0xaa };
  uint8_t dataToSend[sizeof(data2) + sizeof(readData)];

  memcpy(dataToSend, data2, sizeof(data2));
  memcpy(dataToSend + sizeof(data2), readData, sizeof(readData));

  Wire.beginTransmission(deviceAddress);
  Wire.write(dataToSend, sizeof(dataToSend));

  byte error2 = Wire.endTransmission();

  if (error2 == 0) {
    Serial.println("Data written successfully!");
  } else {
    Serial.print("Error writing data: ");
    Serial.println(error2);
  }

  delay(100);

  Serial.println(dataLength);

  uint8_t* dynamicReadData = new uint8_t[dataLength];

  Wire.beginTransmission(deviceAddress_R);
  Wire.write(registerPointer);
  result = Wire.endTransmission(false);
  if (result != 0) {
    Serial.println("Device not responding for dynamic read.");
    delete[] dynamicReadData;
    return;
  }

  delay(100);

  Wire.requestFrom(deviceAddress_R, dataLength);
  i = 0;
  uint8_t bytesRead = 0;
  while (Wire.available()) {
    if (bytesRead < dataLength) {
      dynamicReadData[bytesRead++] = Wire.read();
    }
  }

  // String nmeaData = "";
  // for (int i = 0; i < dataLength; i++) {
  //   nmeaData += (char)dynamicReadData[i];
  // }
  // Serial.println(nmeaData);

  delete[] dynamicReadData;
}
```

### Debug Message

```plain
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x109c
load:0x403c9700,len:0x4
load:0x403c9704,len:0xb50
load:0x403cc700,len:0x2fe4
entry 0x403c98ac
Button pressed! Sending I2C data...
Data written successfully!
Data read: 228 3 0 0 
Data written successfully!
996
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x403754bb  PS      : 0x00050031  A0      : 0x40377f10  A1      : 0x3fc93d00  
A2      : 0x3fcec830  A3      : 0x474e4724  A4      : 0x3fc91fe4  A5      : 0x74e47240  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x00000001  A9      : 0x3fcf1380  
A10     : 0xb4ad9224  A11     : 0x3fc952d8  A12     : 0x00000000  A13     : 0x3fcf4c88  
A14     : 0x80000001  A15     : 0xb33fffff  SAR     : 0x00000000  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xb4ad9224  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x403754b8:0x3fc93d00 0x40377f0d:0x3fc93d40 0x40378f53:0x3fcf4ab0 0x4200c41d:0x3fcf4ad0 0x4037bd27:0x3fcf4af0 0x4037d14a:0x3fcf4b10

ELF file SHA256: b4d9493ba10731f2

Rebooting...
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2024-12-16:
Try calling [setBufferSize](https://github.com/espressif/arduino-esp32/blob/master/libraries/Wire/src/Wire.h#L127). The default size is 128.

Y1hsiaochunnn, 2024-12-16:
Hi @lbernstone ,

Thank you for your prompt reply, In fact, I tried the setBufferSize method in the first time of your reply, unfortunately, setBufferSize may not be enough. The length of the NMEA data that may be transmitted can reach tens of thousands.

I used the logic analyzer to compare the process of driving the LC76G with the ESP32-S3 and Raspberry PI 5 respectively, and found that the ESP32-S3 failed to read the byte limit. So should I use esp-idf native i2c library?

I can share my code if you want

[ESP32-S3-LC76G-I2C.zip](https://github.com/user-attachments/files/18148096/ESP32-S3-LC76G-I2C.zip)

Best wishes

me-no-dev, 2024-12-16:
Wire library already uses ESP-IDF's driver underneath. After starting Wire, you can try to use the following HAL method and see if that improves anything (it does not use the buffers in Wire) https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-i2c.h#L36 (i2c_num is 0)

Y1hsiaochunnn, 2024-12-20:
Hi @me-no-dev 
Sorry to reply so late. Two days vacation on the way. Currently calling driver/i2c.h directly and reading with the i2c_master_read function works fine, which is great, many people want to use the ESP32 I2C control LC76G to get the NMEA data, which will be more convenient. Thank you again for your reminding, which provides more possibilities for our product design

closed 2024-12-20 by Y1hsiaochunnn