espressif/arduino-esp32 #10742: zigbeeInit crashing continuously

Bobbylebob, 2024-12-17:
### Board

ESP32-H2-DEV-KIT-N4

### Device Description

ESP32-H2-DEV-KIT-N4 from Waveshare, nothing else connected.
Same with custom board ESP32-H2-MINI-1U

### Hardware Configuration

No additionnal peripheral connected to dev kit 

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE 2.3.4

### Operating System

Windows 11

### Flash frequency

16 or 64MHz

### PSRAM enabled

no

### Upload speed

921600 or 115200

### Description

With stock ZigBee examples Zigbee_On_Off_Switch or Zigbee_On_Off_Light :

Continous reboot with Guru Error

### Sketch

```cpp
/**
 * @brief This example demonstrates simple Zigbee light bulb.
 *
 * The example demonstrates how to use Zigbee library to create a end device light bulb.
 * The light bulb is a Zigbee end device, which is controlled by a Zigbee coordinator.
 *
 * Proper Zigbee mode must be selected in Tools->Zigbee mode
 * and also the correct partition scheme must be selected in Tools->Partition Scheme.
 *
 * Please check the README.md for instructions and more detailed description.
 *
 * Created by Jan Procházka (https://github.com/P-R-O-C-H-Y/)
 */

#ifndef ZIGBEE_MODE_ED
#error "Zigbee end device mode is not selected in Tools->Zigbee mode"
#endif

#include "Zigbee.h"

/* Zigbee light bulb configuration */
#define ZIGBEE_LIGHT_ENDPOINT 10
uint8_t led = RGB_BUILTIN;
uint8_t button = BOOT_PIN;
[... 358 lines cut ...]
4081d5a0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4081d5c0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000

ELF file SHA256: f3b0bf008
Rebooting...
```

### Other Steps to Reproduce

- Tried on another identical HW board
- Same error with custom board on H2-MINI-1 chip.
- Tried downgrading to 3.0.0, 3.0.6 lib and last 3.1.0 release
- Tried removing Arduino IDE and cleaning lib folder and fresh install.
- Tried setting 2MB and 4MB memory with Minimal or Default partition schemes
- Tried Setting flash to 16 or 64MHz, QIO or DIO
- Erase flash before upload enabled
- Tried disabling CDC on boot

- Can't run [ExceptionDecoder]: does not show in Arduino IDE after unzipping in sketches/tool folder.

Examples and custom code based on examples were working fine in november '24 (don't remember precisely lib version at this time, 3.0.5 or 3.0.6)

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

P-R-O-C-H-Y, 2024-12-17:
Hi @Bobbylebob, you need to select the proper partition scheme as written in the readme or in the header of the example:
`Arduino FQBN: ... PartitionScheme=default, ...`
Please select the Zigbee partition on the tools menu.

closed 2024-12-17 by P-R-O-C-H-Y

Bobbylebob, 2024-12-17:
Partition Scheme only offers 4MB options with Zigbee. 
Can it be compatible with 2MB chips such as ESP32-H2FH2 ?

P-R-O-C-H-Y, 2024-12-18:
@Bobbylebob I will take a look and reply to you soon about this :)

Levak, 2024-12-20:
Hi there, I see the issue is closed, but was wondering if there was an ongoing investigation given the last comment.

I am using ESP32C6 SUPER MINI with a 6V battery. I am new to DIY ZigBee so I might be doing something wrong. I have 3 different cases:
 - When connected to only USB: the arduino example works perfectly.
 - When connected to both USB and 6V battery: it works half of the time. It doesn't crash, just gets stuck trying to connect and eventually says connection failed, then I reset it, then it works.
 - When only connected to 6V battery: it crashes at the same line as OP, just after ZigBee initialization. Because it's not connected to USB, I had to use another esp32 to get the logs via RX/TX pins. No stack trace, just straight up resets after ZigBee initializing log, like OP.

Verbose mode
Zigbee-ED
Partition ZigBee ED 4MB with spiffs

I am aware that this might be a totally different issue than OP, though this is the only thread I found with the same logs as in my case, trying to find ways to debug further. Any hint?

Best regards,

P-R-O-C-H-Y, 2025-01-06:
@Bobbylebob Can you please test the 2MB partition scheme from this PR if you have any 2MB flash H2? https://github.com/espressif/arduino-esp32/pull/10817