espressif/arduino-esp32 #10743: Upgrading to latest zigbee build. Building errors

HexiDev, 2024-12-17:
### Board

ESP32-C6

### Device Description

ESP32C6DevkitM

### Hardware Configuration

Nothing connected just using the zigbee protocol

### Version

latest master (checkout manually)

### IDE Name

VSCode PlatformIO

### Operating System

Windows 11

### Flash frequency

default

### PSRAM enabled

yes

### Upload speed

115200

### Description

After going from platformio/framework-arduinoespressif32#3.0.7 to platformio/framework-arduinoespressif32#3.1.0 I expierenced building issues. 

Also Important I manually changed pioarduino-build.py to platformio-build.py. As platformio was trying to access platformio-build.py. in .platformio\packages\framework-arduinoespressif32\tools

### Sketch

```cpp
My ini file is:
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:esp32-c6-devkitm-1]
platform = espressif32
platform_packages = 
	platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#3.1.0
	platformio/framework-arduinoespressif32-libs @ https://github.com/espressif/esp32-arduino-lib-builder/releases/download/idf-release_v5.1/esp32-arduino-libs-idf-release_v5.1-632e0c2a.zip
framework = arduino
board = esp32-c6-devkitm-1
monitor_filters = direct
board_build.partitions = partitions.csv
monitor_speed = 115200
build_flags = 
	-D CORE_DEBUG_LEVEL=5
	-D ARDUINO_USB_MODE=1
	-D CONFIG_ZB_ENABLED=1
	-D ARDUINO_USB_CDC_ON_BOOT=1
	-D ZIGBEE_MODE_ED
        -lesp_zb_api_ed -lesp_zb_cli_command -lzboss_stack.ed -lzboss_port

; Depending on the Zigbee mode add this
; -D ZIGBEE_MODE_ZCZR
; -lesp_zb_api_zczr -lesp_zb_cli_command -lzboss_stack.zczr -lzboss_port
; -D ZIGBEE_MODE_ED
; -lesp_zb_api_ed -lesp_zb_cli_command -lzboss_stack.ed -lzboss_port
; -D ZIGBEE_MODE_GPD
; -lesp_zb_api_gpd  -lesp_zb_cli_command -lzboss_stack.gpd -lzboss_port
; -D ZIGBEE_MODE_RCP
; -lesp_zb_api_rcp -lesp_zb_cli_command -lzboss_stack.rcp -lzboss_port
board_build.embed_txtfiles = 
	managed_components/espressif__esp_insights/server_certs/https_server.crt
	managed_components/espressif__esp_rainmaker/server_certs/rmaker_mqtt_server.crt
	managed_components/espressif__esp_rainmaker/server_certs/rmaker_claim_service_server.crt
	managed_components/espressif__esp_rainmaker/server_certs/rmaker_ota_server.crt
lib_deps = adafruit/Adafruit NeoPixel@^1.12.3

and the main.cpp is just a simple hello world sketch using log_i.
```

### Debug Message

```plain
In file included from C:/Users/Youri/.platformio/packages/framework-arduinoespressif32/cores/esp32/HWCDC.cpp:18:
C:/Users/Youri/.platformio/packages/framework-arduinoespressif32/cores/esp32/HWCDC.cpp: In static member function 'static bool HWCDC::deinit(void*)':
C:/Users/Youri/.platformio/packages/framework-arduinoespressif32/cores/esp32/HWCDC.cpp:289:33: error: 'USB_INT_PHY0_DM_GPIO_NUM' was not declared in this scope
  289 |   retCode &= perimanClearPinBus(USB_INT_PHY0_DM_GPIO_NUM);
      |                                 ^~~~~~~~~~~~~~~~~~~~~~~~
C:/Users/Youri/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-periman.h:18:48: note: in definition of macro 'perimanClearPinBus'
   18 | #define perimanClearPinBus(p) perimanSetPinBus(p, ESP32_BUS_TYPE_INIT, NULL, -1, -1)
      |                                                ^
C:/Users/Youri/.platformio/packages/framework-arduinoespressif32/cores/esp32/HWCDC.cpp:290:33: error: 'USB_INT_PHY0_DP_GPIO_NUM' was not declared in this scope
  290 |   retCode &= perimanClearPinBus(USB_INT_PHY0_DP_GPIO_NUM);
      |                                 ^~~~~~~~~~~~~~~~~~~~~~~~
C:/Users/Youri/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-periman.h:18:48: note: in definition of macro 'perimanClearPinBus'
   18 | #define perimanClearPinBus(p) perimanSetPinBus(p, ESP32_BUS_TYPE_INIT, NULL, -1, -1)
      |                                                ^
Compiling .pio\build\esp32-c6-devkitm-1\FrameworkArduino\esp32-hal-cpu.c.o
C:/Users/Youri/.platformio/packages/framework-arduinoespressif32/cores/esp32/HWCDC.cpp: In member function 'void HWCDC::begin(long unsigned int)':
C:/Users/Youri/.platformio/packages/framework-arduinoespressif32/cores/esp32/HWCDC.cpp:326:17: error: 'USB_INT_PHY0_DM_GPIO_NUM' was not declared in this scope
  326 |   uint8_t pin = USB_INT_PHY0_DM_GPIO_NUM;
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~
Compiling .pio\build\esp32-c6-devkitm-1\FrameworkArduino\esp32-hal-dac.c.o
C:/Users/Youri/.platformio/packages/framework-arduinoespressif32/cores/esp32/HWCDC.cpp:330:9: error: 'USB_INT_PHY0_DP_GPIO_NUM' was not declared in this scope
  330 |   pin = USB_INT_PHY0_DP_GPIO_NUM;
      |         ^~~~~~~~~~~~~~~~~~~~~~~~
Compiling .pio\build\esp32-c6-devkitm-1\FrameworkArduino\esp32-hal-gpio.c.o
*** [.pio\build\esp32-c6-devkitm-1\FrameworkArduino\HWCDC.cpp.o] Error 1
C:/Users/Youri/.platformio/packages/framework-arduinoespressif32/cores/esp32/HardwareSerial.cpp: In member function 'void HardwareSerial::begin(long unsigned int, uint32_t, int8_t, int8_t, bool, long unsigned int, uint8_t)':
C:/Users/Youri/.platformio/packages/framework-arduinoespressif32/cores/esp32/HardwareSerial.cpp:301:19: error: 'SOC_UART_HP_NUM' was not declared in this scope; did you mean 'SOC_UART_NUM'?
  301 |   if (_uart_nr >= SOC_UART_HP_NUM) {
      |                   ^~~~~~~~~~~~~~~
      |                   SOC_UART_NUM
*** [.pio\build\esp32-c6-devkitm-1\FrameworkArduino\HardwareSerial.cpp.o] Error 1
C:/Users/Youri/.platformio/packages/framework-arduinoespressif32/cores/esp32/chip-debug-report.cpp: In function 'void printChipInfo()':
C:/Users/Youri/.platformio/packages/framework-arduinoespressif32/cores/esp32/chip-debug-report.cpp:90:10: error: 'CHIP_ESP32P4' was not declared in this scope; did you mean 'CHIP_ESP32S3'?
   90 |     case CHIP_ESP32P4: chip_report_printf("ESP32-P4\n"); break;
      |          ^~~~~~~~~~~~
      |          CHIP_ESP32S3
*** [.pio\build\esp32-c6-devkitm-1\FrameworkArduino\chip-debug-report.cpp.o] Error 1
C:/Users/Youri/.platformio/packages/framework-arduinoespressif32/cores/esp32/Esp.cpp: In member function 'const char* EspClass::getChipModel()':
C:/Users/Youri/.platformio/packages/framework-arduinoespressif32/cores/esp32/Esp.cpp:303:10: error: 'CHIP_ESP32P4' was not declared in this scope; did you mean 'CHIP_ESP32S3'?
  303 |     case CHIP_ESP32P4: return "ESP32-P4";
      |          ^~~~~~~~~~~~
      |          CHIP_ESP32S3
*** [.pio\build\esp32-c6-devkitm-1\FrameworkArduino\Esp.cpp.o] Error 1
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

P-R-O-C-H-Y, 2024-12-17:
@HexiDev Also for Zigbee examples you need to select proper partition. For Zigbee ED its` zigbee.csv` partition scheme and for Coordinator/router its `zigbee_zczr.csv`partiton scheme.

HexiDev, 2024-12-17:
May I ask where I could find those partition scheme's? I just used some random zigbee partition from some random forum post.

P-R-O-C-H-Y, 2024-12-17:
> May I ask where I could find those partition scheme's? I just used some random zigbee partition from some random forum post.

https://github.com/espressif/arduino-esp32/tree/release/v3.1.x/tools/partitions

Jason2866, 2024-12-17:
Exactly this partitions can and has to be used. See zigbee example
https://github.com/pioarduino/platform-espressif32/blob/main/examples/arduino-zigbee-light/platformio.ini

**Latest Arduino core 3.1.0 is not compatible to Platformio anymore**
Platformio decided to drop support. You can use with the community version `pioarduino`
You just have to change your platform to
```
platform = https://github.com/pioarduino/platform-espressif32/releases/download/53.03.10/platform-espressif32.zip
```
nothing else more. Remove `platform_packages` completely!

HexiDev, 2024-12-17:
Hi I tried the following platformio ini file:
```ini
[env:esp32-c6-devkitm-1]
platform = https://github.com/pioarduino/platform-espressif32/releases/download/53.03.10/platform-espressif32.zip
framework = arduino
board = esp32-c6-devkitm-1
monitor_filters = direct
board_build.partitions = partitions.csv
monitor_speed = 115200
build_flags = 
	-D CORE_DEBUG_LEVEL=5
	-D ARDUINO_USB_MODE=1
	-D CONFIG_ZB_ENABLED=1
	-D ARDUINO_USB_CDC_ON_BOOT=1
	-D ZIGBEE_MODE_ED
        -lesp_zb_api_ed -lesp_zb_cli_command -lzboss_stack.ed -lzboss_port

; Depending on the Zigbee mode add this
; -D ZIGBEE_MODE_ZCZR
; -lesp_zb_api_zczr -lesp_zb_cli_command -lzboss_stack.zczr -lzboss_port
; -D ZIGBEE_MODE_ED
; -lesp_zb_api_ed -lesp_zb_cli_command -lzboss_stack.ed -lzboss_port
; -D ZIGBEE_MODE_GPD
; -lesp_zb_api_gpd  -lesp_zb_cli_command -lzboss_stack.gpd -lzboss_port
; -D ZIGBEE_MODE_RCP
; -lesp_zb_api_rcp -lesp_zb_cli_command -lzboss_stack.rcp -lzboss_port
board_build.embed_txtfiles = 
	managed_components/espressif__esp_insights/server_certs/https_server.crt
	managed_components/espressif__esp_rainmaker/server_certs/rmaker_mqtt_server.crt
	managed_components/espressif__esp_rainmaker/server_certs/rmaker_claim_service_server.crt
	managed_components/espressif__esp_rainmaker/server_certs/rmaker_ota_server.crt
```
But now I get other error's about missing files. I also deleted all the packages to let platformio reinstall everything but I still get the same error:
```
Successfully created esp32c6 image.
riscv32-esp-elf-gcc: fatal error: cannot execute 'C:/Users/Youri/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/as.exe': CreateProcess: No such file or directory
compilation terminated.
Compiling .pio\build\esp32-c6-devkitm-1\FrameworkArduino\Esp.cpp.o
*** [.pio\build\esp32-c6-devkitm-1\FrameworkArduino\ColorFormat.c.o] Error 1
riscv32-esp-elf-g++: fatal error: cannot execute 'C:/Users/Youri/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/as.exe': CreateProcess: No such file or directory
compilation terminated.
*** [.pio\build\esp32-c6-devkitm-1\src\main.cpp.o] Error 1
riscv32-esp-elf-g++: fatal error: cannot execute 'C:/Users/Youri/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/as.exe': CreateProcess: No such file or directory
compilation terminated.
*** [.pio\build\esp32-c6-devkitm-1\FrameworkArduino\Esp.cpp.o] Error 1
```

HexiDev, 2024-12-17:
Tried reinstalling it again and now I get a package mirror error. Seems like it is offline?

```
oolchain-riscv32-esp-windows_amd64-13.2.0+20240530.tar.gz (Caused by NameResolutionError("<urllib3.connection.HTTPSConnection object at 0x0000021314A3EA10>: Failed to resolve 'eu2.contabostorage.com' ([Errno 11004] getaddrinfo failed)"))
Tool Manager: Looking for another mirror...
Tool Manager: Error: Please read https://bit.ly/package-manager-ioerror
Tool Manager: Warning! Package Mirror: HTTPSConnectionPool(host='eu2.contabostorage.com', port=443): Max retries exceeded with url: /c0a5502915964bc09d2607baa2984f98:pioregmirroreu/tools/95/a6/f40160316be00f77e024013647fda363087c23d4a819c74641e63e0ec7a1/toolchain-riscv32-esp-windows_amd64-13.2.0+20240530.tar.gz (Caused by NameResolutionError("<urllib3.connection.HTTPSConnection object at 0x0000021314A15910>: Failed to resolve 'eu2.contabostorage.com' ([Errno 11004] getaddrinfo failed)"))
Tool Manager: Looking for another mirror...
Tool Manager: Error: Please read https://bit.ly/package-manager-ioerror
Tool Manager: Warning! Package Mirror: HTTPSConnectionPool(host='eu2.contabostorage.com', port=443): Max retries exceeded with url: /c0a5502915964bc09d2607baa2984f98:pioregmirroreu/tools/95/a6/f40160316be00f77e024013647fda363087c23d4a819c74641e63e0ec7a1/toolchain-riscv32-esp-windows_amd64-13.2.0+20240530.tar.gz (Caused by NameResolutionError("<urllib3.connection.HTTPSConnection object at 0x0000021314A21990>: Failed to resolve 'eu2.contabostorage.com' ([Errno 11004] getaddrinfo failed)"))
Tool Manager: Looking for another mirror...
Tool Manager: Error: Please read https://bit.ly/package-manager-ioerror
Tool Manager: Warning! Package Mirror: HTTPSConnectionPool(host='eu2.contabostorage.com', port=443): Max retries exceeded with url: /c0a5502915964bc09d2607baa2984f98:pioregmirroreu/tools/95/a6/f40160316be00f77e024013647fda363087c23d4a819c74641e63e0ec7a1/toolchain-riscv32-esp-windows_amd64-13.2.0+20240530.tar.gz (Caused by NameResolutionError("<urllib3.connection.HTTPSConnection object at 0x00000213149FC8D0>: Failed to resolve 'eu2.contabostorage.com' ([Errno 11004] getaddrinfo failed)"))
Tool Manager: Looking for another mirror...
Tool Manager: Error: Please read https://bit.ly/package-manager-ioerror
Tool Manager: Warning! Package Mirror: HTTPSConnectionPool(host='eu2.contabostorage.com', port=443): Max retries exceeded with url: /c0a5502915964bc09d2607baa2984f98:pioregmirroreu/tools/95/a6/f40160316be00f77e024013647fda363087c23d4a819c74641e63e0ec7a1/toolchain-riscv32-esp-windows_amd64-13.2.0+20240530.tar.gz (Caused by NameResolutionError("<urllib3.connection.HTTPSConnection object at 0x0000021314988190>: Failed to resolve 'eu2.contabostorage.com' ([Errno 11004] getaddrinfo failed)"))
Tool Manager: Looking for another mirror...
Tool Manager: Error: Please read https://bit.ly/package-manager-ioerror
Tool Manager: Warning! Package Mirror: HTTPSConnectionPool(host='eu2.contabostorage.com', port=443): Max retries exceeded with url: /c0a5502915964bc09d2607baa2984f98:pioregmirroreu/tools/95/a6/f40160316be00f77e024013647fda363087c23d4a819c74641e63e0ec7a1/toolchain-riscv32-esp-windows_amd64-13.2.0+20240530.tar.gz (Caused by NameResolutionError("<urllib3.connection.HTTPSConnection object at 0x00000213149E7CD0>: Failed to resolve 'eu2.contabostorage.com' ([Errno 11004] getaddrinfo failed)"))
Tool Manager: Looking for another mirror...
```

Jason2866, 2024-12-17:
@HexiDev Welcome in the wonderful world of Microsoft path length limit. This is a Windows only issue caused from limitations in Windows not solved since decades. To workaround and even enjoy a faster compile speed install and use WSL

HexiDev, 2024-12-17:
Oh, well thats great... I already have wsl but it tends to crash on medium to high loads. might need to go from windows to linux ^^.

Jason2866, 2024-12-17:
@HexiDev Seems your hardware system setup is not reliable. WSL works fast and reliable in general. MS has done a good job with WSL!

HexiDev, 2024-12-17:
Just compiled it using WSL on my laptop. And yeez what is this fast and easy! Thanks for helping me I appreciate a lot.

closed 2024-12-17 by HexiDev