espressif/arduino-esp32 #10746: Zigbee: Improve the sleep sketch with regard to energy requirements

michapr, 2024-12-17:
### Board

ESP32H2

### Device Description

Devboard, Supermini

### Hardware Configuration

nothing connected

### Version

latest development Release Candidate (RC-X)

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

The device require too much energy in active and sleeping mode.
[image]

1) not clear, why there are so many peaks with 100mA
2) sleeping current is in my case about 450uA - too much. Any idea how to minimize it for ESP32H2?
3) - maybe other request - provide a light sleep example without needed rejoin

### Sketch

```cpp
A bit modified sleeping sketch:
----------------------------------

#ifndef ZIGBEE_MODE_ED
#error "Zigbee end device mode is not selected in Tools->Zigbee mode"
#endif

#include "Zigbee.h"
#include <rom/rtc.h>

#define BOOT_PIN                  9  //Boot button for C6/H2
/* Zigbee temperature + humidity sensor configuration */
#define TEMP_SENSOR_ENDPOINT_NUMBER 10

#define uS_TO_S_FACTOR 1000000ULL /* Conversion factor for micro seconds to seconds */
#define TIME_TO_SLEEP  55         /* Sleep for 55s will + 5s delay for establishing connection => data reported every 1 minute */

uint8_t button = BOOT_PIN;

ZigbeeTempSensor zbTempSensor = ZigbeeTempSensor(TEMP_SENSOR_ENDPOINT_NUMBER);

/********************* Arduino functions **************************/
void setup() {
  Serial.begin(115200);

  // Init button switch
  pinMode(button, INPUT_PULLUP);

  // Configure the wake up source and set to wake up every 5 seconds
  esp_sleep_enable_timer_wakeup(TIME_TO_SLEEP * uS_TO_S_FACTOR);

  // Optional: set Zigbee device name and model
  zbTempSensor.setManufacturerAndModel("Espressif", "SleepyZigbeeTempSensorTest");

  // Set minimum and maximum temperature measurement value (10-50°C is default range for chip temperature measurement)
  zbTempSensor.setMinMaxValue(10, 50);

  // Set tolerance for temperature measurement in °C (lowest possible value is 0.01°C)
  zbTempSensor.setTolerance(1);

  // Set power source to battery and set battery percentage to measured value (now 100% for demonstration)
  // The value can be also updated by calling zbTempSensor.setBatteryPercentage(percentage) anytime
  zbTempSensor.setPowerSource(ZB_POWER_SOURCE_BATTERY, 100);

  // Add humidity cluster to the temperature sensor device with min, max and tolerance values
  zbTempSensor.addHumiditySensor(0, 100, 1);

  // Add endpoint to Zigbee Core
  Zigbee.addEndpoint(&zbTempSensor);

  // Create a custom Zigbee configuration for End Device with keep alive 10s to avoid interference with reporting data
  esp_zb_cfg_t zigbeeConfig = ZIGBEE_DEFAULT_ED_CONFIG();
  zigbeeConfig.nwk_cfg.zed_cfg.keep_alive = 10000;

  // When all EPs are registered, start Zigbee in End Device mode
  if (!Zigbee.begin(&zigbeeConfig, false)) {
    Serial.println("Zigbee failed to start!");
    Serial.println("Rebooting...");
    ESP.restart();
  }
  Serial.println("Connecting to network");
  while (!Zigbee.connected()) {
    Serial.print(".");
    delay(100);
  }
  Serial.println();
  Serial.println("Successfully connected to Zigbee network");

  // Delay approx 1s (may be adjusted) to allow establishing proper connection with coordinator, needed for sleepy devices
   if ( (int)rtc_get_reset_reason(0) != 5 )  { // =  SW_CPU_RESET=12  // POWERON_RESET=1 //DEEPSLEEP_RESET=5
     delay(20000);
   }
}

void loop() {
  // Checking button for factory reset
  if (digitalRead(button) == LOW) {  // Push button pressed
    // Key debounce handling
    delay(100);
    int startTime = millis();
    while (digitalRead(button) == LOW) {
      delay(50);
      if ((millis() - startTime) > 3000) {
        // If key pressed for more than 3secs, factory reset Zigbee and reboot
        Serial.println("Resetting Zigbee to factory and rebooting in 1s.");
        delay(1000);
        Zigbee.factoryReset();
      }
    }
  }

  // Call the function to measure temperature and put the device to sleep
  meausureAndSleep();
}
```

### Debug Message

```plain
No errors in debug
```

### Other Steps to Reproduce

disabled USB CDC, debugging for saving energy.
Have removed the internal temperature procedure and replaced with random value.

Have used Nordic Power Profiler Kit II for measurements with 3.3V power

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

michapr, 2024-12-18:
Have added before `Zigbee.begin` (because was thinking about scanning):
`Zigbee.setPrimaryChannelMask(0x800);`
but no change.  (channel 11 in my case)

P-R-O-C-H-Y, 2024-12-18:
I will be able to test the consumption at the start of new year again to check if there is anything that can help.
But I can tell that if you are willing to do the LightSleep you have to switch to use the esp-zigbee-sdk directly. They have the light sleep example there. For Arduino it's not possible now, as there need to be a tickles RTOS config set, which we won't have anytime soon, as it may bring unexpected behavior.

P-R-O-C-H-Y, 2024-12-18:
> Have added before `Zigbee.begin` (because was thinking about scanning): `Zigbee.setPrimaryChannelMask(0x800);` but no change. (channel 11 in my case)

What you can do that can help with optimizing stuff is to provide a pcap file from Wireshark, to see what is happening at that time. If there are some transmissions going on in the peaks. Otherwise for now there is nothing you can set/select to make it more power efficient.

P-R-O-C-H-Y, 2024-12-18:
Btw the energy consumption in the deep sleep is probably related to the board design. I can't tell what board I used if it was C6 or H2 Espressifs devout, but in deep sleep the power consumption was 9,8 uA. That's a big difference.

michapr, 2024-12-18:
My Wireshark setup is not working, haven't found time to find out what happen...

About spikes: what I see in the Gateway logs is, that looks like a routing problem after rejoin (get messages like `ezspIncomingNetworkStatusHandler(): callback called with: [errorCode=ROUTE_ERROR_TREE_LINK_FAILURE], [target=12084]` ).
Same effect in ZHA (other log lines in debug mode) and in Zigbee2MQTT 

> Otherwise for now there is nothing you can set/select to make it more power efficient.

Why it is not possible to save all session information in RTC RAM and use it after reboot from deep sleep mode (if valid)?
Same was realized for ESP32 Lora modules and is working fine.

> I can't tell what board I used if it was C6 or H2 Espressifs devout, but in deep sleep the power consumption was 9,8 uA. That's a big difference.

without any additional settings? like pin setting to analog input or disable anything?

P-R-O-C-H-Y, 2024-12-18:
> without any additional settings? like pin setting to analog input or disable anything?
> Something like Zigbee.end() is not needed to switch off the modem part?

No additional settings, I am just using the sleepy example how its written.
In Deep-sleep mode, the CPUs, most of the RAM, and all digital peripherals that are clocked from APB_CLK are powered off. The only parts of the chip that remain powered on are:

- RTC controller
- ULP coprocessor
- RTC FAST memory

So there is no need for `Zigbee.end()`

> Why it is not possible to save all session information in RTC RAM and use it after reboot from deep sleep mode (if valid)?
> Same was realized for ESP32 Lora modules and is working fine.

I will need to investigate this, if there is anything that can help. But the stuff is saved in Zigbee partitions, and it should be recalled. But the device still has to reconnect to the network.

> My Wireshark setup is not working, haven't found time to find out what happen...

If you have some time, would b nice to have the logs from the Wireshark.

> About spikes: what I see in the Gateway logs is, that looks like a routing problem after rejoin (get messages like ezspIncomingNetworkStatusHandler(): callback called with: [...]

I can ask about this, but for sure more informations would be needed (WireShark traffic).

michapr, 2024-12-18:
>I will need to investigate this, if there is anything that can help. But the stuff is saved in Zigbee partitions, and it should be recalled. But the device still has to reconnect to the network.

But the "rejoin process" should be other and much faster.
If you compare with commercial temperature sensors - the online time is here < 1second (if taken the logs from Zigbee gateway to compare it)
Device will be recognized without a real rejoin.

michapr, 2024-12-21:
>>    About spikes: what I see in the Gateway logs is, that looks like a routing problem after rejoin (get messages like ezspIncomingNetworkStatusHandler(): callback called with: [errorCode=ROUTE_ERROR_TREE_LINK_FAILURE], [target=12084] ).
    Same effect in ZHA (other log lines in debug mode) and in Zigbee2MQTT

>I can ask about this, but for sure more informations would be needed (WireShark traffic).

Cannot confirm my thoughts.... the routing errors was something else...

In Wireshark see only short Zigbee time. So the other code before need so much time for "something"...

[image]
and
[image]

This difference is interesting, but not the top important thing....

P-R-O-C-H-Y, 2025-01-06:
> But the "rejoin process" should be other and much faster. If you compare with commercial temperature sensors - the online time is here < 1second (if taken the logs from Zigbee gateway to compare it) [...]

The difference is probably the deep sleep (we use) vs light sleep (they use).
If you need to make it more power efficient you should move to use esp-idf and the zigbee-sdk directly and use the light sleep mode there. I will keep track on this and if I have any news, I will post an update if there is any improvement.

P-R-O-C-H-Y, 2025-01-06:
> @P-R-O-C-H-Y Light sleep is not a solution for ESP32 - too high sleeping current. What I really do not understand is - what happen all the time after wakeup and first Zigbee packet. Zigbee live time is ok, wakeup time also - but if you look to debug log, you see that a long time "nothing" happen (ZBOSS), only the spikes are visible. This is the most time of all the process.

@michapr Would you be able to do some timeline (I can do it also, but it would help a lot if you can) to see the times from debug log and the packets from Wireshark to have them side by side to really see, the "nothing" happen in any of those logs and the spikes(peaks) are there? I can then report it and discuss more deeply with the Zigbee team. Don't forget to select the Zigbee ED debug libs :)

michapr, 2025-01-06:
Used the sample sketch above.

Debug log:

```
=========== Before Setup Start ===========
Chip Info:
------------------------------------------
  Model             : ESP32-H2
  Package           : 0
  Revision          : 0.01
  Cores             : 1
  CPU Frequency     : 96 MHz
  XTAL Frequency    : 32 MHz
  Features Bitfield : 0x00000050
  Embedded Flash    : No
  Embedded PSRAM    : No
  2.4GHz WiFi       : No
  Classic BT        : No
  BT Low Energy     : Yes
  IEEE 802.15.4     : Yes
------------------------------------------
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   235808 B ( 230.3 KB)
  Free Bytes        :   202452 B ( 197.7 KB)
  Allocated Bytes   :    28204 B (  27.5 KB)
  Minimum Free Bytes:   197640 B ( 193.0 KB)
  Largest Free Block:   184308 B ( 180.0 KB)
------------------------------------------
Flash Info:
------------------------------------------
  Chip Size         :  4194304 B (4 MB)
  Block Size        :    65536 B (  64.0 KB)
  Sector Size       :     4096 B (   4.0 KB)
  Page Size         :      256 B (   0.2 KB)
  Bus Speed         : 64 MHz
  Bus Mode          : QIO
------------------------------------------
Partitions Info:
------------------------------------------
                nvs : addr: 0x00009000, size:    20.0 KB, type: DATA, subtype: NVS
            otadata : addr: 0x0000E000, size:     8.0 KB, type: DATA, subtype: OTA
               app0 : addr: 0x00010000, size:  1280.0 KB, type:  APP, subtype: OTA_0
               app1 : addr: 0x00150000, size:  1280.0 KB, type:  APP, subtype: OTA_1
             spiffs : addr: 0x00290000, size:  1388.0 KB, type: DATA, subtype: SPIFFS
         zb_storage : addr: 0x003EB000, size:    16.0 KB, type: DATA, subtype: FAT
             zb_fct : addr: 0x003EF000, size:     4.0 KB, type: DATA, subtype: FAT
           coredump : addr: 0x003F0000, size:    64.0 KB, type: DATA, subtype: COREDUMP
------------------------------------------
Software Info:
------------------------------------------
  Compile Date/Time : Jan  6 2025 11:29:50
  Compile Host OS   : windows
  ESP-IDF Version   : v5.3.2-174-g083aad99cf-dirty
  Arduino Version   : 3.1.0
------------------------------------------
Board Info:
------------------------------------------
  Arduino Board     : ESP32H2_DEV
  Arduino Variant   : esp32h2
  Arduino FQBN      : esp32:esp32:esp32h2:UploadSpeed=921600,CDCOnBoot=cdc,FlashFreq=64,FlashMode=qio,FlashSize=4M,PartitionScheme=zigbee,DebugLevel=verbose,EraseFlash=all,JTAGAdapter=default,ZigbeeMode=ed_debug
============ Before Setup End ============
[   990][I][esp32-hal-periman.c:141] perimanSetPinBus(): Pin 26 already has type USB_DM (38) with bus 0x408114d4
[   991][I][esp32-hal-periman.c:141] perimanSetPinBus(): Pin 27 already has type USB_DP (39) with bus 0x408114d4
[   992][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type GPIO (1) successfully set to 0x42003c24
[   993][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 9 successfully set to type GPIO (1) with bus 0xa
ZBOSS: zcl/zcl_common.c:215   >> zb_zcl_get_attribute_size: attr_type 0x42 attr_value 0x4081a44c
ZBOSS: zcl/zcl_common.c:348   << zb_zcl_get_attribute_size: result 10
ZBOSS: zcl/zcl_common.c:215   >> zb_zcl_get_attribute_size: attr_type 0x42 attr_value 0x4081a468
[... 84 lines cut ...]
[  3496][V][ZigbeeTempSensor.cpp:71] setTemperature(): Updating temperature sensor value...
[  3498][D][ZigbeeTempSensor.cpp:73] setTemperature(): Setting temperature to 2000
[  3499][V][ZigbeeTempSensor.cpp:112] setHumidity(): Updating humidity sensor value...
[  3501][D][ZigbeeTempSensor.cpp:114] setHumidity(): Setting humidity to 2000
[  3506][V][ZigbeeTempSensor.cpp:93] reportTemperature(): Temperature report sent
[  3512][V][ZigbeeTempSensor.cpp:135] reportHumidity(): Humidity report sent
Reported temperature: 20.00°C, Humidity: 20.00%
Going to sleep now

[  3700][V][ZigbeeHandlers.cpp:134] zb_cmd_default_resp_handler(): Received default response: from address(0x0), src_endpoint(1) to dst_endpoint(10), cluster(0x402) with status 0x0
[  3717][V][ZigbeeHandlers.cpp:134] zb_cmd_default_resp_handler(): Received default response: from address(0x0), src_endpoint(1) to dst_endpoint(10), cluster(0x405) with status 0x0
```

Wireshark:

[image]

Data from Nordic Power Profiler v4.2.1
[image]

Zigbee online time:
[image]

Over all:
[image]

michapr, 2025-01-23:
@P-R-O-C-H-Y Are here any news about the ZBOSS procedure? What happens here during this time?

Thanks!

P-R-O-C-H-Y, 2025-01-28:
@michapr No news yet, once I have anything I will let you know :)

psi-4ward, 2025-02-01:
I'd a cheap temp sensor a while ago and it mostly chooses a bulb as router but the bulb was not reachable for some months. Zigbee2mqtt map showed that the sensor uses another router after some time but the battery drained every 2 months. After powering the bulb again the battery holds a year.

So long story short, could it be the case the device has to find another router after wakeup?

def1149, 2025-02-23:
I've played around with this Zigbee example and determined that there's some kind of quirk (bug?)  when calling esp_deep_sleep_start(); from a function. I see the same high current behavior you reported. I moved esp_deep_sleep_start(); to the main loop and I see the expected about 19 uA during deep sleep.  I also made a few other changes to optimize deep sleep power saving.

It works pretty well 

But, I have a few open issues to solve:
1. If I power off and reconnect I have to remove the device from Home Assistant and re-enroll it unless I wait a while to reconnect then it works without re-enrolling???. 
2. There are a couple of errors in the log I need to sort out.

**During operation**
```
09:53:06.303 -> [   954][D][ZigbeeCore.cpp:84] addEndpoint(): Endpoint: 10, Device ID: 0x0302
09:53:06.303 -> [   955][D][ZigbeeCore.cpp:123] zigbeeInit(): Initialize Zigbee stack
09:53:06.303 -> [   978][D][ZigbeeCore.cpp:130] zigbeeInit(): Register all Zigbee EPs in list
09:53:06.303 -> [   980][I][ZigbeeCore.cpp:138] zigbeeInit(): List of registered Zigbee EPs:
09:53:06.303 -> [   981][I][ZigbeeCore.cpp:140] zigbeeInit(): Device type: Temperature Sensor, Endpoint: 10, Device ID: 0x0302
ERROR >>>> 09:53:06.335 -> [   996][V][ZigbeeCore.cpp:352] esp_zb_app_signal_handler(): ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL
09:53:06.335 -> [   998][I][ZigbeeCore.cpp:219] esp_zb_app_signal_handler(): Zigbee stack initialized
09:53:08.674 -> [  3356][I][ZigbeeCore.cpp:225] esp_zb_app_signal_handler(): Device started up in non factory-reset mode
09:53:08.674 -> [  3357][I][ZigbeeCore.cpp:238] esp_zb_app_signal_handler(): Device rebooted
09:53:08.674 -> [  3358][D][ZigbeeCore.cpp:479] searchBindings(): Requesting binding table for address 0x20d8
09:53:08.674 -> [  3360][D][ZigbeeCore.cpp:419] bindingTableCb(): Binding table callback for address 0x20d8 with status 0
09:53:08.674 -> [  3361][D][ZigbeeCore.cpp:422] bindingTableCb(): Binding table info: total 2, index 0, count 2
09:53:08.718 -> [  3362][D][ZigbeeCore.cpp:432] bindingTableCb(): Binding table record: src_endp 10, dst_endp 1, cluster_id 0x0402, dst_addr_mode 3
09:53:08.718 -> [  3363][D][ZigbeeCore.cpp:449] bindingTableCb(): Device bound to EP 10 -> device endpoint: 1, short addr: 0x0000, ieee addr: F0:82:C0:FF:FE:6B:DB:EC
09:53:08.718 -> [  3365][D][ZigbeeCore.cpp:432] bindingTableCb(): Binding table record: src_endp 10, dst_endp 1, cluster_id 0x0001, dst_addr_mode 3
09:53:08.718 -> [  3366][D][ZigbeeCore.cpp:449] bindingTableCb(): Device bound to EP 10 -> device endpoint: 1, short addr: 0x0000, ieee addr: F0:82:C0:FF:FE:6B:DB:EC
09:53:08.718 -> [  3368][D][ZigbeeCore.cpp:470] bindingTableCb(): Filling bounded devices finished
09:53:08.718 -> Connecting to network
09:53:08.718 -> 
09:53:08.718 -> Successfully connected to Zigbee network
09:53:08.923 -> [  3570][V][ZigbeeEP.cpp:97] setBatteryPercentage(): Battery percentage updated
09:53:08.923 -> [  3571][V][ZigbeeTempSensor.cpp:58] setTemperature(): Updating temperature sensor value...
09:53:08.923 -> [  3573][D][ZigbeeTempSensor.cpp:60] setTemperature(): Setting temperature to 900
09:53:08.923 -> [  3574][V][ZigbeeEP.cpp:112] reportBatteryPercentage(): Battery percentage reported
09:53:08.923 -> [  3578][V][ZigbeeTempSensor.cpp:80] reportTemperature(): Temperature report sent
09:53:08.923 -> Reported temperature: 9.00°C, Battery Pct: 51%
09:53:09.095 -> Going to sleep now
```
**During enrollment:**
```
0:10:53.804 -> [   802][D][ZigbeeCore.cpp:84] addEndpoint(): Endpoint: 10, Device ID: 0x0302
10:10:53.804 -> [   803][D][ZigbeeCore.cpp:123] zigbeeInit(): Initialize Zigbee stack
10:10:53.837 -> [   877][D][ZigbeeCore.cpp:130] zigbeeInit(): Register all Zigbee EPs in list
10:10:53.882 -> [   879][I][ZigbeeCore.cpp:138] zigbeeInit(): List of registered Zigbee EPs:
10:10:53.882 -> [   880][I][ZigbeeCore.cpp:140] zigbeeInit(): Device type: Temperature Sensor, Endpoint: 10, Device ID: 0x0302
10:10:53.882 -> [   898][V][ZigbeeCore.cpp:352] esp_zb_app_signal_handler(): ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL
10:10:53.882 -> [   900][I][ZigbeeCore.cpp:219] esp_zb_app_signal_handler(): Zigbee stack initialized
10:10:56.268 -> [  3264][I][ZigbeeCore.cpp:225] esp_zb_app_signal_handler(): Device started up in non factory-reset mode
10:10:56.268 -> [  3265][I][ZigbeeCore.cpp:238] esp_zb_app_signal_handler(): Device rebooted
10:10:56.268 -> [  3266][D][ZigbeeCore.cpp:479] searchBindings(): Requesting binding table for address 0x20d8
10:10:56.268 -> [  3268][D][ZigbeeCore.cpp:419] bindingTableCb(): Binding table callback for address 0x20d8 with status 0
10:10:56.268 -> [  3269][D][ZigbeeCore.cpp:422] bindingTableCb(): Binding table info: total 2, index 0, count 2
10:10:56.268 -> [  3270][D][ZigbeeCore.cpp:432] bindingTableCb(): Binding table record: src_endp 10, dst_endp 1, cluster_id 0x0402, dst_addr_mode 3
10:10:56.268 -> [  3271][D][ZigbeeCore.cpp:449] bindingTableCb(): Device bound to EP 10 -> device endpoint: 1, short addr: 0x0000, ieee addr: F0:82:C0:FF:FE:6B:DB:EC
10:10:56.268 -> [  3273][D][ZigbeeCore.cpp:432] bindingTableCb(): Binding table record: src_endp 10, dst_endp 1, cluster_id 0x0001, dst_addr_mode 3
10:10:56.268 -> [  3274][D][ZigbeeCore.cpp:449] bindingTableCb(): Device bound to EP 10 -> device endpoint: 1, short addr: 0x0000, ieee addr: F0:82:C0:FF:FE:6B:DB:EC
10:10:56.268 -> [  3275][D][ZigbeeCore.cpp:470] bindingTableCb(): Filling bounded devices finished
[... cut ...]
 ERROR >>>> 10:10:56.778 -> [  3817][W][ZigbeeHandlers.cpp:24] zb_action_handler(): Receive unhandled Zigbee action(0x12) callback
```
**My  version**
[... 139 lines cut ...]
        //ESP_LOGV("Resetting Zigbee to factory and rebooting in 1s.");
        delay(1000);
        Zigbee.factoryReset();
      }
    }
  }
  // Read temp and report if warranted 
  checkTemp();
  // Put device to deep sleep
  Serial.println("Going to sleep now");
  //delay(100);
  esp_deep_sleep_start();
}
```
**Sleep Power Consumption**

[image]

**One-hour Average** 

[image]

Home Assistant Zigbee device page

[image]

michapr, 2025-02-24:
@def1149 - can you please show the power usage (chart from PPK) for the active process for your device?

Would be interesting to compare it to my H2

def1149, 2025-02-24:
I'm using the Adafruit Feather ESP32-C6 which has a MAX17048 battery charger/monitor chip that I'm now using to measure the battery voltage and percent remaining since my previous post.  https://www.adafruit.com/product/5933
I'll next be substituting a real temperature sensor for the very inaccurate CPU_temp sensor. I haven't decided on which one.
I have Log_V (verbose) enabled in the settings 
CDC on boot enabled
Clock 80 MHz 

**Wake-up temp check cycle without a Zigbee connection.** 

[image]

**Wake-up temp check cycle with a Zigbee connection**

[image]

michapr, 2025-02-25:
It's the same what I got with C6.
Main problem (in my opinion) is the ZBOSS function - it take too much time/energy in this process.
Would be really interesting what happen here in this time frame - and how to optimize it (see above).

about sensor - any good low power I2C sensor will be fine...

@P-R-O-C-H-Y - still no answer from ZBOSS team? Does it make sense for me to contact them - will I get an answer?
As far as I understand this is a special version for Espressif, isn't it?

def1149, 2025-02-25:
I'm using the TMP36 analog temperature sensor or a DS18B20 depending on the use case. The DS18B20 is better if you need a long connection lead.

I found out why I had to re-join network after making changes. I had set erase flash before upload. It seems that the Arduino Zigbee library functions save some network configuration on the data partion it creates. If that gets erased, you have to rejoin the network. I erase flash the first time I program a device then I reset not to erase flash. Now I don't have to keep rejoining after making changes.

 
[image]

[image]

michapr, 2025-02-25:
@def1149 - about erase flash - it is well known that you have to do it this way. You can often read about it here ;)
without this it will not work, onlc if you will not change Zigbee endpoints etc. it can work again.

About sensor - DS18B20 is only good for long distance, but it take too much time (and energy) in most cases for battery devices. And the accuracy is not so good either...

def1149, 2025-02-26:
After setting Core Debug to None I've minimized power use.

**Wake up for a temperature  check w/o making a Zigbee connection** 

[image]

**Wake up and report**

[image]

michapr, 2025-02-26:
yes,... it's the same about as for my devices (I get about 2850ms)

In this discussion we are looking for the spikes before Zigbee communication - see here: https://github.com/espressif/arduino-esp32/issues/10746#issuecomment-2573076829
This is inside the ZBOSS module.

xieqinan, 2025-02-27:
Hi @michapr,

> What I really do not understand is - what happen all the time after wakeup and first Zigbee packet.

Regarding this issue, if the device is not factory reset, it will register its ZCL model with the stack, restore network information from NVRAM, and scan the network based on the `channelmask` to locate its parent for rejoining. If a parent is found, the device will initiate a `rejoin request` to attempt to rejoin the network.  

From the figure above, the peaks suggest that the stack is scanning channels 11 to 26—a total of 16 times—which matches the number of peaks. Could you double-check this? If you have access to `esp_ieee802154_set_channel()`, you can print which channels are being scanned when the device attempts to rejoin.  

I will also debug this issue simultaneously, and any updates will be shared here.

P-R-O-C-H-Y, 2025-02-28:
> > Using void setPrimaryChannelMask(uint32_t mask); have not changed anything.
> 
> Please ensure that `setPrimaryChannelMask()` is called before `zigbeeInit()`, and the `setPrimaryChannelMask()` only takes effect for factory-new devices (which has not any the network information).
> 
> Could you share the code that includes the Zigbee configurations?
> 
> > Where I can find the esp_ieee802154_set_channel() ?
> 
> You can find it [here](https://github.com/espressif/esp-idf/blob/0461e2ff88369c3da0d4caced31e8488f53376cd/components/ieee802154/esp_ieee802154.c#L55).

@xieqinan Can you please check if the call of esp_zb_set_primary_network_channel_set() in the right place (order) of Zigbee Initialization? https://github.com/espressif/arduino-esp32/blob/7575fa0ce8fc8eb3cf886fe28ef1bd881cbbe01d/libraries/Zigbee/src/ZigbeeCore.cpp#L151

Do we need to also call the `esp_ieee802154_set_channel()`?

michapr, 2025-02-28:
> and the `setPrimaryChannelMask()` only takes effect for factory-new devices (which has not any the network information).

That was the key point ... (I have tested this without creating new device)
Great!
Without defined channel:
[image]

**With** defined channel:
[image]

**We save with a defined channel about 82% energy (!!!)**  (35mAs <-> 199mAs)
This is a great and significant improvement for battery-powered devices - just as I expected!

I always thought that this must be possible - even with an ESP32 and its "special deep sleep properties" ;)

BUT:
> and the setPrimaryChannelMask() only takes effect for factory-new devices (which has not any the network information).

Does that really make sense?
We only know the channel after the join process - so we could remember it for the rejoin and save a lot of time/energy.

If the channel changes (new configuration of the gateway) - then you can delete the memorized channel after x failed attempts and perform a full scan again and save the new channel if necessary.

In this case, this makes the development of battery-operated devices much easier.

What do you think?

michapr, 2025-02-28:
Additional info for my ESP32C6 energy use (with sleeping current) - without any external sensor:

[image]

So optimizing this procedure really makes sense, as we can see...

xieqinan, 2025-03-03:
Hi @michapr ，

> Does that really make sense?
> We only know the channel after the join process - so we could remember it for the rejoin and save a lot of time/energy.

I haven't fully understood your point, but I think it makes sense. Typically, a Zigbee network does not change its channel. When a device rejoins, it should scan only the known and available channels instead of all channels, which is more efficient and consumes less energy, right? If the channel changes (i.e., the network configuration changes), the device should leave the current network, scan the channels specified by the primary channel mask, and select the appropriate network to join, rather than simply deleting the memorized channel (which would be insecure).

Could you please share more details about your idea?

P-R-O-C-H-Y, 2025-03-03:
I guess we can implement this behavior in the Zigbee library, but having this implemented in the esp-zigbee-sdk directly would be best for everyone. As shown in the info by @michapr it enhances and optimizes a power consumption a lot.

def1149, 2025-03-03:
I've created Zigbee temperature sensor EP using an ESP32-C6 and the Arduino Zigbee library from Express. I'm using the Express example to test: https://github.com/espressif/arduino-esp32/tree/master/libraries/Zigbee/examples/Zigbee_Temperature_Sensor
The device joins my ZHA network and reports in but the temperature reported never changes in HA after the initial report when I subject the device to temperature change using a coffee mug warmer. I have logging set to verbose. The log shows Zigbee updates being sent without error.  The RSSI and LQI values are updating, as is the last seen time,  so the device is reporting in over Zigbee.  If I go into Manage Zigbee device in the HA device page and read the temperature cluster the temperature updates in HA.  Not sure where to go from here to figure out what's going on

rdorys, 2025-03-03:
@michapr Thx for the info and your tenacity ! Now... Time to store network info in RTC_ATTR to skip annoucement and have sub 500ms reports ! 😉

For future reference and people not very confortable with this code:
I needed to get this version of ZigbeeCore.cpp
[arduino-esp32/libraries/Zigbee/src/ZigbeeCore.cpp](https://github.com/espressif/arduino-esp32/blob/7575fa0ce8fc8eb3cf886fe28ef1bd881cbbe01d/libraries/Zigbee/src/ZigbeeCore.cpp#L151)

And then, use `#define CHANNEL_MASK 0x00000800 //channel 11
` at the beginning of your sleepy example code and setting the mask just after adding the Endpoint

`Zigbee.addEndpoint(&zbTempSensor);
  Zigbee.setPrimaryChannelMask(CHANNEL_MASK);`

I reflashed the code erasing the FLASH to ensure my device is "new" (meanwhile I also force removed it from Z2M and restarted Z2M)

[image].

Thus, I managed to get this from deep sleep.

[image]

One of the issue I'm still facing is that I first need to add a huge delay to ensure proper pairing in Z2M, then, reupload (without erasing flash) a "fast" version of the code with lower delay just after the initial connection.

Going from this : 
`while (!Zigbee.connected()) {
    Serial.print(".");
    delay(10);
  }
  Serial.println();
  Serial.println("Successfully connected to Zigbee network");
  meausure();
  delay(10000); //allowing proper pairing`

To this :
`while (!Zigbee.connected()) {
    Serial.print(".");
    delay(10);
  }
  Serial.println();
  Serial.println("Successfully connected to Zigbee network");
  meausure();
  // Short delay to assure proper connection
  delay(70);`

@P-R-O-C-H-Y 
Any idea how to properly implement this and then improve existing SED examples ?

xieqinan, 2025-03-04:
Hi @michapr,

> If I have to reconfigure the Zigbee network because of overloaded channel (WiFi etc.) all devices will be "lost".
In this time my other (commercial) Zigbee devices reconnect to the Zigbee network by themselves - just with a different channel.
Is this really insecure? The channel is not part of the key, as far as I know, correct?

The original issue is why a non-factory-new device only scans the current channel instead of all channels. Scanning all channels is intended to support network migration. However, for a non-factory-new device, Zigbee provides the `Mgmt_NWK_Update_req()` command to update network information for remote devices, rather than erasing the memorized channel and resetting the primary channel mask. Doing so could create orphan networks and cause short-term network failures, which would be insecure.

By the way, the `Mgmt_NWK_Update_req()` command is optional and does not have an available API in the current Zigbee libraries (v1.6.4). If you need this feature, please submit an issue [here](https://github.com/espressif/esp-zigbee-sdk/issues) to request it.

michapr, 2025-03-04:
> Zigbee provides the `Mgmt_NWK_Update_req()` command to update network information for remote devices, rather than erasing the memorized channel and resetting the primary channel mask. Doing so could create orphan networks and cause short-term network failures, which would be insecure.

ok, I have to admit that I don't quite understand this (compared to the current solution - we scan all channels every time), but of course we should choose the method according to the Zigbee standard.

Can the above suggestion be realized to save the channel used during the join and then use it during the rejoin - and thus save energy?

michapr, 2025-03-04:
@rdorys 
> One of the issue I'm still facing is that I first need to add a huge delay to ensure proper pairing in Z2M, then, reupload (without erasing flash) a "fast" version of the code with lower delay just after the initial connection.

Please check the sample sketch above (first post) - here it is working as expected
```
   if ( (int)rtc_get_reset_reason(0) != 5 )  { // =  SW_CPU_RESET=12  // POWERON_RESET=1 //DEEPSLEEP_RESET=5
     delay(20000);
   }
```
> Now... Time to store network info in RTC_ATTR to skip annoucement and have sub 500ms reports !

It does not help, because the channel mask will be taken only before the first join, and not more... (this is the current status)
see above:
>and the setPrimaryChannelMask() only takes effect for factory-new devices (which has not any the network information).

Using the sample code above and remove the comment for channel mask(for ***first*** join) will be enough.

def1149, 2025-03-04:
> I guess we can implement this behavior in the Zigbee library, but having this implemented in the esp-zigbee-sdk directly would be best for everyone. As shown in the info by [@michapr](https://github.com/michapr) it enhances and optimizes a power consumption a lot.

First:  I appreciate the effort by EXPRESSIF to bring Zigbee to the masses by creating an Arduino library and the Zigbee-capable ESP32 devices

But, what we (I) need is documentation of all the Zigbee Library functions that are exposed in the Arduino Libary and guidance on how they all work in relation to each other. The examples are useful but very minimal and don't demonstrate the use of all the API. The comments in the code explain functions briefly, but there is no overall example of how all functions are used in what sequence or, dependencies on, or interactions with other functions. All this may be obvious to Zigbee experts but not to non-Zigbee experts (like me) who are using Arduino and trying to implement Zigbee for the first time. 

I tried some time ago using the official Zigbee documentation on non-ESP32 devices and the effort to master wasn't worth it. 

The whole channel mask discussion above exemplifies the lack of comprehensive documentation for the Arduino library. 
Minimizing power use for battery-operated devices is the main reason why I'm trying to use Zigbee on my favorite devices, the EPP32 family. Previously I used deep sleep ESP32-S2 and S3 devices and Wi-fi. In my limited experience Wi-fi uses at least 4x the power to connect and send an equivalent data packet than does Zigbee on the C6.

def1149, 2025-03-07:
After figuring out how to stop the scanning every time the Feather ESP32-C6 connects I'm able to optimize power consumption for a Sleepy Arduino temperature sensor.  I'm good 👍

- Sleep  = 18 uA / 4.5 uC
- Wake up and check the temperature using a TMP36 and decide if to report = 24.7 mA avg for 58 mS  / 1.43mC
- Wake up check temperature, measure battery Pct. using a MAX17043, and report  = 65.2 mA avg for 663 mS / 43.21 mC

My report logic is:  Wake up every 60 seconds and check the temperature. If the new temperature varies from the previous temperature (+/-) by more than N.N degrees or it's been > than M check cycles report.

psi-4ward, 2025-03-07:
Sounds very good. With this a expected battery on lifetime on two AAA boils down to at least a year

rdorys, 2025-03-07:
@def1149 I got about the exact same power consomption. I don’t know if we have reached the limit for this board. 
I am also trying to accomplish the same thing using a nrf52840. Report is 400uC and absolutely instantaneous (and doesn’t announce itself in Z2M). 
Idle consumption is about 16uA for now but it seems some people have reached the advertised 1,5uA (i’m still tweaking config file and searching for hw leaks). 
Even, more, the nez nrf54L15 series is supposed to reach 0,8uA in rtc deep sleep.

def1149, 2025-03-07:
I use 2500 mAH rechargeable li-poly batteries ... I'm good for about 2 years 😁

def1149, 2025-03-07:
@rdorys if a suitable nrf52840 board and Arduino library comes along I'll take a look.
For now I'm good. The Zigbee radios and radio code in the C6 seem to be pretty good. I have a bunch of test mules located so they are running down at about -85 to -75 dBm and they've been solid and haven't gone AWOL

michapr, 2025-03-07:
@rdorys @def1149 
I also think that we could realize now nice battery powered devices (or solar powered with Lipo) if the fixed channel after join will be integrated in library.

The most important thing now is to select the right sensors that are both energy-efficient and have a short processing time.
As I have shown in the table above, the values (without sensor) are very promising.

def1149, 2025-03-07:
From the datasheet, the H2 looks to be about 2x as efficient when sending Zigbee as the C6 
I'm looking for an H2 board to try out

psi-4ward, 2025-03-07:
Things like removing ldo from dev boards makes a huge difference: https://github.com/TomMajor/SmartHome/tree/master/Info/Ruhestrom#ruhestrom-mit-sensor-boards

Example bme280

rdorys, 2025-03-07:
@psi-4ward you're right but embedded ldo is already really good on the xiao C6 board (SGM6029 with a typical 2.3µA quiescent current). So imo it isn't worth it to remove it (btw you would need to remove the metal shield).

michapr, 2025-03-07:
@psi-4ward 
> Things like removing ldo from dev boards makes a huge difference

It really depends a lot on the board.
Different boards use different LDOs.

The ESP32C6 Supermini, for example, achieves 31uA deep sleep from the Batt input without RGB LED - but with LDO - and 23uA from the 3V input.
With the ESP32H2 Supermini, the values with LDO are much worse.
For XIAO @rdorys told... - BTW: I have got 10uA there from BATT input without removing anything - so no bad....

Real differences with/without LDO can be seen, for example, with the ESP32S3Zero (190uA <-> 8.5uA)

So yes, it can make sense to replace the LDO with something better, depending on how you want to power your device and which board you use.

michapr, 2025-03-07:
@def1149 
> From the datasheet, the H2 looks to be about 2x as efficient when sending Zigbee as the C6

What do you mean with **_2x as efficient_** ? Used electrical power for same RF power (with same CPU frequency)?

def1149, 2025-03-07:
@michapr  from the data sheets 

[image]

[image]

michapr, 2025-03-07:
@def1149 - so I will remove the LDO from my H2 and make some tests ;)
Have seen before that active current is lower,... right... but because sleeping was not "effective", made all other tests with C6

def1149, 2025-03-07:
I'm guessing that by leaving out Wi-Fi they were able to make the radio in the H2 more efficient

michapr, 2025-03-08:
I have tested same sketch with ESP32H2 - it saves 55% energy compared to the C6 !

[image]

[image]

ESP32-H2 powered with 3V, LDO removed. Sleeping current in my case 9µA

**BUT:** To do it correctly, you would actually have to wait for the responses from the Zigbee gateway - the timeout of 100ms before the deepsleep is not enough, it will probably be about 300ms.  This may not be critical for some sensors...
See also https://github.com/espressif/arduino-esp32/issues/11038
Will add final results when the function is available.

But in any case, it is a very good value for battery-powered devices!

def1149, 2025-03-08:
That's what I expected from the datasheet. And I agree for optimal power uses, there needs to be some type of completion status that can be polled rather than just using a long delay to ensure that the data was delivered.

Now we need some H2 boards with a Qwiic connector, a battery connector for Li-ion and Li-Poly batteries,  and a low-power battery monitor (gas gauge) chip like the MAX17048

Jason2866, 2025-03-13:
Please open a discussion about how to implement / build a low energy device. Thx
Github issues is not ment for. It is hard to follow here if the original issue is now solved or not.
Is there still an issue? If yes please close this issue and open a new one (with a reference to this one) where the issue is shortly described. Thx.

michapr, 2025-03-13:
@xieqinan @P-R-O-C-H-Y 
Should I open a new issue for this, or are you working on this issue?

>Typically, a Zigbee network does not change its channel. When a device rejoins, it should scan only the known and available channels instead of all channels, which is more efficient and consumes less energy, right?

> Can the above suggestion be realized to save the channel used during the join and then use it during the rejoin - and thus save energy?

If you remember, the problem is, that after wakeup from deep sleep the device will scan all channels for the rejoin - and consumes a lot of energy

P-R-O-C-H-Y, 2025-03-14:
@michapr Lets keep this open, I will be testing currently the scanning after rejoin, if that works as expected and link this issue to the PR with the changes :)

michapr, 2025-03-14:
@P-R-O-C-H-Y  Can confirm that it is working fine! Thank you for your work!

If somebody else want to confirm, would be nice.
(line with channel definition in sketch above must be removed)

P-R-O-C-H-Y, 2025-03-14:
> @P-R-O-C-H-Y  Can confirm that it is working fine! Thank you for your work!
> 
> If somebody else want to confirm, would be nice.
> (line with channel definition in sketch above must be removed)
> 

Just to clarify, any example can be used to test that :) just hit the reset button to see the "fast" reconnection

P-R-O-C-H-Y, 2025-03-14:
> I have confirmed the PR works. During initial enrollment, all 16 channels are scanned. On restart, or power off and power on, only one channel is scanned.
> 
> What happens if the channel on the controller is changed? Will the ED perform a full scan?
> 
> I changed the channel on the controller and the ED will repeatedly scan and never connect. Commercial Zibee devices will rescan reconnect after a channel change. I confirmed reconnecting on channel change with a Lumi temperature sensor, Third Reality Mains Switch, and Third Reality Door Sensor. Is there a function call to reinitiate a full scan if a connected attempt fails on the current saved channel? Maybe a call to erase the saved channel information. Would that force a new full channel scan ?

Thank you for additional info and testing :) I did not cover this scenario. I was testing only if the device gets factory reset, this scenario works. I will take a look and add some function or check if there is any callback available that channel was changed.

michapr, 2025-03-14:
I wanted to write about the same just now :))

as @xieqinan wrote:
>If the channel changes (i.e., the network configuration changes), the device should leave the current network, scan the channels specified by the primary channel mask, and select the appropriate network to join, rather than simply deleting the memorized channel (which would be insecure).

Would be nice to realize this too.

def1149, 2025-03-14:
If a device fails to connect it should clear the current channel and rescan for a new channel on the same controller. That would be secure

michapr, 2025-03-14:
> If a device fails to connect

I think there should be a definable timeout so that a full scan is not carried out immediately after the first connection problems. Or a definable number of failed attempts...

P-R-O-C-H-Y, 2025-03-14:
> > If a device fails to connect
> 
> I think there should be a definable timeout so that a full scan is not carried out immediately after the first connection problems. Or a definable number of failed attempts...

I agree. Do you think this will be good to handle in user code in check if(Zigbee.begin())? You can simply add a number of tries + I will add a API to reset the channel scan mask.

[11 more replies]

PR #11123 (2025-03-19): feat(zigbee): Save network channel after 1st joining for faster rejoin
## Description of Change
This PR improves Zigbee rejoining process, as the network channel is saved to zb NVRAM, so after device reset the scanning will occur only on the saved channel saving time and energy.

## Tests scenarios
Tested using sleepy example and HA. (did not test the actual power consumption, but the reconnection is a lot faster)

## Related links

Closes #10746
closed 2025-03-19 by me-no-dev