espressif/arduino-esp32 #10751: Wrong idf dependecies for release 3.1.0 on ESP Component Registry

Tobi177, 2024-12-18:
### Board

None

### Device Description

None

### Hardware Configuration

None

### Version

latest master (checkout manually)

### IDE Name

ESP-IDF

### Operating System

Ubuntu

### Flash frequency

None

### PSRAM enabled

no

### Upload speed

None

### Description

On the latest Arduino-esp32 release v3.1.0  for the ESP Component Registry, the idf dependecies are wrong. The release requires ESP-IDF >=5.1,<5.2 which needs to be corrected. The arduino release v3.1.0 is based on ESP-IDF >=5.3.0 or even ESP-IDF >=5.3.2.

The issue can be checked easily at
https://components.espressif.com/components/espressif/arduino-esp32/versions/3.1.0/dependencies

For the v3.1.0 arduino release the branch "0515264" is used, which display the wrong idf requirements as well:
https://github.com/espressif/arduino-esp32/tree/0515264aa6316a7ed2388300000fb78bf5da9c7e/
At least "CMakeLists.txt" and "idf_component.yml" needs to be updated to allow usage with ESP-IDF >=5.3.

Currently, it ist not possible to use the arduino-esp32 component v3.1.0 on esp-idf as component. from registry. Hopefully, this information helps to fix this issue. 

### Sketch

```cpp
None
```

### Debug Message

```plain
None
```

### Other Steps to Reproduce

None

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-12-18:
I can confirm the issue. For some reason the package was generated from master and not from the branch that it was supposed to be. @lucasssvaz PTAL

me-no-dev, 2024-12-18:
@Tobi177 in a mean time you can just clone arduino from the repository. current master is what 3.1.0 is

me-no-dev, 2024-12-18:
We have fixed the scripts that upload the component, but are waiting for our colleagues to allow re-upload of the new file. Will close this once that happens

me-no-dev, 2024-12-19:
Fixed by releasing a new component with version `3.1.0~1` and yanking the old one from the registry. @Tobi177 please try again! https://components.espressif.com/components/espressif/arduino-esp32/versions/3.1.0~1/dependencies?language=en

PR #10752 (2024-12-18): fix(component): Checkout proper branch for uploading component
## Description of Change

Fix uploading component to IDF registry.

## Tests scenarios

CI

## Related links

Fixes #10751
closed 2024-12-19 by me-no-dev