espressif/arduino-esp32 #10754: 3.1.0 with ESP32 C6 - SimpleTime.ino produces mutex not ready errors

Daedelos, 2024-12-18:
### Board

M5Stack M5NanoC6

### Device Description

None

### Hardware Configuration

Nothing

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE 2.3.4

### Operating System

Windows 11

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

3.1.0 with ESP32 C6 - SimpleTime produces mutex not ready errors

Running SimpleTime.ino example under 3.1.0 with a M5NanoC6 (ESP32 C6) produces repeated:

E (50100) OPENTHREAD: esp_openthread_task_switching_lock_acquire(38): Failed to acquire the lock because the mutex is not ready
E (50100) OPENTHREAD: esp_openthread_task_switching_lock_release(46): Failed to release the lock because the mutex is not ready
E (50111) OT_DNS64: Cannot find NAT64 prefix
No time available (yet)

### Sketch

```cpp
Example SimpleTime.ino code
```

### Debug Message

```plain
E (50100) OPENTHREAD: esp_openthread_task_switching_lock_acquire(38): Failed to acquire the lock because the mutex is not ready
E (50100) OPENTHREAD: esp_openthread_task_switching_lock_release(46): Failed to release the lock because the mutex is not ready
E (50111) OT_DNS64: Cannot find NAT64 prefix
```

### Other Steps to Reproduce

SimpleTime.ino on M5NanoC6 with 3.0.7 works correctly.
ESP32 & ESP32 S2 work correctly on 3.1.0.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2024-12-19:
@me-no-dev - I confirm the issue.
The error message points to some issue  with SDK Config settings related to IPv6 and OpenThread.

KamadoTanjiro-beep, 2024-12-20:
I am getting the same error. Is there any fix to it? I am unable to make HTTP GET calls. Even the Async Server is broken now. Everything seems broken with XIAO ESP32C6 moving from XIAO ESP32C3.

KamadoTanjiro-beep, 2024-12-20:
Another Example. While running this program, the same error on XIAO ESP32C6.
Program:
```
#include <NTPClient.h>
// change next line to use with another board/shield
//#include <ESP8266WiFi.h>
#include <WiFi.h> // for WiFi shield
//#include <WiFi101.h> // for WiFi 101 shield or MKR1000
#include <WiFiUdp.h>

const char *ssid     = "SSID";
const char *password = "PASS";

WiFiUDP ntpUDP;
NTPClient timeClient(ntpUDP);

void setup(){
  Serial.begin(115200);

  WiFi.begin(ssid, password);

  while ( WiFi.status() != WL_CONNECTED ) {
    delay ( 500 );
    Serial.print ( "." );
  }

  timeClient.begin();
}

void loop() {
  timeClient.update();

  Serial.println(timeClient.getFormattedTime());

  delay(1000);
}

```

Error:
```
E (36035) OPENTHREAD: esp_openthread_task_switching_lock_acquire(38): Failed to acquire the lock because the mutex is not ready
E (36035) OPENTHREAD: esp_openthread_task_switching_lock_release(46): Failed to release the lock because the mutex is not ready
E (36046) OT_DNS64: Cannot find NAT64 prefix
```

KamadoTanjiro-beep, 2024-12-20:
I can confirm that any kind of network activity, mostly requests are leading to this issue.

SuGlider, 2024-12-20:
The issue has been found, but the fix will require a new Arduino release because it is embedded into OpenThread static precompiled library.

We are working to solve it. I'll post the progress in this issue thread.
Fell free to ask about it.

SuGlider, 2024-12-20:
Lib Builder PR https://github.com/espressif/esp32-arduino-lib-builder/pull/257 fixes this issue.
A new ESP32 Arduino Release will address and solve the problem.
Thanks for the report!

KamadoTanjiro-beep, 2024-12-21:
@SuGlider Wow, you people are like `heroes without the cape` of the open-source world. Thank You very much.

desiFish, 2024-12-23:
> Lib Builder PR [espressif/esp32-arduino-lib-builder#257](https://github.com/espressif/esp32-arduino-lib-builder/pull/257) fixes this issue. A new ESP32 Arduino Release will address and solve the problem. Thanks for the report!

@SuGlider I am new to these things, I explored few things, so a new ESP32 Arduino Release will happen when these milestones are completed right? These: [Milestones](https://github.com/espressif/arduino-esp32/milestone/31)

motamman, 2024-12-24:
I have reproduced this error on all the C6 boards I have laying around. Basically, anytime I use a domain name for anything, it blows up. If I shift to IP addresses, the problem disappears. It makes it a little cumbersome switching back and forth between chips.

SuGlider, 2024-12-25:
The issue is directly related to DNS resolution.

ESP32 Arduino will solve it by releasing a new package version that fixes it by removing the OpenThread protocol DNS64 client feature. 

This shall be done soon in the beginning of 2025.

At the same time OpenThread IDF Team is investigating the issue in order to fix DNS64 client. When this is solved, we shall reactivate it in the ESP32 Arduino Configuration.

That shall be done by releasing again a new version package.

SuGlider, 2024-12-25:
FYI:  An IDF fix has been tested and I confirmed that it solves this issue.

Therefore, it will be merged soon to IDF and we will be able to enable back OpenThread DNS64 Client support with no problems related to WiFi DNS Client resolution as here reported.

SuGlider, 2024-12-25:
> @SuGlider I am new to these things, I explored few things, so a new ESP32 Arduino Release will happen when these milestones are completed right? These: [...]

I think that a fix release will be released first.
I'll confirm that on the Jan 6th when the team will back from vacation.

SuGlider, 2025-01-05:
> Is there a workaround for this bug? I tried using an IP address vs. hostname as a string (e.g. "162.159.200.123" instead of "pool.ntp.org") as the third argument to `configTime` but got the same error.

The solution is ready. We will release as soon as possible a new ESP32 Arduino version that solves it.

pohlt, 2025-01-06:
> [Tracking image][https://github.com/notifications/beacon/ABIOQE6ZGHCMGTK2F6TDCOL2JFHR3A5CNFSM6AAAAABT3NTJY6WGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTUZJCE5C.gif]

You can always go back to 3.0.x as I did.

dirkvranckaert, 2025-01-06:
Facing the same issue here. Seems like 3.1.0 fixes some other issue I'm facing but introducing this mutex stuff...

Hoping for a quick new release... 🤞 Any updat on that matter @SuGlider?

SuGlider, 2025-01-07:
> Facing the same issue here. Seems like 3.1.0 fixes some other issue I'm facing but introducing this mutex stuff...
> 
> Hoping for a quick new release... 🤞 Any updat on that matter @SuGlider?

We will release a new version, Arduino Core 3.1.1, up to the of this week with a fix to this issue.

desiFish, 2025-01-08:
> Another Example. While running this program, the same error on XIAO ESP32C6. Program:
>
> ```
> #include <NTPClient.h>
> // change next line to use with another board/shield
> //#include <ESP8266WiFi.h> [...]

I can confirm that this code now works normally in ESP32 C6, I will confirm other codes as well soon.

REMcD, 2025-01-09:
Uploading to thingspeak again - phew! Thanks to all involved

desiFish, 2025-01-10:
Rest of the programs are working fine. Thanks for fixing this.
Thank You,
Aniket Patra

SuGlider, 2025-01-13:
IDF Team has fixed the issue with OpenThread DNS Hook.
Therefore we shall enable OpenThread DNS64 again for the next releases (3.1.2, 3.2.0 and so on).

I'll test the new IDF library for Arduino before we release next versions.
In case the issue here presented doesn't happen again, we will move forward with the reactivation of OpenThread DNS64.

desiFish, 2025-01-14:
> > IDF Team has fixed the issue with OpenThread DNS Hook. Therefore we shall enable OpenThread DNS64 again for the next releases (3.1.2, 3.2.0 and so on).
> > I'll test the new IDF library for Arduino before we release next versions. In case the issue here presented doesn't happen again, we will move forward with the reactivation of OpenThread DNS64.
> 
> FYI: A decision was made that ESP32 Arduino won't enable Open Thread DNS64 Client feature due to issues with Zigbee and WiFi coexistant applications. It seems that enabling DNS64 Client in the `sdkconfig` leads to linking twice some functions that prevent building Zigbee + WiFi Arduino sketches.
> 
> Therefore, no changes will be done, based on the present scenario. No DNS issues shall happen as well.

I am not using Thread for any project right now, but what will I do to use it in the future? Just curious. Because while purchasing these microcontrollers they list these in features.