espressif/arduino-esp32 #10756: IRAM error on compiling

PDE-Paul, 2024-12-19:
### Board

ESP32 Custom Board

### Device Description

Final Product (Custom Board) in market for 6 years

### Hardware Configuration

Final Product in market for 6 years

### Version

other

### IDE Name

2.3.4

### Operating System

Windows 11

### Flash frequency

40

### PSRAM enabled

no

### Upload speed

115200

### Description

Upgrading from 3.0.6 to 3.0.7 or 3.10 I now get an IRAM error

section `.iram0.text' will not fit in region `iram0_0_seg'

If I roll back to 3.0.6 this error disappears.

What has changed in the compiler?
Do I need to add something to my code?

I need to add some features and wish to use the latest version compiler, for now I will stay on 3.0.6

### Sketch

```cpp
Unable to post code as it is for a product that is being sold into the market. Compiled sketch is 1.8Mb using 3.0.6
```

### Debug Message

```plain
IRAM0 segment data does not fit.
region `iram0_0_seg' overflowed by 1040 bytes
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-12-19:
so both 3.0.7 and 3.1.0 cause this error? There are insignificant changes between 3.0.6 and 3.0.7. You can go through the code of your project and remove `IRAM_ATTR` from functions/methods to minimize the IRAM usage from your code. It is no longer required in Arduino anyway.

PDE-Paul, 2024-12-19:
Hi, Thank you for the quick response. Ok found the only reference to IRAM_ATTR are all my interrupts, should these now be ARDUINO_ISR_ATTR?

PDE-Paul, 2024-12-19:
ok, confirmed compile successful, thank you, you fixed my code.

closed 2024-12-19 by me-no-dev