espressif/arduino-esp32 #10758: Missing TCPIP Lock in AsyncUDP multicast

wladwnt, 2024-12-19:
### Board

ESP32dev

### Device Description

Custom ESP32 with ESP32-WROOM-32E-N4

### Hardware Configuration

Not relevant

### Version

v3.1.0

### IDE Name

VSCode

### Operating System

Windows 11

### Flash frequency

40 MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

Call AsyncUDP.listenMulticast(addr,port) lead to reboot with message *Required to lock TCPIP core functionality!*
AsyncUDP.listenMulticast(addr,port) fails apparently because there is no TCPIP lock implemented for call igmp_joingroup()

No issue with v3.1.0-rc1 (before TCPIP lock introduced)

### Sketch

```cpp
#include "WiFi.h"
#include "AsyncUDP.h"

const char *ssid = "*******";
const char *password = "*******";

AsyncUDP udp;

AsyncUDP udp_multicast;
static const IPAddress SSDP_MULTICAST_ADDR(239, 255, 255, 250);

void setup() {
   Serial.begin(115200);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  while (WiFi.waitForConnectResult() != WL_CONNECTED) {
    Serial.println("WiFi Failed");
    delay(1000);
  }
  Serial.println("WiFi connected");
  udp_multicast.listenMulticast(SSDP_MULTICAST_ADDR, 1900);
  udp_multicast.onPacket([&](AsyncUDPPacket packet) {
      Serial.println("Got packet!");
  });
}

void loop() {
  delay(1000);
}
```

### Debug Message

```plain
Backtrace: 0x400832e5:0x3ffb2020 0x4008c171:0x3ffb2040 0x4009234e:0x3ffb2060 0x400ec747:0x3ffb2190 0x400d2ff1:0x3ffb21b0 0x400d31d1:0x3ffb21e0 0x400d293d:0x3ffb2220 0x400d60bb:0x3ffb2270 0x4008cb76:0x3ffb2290
  #0  0x400832e5 in panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c:463
  #1  0x4008c171 in esp_system_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/port/esp_system_chip.c:92
  #2  0x4009234e in __assert_func at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/assert.c:80
  #3  0x400ec747 in igmp_joingroup at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/ipv4/igmp.c:460 (discriminator 1)
  #4  0x400d2ff1 in AsyncUDP::listenMulticast(ip_addr const*, unsigned short, unsigned char, tcpip_adapter_if_t) at C:/Users/xxx/.platformio/packages/framework-arduinoespressif32/libraries/AsyncUDP/src/AsyncUDP.cpp:642
      (inlined by) AsyncUDP::listenMulticast(ip_addr const*, unsigned short, unsigned char, tcpip_adapter_if_t) at C:/Users/xxx/.platformio/packages/framework-arduinoespressif32/libraries/AsyncUDP/src/AsyncUDP.cpp:681
  #5  0x400d31d1 in AsyncUDP::listenMulticast(IPAddress, unsigned short, unsigned char, tcpip_adapter_if_t) at C:/Users/xxx/.platformio/packages/framework-arduinoespressif32/libraries/AsyncUDP/src/AsyncUDP.cpp:764
  #6  0x400d293d in setup() at src/test_udp.cpp:21 (discriminator 2)
  #7  0x400d60bb in loopTask(void*) at C:/Users/xxx/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:59   
  #8  0x4008cb76 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Tony763, 2024-12-20:
Hello, we observe a same error after update to `v3.1.0`. Our code works fine with previous versions.

me-no-dev, 2024-12-20:
will be fixed. i guess we missed to add a lock somewhere

maddisondesigns, 2024-12-28:
Normally I'd raise a my own issue, but this error looks similar to one that I'm also getting. I'm using [ESPAsyncWebServer](https://github.com/mathieucarbou/ESPAsyncWebServer) and my ESP32 device crashes and reboots as soon as it calls `server.begin()`. The error I'm getting is similar to the one mentioned above... 

```
assert failed: tcp_alloc /IDF/components/lwip/lwip/src/core/tcp.c:1851 (Required to lock TCPIP core functionality!)
```

The exact same code worked fine using `v3.0.6` and `v3.0.7` but crashed as soon as I updated and recompiled using `v3.1.0`.

Happy to open a separate issue if you don't think it's related.

closed 2025-01-07 by me-no-dev