espressif/arduino-esp32 #10760: Compilation error when using PlatformIO and Ethernet.h

MicSG-dev, 2024-12-19:
### Board

ESP Dev Module

### Device Description

Module on breadboard.

### Hardware Configuration

Software problem. 
There is no hardware as the problem is in the software.

### Version

v3.1.0

### IDE Name

VSCode PlatformIO

### Operating System

Windows 11 Pro 23H2 - Compilation 22631.4037

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

When using PlatformIO with the following platformio.ini file:
```
[env:esp32dev]
platform = espressif32@ 6.9.0
board = esp32dev
framework = arduino
lib_deps =
    arduino-libraries/Ethernet@^2.0.2
```
And running the [WebServer](https://github.com/arduino-libraries/Ethernet/blob/master/examples/WebServer/WebServer.ino) example, several compilation errors occur.

### Sketch

```cpp
/*
 Web Server

 A simple web server that shows the value of the analog input pins.
 using an Arduino WIZnet Ethernet shield.

 Circuit:
 * Ethernet shield attached to pins 10, 11, 12, 13
 * Analog inputs attached to pins A0 through A5 (optional)

 created 18 Dec 2009
 by David A. Mellis
 modified 9 Apr 2012
 by Tom Igoe
 modified 02 Sept 2015
 by Arturo Guadalupi
 
[... 149 lines cut ...]
      |                                                      ^
.pio/libdeps/esp32dev/Ethernet/src/EthernetServer.cpp:72:40: error: invalid cast to abstract class type 'EthernetClient'
   72 |         return EthernetClient(sockindex);
      |                                        ^
.pio/libdeps/esp32dev/Ethernet/src/EthernetServer.cpp: At global scope:
.pio/libdeps/esp32dev/Ethernet/src/EthernetServer.cpp:75:16: error: invalid abstract return type for member function 'EthernetClient EthernetServer::accept()'
   75 | EthernetClient EthernetServer::accept()
      |                ^~~~~~~~~~~~~~
.pio/libdeps/esp32dev/Ethernet/src/EthernetServer.cpp: In member function 'EthernetClient EthernetServer::accept()':
.pio/libdeps/esp32dev/Ethernet/src/EthernetServer.cpp:82:54: error: invalid cast to abstract class type 'EthernetClient'
   82 |         if (!chip) return EthernetClient(MAX_SOCK_NUM);
      |                                                      ^
.pio/libdeps/esp32dev/Ethernet/src/EthernetServer.cpp:104:40: error: invalid cast to abstract class type 'EthernetClient'
  104 |         return EthernetClient(sockindex);
      |                                        ^
.pio/libdeps/esp32dev/Ethernet/src/EthernetServer.cpp: In member function 'virtual size_t EthernetServer::write(const uint8_t*, size_t)':
.pio/libdeps/esp32dev/Ethernet/src/EthernetServer.cpp:170:18: error: cannot allocate an object of abstract type 'EthernetClient'
  170 |         available();
      |         ~~~~~~~~~^~
*** [.pio\build\esp32dev\libe72\Ethernet\EthernetServer.cpp.o] Error 1
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2024-12-20:
You are using a library designed for Arduino.  Use the [native esp32 library](https://github.com/espressif/arduino-esp32/blob/master/libraries/Ethernet/examples/ETH_LAN8720/ETH_LAN8720.ino)

JAndrassy, 2024-12-20:
@lbernstone this is the TinyGSM library's HttpClient example. what is your advice for that?
```
..\HttpClient.ino:134:15: error: cannot declare variable 'client' to be of abstract type 'TinyGsmSim800::GsmClientSim800'
  134 | TinyGsmClient client(modem);
      |               ^~~~~~
In file included from C:\lab\Sloeber\arduinoPlugin\libraries\TinyGSM\0.12.0\src/TinyGsmClient.h:13,
                 from ..\sloeber.ino.cpp:26:
C:\lab\Sloeber\arduinoPlugin\libraries\TinyGSM\0.12.0\src/TinyGsmClientSIM800.h:86:9: note:   because the following virtual functions are pure within 'TinyGsmSim800::GsmClientSim800':
   86 |   class GsmClientSim800 : public GsmClient {
      |         ^~~~~~~~~~~~~~~
In file included from C:\lab\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\3.1.0\cores\esp32/Arduino.h:197,
                 from ..\sloeber.ino.cpp:7:
C:\lab\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\3.1.0\cores\esp32/Client.h:29:15: note:     'virtual int Client::connect(IPAddress, uint16_t, int32_t)'
   29 |   virtual int connect(IPAddress ip, uint16_t port, int32_t timeout) = 0;
      |               ^~~~~~~
C:\lab\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\3.1.0\cores\esp32/Client.h:31:15: note:     'virtual int Client::connect(const char*, uint16_t, int32_t)'
   31 |   virtual int connect(const char *host, uint16_t port, int32_t timeout) = 0;
      |               ^~~~~~~
..\HttpClient.ino:135:15: error: cannot declare variable 'http' to be of abstract type 'HttpClient'
  135 | HttpClient    http(client, server, port);
      |               ^~~~
In file included from C:\Users\andrassy\Documents\Arduino\libraries\ArduinoHttpClient\src/ArduinoHttpClient.h:8,
                 from ..\sloeber.ino.cpp:27:
C:\Users\andrassy\Documents\Arduino\libraries\ArduinoHttpClient\src/HttpClient.h:41:7: note:   because the following virtual functions are pure within 'HttpClient':
   41 | class HttpClient : public Client
      |       ^~~~~~~~~~
C:\lab\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\3.1.0\cores\esp32/Client.h:29:15: note:     'virtual int Client::connect(IPAddress, uint16_t, int32_t)'
   29 |   virtual int connect(IPAddress ip, uint16_t port, int32_t timeout) = 0;
      |               ^~~~~~~
C:\lab\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\3.1.0\cores\esp32/Client.h:31:15: note:     'virtual int Client::connect(const char*, uint16_t, int32_t)'
   31 |   virtual int connect(const char *host, uint16_t port, int32_t timeout) = 0;
      |               ^~~~~~~
```

Jason2866, 2024-12-20:
@JAndrassy The TinyGSM library needs to be adopted. Native SIM PPP support is integrated in espressif32 Arduino core.

me-no-dev, 2024-12-20:
libs need to be updated for 3.1

me-no-dev, 2024-12-20:
but as @Jason2866 said. we now have native ppp support

JAndrassy, 2024-12-20:
so now the esp32 Arduino team defines the Arduino API and all libraries should reflect the changes?

me-no-dev, 2024-12-20:
we define additions as required. New gcc versions that come with IDF drive some of those changes. We also provide a way to detect the version and extend the classes only when necessary. We have extended many Arduino classes so far. Libs do not need to update, but they will be incompatible with newer versions of our core. Using old version is always an option

Jason2866, 2024-12-20:
Not enhancing the API would mean not fully supporting the new available features from new MCUs (like the C6 and P4). Not doing this would mean, people using Arduino couldn't use this new great features of this chips. Not the way development works. So yes espressif is doing enhancements when needed. Arduino enhancement is not a one way train.

MicSG-dev, 2024-12-20:
> You are using a library designed for Arduino/AVR. Use the [native esp32 library](https://github.com/espressif/arduino-esp32/blob/master/libraries/Ethernet/examples/ETH_LAN8720/ETH_LAN8720.ino)

This statement that the arduino-libraries/Ethernet library is not compatible with ESP32 is very WRONG.

Below is the proof that the arduino-libraries/Ethernet library IS compatible with ESP32:

1. On the PlatformIO Registration page it clearly says that the arduino-libraries/Ethernet library is compatible with ANY PLATFORM.
[image]
Source: https://registry.platformio.org/libraries/arduino-libraries/Ethernet/compatibility

2. When searching for ESP32 in the library repository, there are several results, clearly indicating that the use of the library with ESP32 is COMPATIBLE. 
[image]
Source: https://github.com/search?q=repo%3Aarduino-libraries%2FEthernet%20esp32&type=code

3. In the examples of the arduino-libraries/Ethernet library, there is a clear indication of use of ESP32 (including platforms other than AVR), clearly indicating that the use of the library with ESP32 is COMPATIBLE.
[image]
Source: https://github.com/arduino-libraries/Ethernet/blob/b16c38a5bd0e25143da8d0e9cdeba5b90db62d0a/examples/WebServer/WebServer.ino#L42

4. In the properties file of the arduino-libraries/Ethernet library (library.properties) it clearly says that it is COMPATIBLE with any architecture. 
[image]
Source: https://github.com/arduino-libraries/Ethernet/blob/b16c38a5bd0e25143da8d0e9cdeba5b90db62d0a/library.properties#L9

5. I successfully compiled and uploaded the [WebServer](https://github.com/arduino-libraries/Ethernet/blob/master/examples/WebServer/WebServer.ino) example of the arduino-libraries/Ethernet library to an ESP32!! (using an Arduino IDE).
[image]

With all this irrefutable evidence, I hope you (@lbernstone and @Jason2866 [[reply](https://github.com/platformio/platform-espressif32/issues/1520#issuecomment-2556860733)]) stop claiming that the arduino-libraries/Ethernet library is not compatible with the ESP32, when in fact it IS COMPATIBLE with the ESP32.

And about the ETH.h library, I already used this library a few days ago with ESP32 and W5500 and there were always random disconnections (triggering the `ARDUINO_EVENT_ETH_DISCONNECTED` event). With the trusted Ethernet.h library there are no such disconnections. That's why I would prefer to use Ethernet.h. but still, thanks for the suggestion.

MicSG-dev, 2024-12-20:
I don't know what happened with the [Client.h](https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/Client.h) file (from the esp32 core), but according to @per1234 (https://github.com/arduino-libraries/Ethernet/issues/277#issuecomment-2555774494) there seems to have been some setback.

MicSG-dev, 2024-12-20:
On my local machine I made the changes as per the pull request #3191 and it temporarily resolved the [issue](https://github.com/espressif/arduino-esp32/issues/10760#issue-2751313443) that I was facing.

Changes made: https://github.com/arduino-libraries/Ethernet/issues/277#issuecomment-2555815432

JAndrassy, 2024-12-20:
what is the benefit for esp32 core adding virtual functions to Arduino's Client class?

Jason2866, 2024-12-20:
@MicSG-dev Platformio docu is correct, since there last supported espressif32 Arduino core is v2.0.17.
With this version the library is compatible. This was mentioned from me-no-dev -> You can use older version.

Jason2866, 2024-12-20:
@MicSG-dev 
The setup for Platformio you show is not using Arduino Core 3.1.0 at all. 
```
[env:esp32dev]
platform = espressif32@ 6.9.0
board = esp32dev
framework = arduino
lib_deps =
    arduino-libraries/Ethernet@^2.0.2
```
It uses Arduino Core 2.0.17 the last supported version from Platformio. Don't know the reason why it does not work with this old core, it just shows the library never worked well with Arduino espressif32

MicSG-dev, 2024-12-20:
> The setup for Platformio you show is not using Arduino Core 3.1.0 at all.
It uses Arduino Core 2.0.17 [...]

Oh yes, thanks for the information. I just saw at https://github.com/platformio/platform-espressif32/releases/tag/v6.9.0 that `espressif32@6.9.0` uses Arduino Core version 2.0.17.

---

> [...], it just shows the library never worked well with Arduino espressif32

In my opinion this is not the fault of the Ethernet.h library. Where were the changes that broke Ethernet.h? I think it was in [Client.h](https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/Client.h). This problem is caused by the ESP32 platform deviating from the standardized Arduino Client class API.

Jason2866, 2024-12-20:
@MicSG-dev Finally we agree the library is not compatible as we told. Since core 2.0.17 is outdated and out of support there will be no fix. For Arduino core 3.1.0 the currently only solution is to use the provided included as initially suggested.

MicSG-dev, 2024-12-20:
> the currently only solution is to use the provided included as initially suggested.
> Just tested with my W5500 and it does work perfectly fine.

I understand that ETH.h is working. But I have already said that I will not use it for the reasons mentioned above. I prefer to use Ethernet.h. Thanks again for the suggestion.

For now, to use Ethernet.h, I will continue to modify the package locally in the [Client.h](https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/Client.h) file (removing [line 29](https://github.com/espressif/arduino-esp32/blob/9eb7dc6f7bddf98600f6e8de3ccc19cc8ba5281b/cores/esp32/Client.h#L29) and [line 31](https://github.com/espressif/arduino-esp32/blob/9eb7dc6f7bddf98600f6e8de3ccc19cc8ba5281b/cores/esp32/Client.h#L31) ),

I will continue to wait for the problem to be fixed in version 3.1.0 (or another future version) of Arduino Core ESP32.

MicSG-dev, 2024-12-20:
> Don't know the reason why it does not work with this old core

I know the reason why it doesn't work in version 2.0.17. Read below:

As I wrote above in https://github.com/espressif/arduino-esp32/issues/10760#issuecomment-2557428409, the explanation is in the link https://github.com/arduino-libraries/Ethernet/issues/277#issuecomment-2555774494. See:

> It is caused by the ESP32 platform deviating from the standardized Arduino Client class API:
https://github.com/espressif/arduino-esp32/issues/2755
They brought the platform back into compliance some time ago: https://github.com/espressif/arduino-esp32/pull/3191
But then there was a recent regression: https://github.com/espressif/arduino-esp32/pull/10202

MicSG-dev, 2024-12-20:
> For Arduino core 3.1.0 the currently only solution is to use the provided included as initially suggested.

I also see as a solution to simply apply the fix as per pull request #3191 .

me-no-dev, 2024-12-21:
@JAndrassy try removing them and you will find out. I gave you a hint already (gcc)

JAndrassy, 2024-12-21:
> @JAndrassy try removing them and you will find out. I gave you a hint already (gcc)
I don't get an error (with the BasicHttpClient.ino example). 
The only implementation of Client bundled with the esp32 core is NetworkClient

me-no-dev, 2024-12-23:
@JAndrassy https://github.com/espressif/arduino-esp32/actions/runs/12465525012/job/34791469849?pr=10773#step:6:111

rftafas, 2025-01-14:
Core 2.0.17 is out of support.

User seems to have made a PR, hopefully it solves their issue. Nothing more we can do here. Closing.

closed 2025-01-14 by rftafas