espressif/arduino-esp32 #10761: error: 'esp_sleep_enable_ext1_wakeup_io' was not declared in this scope

Levanterman, 2024-12-19:
### Board

TTGO LoRa32-OLED

### Device Description

TTGO LoRa32-OLED , 10K pull up connected to GPIO 13 and 25, works fine in Arduino IDE but not Platform IO .

### Hardware Configuration

#define WAKEUP_GPIO_13 GPIO_NUM_13 
#define WAKEUP_GPIO_25 GPIO_NUM_25    

### Version

latest master (checkout manually)

### IDE Name

VSCode  Version: 1.95.3 (Universal) Commit: f1a4fb101478ce6ec82fe9627c43efbf9e98c813 Date: 2024-11-13T14:50:04.152Z (1 mo ago) Electron: 32.2.1 ElectronBuildId: 10427718 Chromium: 128.0.6613.186 Node.js: 20.18.0 V8: 12.8.374.38-electron.0 OS: Darwin arm64 24.1.0

### Operating System

Mac 

### Flash frequency

80

### PSRAM enabled

yes

### Upload speed

115200

### Description

I have the same problem as [https://github.com/espressif/arduino-esp32/pull/9904](url) and [https://github.com/espressif/arduino-esp32/issues/9809](url)

I have this [https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/DeepSleep/ExternalWakeUp/ExternalWakeUp.ino](url) this example working in the Arduino IDE with TTGO LoRa32-OLED, waking up on either Pin 

#define WAKEUP_GPIO_13 GPIO_NUM_13 
#define WAKEUP_GPIO_25 GPIO_NUM_25    

I tried to convert the project to PlatformIO but I get the error 

"src/main.cpp:196:5: error: 'esp_sleep_enable_ext1_wakeup_io' was not declared in this scope"

[env:ttgo-lora32-v2]
platform = espressif32
board = ttgo-lora32-v2
framework = arduino
lib_deps = sandeepmistry/LoRa@^0.8.0
platform_packages = platformio/framework-arduinoespressif32

### Sketch

```cpp
I have this [https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/DeepSleep/ExternalWakeUp/ExternalWakeUp.ino](url) this example working in the Arduino IDE with TTGO LoRa32-OLED, waking up on either Pin
```

### Debug Message

```plain
src/main.cpp: In function 'void setup()':
src/main.cpp:196:5: error: 'esp_sleep_enable_ext1_wakeup_io' was not declared in this scope
     esp_sleep_enable_ext1_wakeup_io(bitmask, ESP_EXT1_WAKEUP_ANY_HIGH);
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/main.cpp:196:5: note: suggested alternative: 'esp_sleep_enable_ext1_wakeup'
     esp_sleep_enable_ext1_wakeup_io(bitmask, ESP_EXT1_WAKEUP_ANY_HIGH);
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2024-12-19:
platformio is using version 2.0.17 (framework-arduinoespressif32 @ 3.20017.241212 <- the second set of numbers in there)
You can use the [community version](https://github.com/espressif/arduino-esp32/issues/10721#issuecomment-2539567775) to run arduino-esp32 v3.X

PetrVobor, 2024-12-20:
I have the same issue.

Levanterman, 2024-12-20:
> platformio is using version 2.0.17 (framework-arduinoespressif32 @ 3.20017.241212 <- the second set of numbers in there) You can use the [community version](https://github.com/espressif/arduino-esp32/issues/10721#issuecomment-2539567775) to run arduino-esp32 v3.X

AH thank you , I will try this later and see if that fixes it.

Jason2866, 2024-12-20:
With pioarduino (Platformio community fork) Arduino espressif32 core 3.1.0 can be used with
```
platform = https://github.com/pioarduino/platform-espressif32/releases/download/53.03.10/platform-espressif32.zip
```

Levanterman, 2024-12-21:
@lbernstone lbernstone / @Jason2866 Thank you both, this worked ! Very happy, thank you.

closed 2024-12-21 by Levanterman