espressif/arduino-esp32 #10762: Bound devices not showing up when using printbounddevices()

HexiDev, 2024-12-20:
### Board

ESP32 C6

### Device Description

esp32 c6 DevkitM 

### Hardware Configuration

Just a dev board with nothing attached.

### Version

latest master (checkout manually)

### IDE Name

VScode, platformio

### Operating System

Linux Ubuntu server

### Flash frequency

default

### PSRAM enabled

yes

### Upload speed

115200

### Description

Basically whenever I use ZHA to bind a switch and a light to each other it never shows it when I print the bound devices. But it does work. As when I click the button the light switch toggles.

Now I wanted to connect my esp32 c6 to a moes bht-006 thermostat so I can use that for setting the temp and reading the temp. However I want to implement opentherm for my heating setup. 

So for a proof of concept I tried to use the thermostat example scripts, with changing the coordinator to a end device. Only problem is its hard to debug if I am not sure if it has bound or not.

### Sketch

```cpp
Default example scripts, except changing the thermostat to a end device and binding them through ZHA toolkit
```

### Debug Message

```plain
[     0][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Dein[    57][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x42006408
[    68][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x420063cc
[    80][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x42006390
[    91][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type USB_DM (38) successfully set to 0x42002e54
[   102][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type USB_DP (39) successfully set to 0x42002e54
[   114][V][ZigbeeEP.cpp:20] ZigbeeEP(): Endpoint: 6
[   123][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 17 successfully set to type UART_RX (2) with bus 0x4080f2c8
[   134][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 16 successfully set to type UART_TX (3) with bus 0x4080f2c8
[   145][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 12 successfully set to type USB_DM (38) with bus 0x408118d4
[   156][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 13 successfully set to type USB_DP (39) with bus 0x408118d4
=========== Before Setup Start ===========
Chip Info:
------------------------------------------
  Model             : ESP32-C6
  Package           : 0
  Revision          : 0.00
  Cores             : 1
  CPU Frequency     : 160 MHz
  XTAL Frequency    : 40 MHz
  Features Bitfield : 0x00000052
  Embedded Flash    : No
  Embedded PSRAM    : No
  2.4GHz WiFi       : Yes
  Classic BT        : No
  BT Low Energy     : Yes
  IEEE 802.15.4     : Yes
------------------------------------------
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   442092 B ( 431.7 KB)
  Free Bytes        :   401396 B ( 392.0 KB)
  Allocated Bytes   :    33504 B (  32.7 KB)
  Minimum Free Bytes:   401340 B ( 391.9 KB)
  Largest Free Block:   368628 B ( 360.0 KB)
------------------------------------------
Flash Info:
------------------------------------------
  Chip Size         :  4194304 B (4 MB)
  Block Size        :    65536 B (  64.0 KB)
  Sector Size       :     4096 B (   4.0 KB)
  Page Size         :      256 B (   0.2 KB)
  Bus Speed         : 40 MHz
  Bus Mode          : QIO
------------------------------------------
Partitions Info:
------------------------------------------
                nvs : addr: 0x00009000, size:    20.0 KB, type: DATA, subtype: NVS
            otadata : addr: 0x0000E000, size:     8.0 KB, type: DATA, subtype: OTA
               app0 : addr: 0x00010000, size:  1280.0 KB, type:  APP, subtype: OTA_0
               app1 : addr: 0x00150000, size:  1280.0 KB, type:  APP, subtype: OTA_1
             spiffs : addr: 0x00290000, size:  1388.0 KB, type: DATA, subtype: SPIFFS
         zb_storage : addr: 0x003EB000, size:    16.0 KB, type: DATA, subtype: FAT
             zb_fct : addr: 0x003EF000, size:     4.0 KB, type: DATA, subtype: FAT
           coredump : addr: 0x003F0000, size:    64.0 KB, type: DATA, subtype: COREDUMP
------------------------------------------
Software Info:
------------------------------------------
  Compile Date/Time : Dec 19 2024 14:20:28
  ESP-IDF Version   : v5.3.2-174-g083aad99cf-dirty
  Arduino Version   : 3.1.0
------------------------------------------
Board Info:
------------------------------------------
  Arduino Board     : Espressif ESP32-C6-DevKitM-1
  Arduino Variant   : esp32c6
  Core Debug Level  : 5
  Arduino Runs Core : 0
  Arduino Events on : 0
  Arduino USB Mode  : 1
  CDC On Boot       : 1
============ Before Setup End ============
[   280][I][esp32-hal-periman.c:141] perimanSetPinBus(): Pin 12 already has type USB_DM (38) with bus 0x408118d4
[   281][I][esp32-hal-periman.c:141] perimanSetPinBus(): Pin 13 already has type USB_DP (39) with bus 0x408118d4
[   281][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type GPIO (1) successfully set to 0x42003caa
[   282][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 8 successfully set to type GPIO (1) with bus 0x9
[   282][V][esp32-hal-rmt.c:454] rmtInit(): GPIO 8 - TX MODE - MemSize[48] - Freq=10000000Hz
[   283][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type RMT_TX (10) successfully set to 0x420052be
[   284][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type RMT_RX (11) successfully set to 0x420052be
[   284][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 8 successfully set to type INIT (0) with bus 0x0
[   285][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 8 successfully set to type RMT_TX (10) with bus 0x4081b5cc
[   286][V][esp32-hal-rmt.c:305] _rmtWrite(): GPIO: 8 - Request: 24 RMT Symbols - Blocking - Timeout: -1
[   287][V][esp32-hal-rmt.c:306] _rmtWrite(): GPIO: 8 - Currently in Loop Mode: [NO] | Asked to Loop: NO, LoopCancel: NO
[   287][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type GPIO (1) successfully set to 0x42003caa
[   288][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 9 successfully set to type GPIO (1) with bus 0xa
[   289][D][ZigbeeCore.cpp:84] addEndpoint(): Endpoint: 6, Device ID: 0x0301
[   289][D][main.cpp:115] setup(): Calling Zigbee.begin()
[   290][D][ZigbeeCore.cpp:123] zigbeeInit(): Initialize Zigbee stack
[   325][D][ZigbeeCore.cpp:130] zigbeeInit(): Register all Zigbee EPs in list
[   326][I][ZigbeeCore.cpp:138] zigbeeInit(): List of registered Zigbee EPs:
[   327][I][ZigbeeCore.cpp:140] zigbeeInit(): Device type: Thermostat Device, Endpoint: 6, Device ID: 0x0301
[   340][V][ZigbeeCore.cpp:339] esp_zb_app_signal_handler(): ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL
[   340][I][ZigbeeCore.cpp:219] esp_zb_app_signal_handler(): Zigbee stack initialized
[  2696][I][ZigbeeCore.cpp:225] esp_zb_app_signal_handler(): Device started up in non factory-reset mode
[  2697][I][ZigbeeCore.cpp:238] esp_zb_app_signal_handler(): Device rebooted
[  2697][D][ZigbeeCore.cpp:466] searchBindings(): Requesting binding table for address 0x8ce5
[  2699][D][ZigbeeCore.cpp:406] bindingTableCb(): Binding table callback for address 0x8ce5 with status 0
[  2699][D][ZigbeeCore.cpp:409] bindingTableCb(): Binding table info: total 1, index 0, count 1
[  2700][D][ZigbeeCore.cpp:419] bindingTableCb(): Binding table record: src_endp 6, dst_endp 1, cluster_id 0x0201, dst_addr_mode 3
[  2701][D][ZigbeeCore.cpp:436] bindingTableCb(): Device bound to EP 6 -> device endpoint: 1, short addr: 0x0000, ieee addr: BC:33:AC:FF:FE:93:E9:44
[  2701][D][ZigbeeCore.cpp:457] bindingTableCb(): Filling bounded devices finished
Waiting for Sensor to bound to the thermostat
=========== After Setup Start ============
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   442092 B ( 431.7 KB)
  Free Bytes        :   374220 B ( 365.4 KB)
  Allocated Bytes   :    59576 B (  58.2 KB)
  Minimum Free Bytes:   374216 B ( 365.4 KB)
  Largest Free Block:   344052 B ( 336.0 KB)
------------------------------------------
GPIO Info:
------------------------------------------
  GPIO : BUS_TYPE[bus/unit][chan]
  --------------------------------------  
     8 : RMT_TX
     9 : GPIO
    12 : USB_DM
    13 : USB_DP
    16 : UART_TX[0]
    17 : UART_RX[0]
============ After Setup End =============
[  6941][I][ZigbeeEP.cpp:190] printBoundDevices(): Bound devices:
[  6942][I][ZigbeeEP.cpp:193] printBoundDevices(): Device on endpoint 1, short address: 0x0, ieee address: bc:33:ac:ff:fe:93:e9:44
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

P-R-O-C-H-Y, 2025-02-25:
Hi @HexiDev, can you please retest this on latest version to check if this is still valid? Thanks

P-R-O-C-H-Y, 2025-04-02:
Closing as outdated and no info was provided. I did test on latest versions and it works fine.

closed 2025-04-02 by P-R-O-C-H-Y