espressif/arduino-esp32 #10763: Register / Bits not supported for ESP32C3

jh-elec, 2024-12-20:
### Board

ESP32C3 Super Mini

### Device Description

nope

### Hardware Configuration

nope

### Version

latest master (checkout manually)

### IDE Name

VSD

### Operating System

WIN11

### Flash frequency

80

### PSRAM enabled

yes

### Upload speed

115200

### Description

https://github.com/espressif/arduino-esp32/blob/9eb7dc6f7bddf98600f6e8de3ccc19cc8ba5281b/cores/esp32/esp32-hal-spi.c#L677

`
#if defined(CONFIG_IDF_TARGET_ESP32S3) || defined(CONFIG_IDF_TARGET_ESP32C2) || defined(CONFIG_IDF_TARGET_ESP32C3)
  spi->dev->dma_conf.tx_seg_trans_clr_en = 1;
  spi->dev->dma_conf.rx_seg_trans_clr_en = 1;
  spi->dev->dma_conf.dma_seg_trans_en = 0;
`
this bits not supported for a C3

### Sketch

```cpp
nope
```

### Debug Message

```plain
nope
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-12-21:
did you get an error?

jh-elec, 2024-12-21:
No error.
But i want that the Library is Perfect 😙 greets from germany

me-no-dev, 2024-12-21:
why do you think that it's not supported then?

jh-elec, 2024-12-21:
I dont have see this bits in c3 datasheet?! or i am blind?

SuGlider, 2024-12-21:
> I dont have see this bits in c3 datasheet?! or i am blind?

https://www.espressif.com/sites/default/files/documentation/[esp32-c3_technical_reference_manual_en.pdf](https://www.espressif.com/sites/default/files/documentation/esp32-c3_technical_reference_manual_en.pdf)

Page 636.

[image]

SuGlider, 2024-12-21:
@jh-elec - You wiil find the SPI Register Bits related to it in https://github.com/espressif/esp-idf/blob/release/v5.3/components/soc/esp32c3/include/soc/spi_struct.h#L170-L185

``` cpp
    union {
        struct {
            uint32_t reserved0                     :    18;  /*reserved*/
            uint32_t dma_seg_trans_en              :    1;  /*Enable dma segment transfer in spi dma half slave mode. 1: enable. 0: disable.*/
            uint32_t rx_seg_trans_clr_en           :    1;  /*1: spi_dma_infifo_full_vld is cleared by spi slave cmd 5. 0: spi_dma_infifo_full_vld is cleared by spi_trans_done.*/
            uint32_t tx_seg_trans_clr_en           :    1;  /*1: spi_dma_outfifo_empty_vld is cleared by spi slave cmd 6. 0: spi_dma_outfifo_empty_vld is cleared by spi_trans_done.*/
            uint32_t rx_eof_en                     :    1;  /*1: spi_dma_inlink_eof is set when the number of dma pushed data bytes is equal to the value of spi_slv/mst_dma_rd_bytelen[19:0] in spi dma transition.  0: spi_dma_inlink_eof is set by spi_trans_done in non-seg-trans or spi_dma_seg_trans_done in seg-trans.*/
            uint32_t reserved22                    :    5;  /*reserved*/
            uint32_t dma_rx_ena                    :    1;  /*Set this bit to enable SPI DMA controlled receive data mode.*/
            uint32_t dma_tx_ena                    :    1;  /*Set this bit to enable SPI DMA controlled send data mode.*/
            uint32_t rx_afifo_rst                  :    1;  /*Set this bit to reset RX AFIFO, which is used to receive data in SPI master and slave mode transfer.*/
            uint32_t buf_afifo_rst                 :    1;  /*Set this bit to reset BUF TX AFIFO, which is used send data out in SPI slave CPU controlled mode transfer and master mode transfer.*/
            uint32_t dma_afifo_rst                 :    1;  /*Set this bit to reset DMA TX AFIFO, which is used to send data out in SPI slave DMA controlled mode transfer.*/
        };
        uint32_t val;
    } dma_conf;
```

SuGlider, 2024-12-29:
Closing this issue as completed. Feel free to open it again if necesary.

closed 2024-12-29 by SuGlider