espressif/arduino-esp32 #10771: otNcpHdlcInit missing

lsroka76, 2024-12-22:
### Board

ESP32 C6

### Device Description

ESP32 C6 Dev Module

### Hardware Configuration

nope

### Version

v3.1.0

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

80

### PSRAM enabled

yes

### Upload speed

115200

### Description

I'm trying to port Zigbee Gateway example to Arduino but I've found some problems with ot_rcp part.
There was no otAppNcpInit function:

```
// Initialize the OpenThread ncp
    otAppNcpInit(esp_openthread_get_instance());

```

so I tried to port it:

```
void otAppNcpInit(otInstance *aInstance)
{
#if CONFIG_OPENTHREAD_RCP_SPI
    otNcpSpiInit(aInstance);
#else
    otPlatUartEnable();

    otNcpHdlcInit(aInstance, NcpSend);
#endif
}

but next problem arised :

there is declaration of otNcpHdlcInint function in `openthread/ncp.h` but linker is missing body definition.

Any ideas how I can fix it?

Lucas

```

### Sketch

```cpp
as above
```

### Debug Message

```plain
linker is missing otNcpHdlcIinit
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2024-12-27:
@lsroka76 - Arduino OpenThread Library doesn't support NCP natively, beause it works as a Full Thread Device.

You will need to create an Arduino as IDF Component project in order to configure the sdkconfig file.
It is necessary to set OpenThread in RCP (Radio CoProcessor) mode and enable NCP Hook.

This can be done by adding `CONFIG_OPENTHREAD_ENABLE_NCP_VENDOR_HOOK=y` using IDF MenuConfig:

`(Top) → Component config → OpenThread → OpenThread → Thread Core Features → Thread device type` --> (X) Radio Only Device
`(Top) → Component config → OpenThread → OpenThread → Thread Core Features → Thread Radio Co-Processor Feature` --> [*] Enable vendor command for RCP

This will enable `OPENTHREAD_ENABLE_NCP_VENDOR_HOOK=y` and `otAppNcpInit()` will be visible in the application.

SuGlider, 2024-12-27:
Another way to get it, would be by developing an IDF project directly. ESP32 Arduino OpenThread won't help for this task.

closed 2025-01-14 by rftafas