espressif/arduino-esp32 #10772: boards.txt is missing a "Custom" partition scheme entry for the ESP32S3 Dev Module Octal (WROOM2)

zingzog, 2024-12-22:
### Board

ESP32S3 Dev Module Octal (WROOM2), e.g. ESP32-S3-DevKitC-1-N32R8V

### Device Description

ESP32-S3-DevKitC-1-N32R8V

### Hardware Configuration

No

### Version

v3.1.0

### IDE Name

Arduino IDE 2.3.4

### Operating System

Windows 10

### Flash frequency

80MHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

The boards.txt is missing a "Custom" partition scheme entry for the ESP32S3 Dev Module Octal (WROOM2). This makes it impossible to use a Custom partition scheme for this board type.

Code something like the following should be added to the partition section for this board:
esp32s3-octal.menu.PartitionScheme.custom=Custom
esp32s3-octal.menu.PartitionScheme.custom.build.partitions=
esp32s3-octal.menu.PartitionScheme.custom.upload.maximum_size=33554432

### Sketch

```cpp
Sketch independent
```

### Debug Message

```plain
There is no specific error message as such. Any custom partitions.csv files are ignored, and the default partion scheme is used because the custom option cannot be selected.
```

### Other Steps to Reproduce

If the following entry is added to the partition section of boards.txt for the "ESP32S3 Dev Module Octal (WROOM2)" board the issue is solved.
esp32s3-octal.menu.PartitionScheme.custom=Custom
esp32s3-octal.menu.PartitionScheme.custom.build.partitions=
esp32s3-octal.menu.PartitionScheme.custom.upload.maximum_size=33554432

The above fixes the issue, but I would appreciate someone checking the detail and the max_size number.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2024-12-22:
Note that you can add a boards.local.txt to your esp32 directory with these additional lines.

zingzog, 2024-12-22:
Hi lbernstone, thanks for the tip. I made a boards.local.txt file with the three lines in it, put it next to the original boards.txt (without my mod), and it compiles fine.

It's a good temporary fix for me until boards.txt is updated.