espressif/arduino-esp32 #10775: The previous normal camera driver now does not recognize the camera

lztmfx, 2024-12-23:
### Board

ESP32S3

### Device Description

Self designed camera adapter board based on LUATOS ESP32S3-CORE to adjust camera pins to avoid PSRAM conflicts

### Hardware Configuration

nothing

### Version

latest master (checkout manually)

### IDE Name

VSCode based PlatformIO

### Operating System

WIN10

### Flash frequency

240

### PSRAM enabled

yes

### Upload speed

115200

### Description

I have used this code multiple times in the past few months and it has been normal, but now when I use it again, I encounter this error

### Sketch

```cpp
#include <esp_camera.h>
#include <WiFi.h>
#include <esp32-hal-psram.h>
#include "FS.h"
#include "SD_MMC.h"
#include "AsyncUDP.h"

#define USE_UDP

#define CAM_PIN_PWDN -1
#define CAM_PIN_RESET 45
#define CAM_PIN_XCLK 39
#define CAM_PIN_SIOD 21
#define CAM_PIN_SIOC 46
#define CAM_PIN_D7 40
#define CAM_PIN_D6 38
#define CAM_PIN_D5 10
#define CAM_PIN_D4 12
#define CAM_PIN_D3 7
#define CAM_PIN_D2 48
#define CAM_PIN_D1 47
#define CAM_PIN_D0 6
#define CAM_PIN_VSYNC 42
#define CAM_PIN_HREF 41
#define CAM_PIN_PCLK 11

static const char *WIFI_SSID = "xxxx";
static const char *WIFI_PASS = "xxxx";

IPAddress ServerIP(192, 168, 1, 101);
unsigned int ServerPort = 888;

char jpgName[20];
int MAXBUFSIZE = 800;
int imgIndex = 0;
int isCamInitSucess = 1;

static camera_config_t camera_config = {
    .pin_pwdn = CAM_PIN_PWDN,
    .pin_reset = CAM_PIN_RESET,
    .pin_xclk = CAM_PIN_XCLK,
    .pin_sccb_sda = CAM_PIN_SIOD,
    .pin_sccb_scl = CAM_PIN_SIOC,
    .pin_d7 = CAM_PIN_D7,
    .pin_d6 = CAM_PIN_D6,
    .pin_d5 = CAM_PIN_D5,
    .pin_d4 = CAM_PIN_D4,
    .pin_d3 = CAM_PIN_D3,
    .pin_d2 = CAM_PIN_D2,
    .pin_d1 = CAM_PIN_D1,
    .pin_d0 = CAM_PIN_D0,
    .pin_vsync = CAM_PIN_VSYNC,
    .pin_href = CAM_PIN_HREF,
    .pin_pclk = CAM_PIN_PCLK,
    // XCLK 20MHz or 10MHz for OV2640 double FPS (Experimental)
    .xclk_freq_hz = 36000000,
    .ledc_timer = LEDC_TIMER_0,
    .ledc_channel = LEDC_CHANNEL_0,
    .pixel_format = PIXFORMAT_JPEG,
    .frame_size = FRAMESIZE_QVGA, // FRAMESIZE_HD
    .jpeg_quality = 5,
    .fb_count = 2,
    .fb_location = CAMERA_FB_IN_PSRAM,
    .grab_mode = CAMERA_GRAB_LATEST,
};

camera_fb_t *fb = NULL;

#ifdef USE_UDP
AsyncUDP udp;
#else
AsyncClient tcp;
#endif

static esp_err_t init_camera(void)
{
  // initialize the camera
  esp_err_t err = esp_camera_init(&camera_config);
  if (err != ESP_OK)
  {
    Serial.println("Camera Init Failed");
    return err;
  }

  sensor_t *s = esp_camera_sensor_get();
  // initial sensors are flipped vertically and colors are a bit saturated
  if (s->id.PID == OV5640_PID)
  {
    s->set_vflip(s, 1);
    // s->set_hmirror(s, 1);
    s->set_brightness(s, -2);
    s->set_sharpness(s, 2);
    s->set_saturation(s, 2);
    // s->set_contrast(s, 2);
    s->set_awb_gain(s, 1);
  }

  return ESP_OK;
}

void connWifi(void)
{
  int conn_times = 0;
  WiFi.persistent(false);
  WiFi.mode(WIFI_STA);
  WiFi.begin(WIFI_SSID, WIFI_PASS);
  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    conn_times++;
    Serial.print(".");
    if (conn_times > 20)
    {
      Serial.printf("\nWIFI conn fail,will reboot!!!\n");
      esp_restart();
    }
  }
  Serial.print("WIFI IP:");
  Serial.println(WiFi.localIP());
}

void setup()
{
  Serial.begin(115200);
  Serial.println("**********ESPCAM_LCD**********");
  setCpuFrequencyMhz(240);
  psramInit();
  connWifi();

  if (init_camera() != 0)
  {
    isCamInitSucess = 0;
  }
  udp.connect(ServerIP, ServerPort);
}

void loop()
{
  if (isCamInitSucess)
  {
    fb = esp_camera_fb_get();
    uint8_t *temp = fb->buf;
    Serial.println(fb->len);

    int len = fb->len;
    int times = len / MAXBUFSIZE;
    int extra = len % MAXBUFSIZE;
    uint8_t HEAD[4] = {'H', 'E', 'A', 'D'};
    uint8_t END[3] = {'E', 'N', 'D'};

    udp.write(HEAD, sizeof(HEAD));
    for (int i = 0; i < times; i++)
    {
      udp.write(fb->buf, MAXBUFSIZE);
      for (int j = 0; j < MAXBUFSIZE; j++)
      {
        fb->buf++;
      }
    }
    udp.write(fb->buf, extra);
    udp.write(END, sizeof(END));
    fb->buf = temp;
    esp_camera_fb_return(fb);
  }
}
```

### Debug Message

```plain
E (1107) camera: Camera probe failed with error 0x105(ESP_ERR_NOT_FOUND)
```

### Other Steps to Reproduce

I have compiled this program on different computers, and the error phenomenon is the same. I am wondering if the latest SDK update has caused this error?

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-12-23:
if you are using PlatformIO, then we have not touched a thing. If you have an issue, check your hardware

lztmfx, 2024-12-24:
Upon investigation, it was found that the I2C address could not be detected. I will first investigate the hardware issue

closed 2024-12-24 by lztmfx