espressif/arduino-esp32 #10777: Zigbee readManufacturer fail leads to reboot

AndunHH, 2024-12-23:
### Board

seeed xiao esp32-c6

### Device Description

I use the [Seeeduino-XIAO-Expansion-Board/](https://wiki.seeedstudio.com/Seeeduino-XIAO-Expansion-Board/)

### Hardware Configuration

nothing added

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE 2 and pioarduino

### Operating System

Ubuntu 2024.04

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

921600

### Description

When calling `zbSwitch.readManufacturer(device->endpoint, device->short_addr, device->ieee_addr)` within the [zigbee Switch example](https://github.com/espressif/arduino-esp32/tree/master/libraries/Zigbee/examples/Zigbee_On_Off_Switch) the manufacturer is reported.

Unfortunately, the timeout within this function leads to a rebooting device, because it is not handled, when the device is not answering.

I expect, that the function `readManufacturer()` returns a "n/a" or likewise, but is not resetting the controller, when no information could be obtained.

I tried to avoid the problem by checking `zbSwitch.bound()` before calling `readManufacturer()`, but this doesn't solve the problem, because bound() doesn't know, that I just unplugged the light.

### Sketch

```cpp
// Copyright 2024 Espressif Systems (Shanghai) PTE LTD
//
// Licensed under the Apache License, Version 2.0 (the "License");
// you may not use this file except in compliance with the License.
// You may obtain a copy of the License at
//
//     http://www.apache.org/licenses/LICENSE-2.0
//
// Unless required by applicable law or agreed to in writing, software
// distributed under the License is distributed on an "AS IS" BASIS,
// WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
// See the License for the specific language governing permissions and
// limitations under the License.

/**
 * @brief This example demonstrates simple Zigbee light switch.
 *
 * The example demonstrates how to use Zigbee library to control a light bulb.
 * The light bulb is a Zigbee end device, which is controlled by a Zigbee coordinator (Switch).
 * Button switch and Zigbee runs in separate tasks.
 *
[... 185 lines cut ...]
IEEE Address: 84:71:27:FF:FE:DC:28:D0
[   856][I][ZigbeeCore.cpp:326] esp_zb_app_signal_handler(): Network(0x5d24) is open for 180 seconds
[  7647][V][ZigbeeHandlers.cpp:82] zb_cmd_read_attr_resp_handler(): Read attribute response: from address(0x264) src endpoint(1) to dst endpoint(5) cluster(0x0)
[  7648][V][ZigbeeHandlers.cpp:91] zb_cmd_read_attr_resp_handler(): Read attribute response: status(134), cluster(0x0), attribute(0x4), type(0x0), value(0)
[ 10395][E][ZigbeeEP.cpp:148] readManufacturer(): Error while reading manuGuru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.
```

### Other Steps to Reproduce

In the pasted example I changed the xQueueReceive timeout in line 161 to 200 ticks to get to the end of the loop().
The really interesting part, is that I added the for loop in line 193 from the printBoundDevice function.

I'm using the latest master for arduino-esp32: commit 9eb7dc6f7bddf98600f6e8de3ccc19cc8ba5281b

The behavior is independent of a factory-reset or not. 

When the light is not available the readManufacturer runs into the unhandled exception.

I can reproduce this problems when
a) I disconnected the light bulb after successfull connection and toggling.
b) if the esp32c6 restarts and remembers the bound devices, but the actual reconnection to the light needs some time, but I'm already asking for a manufacturer.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

P-R-O-C-H-Y, 2025-01-06:
@AndunHH Thank you for the report, I will work on fixing it.

P-R-O-C-H-Y, 2025-01-06:
@AndunHH Can you test with this changed code in the example? Just replace the part of listing devices:

```
  // Optional: List all bound devices and read manufacturer and model name
  std::list<zb_device_params_t *> boundLights = zbSwitch.getBoundDevices();
  for (const auto &device : boundLights) {
    Serial.printf("Device on endpoint %d, short address: 0x%x\r\n", device->endpoint, device->short_addr);
    Serial.printf(
      "IEEE Address: %02X:%02X:%02X:%02X:%02X:%02X:%02X:%02X\r\n", device->ieee_addr[7], device->ieee_addr[6], device->ieee_addr[5], device->ieee_addr[4],
      device->ieee_addr[3], device->ieee_addr[2], device->ieee_addr[1], device->ieee_addr[0]
    );
    char* manufacturer = zbSwitch.readManufacturer(device->endpoint, device->short_addr, device->ieee_addr);
    if(manufacturer != nullptr) {
      Serial.printf("Light manufacturer: %s\r\n", zbSwitch.readManufacturer(device->endpoint, device->short_addr, device->ieee_addr));
    }
    char* model = zbSwitch.readModel(device->endpoint, device->short_addr, device->ieee_addr);
    if(model != nullptr) {
      Serial.printf("Light model: %s\r\n", zbSwitch.readModel(device->endpoint, device->short_addr, device->ieee_addr));
    }
  }
```

AndunHH, 2025-01-06:
Thanks for the fast solution approach. The first response: Yes, the device is not rebooting endlessly because of the readModel() or readManufacturer().

But sadly, it's also not working as expected. 

I refer the millisecond counter from the log file:fg
1. After ca. 12 seconds the ikea plug was found and bound.
2. At 30s the device trys to read the manfacturer and model successfully (see 30515 and 30668). Here the list of bound devices is searched for TWO devices: short adress 0x0 and 0xfa2d.
3. At 35s I  toggle the plug successfully four times.
4. I observe the behavior...
5. After 60s I unplug the outlet and the device is not able to get the information, see 60310 to 70290.
6. After 75s I reconnect the plug, see 75205 and from the debugging we see, that the information is in general present. Unfortunately, asking for this information in our application is not working: At 77594 the list of bound devices is checked again and contains THREE devices now: 0x0, 0xfa2d and again 0xfa2d. Sadly, none of them reports the manufacturer to our application (only the debug info is visible).
7. I let the application run to see if something changes after 180s with the open network, but this is not the case.

Therefore: reading the manufacturer in the application is not possible after un- and replugging the outlet.

Here is the complete log and below my actual sketch:

<details>
  <summary>complete log</summary>

```
=========== Before Setup Start ===========
Chip Info:
------------------------------------------
  Model             : ESP32-C6
  Package           : 1
  Revision          : 0.01
  Cores             : 1
  CPU Frequency     : 160 MHz
  XTAL Frequency    : 40 MHz
  Features Bitfield : 0x00000052
  Embedded Flash    : No
  Embedded PSRAM    : No
  2.4GHz WiFi       : Yes
  Classic BT        : No
  BT Low Energy     : Yes
  IEEE 802.15.4     : Yes
------------------------------------------
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   446348 B ( 435.9 KB)
  Free Bytes        :   414488 B ( 404.8 KB)
  Allocated Bytes   :    24652 B (  24.1 KB)
  Minimum Free Bytes:   409676 B ( 400.1 KB)
  Largest Free Block:   376820 B ( 368.0 KB)
------------------------------------------
Flash Info:
------------------------------------------
  Chip Size         :  4194304 B (4 MB)
  Block Size        :    65536 B (  64.0 KB)
  Sector Size       :     4096 B (   4.0 KB)
  Page Size         :      256 B (   0.2 KB)
  Bus Speed         : 40 MHz
  Bus Mode          : QIO
------------------------------------------
Partitions Info:
------------------------------------------
                nvs : addr: 0x00009000, size:    20.0 KB, type: DATA, subtype: NVS
            otadata : addr: 0x0000E000, size:     8.0 KB, type: DATA, subtype: OTA
               app0 : addr: 0x00010000, size:  1280.0 KB, type:  APP, subtype: OTA_0
               app1 : addr: 0x00150000, size:  1280.0 KB, type:  APP, subtype: OTA_1
             spiffs : addr: 0x00290000, size:  1388.0 KB, type: DATA, subtype: SPIFFS
         zb_storage : addr: 0x003EB000, size:    16.0 KB, type: DATA, subtype: FAT
             zb_fct : addr: 0x003EF000, size:     4.0 KB, type: DATA, subtype: FAT
           coredump : addr: 0x003F0000, size:    64.0 KB, type: DATA, subtype: COREDUMP
------------------------------------------
Software Info:
------------------------------------------
  Compile Date/Time : Jan  6 2025 19:40:26
  Compile Host OS   : linux
[... 833 lines cut ...]

  // Optional: List all bound devices and read manufacturer and model name
  Serial.println("Trying to read every 10s");
  std::list<zb_device_params_t *> boundLights = zbSwitch.getBoundDevices();
  for (const auto &device : boundLights) {
    Serial.printf("Device on endpoint %d, short address: 0x%x\r\n", device->endpoint, device->short_addr);
    Serial.printf(
      "IEEE Address: %02X:%02X:%02X:%02X:%02X:%02X:%02X:%02X\r\n", device->ieee_addr[7], device->ieee_addr[6], device->ieee_addr[5], device->ieee_addr[4],
      device->ieee_addr[3], device->ieee_addr[2], device->ieee_addr[1], device->ieee_addr[0]
    );
    char* manufacturer = zbSwitch.readManufacturer(device->endpoint, device->short_addr, device->ieee_addr);
    if(manufacturer != nullptr) {
      Serial.printf("Light manufacturer: %s\r\n", zbSwitch.readManufacturer(device->endpoint, device->short_addr, device->ieee_addr));
    }
    char* model = zbSwitch.readModel(device->endpoint, device->short_addr, device->ieee_addr);
    if(model != nullptr) {
      Serial.printf("Light model: %s\r\n", zbSwitch.readModel(device->endpoint, device->short_addr, device->ieee_addr));
    }
  }

  }
}
```

</details>

P-R-O-C-H-Y, 2025-01-07:
@AndunHH The reading there is duplicated, I have fixed it in the PR I opened.
I was able to reproduce the issue that the device is added multiple times to the bounded device list. I will add a fix for it. Thank you for nice report.

P-R-O-C-H-Y, 2025-01-07:
@AndunHH I have added a fix to the PR #10817

PR #10817 (2025-01-08): fix(zigbee): Fixes of timeout, example, warnings and bounding + add a 2MB part scheme for Zigbee ED
## Description of Change
This PR:
- fixes a blocking `getSensorSettings()` function for a Thermostat EP, which is now set to 10s.
- adds a check in the on/off switch example manufacturer/model read, so if reading fails it does not crash.
- adds a 2MB partition scheme for Zigbee End Device for C6 and H2
- fixes a device bounding multiple times, if its rejoining the network. If it was bounded it won't bound again.
- fixes warnings of missing initializer for member in EPs reporting function

## Tes [... cut]
closed 2025-01-08 by me-no-dev