espressif/arduino-esp32 #10778: ESP32-C6 doesn't compile with zigbee and WiFi.h #9978 - once again on 3.1.0

lsroka76, 2024-12-23:
### Board

ESP32 C6

### Device Description

plain

### Hardware Configuration

no

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

80

### PSRAM enabled

no

### Upload speed

115200

### Description

ESP32-C6 doesn't compile with zigbee and WiFi.h #9978 - the same issue

### Sketch

```cpp
as above
```

### Debug Message

```plain
as above
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

AndunHH, 2025-01-05:
I'm having the same problem. I'm not sure if it helps, but the linked ticket is closed, therefore here is a minimal example how to reproduce the problem:

My sketch:
```
#include "Zigbee.h"
#include "WiFi.h"

/********************* Arduino functions **************************/
void setup() {
}

void loop() {
  // Checking button for factory reset
  delay(100);
}
```

Board: XIAO_ESP32C6
Partition Scheme: Zigbee 4MB with spiffs
Zigbee mode: end device

Used arduino-esp32: 9eb7dc6f7bddf98600f6e8de3ccc19cc8ba5281b 

Compile error:
```
WARNING: library Zigbee claims to run on esp32 architecture(s) and may be incompatible with your current board which runs on arduino-esp32 architecture(s).
WARNING: library WiFi claims to run on esp32 architecture(s) and may be incompatible with your current board which runs on arduino-esp32 architecture(s).
WARNING: library Networking claims to run on esp32 architecture(s) and may be incompatible with your current board which runs on arduino-esp32 architecture(s).
/home/johannes/Arduino/hardware/espressif/arduino-esp32/tools/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld: /home/johannes/Arduino/hardware/espressif/arduino-esp32/tools/esp32-arduino-libs/esp32c6/lib/libopenthread.a(esp_openthread_radio.c.obj): in function `esp_ieee802154_transmit_done':
(.iram1.0+0x0): multiple definition of `esp_ieee802154_transmit_done'; /home/johannes/Arduino/hardware/espressif/arduino-esp32/tools/esp32-arduino-libs/esp32c6/lib/libzboss_port.a(zb_esp_mac.c.obj):(.iram1.1+0x0): first defined here
/home/johannes/Arduino/hardware/espressif/arduino-esp32/tools/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld: /home/johannes/Arduino/hardware/espressif/arduino-esp32/tools/esp32-arduino-libs/esp32c6/lib/libopenthread.a(esp_openthread_radio.c.obj): in function `esp_ieee802154_receive_done':
(.iram1.4+0x0): multiple definition of `esp_ieee802154_receive_done'; /home/johannes/Arduino/hardware/espressif/arduino-esp32/tools/esp32-arduino-libs/esp32c6/lib/libzboss_port.a(zb_esp_mac.c.obj):(.iram1.2+0x0): first defined here
/home/johannes/Arduino/hardware/espressif/arduino-esp32/tools/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld: /home/johannes/Arduino/hardware/espressif/arduino-esp32/tools/esp32-arduino-libs/esp32c6/lib/libopenthread.a(esp_openthread_radio.c.obj): in function `esp_ieee802154_transmit_failed':
(.iram1.5+0x0): multiple definition of `esp_ieee802154_transmit_failed'; /home/johannes/Arduino/hardware/espressif/arduino-esp32/tools/esp32-arduino-libs/esp32c6/lib/libzboss_port.a(zb_esp_mac.c.obj):(.iram1.4+0x0): first defined here
/home/johannes/Arduino/hardware/espressif/arduino-esp32/tools/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld: /home/johannes/Arduino/hardware/espressif/arduino-esp32/tools/esp32-arduino-libs/esp32c6/lib/libopenthread.a(esp_openthread_radio.c.obj): in function `esp_ieee802154_receive_sfd_done':
(.iram1.6+0x0): multiple definition of `esp_ieee802154_receive_sfd_done'; /home/johannes/Arduino/hardware/espressif/arduino-esp32/tools/esp32-arduino-libs/esp32c6/lib/libzboss_port.a(zb_esp_mac.c.obj):(.iram1.5+0x0): first defined here
/home/johannes/Arduino/hardware/espressif/arduino-esp32/tools/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld: /home/johannes/Arduino/hardware/espressif/arduino-esp32/tools/esp32-arduino-libs/esp32c6/lib/libopenthread.a(esp_openthread_radio.c.obj): in function `esp_ieee802154_transmit_sfd_done':
(.iram1.7+0x0): multiple definition of `esp_ieee802154_transmit_sfd_done'; /home/johannes/Arduino/hardware/espressif/arduino-esp32/tools/esp32-arduino-libs/esp32c6/lib/libzboss_port.a(zb_esp_mac.c.obj):(.iram1.6+0x0): first defined here
/home/johannes/Arduino/hardware/espressif/arduino-esp32/tools/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld: /home/johannes/Arduino/hardware/espressif/arduino-esp32/tools/esp32-arduino-libs/esp32c6/lib/libopenthread.a(esp_openthread_radio.c.obj): in function `esp_ieee802154_energy_detect_done':
(.iram1.8+0x0): multiple definition of `esp_ieee802154_energy_detect_done'; /home/johannes/Arduino/hardware/espressif/arduino-esp32/tools/esp32-arduino-libs/esp32c6/lib/libzboss_port.a(zb_esp_mac.c.obj):(.iram1.7+0x0): first defined here
/home/johannes/Arduino/hardware/espressif/arduino-esp32/tools/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld: /home/johannes/Arduino/hardware/espressif/arduino-esp32/tools/esp32-arduino-libs/esp32c6/lib/libopenthread.a(esp_openthread_radio.c.obj): in function `esp_ieee802154_cca_done':
(.iram1.9+0x0): multiple definition of `esp_ieee802154_cca_done'; /home/johannes/Arduino/hardware/espressif/arduino-esp32/tools/esp32-arduino-libs/esp32c6/lib/libzboss_port.a(zb_esp_mac.c.obj):(.iram1.8+0x0): first defined here
collect2: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1
```

The summary:
esp_ieee802154_cca_done and other symbols are both defined by 
- libzboss_port.a(zb_esp_mac.c.obj) and
- libopenthread.a(esp_openthread_radio.c.obj)
which is to much ;)

AndunHH, 2025-01-05:
One addition, because I actually don't plan to use WiFi and Zigbee in parallel.

I'm wrote a library which utilizes a RTC which checks where to find the "best" actual data. Therefore, I also wrote the following debug function:
```
/// @brief prints the localTime to the serial console
void RTCHelper::printLocalTime()
{
  Serial.print("Local: ");
  struct tm timeinfo;
  *if (!getLocalTime(&timeinfo))
  {
    Serial.println("No time available (yet)");
    return;
  }
  Serial.println(&timeinfo, "%A, %B %d %Y %H:%M:%S");
  
}
```

Calling `getLocalTime()` from esp32-hal-time.c also leads to the inclusion of many libraries, leading to the error above, even without using wifi in the first place.

P-R-O-C-H-Y, 2025-01-08:
Closing as completed by merging the linked PR.

closed 2025-01-08 by P-R-O-C-H-Y

sonirohit3, 2025-01-08:
@P-R-O-C-H-Y , Can you help me with ZBZCR 8MB partition scheme. I am facing 108% program storage when both zigbee and wifi is enabled.  I see the Zigbee ZCZR 8MB configuration missing in partition scheme type
[image]

compilation error: 
Sketch too big; see https://support.arduino.cc/hc/en-us/articles/360013825179 for tips on reducing it.
Sketch uses 1424651 bytes (108%) of program storage space. Maximum is 1310720 bytes.
Global variables use 61312 bytes (18%) of dynamic memory, leaving 266368 bytes for local variables. Maximum is 327680 bytes.
text section exceeds available space in board

Selecting 8M partition scheme keeps the device rebooting :
Rebooting...
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0xc (SW_CPU),boot:0x6c (SPI_FAST_FLASH_BOOT)
Saved PC:0x4001975a
SPIWP:0xee
mode:DIO, clock div:2
load:0x40875720,len:0x1228
load:0x4086c110,len:0xd9c
load:0x4086e610,len:0x2f74
entry 0x4086c110
[0;31mE (947) ESP_ZIGBEE_CLUSTER: The requested add cluster ID:0x3 is already existed[0m
Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

sonirohit3, 2025-01-11:
@P-R-O-C-H-Y , can you help me on above?