espressif/arduino-esp32 #10780: 13.1.0 'zigbee_role_t' does not name a type

Da10Meg, 2024-12-24:
### Board

ESP32 DEV Module

### Device Description

Module direct to usb c...no breadboard

### Hardware Configuration

N/A

### Version

v3.1.0

### IDE Name

Arduino

### Operating System

Windows 11

### Flash frequency

N/A

### PSRAM enabled

yes

### Upload speed

N/A

### Description

I followed the instructions @ https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html

installing v2.3.4 of the Arduino IDE on windows 11

used this link https://espressif.github.io/arduino-esp32/package_esp32_index.json in preferences > additional board manager urls

then via boards manager installed v3.1.0 of the esp32 by espressif systems.

selected 'esp32 dev module' then 

I opened example zigbee > scan_networks

set zigbee mode

did the verify go error

"Compilation error: 'zigbee_role_t' does not name a type"

I did reinstall of the IDE as per [3.1.0-RC3] Compilation error: 'ZigbeeTempSensor' does not name a type #10608 and tried other examples but get other type errors.

### Sketch

```cpp
// Copyright 2024 Espressif Systems (Shanghai) PTE LTD
//
// Licensed under the Apache License, Version 2.0 (the "License");
// you may not use this file except in compliance with the License.
// You may obtain a copy of the License at

//     http://www.apache.org/licenses/LICENSE-2.0
//
// Unless required by applicable law or agreed to in writing, software
// distributed under the License is distributed on an "AS IS" BASIS,
// WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
// See the License for the specific language governing permissions and
// limitations under the License.

/**
 * @brief This example demonstrates Zigbee Network Scanning.
 *
 * The example demonstrates how to use ESP Zigbee stack to scan for Zigbee networks.
 *
 * Any Zigbee mode can be selected in Tools->Zigbee mode
 * with proper Zigbee partition scheme in Tools->Partition Scheme.
 *
 * Please check the README.md for instructions and more detailed description.
 *
 * Created by Jan Procházka (https://github.com/P-R-O-C-H-Y/)
 */

#if !defined(ZIGBEE_MODE_ED) && !defined(ZIGBEE_MODE_ZCZR)
#error "Zigbee device mode is not selected in Tools->Zigbee mode"
#endif

#include "Zigbee.h"

#ifdef ZIGBEE_MODE_ZCZR
zigbee_role_t role = ZIGBEE_ROUTER;  // or can be ZIGBEE_COORDINATOR, but it won't scan itself
#else
zigbee_role_t role = ZIGBEE_END_DEVICE;
#endif

void printScannedNetworks(uint16_t networksFound) {
  if (networksFound == 0) {
    Serial.println("No networks found");
  } else {
    zigbee_scan_result_t *scan_result = Zigbee.getScanResult();
    Serial.println("\nScan done");
    Serial.print(networksFound);
    Serial.println(" networks found:");
    Serial.println("Nr | PAN ID | CH | Permit Joining | Router Capacity | End Device Capacity | Extended PAN ID");
    for (int i = 0; i < networksFound; ++i) {
      // Print all available info for each network found
      Serial.printf("%2d", i + 1);
      Serial.print(" | ");
      Serial.printf("0x%04hx", scan_result[i].short_pan_id);
      Serial.print(" | ");
      Serial.printf("%2d", scan_result[i].logic_channel);
      Serial.print(" | ");
      Serial.printf("%-14.14s", scan_result[i].permit_joining ? "Yes" : "No");
      Serial.print(" | ");
      Serial.printf("%-15.15s", scan_result[i].router_capacity ? "Yes" : "No");
      Serial.print(" | ");
      Serial.printf("%-19.19s", scan_result[i].end_device_capacity ? "Yes" : "No");
      Serial.print(" | ");
      Serial.printf(
        "%02x:%02x:%02x:%02x:%02x:%02x:%02x:%02x", scan_result[i].extended_pan_id[7], scan_result[i].extended_pan_id[6], scan_result[i].extended_pan_id[5],
        scan_result[i].extended_pan_id[4], scan_result[i].extended_pan_id[3], scan_result[i].extended_pan_id[2], scan_result[i].extended_pan_id[1],
        scan_result[i].extended_pan_id[0]
      );
      Serial.println();
      delay(10);
    }
    Serial.println("");
    // Delete the scan result to free memory for code below.
    Zigbee.scanDelete();
  }
}

void setup() {
  Serial.begin(115200);

  // Initialize Zigbee stack without any EPs just for scanning
  if (!Zigbee.begin(role)) {
    Serial.println("Zigbee failed to start!");
    Serial.println("Rebooting...");
    ESP.restart();
  }

  Serial.println("Setup done, starting Zigbee network scan...");
  // Start Zigbee Network Scan with default parameters (all channels, scan time 5)
  Zigbee.scanNetworks();
}

void loop() {
  // check Zigbee Network Scan process
  int16_t ZigbeeScanStatus = Zigbee.scanComplete();
  if (ZigbeeScanStatus < 0) {  // it is busy scanning or got an error
    if (ZigbeeScanStatus == ZB_SCAN_FAILED) {
      Serial.println("Zigbee scan has failed. Starting again.");
      Zigbee.scanNetworks();
    }
    // other option is status ZB_SCAN_RUNNING - just wait.
  } else {  // Found Zero or more Wireless Networks
    printScannedNetworks(ZigbeeScanStatus);
    Zigbee.scanNetworks();  // start over...
  }

  // Loop can do something else...
  delay(500);
  Serial.println("Loop running...");
}
```

### Debug Message

```plain
N/A
```

### Other Steps to Reproduce

Did full uninstall and re-install

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2024-12-27:
>then via boards manager installed v3.1.0 of the esp32 by espressif systems.
>selected 'esp32 dev module' then

@Da10Meg - Zigbee and OpenThread are only available for ESP32-C6 and ESP32-H2.
It is not possible to build any zigbee sketch using the ESP32.

Da10Meg, 2024-12-28:
thanks for that.

this is all new to me...will order a board that is compatible.

closed 2024-12-28 by Da10Meg