espressif/arduino-esp32 #10783: Using pins 18 & 19 as GPIO on ESP32-C3 (disabling USB-JTAG)

Rob58329, 2024-12-25:
### Board

Waveshare ESP32-C3 Zero Mini Development Board

### Device Description

Basic board only, an oscilloscope attached to GPIO-18

### Hardware Configuration

Basic board only, an oscilloscope attached to GPIO-18

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE v1.8.19

### Operating System

Windows 10

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

(This issue is a bit similar to https://github.com/espressif/arduino-esp32/issues/10587 which was for ESP32-S3.)

Using Arduino-IDE and latest "https://github.com/espressif/arduino-esp32" as at 24Dec24.

If you try to use pin 18 as GPIO on ESP32-C3 (either doing “`pinMode(18,INPUT)`” or “`pinMode(18,OUTPUT)`” you still get unwanted fast pulse spikes on pin 18 (USB-JTAG signals).

In order to disconnect the USB-JTAG from GPIO-18, you actually need to do a “`pinMode(19,INPUT)`” or “`pinMode(19,OUTPUT)`” (which appears to disconnect the USB-JTAG from BOTH GPIO-19 AND GPIO-18).

It would appear to be a bug that doing “`pinMode(18,INPUT)`” or “`pinMode(18,OUTPUT)`” does not do the same?

### Sketch

```cpp
// NB. "USB CDC on Boot=Disabled"
void setup() {
  delay(5000);
  pinMode(18,INPUT);    // does NOT remove USB-JTAG pulses from GPIO-18 
  // pinMode(19,INPUT); // DOES remove USB-JTAG pulses from BOTH GPIO-18 & GPIO-19    
}

void loop() {}
```

### Debug Message

```plain
n/a
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2024-12-27:
@Rob58329 - Please confirm some information related to this issue:

1- Is this board: https://www.waveshare.com/wiki/ESP32-C3-Zero ?
2- What is the board selected in the Arduino IDE? I only see Waveshare S3 in the menu.
3- Please enable the debug level using `IDE Menu->Tools->Core Debug Level: "Verbose"` and report back the log information after running the sketch provided in this issue.

SuGlider, 2024-12-27:
No Issue found with Arduino Core 3.1.0.

-------
### Board Configuration:
[image]

-------
### Sketch executed:
``` cpp
// NB. "USB CDC on Boot=Disabled"
// ESP32-C3

void setup() {
  delay(5000);
  pinMode(18,INPUT);    // does NOT remove USB-JTAG pulses from GPIO-18 
  pinMode(19,INPUT);    // DOES remove USB-JTAG pulses from BOTH GPIO-18 & GPIO-19    
}

void loop() {}
```
-------
### Debug Output:

```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0x1144
load:0x403cc710,len:0xad8
load:0x403ce710,len:0x2f80
entry 0x403cc710
[     0][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x42002cfa
[    12][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x42002cbe
[    23][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x42002c82
[    34][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x42002c46
[    46][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x42002cfa
[    57][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x42002cbe
[    68][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x42002c82
[    80][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x42002c46
[    95][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 20 successfully set to type UART_RX (2) with bus 0x3fc8ba74
[   105][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 21 successfully set to type UART_TX (3) with bus 0x3fc8ba74
=========== Before Setup Start ===========
Chip Info:
------------------------------------------
  Model             : ESP32-C3
  Package           : 0
  Revision          : 0.03
  Cores             : 1
  CPU Frequency     : 160 MHz
  XTAL Frequency    : 40 MHz
  Features Bitfield : 0x00000012
  Embedded Flash    : No
  Embedded PSRAM    : No
  2.4GHz WiFi       : Yes
  Classic BT        : No
  BT Low Energy     : Yes
  IEEE 802.15.4     : No
------------------------------------------
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   337496 B ( 329.6 KB)
  Free Bytes        :   299544 B ( 292.5 KB)
  Allocated Bytes   :    34176 B (  33.4 KB)
  Minimum Free Bytes:   299544 B ( 292.5 KB)
  Largest Free Block:   163828 B ( 160.0 KB)
------------------------------------------
Flash Info:
------------------------------------------
  Chip Size         :  4194304 B (4 MB)
  Block Size        :    65536 B (  64.0 KB)
  Sector Size       :     4096 B (   4.0 KB)
  Page Size         :      256 B (   0.2 KB)
  Bus Speed         : 80 MHz
  Bus Mode          : QIO
------------------------------------------
Partitions Info:
------------------------------------------
                nvs : addr: 0x00009000, size:    20.0 KB, type: DATA, subtype: NVS
            otadata : addr: 0x0000E000, size:     8.0 KB, type: DATA, subtype: OTA
               app0 : addr: 0x00010000, size:  1280.0 KB, type:  APP, subtype: OTA_0
               app1 : addr: 0x00150000, size:  1280.0 KB, type:  APP, subtype: OTA_1
             spiffs : addr: 0x00290000, size:  1408.0 KB, type: DATA, subtype: SPIFFS
           coredump : addr: 0x003F0000, size:    64.0 KB, type: DATA, subtype: COREDUMP
------------------------------------------
Software Info:
------------------------------------------
  Compile Date/Time : Dec 27 2024 10:17:16
  Compile Host OS   : windows
  ESP-IDF Version   : v5.3.2-174-g083aad99cf-dirty
  Arduino Version   : 3.1.0
------------------------------------------
Board Info:
------------------------------------------
  Arduino Board     : ESP32C3_DEV
  Arduino Variant   : esp32c3
  Arduino FQBN      : esp32:esp32:esp32c3:JTAGAdapter=default,CDCOnBoot=default,PartitionScheme=default,CPUFreq=160,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=verbose,EraseFlash=none,ZigbeeMode=default
============ Before Setup End ============
[  5432][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type GPIO (1) successfully set to 0x4200003e
[  5443][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 18 successfully set to type GPIO (1) with bus 0x13
[  5453][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type GPIO (1) successfully set to 0x4200003e
[  5465][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 19 successfully set to type GPIO (1) with bus 0x14
=========== After Setup Start ============
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   337496 B ( 329.6 KB)
  Free Bytes        :   304444 B ( 297.3 KB)
  Allocated Bytes   :    29276 B (  28.6 KB)
  Minimum Free Bytes:   299208 B ( 292.2 KB)
  Largest Free Block:   163828 B ( 160.0 KB)
------------------------------------------
GPIO Info:
------------------------------------------
  GPIO : BUS_TYPE[bus/unit][chan]
  --------------------------------------  
    18 : GPIO
    19 : GPIO
    20 : UART_RX[0]
    21 : UART_TX[0]
============ After Setup End =============

```

SuGlider, 2024-12-27:
@Rob58329 - I can't reproduce the reported issue.

Using this sketch:
``` cpp
// NB. "USB CDC on Boot=Disabled"
// ESP32-C3

void setup() {
  pinMode(18, INPUT_PULLUP);   // does NOT remove USB-JTAG pulses from GPIO-18
  pinMode(19, INPUT_PULLUP);   // DOES remove USB-JTAG pulses from BOTH GPIO-18 & GPIO-19
  Serial.begin(115200);
}

// Use a wire to connect Pins 18 and 19 to +3.3v or GND
void loop() {
  Serial.printf("Pin 18 = %s || Pin 19 = %s\r\n",
                digitalRead(18) == HIGH ? "High" : "Low",
                digitalRead(19) == HIGH ? "High" : "Low"
               );
  delay(1000);
}
```

I can ground each pin and read the correct input:

```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0x1144
load:0x403cc710,len:0xad8
load:0x403ce710,len:0x2f80
entry 0x403cc710
Pin 18 = High || Pin 19 = High
Pin 18 = High || Pin 19 = High
Pin 18 = High || Pin 19 = High
Pin 18 = High || Pin 19 = High
Pin 18 = High || Pin 19 = Low
Pin 18 = High || Pin 19 = Low
Pin 18 = High || Pin 19 = Low
Pin 18 = High || Pin 19 = Low
Pin 18 = High || Pin 19 = High
Pin 18 = High || Pin 19 = High
Pin 18 = Low || Pin 19 = High
Pin 18 = Low || Pin 19 = High
Pin 18 = Low || Pin 19 = High
Pin 18 = Low || Pin 19 = High
Pin 18 = High || Pin 19 = High
Pin 18 = High || Pin 19 = High
Pin 18 = High || Pin 19 = High
Pin 18 = High || Pin 19 = High

```

Rob58329, 2024-12-27:
@SuGlider - thank you for getting back to me!

As I said above "In order to disconnect the USB-JTAG from GPIO-18, you actually need to do a “`pinMode(19,INPUT)`” or “`pinMode(19,OUTPUT)`” - IE. GPIO-18 works OK if you do a “`pinMode(19,INPUT)`” or “`pinMode(19,OUTPUT)`”.

(In your above test sketch you are using GPIO-19 to obtain your DEBUG-OUTPUT, so this will remove the "USB-JTAG" signals from both GPIO-19 AND GPIO-18. - if you instead use GPIO-18 as your Serial-Tx, and set the Serial-Rx to say GPIO-1 [ie. NOT GPIO-19], you would probably see a significantly corrupted output on your Serial-monitor, but I haven't tried this!)

IF you ONLY do a "`pinMode(18,INPUT)`" the USB-JTAG is STILL-CONNECTED to GPIO-18, and if you monitor this pin with an oscilloscope, you will see these as fast (<1us) spikes.  If you instead use "`pinMode(18,OUTPUT)`" with a pulsed output (say a 20us = 50kHz square-wave) you will see this USB-JTAG interference as intermittent disortions of the square-wave (again visible on an oscilloscope, or for example as a corrupted display on a HUB-8 or similar display).

(The bug I am reporting is that trying to use "`pinMode(18,INPUT)`" on is own (without touching GPIO-19) does NOT disconnect the USB-JTAG from GPIO-18 and the so you get unwanted pulses on GPIO-18.)

SuGlider, 2024-12-27:
>In your above test sketch you are using GPIO-19 to obtain your DEBUG-OUTPUT

This is incorrect.
When I set "USB CDC on Boot=Disabled", Serial will be UART0, not JTAG/USB-CDC.

The debug output I got is from pin 21 which is UART0 Tx.

SuGlider, 2024-12-27:
I'll try to check my ESP32-C3 Devkit board in the osciloscope in order to verify if this spike also happens here.

Rob58329, 2024-12-27:
>The debug output I got is from pin 21 which is UART0 Tx.

Sorry - my bad.

I connected a Serial-to-USB adapter-Rx to GPIO-21 and got the following differences in "Core Debug Level=Verbose" outputs

(A) With "`pinMode(18,INPUT);`" ONLY (which gives the discussed 1us unwanted pulse on GPIO-18): 

[     0][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x42002cf2
[    12][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x42002cb6
[    23][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x42002c7a
[    34][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x42002c3e
[    46][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x42002cf2
[    57][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x42002cb6
[    69][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x42002c7a
[    80][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x42002c3e

============ Before Setup End ============
[  9443][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type GPIO (1) successfully set to 0x42000038
[  9454][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 18 successfully set to type GPIO (1) with bus 0x13
=========== After Setup Start ============

(B) With "`pinMode(18,INPUT);"` AND "`pinMode(19,INPUT);`" which fully dissconnects the USB-JTAG (no unwanted pulses on GPIO-18): 

[     0][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x42002cfa
[    12][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x42002cbe
[    23][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x42002c82
[    34][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x42002c46
[    46][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x42002cfa
[    57][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x42002cbe
[    68][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x42002c82
[    80][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x42002c46

============ Before Setup End ============
[  9443][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type GPIO (1) successfully set to 0x4200003e
[  9454][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 19 successfully set to type GPIO (1) with bus 0x14
[  9464][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type GPIO (1) successfully set to 0x4200003e
[  9475][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 18 successfully set to type GPIO (1) with bus 0x13
=========== After Setup Start ============

Interestingly if you use "pinMode(18,INPUT_PULLUP);" ONLY you get a really weird waveform (as shown below) which is nominally about 1V only  (ie NOT the 3.3V pulled-up value it should be), and intermittent 1us USB-JTAG pulses shooting up to 3.3V down to 0V then ending at about 1V...

(I assume the USB-JTAG circuitry must be putting some type of pull-down resistor onto GPIO-18.) 

[image]

SuGlider, 2024-12-30:
Does this pulse happen one time only?

SuGlider, 2024-12-30:
The debug output is correct.
It means that pins 18 and/or 19 are set to INPUT Mode.

SuGlider, 2024-12-30:
@Rob58329 - I found some information about Boot configuration and ROM Boot Code in the C3 datasheet:
https://www.espressif.com/sites/default/files/documentation/[esp32-c3_datasheet_en.pdf](https://www.espressif.com/sites/default/files/documentation/esp32-c3_datasheet_en.pdf)

[image]

I think that the pulse you have seen is actually the default Boot Message in USB Serial/JTAG port (pins 18 and 19 USB Hardware CDC). The message is "ESP-ROM:esp32c3-api1-20210207"

It is printed by ROM code into USB Serial/JTAG before any bootloader or application code is executed.
It may be disabled by burning an eFuse (which can't be reverted) as detailed in the Datasheet.

eFuse `EFUSE_USB_PRINT_CHANNEL` will disable printing this BOOT Message into USB Serial/JTAG.
**Do not disable the whole USB Serial/JTAG otherwise it won't work anymore (` EFUSE_DIS_USB_SERIAL_JTAG`).**

eFuse can be listed and burned by using `espefuse.py` tool.
For more information: https://docs.espressif.com/projects/esptool/en/latest/esp32/espefuse/index.html

SuGlider, 2024-12-30:
@Rob58329 - Use `espefuse.py` carefully when burning any eFuse. Once burned, it can't be reverted.

Rob58329, 2024-12-30:
@SuGlider - many thanks for your comments.

>Does this pulse happen one time only?

No, you see the pulses continuously (every few seconds or more frequently) on both GPIO-18 and GPIO-19, UNTIL you do a `pinMode(19,INPUT)`.

>------------------------------------------------

I'm surprised it isn't possible to do whatever `pinMode(19,INPUT)` does for `pinMode(18,INPUT)`!

>Use espefuse.py carefully when burning any eFuse. Once burned, it can't be reverted.

Instead I will probably just do a `pinMode(19,INPUT)` on my unused GPIO-19, which solves the pulse issue on GPIO-18.

Interestingly although `pinMode(19,INPUT)` stops the USB-JTAG pulses on GPIO-18, it does NOT remove the approx 18k pulldown on GPIO-18, meaning that the above sketch's "`pinMode(18,INPUT_PULLUP); digitalRead(18);`" still reads LOW.  This doesn't really matter if you are using `pinMode(18,INPUT)` or `pinMode(18,OUTPUT)`, but this pullup means can't rely on `pinMode(18,INPUT_PULLUP) `here.

Also note that this 18k pulldown actually seems to be provided by the D+/D- voltages on the USB-C-connector; if instead I power the USB-C from a USB-Power-Supply, this changes to a 30k pullup, and interestingly you DO NOT get the above unwanted USB-JTAG pulses from GPIO-18 at all!

Also, if you power directly from the 5V pin, then you don't get the unwanted USB-JTAG pulses on GPIO-18, and ALSO `pinMode(18,INPUT_PULLUP)` works properly (as there is no pullup/pulldown resistance or voltages added by the D+/D- pins).

I would therefore suspect that burning an eFuse won't actually effect these external pullup/pulldown resistances, so if you wan't to use a pullup on GPIO-18, either power the ESP32-C3 from it's 5V (or 3.3V?) pin and use `pinMode(18,INPUT_PULLUP)`, or use an external strong pullup.

SuGlider, 2024-12-30:
ROM code enables Hardware Serial (USB CDC) by default. The CDC hardware implementation may try to talk to the USB Host from time to time and it may be the pulse.

When any other device is attached to pins 18 or 19, the HW Serial/JTAG is detached and it stops sending the pulse.

This is done by ROM code and it can be changed using eFuses.

Arduino can't help it, unless by changing these pins function.

SuGlider, 2024-12-30:
>Interestingly although pinMode(19,INPUT) stops the USB-JTAG pulses on GPIO-18, it does NOT remove the approx 18k pulldown on GPIO-18, meaning that the above sketch's "pinMode(18,INPUT_PULLUP); digitalRead(18);" still reads LOW. This doesn't really matter if you are using pinMode(18,INPUT) or pinMode(18,OUTPUT), but this pullup means can't rely on pinMode(18,INPUT_PULLUP) here. (ESP32-C3 powered by it's USB-C connected to a computer,)

This is not what I have seen with
https://github.com/espressif/arduino-esp32/issues/10783#issuecomment-2563711439

me-no-dev, 2024-12-30:
one option is to set the peripheral manager to have the USB set by default, so when you try to reassign one of the pins, it will trigger detach from the driver for both

SuGlider, 2024-12-30:
> one option is to set the peripheral manager to have the USB set by default, so when you try to reassign one of the pins, it will trigger detach from the driver for both

Yes, this is already done for UART0, but not for USB/JTAG.
Actually to be precise, the code should check the eFuses to know what has been enabled or not when running the ROM code.

Rob58329, 2025-01-09:
The below sketch is an easy way to show if the external USB-D+ / USB-D- pins on the USB-C connector are pulling GPIO-18 up or down.

```
#define LED_DATA_PIN 10 // WS2818_LED // RGB 
#define LED_COUNT  1
#include <Adafruit_NeoPixel.h>
Adafruit_NeoPixel strip(LED_COUNT, LED_DATA_PIN, NEO_GRB + NEO_KHZ800);

const uint32_t green=strip.Color(255,0,0);
const uint32_t red  =strip.Color(0,255,0);
const uint32_t blue =strip.Color(0,0,255);
const uint32_t white=strip.Color(128,128,128);

void setup() { // NB. "USB CDC on Boot=Disabled"
  delay(5000); // give enough time to upload new code (otherwise use buttons to put ESP32-C3 into upload-mode)
  strip.begin(); // INITIALIZE NeoPixel strip object
  strip.setBrightness(80); // 0 to 255
  pinMode(19,INPUT); delay(1); // disables USB-JTAG on both GPIO-19 AND GPIO-18
  pinMode(18,INPUT); // does NOT remove USB-JTAG pulses from GPIO-18 (the above is required to do this) 
}

int old=-1;
void loop() {
  int level=digitalRead(18); // read GPIO-18 to see if pulled-up or pulled-down by the external USB-D+ / USB-D- pins 
  if (old!=level) {
     old=level;
     if (old) strip.setPixelColor(0,green); else strip.setPixelColor(0,red); // green=pulled-up, red=pulled-down
     strip.show();
  }
}
```

For me, it shows Red when powered from my PC’s USB port, and Green when powered from a plug-top USB power-supply.

I will close this issue now as it does not sound like it’s possible to make GPIO-18 act the same way as GPIO-19 (in disconnecting the USB-JTAG from both pins), so instead I will just ensure I do a `pinMode(19,INPUT)` to disconnect the USB-JTAG before I use GPIO-18.

TD-er, 2025-01-09:
About the presence of pull-resistors on the D+/D-... 
There needs to be some resistors when using an USB-connector to tell the USB host something is connected and also allow to draw > 100 mA current.
Not sure if those can even be disabled as they are probably stronger than the typical pull resistors in the ESP on the other GPIOs. Or maybe they are not even internal in the ESP chip.

Rob58329, 2025-01-09:
@TD-er : I don’t think the ESP32-C3-Zero board puts any external pullup or pulldown resistors onto GPIO-18 (D+ / D-) because if you power the ESP32-C3-Zero board from it’s 5V and Gnd Pins, (AND do a `pinMode(19,INPUT)` and `pinMode(18,INPUT)`), GPIO-18 is LOW (Red) until you add an external 100k pullup between GPIO-18 and 3.3V, which causes GPIO-18 to go fully HIGH (Green) and measures 3.3V.  I think the pullup/pulldown resistances I see on GPIO-18 are all due to whatever external is connected to the USB-C connector’s D+ and D- pins.

If the USB-JTAG is connected to GPIO-18/ GPIO-19, then I'm guessing the ESP32-C3 enables it’s internal pullup/pulldown resistors as required.  (I've no idea if these would be stronger than normal.)

TD-er, 2025-01-09:
I think it is still something that needs to be enabled before starting the ESP or else the USB controller may not be willing to output the required current.
Well not that many USB hosts are so strict, but it is part of the USB specification.